o-gs/dji-firmware-tools #1: DJI x5 RAW

coptersafe, 2016-09-04:
hi , what about unpacking DJI X5R raw - camera - Xilinx proc Zynq 7020

https://dl.djicdn.com/downloads/osmo/OSMO_FC550R_FW_V01.03.00.40.zip
if you unpack this binary , you can find module C1 - c1_(1.15.7.17) . this file  have few partition for Zynq

coptersafe, 2016-09-04:
Xilinx First Stage Boot Loader
Release 2014.2  Dec  9 2015-11:33:53
Devcfg driver initialized
Silicon Version 3.1
Watchdog driver initialized
Watchdog Timer Disabled
User not allowed to do any system resets
Boot mode is QSPI
Single Flash Information
FlashID=0x20 0xBA 0x20
MICRON 512M Bits
QSPI is in single flash connection
QSPI Init Done
Flash Base Address: 0xFC000000
Reboot status register: 0x60400000
Multiboot Register: 0x0000C000
Image Start Address: 0x00000000
Partition Header Offset:0x00000C80
Partition Count: 7
Partition Number: 1
Header Dump
Image Word Len: 0x0016CEEB
Data Word Len: 0x0016CE28
Partition Word Len:0x0016CEEB
Load Addr: 0x00000000
Exec Addr: 0x00000000
Partition Start: 0x00006690
Partition Attr: 0x00000020
Partition Checksum Offset: 0x00000000
Section Count: 0x00000001
Checksum: 0xFFBB2B00
Bitstream
Encrypted
In FsblHookBeforeBitstreamDload function
PCAP:StatusReg = 0x40000A30
PCAP:device ready
PCAP:Clear done
Level Shifter Value = 0xA
Devcfg Status register = 0x40000A30
PCAP:Fabric is Initialized done
PCAP register dump:
PCAP CTRL 0xF8007000: 0x4E80FE80
PCAP LOCK 0xF8007004: 0x00000012
PCAP CONFIG 0xF8007008: 0x00000508
PCAP ISR 0xF800700C: 0x0802000B
PCAP IMR 0xF8007010: 0xFFFFFFFF
PCAP STATUS 0xF8007014: 0x00070A30
PCAP DMA SRC ADDR 0xF8007018: 0x00100001
PCAP DMA DEST ADDR 0xF800701C: 0xFFFFFFFF
PCAP DMA SRC LEN 0xF8007020: 0x0016CEEB
PCAP DMA DEST LEN 0xF8007024: 0x0016CE28
PCAP ROM SHADOW CTRL 0xF8007028: 0xFFFFFFFF
PCAP MBOOT 0xF800702C: 0x0000C000
PCAP SW ID 0xF8007030: 0x00000000
PCAP UNLOCK 0xF8007034: 0x757BDF0D
PCAP MCTRL 0xF8007080: 0x30800100
...................................................................................................
DMA Done !

FPGA Done !
In FsblHookAfterBitstreamDload function
Partition Number: 2
Header Dump
Image Word Len: 0x00014BDB
Data Word Len: 0x00014B1D
Partition Word Len:0x00014BDB
Load Addr: 0x04000000
Exec Addr: 0x04000000
Partition Start: 0x00173580
Partition Attr: 0x00000010
[... 140 lines cut ...]
Partition Checksum Offset: 0x00000000
Section Count: 0x00000001
Checksum: 0xFD382538
Application
Encrypted
Bank Selection 1
PCAP:StatusReg = 0x40000F30
PCAP:device ready
PCAP:Clear done
PCAP register dump:
PCAP CTRL 0xF8007000: 0x4E80FE80
PCAP LOCK 0xF8007004: 0x00000012
PCAP CONFIG 0xF8007008: 0x00000508
PCAP ISR 0xF800700C: 0x00030004
PCAP IMR 0xF8007010: 0xFFFFFFFF
PCAP STATUS 0xF8007014: 0x00A6EF30
PCAP DMA SRC ADDR 0xF8007018: 0x02000001
PCAP DMA DEST ADDR 0xF800701C: 0x02000001
PCAP DMA SRC LEN 0xF8007020: 0x002FDD9B
PCAP DMA DEST LEN 0xF8007024: 0x002FDCDD
PCAP ROM SHADOW CTRL 0xF8007028: 0xFFFFFFFF
PCAP MBOOT 0xF800702C: 0x0000C000
PCAP SW ID 0xF8007030: 0x00000000
PCAP UNLOCK 0xF8007034: 0x757BDF0D
PCAP MCTRL 0xF8007080: 0x30800100

mefistotelis, 2016-09-04:
Why there's no "Partition Number 7" info in the log you've provided? Was it cut because the comment is too long? Please paste the rest of the log, if you can.

Files names within the Zynq firmware doesn't seem to be plain text - ie. "maruksidami.g.eg" looks more like an encrypted file name; though the algorithm used is likely something simple if encrypted alphanumeric chars are still within alphanumeric range.

coptersafe, 2016-09-04:
no..it is all log, at now u dont know why 7 partitions but only 6 at log.. maybe i need more ne logs.

coptersafe, 2016-09-05:
DMA Done !
Handoff Address: 0x04000000
In FsblHookBeforeHandoff function
SUCCESSFUL_HANDOFF
FSBL Status = 0x1

U-Boot 2014.01 (Sep 17 2015 - 22:10:38)

I2C:   ready
Memory: ECC disabled
DRAM:  256 MiB
MMC:   zynq_sdhci: 0
SF: Detected N25Q512 with page size 256 Bytes, erase size 4 KiB, total 64 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   Gem.e000b000
Hit any key to stop autoboot:  0
Copying Linux from QSPI flash to RAM...
## Booting kernel from Legacy Image at 03000000 ...
   Image Name:   Linux-3.14.0-xilinx
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2421728 Bytes = 2.3 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Loading init Ramdisk from Legacy Image at 02000000 ...
   Image Name:
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    12546865 Bytes = 12 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 02d00000
   Booting using the fdt blob at 0x2d00000
   Loading Kernel Image ... OK
   Loading Ramdisk to 0ef37000, end 0fb2e331 ... OK
   Loading Device Tree to 0ef31000, end 0ef36877 ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0x0
Linux version 3.14.0-xilinx (wk@localhost.localdomain) (gcc version 4.6.1 (Sourcery CodeBench Lite 2011.09-50) ) #7 SMP PREEMPT Wed Dec 2 20:52:00 CST 2015
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Machine model: Xilinx Zynq
bootconsole [earlycon0] enabled
Memory policy: Data cache writealloc
PERCPU: Embedded 7 pages/cpu @cfde2000 s7232 r8192 d13248 u32768
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
Kernel command line: console=ttyPS0,115200 maxcpus=1 root=/dev/ram rw earlyprintk
PID hash table entries: 1024 (order: 0, 4096 bytes)
Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Memory: 242812K/262144K available (3071K kernel code, 174K rwdata, 1112K rodata, 155K init, 158K bss, 19332K reserved, 0K highmem)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    vmalloc : 0xd0800000 - 0xff000000   ( 744 MB)
    lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
      .text : 0xc0008000 - 0xc041dfc0   (4184 kB)
      .init : 0xc041e000 - 0xc0444c40   ( 156 kB)
      .data : 0xc0446000 - 0xc04718c0   ( 175 kB)
       .bss : 0xc04718cc - 0xc04994bc   ( 159 kB)
Preemptible hierarchical RCU implementation.
[... 51 lines cut ...]
dma-pl330 f8003000.ps7-dma:     DBUFF-128x8bytes Num_Chans-8 Num_Peri-4 Num_Events-16
xuartps e0001000.serial: clock name 'aper_clk' is deprecated.
xuartps e0001000.serial: clock name 'ref_clk' is deprecated.
e0001000.serial: ttyPS0 at MMIO 0xe0001000 (irq = 82, base_baud = 3124999) is a xuartps
console [ttyPS0] enabled
console [ttyPS0] enabled
bootconsole [earlycon0] disabled
bootconsole [earlycon0] disabled
xdevcfg f8007000.ps7-dev-cfg: ioremap 0xf8007000 to d0862000
brd: module loaded
loop: module loaded
mousedev: PS/2 mouse device common for all mice
i2c /dev entries driver
cdns-i2c e0004000.ps7-i2c: 100 kHz mmio e0004000 irq 57
zynq-edac f8006000.ps7-ddrc: ecc not enabled
sdhci: Secure Digital Host Controller Interface driver
sdhci: Copyright(c) Pierre Ossman
sdhci-pltfm: SDHCI platform and OF driver helper
Registering SWP/SWPB emulation handler
drivers/rtc/hctosys.c: unable to open rtc device (rtc0)
RAMDISK: gzip image found at block 0
EXT2-fs (ram0): warning: mounting unchecked fs, running e2fsck is recommended
VFS: Mounted root (ext2 filesystem) on device 1:0.
devtmpfs: mounted
Freeing unused kernel memory: 152K (c041e000 - c0444000)