o-gs/dji-firmware-tools #2: Extraction

notsolowki, 2016-12-18:
C:\Users\Usr\Desktop\phantom-firmware-tools-master>python dji_fwcon.py -vv -x -p P3X_FW_V01.08.0080.bin
P3X_FW_V01.08.0080.bin: Opening for extraction
P3X_FW_V01.08.0080.bin: Header:
{   'entry_count': 16,
    'hdrend_offs': 898,
    'magic': 305419896L,
    'magic_ver': 1,
    'manufacturer': 'DJI',
    'model': 'P3X',
    'padding': '00000000000000000000',
    'salt': 1459829053,
    'ver_latest': 17301584,
    'ver_latest_enc': 120364041,
    'ver_rollbk': 17236028,
    'ver_rollbk_enc': 119774309}
P3X_FW_V01.08.0080.bin: Module index 0
{   'decrypted_len': 43776L,
    'decrypted_md5': 'cc3baf52297f2eb67adf98f0270956f4',
    'dt_offs': 898L,
    'encrypt_type': 1,
    'reserved2': 1,
    'spcoding': 16,
    'splvalue': 0,
    'stored_len': 43776L,
    'stored_md5': 'bbd8964489204dfdb8858d5824a116b0',
    'target': 'm0305',
    'target_name': "main controller 'A3' ldr",
    'version': '34.02.0009'}
P3X_FW_V01.08.0080.bin: Module index 1
{   'decrypted_len': 790016L,
    'decrypted_md5': '9ee6f2692d3982e8ce532372789a50cd',
    'dt_offs': 44674L,
    'encrypt_type': 1,
    'reserved2': 1,
    'spcoding': 16,
    'splvalue': 0,
    'stored_len': 790016L,
    'stored_md5': '229617fbd99769da90e394b239621935',
    'target': 'm0306',
    'target_name': "main controller 'A3' app",
    'version': '02.04.3589'}
P3X_FW_V01.08.0080.bin: Module index 2
{   'decrypted_len': 93696L,
    'decrypted_md5': '2b06bdb11b7b813863cb8efe106667d8',
    'dt_offs': 834690L,
    'encrypt_type': 1,
    'reserved2': 1,
    'spcoding': 16,
    'splvalue': 0,
    'stored_len': 93696L,
    'stored_md5': '6a5e48b0a6b81ed24e14c7548b6d9027',
    'target': 'm0400',
    'target_name': 'gimbal mdl 0',
    'version': '01.44.0000'}
P3X_FW_V01.08.0080.bin: Module index 3
{   'decrypted_len': 19140L,
    'decrypted_md5': '09bebba2951faf82ea3f9cb0793f5dd7',
    'dt_offs': 928386L,
    'encrypt_type': 0,
    'reserved2': 1,
    'spcoding': 0,
    'splvalue': 0,
    'stored_len': 19140L,
    'stored_md5': '09bebba2951faf82ea3f9cb0793f5dd7',
    'target': 'm1100',
    'target_name': 'battery controller 1 app',
    'version': '01.07.3841'}
P3X_FW_V01.08.0080.bin: Module index 4
{   'decrypted_len': 42496L,
    'decrypted_md5': '8f773cc3245745ac6861af803fc520b6',
[... 235 lines cut ...]
    'dt_len': 5038864L}, {   'crc32': 'EF4876FE',
    'dt_len': 27001088L}]
P3X_FW_V01.08.0080_mi12.bin: Post Header:
{   'part_size': '00000800 00040000 00040000 00100000 02000000 00500000 00f00000 00a00000 03c00000 01e00000 00000000 00500000 00c00000 00500000 00100000'}
P3X_FW_V01.08.0080_mi12.bin: Entry 0
{   'build_date': '2016-03-24',
    'crc32': '56F034C9',
    'dt_len': 9268800L,
    'flag1': '00000000',
    'flag2': '00000000',
    'magic': 'A324EB90',
    'mem_addr': '0E600000',
    'padding': '00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000',
    'version': '1.0'}
P3X_FW_V01.08.0080_mi12.bin: Extracting entry  0, pos      140, len  9268800 bytes
Traceback (most recent call last):
  File "amba_fwpak.py", line 649, in <module>
    main(sys.argv[1:])
  File "amba_fwpak.py", line 620, in main
    amba_extract(po,fwmdlfile)
  File "amba_fwpak.py", line 405, in amba_extract
    hdcrc = amba_calculate_crc32h_part(copy_buffer, hdcrc)
  File "amba_fwpak.py", line 257, in amba_calculate_crc32h_part
    crc = crc32_tab[(crc ^ octet) & 0xff] ^ (crc >> 8)
TypeError: unsupported operand type(s) for ^: 'long' and 'str'

mefistotelis, 2016-12-18:
Looks like copy_buffer is a string instead of c_ubyte array. You should convert copy_buffer to array before calling amba_calculate_crc32h_part().

mefistotelis, 2016-12-18:
Just to make sure: did you made any modifications to the cloned scripts? If not, this is likely caused by version of the python interpreter.

notsolowki, 2016-12-18:
Actually, i got a configparaser error and changed it to ConfigParaser in all the scripts. im using python 2.7 on widows. what version of python would you recommend i use. i was trying to save myself the embarrassment. id like to thank you for your hard work too even if i don't figure it out

mefistotelis, 2016-12-18:
I can confirm dji_fwcon.py and amba_fwpak.py works fine on Windows, with python from msys2 package:
[code]python --version
Python 3.4.3[/code]
Some scripts will not work on Windows:
- amba_ubifs.sh - uses POSIX tools to mount, most likely will never work on Windows
- amba_sys2elf.py - uses an ARM toolchain which are not easily obtainable on Windows; I hope to remove this dependency in the future, so it will start working on Windows

notsolowki, 2016-12-18:
when i updated to python 3 all worked great thank you ill post on here if i have any more issues

mefistotelis, 2016-12-18:
great; have fun.

closed 2016-12-18 by mefistotelis