o-gs/dji-firmware-tools #3: repacking

notsolowki, 2016-12-19:
i got a phantom 3 pro. from what i understand they don't use a wificard according to hostapd.conf it is a nl8011 chipset card. i would like to enable a shell from the usb port. do you know a quick way or can you point me in the right direction. i suspect ill have a limited number of times i can flash the firmware. right now im on 1.5+ and im not sure if they will let you reflash the same firmware twice so id like to get it right befor i end up at 1.10 and cant upgrade it again. i dont know that to be true im just guessing. Looking at inittab it looks like they have #ttyusb0 to disable its function? i understand the risks of bricking something. do you think i could edit some of the settings it and repack it then load it into the drone at boot it. i jsut want to get shell from the usb port. thanks again

mefistotelis, 2016-12-19:
The re-packing function should work on P3 Pro, this is the firmware on which I developed the tools. You can test it by unpacking with dji_fwcon.py and amba_fwpak.py, and then re-packing everything again and checking if generated file is exactly the same as original (ie. with cmp command).

Upgrading over the same version:
In case the drone will refuse to update to same version, you may try to increase version numbers in the ini files.

WiFi on Ambarella camera board:
I don't know whether such hardware is there; I doubt it, but I never looked at the board. The nl80211 mentioned in config might come from Ambarella SDK or early development board.

Enabling TTY on USB in Ambarella camera board:
I haven't tried that. It should work.

notsolowki, 2016-12-19:
good point about the ini files thankyou. i dont know if you have connected the rc usb port to your computer and tried to see what happens on terminal. when i tried i got a bunch of gibberish  but in between the gibberish you could make since of some of it. what about the binary of the battery. would we see more scripts in the future to to disassemble further. if i got shell from the cameras usb port do you think i could get access to the full file system of the drone. id like to maybe add a couple degrees of pitch to the copter and change change the behavior of the leds to make them strobe faster. im just a hobbyist not looking to mess around with anything that could get me in trouble or anything im not really sure what i want to do with it yet. just looking for things i could do. and thank you again

mefistotelis, 2016-12-19:
There is no "full file system of the drone". All DJI drones have very separate, almost autonomous components. This differentiates DJI from other companies, which often cut costs by reducing the amount of CPUs and inter-connecting components. For example Parrot drones are made on a single CPU, which drives everything - easier to make customizations, but less reliable.

Enabling serial over USB in Ambarella firmware will only give you access to the USB port connected to gimbal. With access to that component you will be able to use internal interface to other parts of the drone - ie. Ambarella firmware can take over control of the LEDs, and is the place from which firmware updates are distributed to all other components. We don't really know how far that control would go. But you won't have direct control over the other components, just the ability to ask them for something over the internal bus. For example, you will not have access to the SD card under flight controller - for that, you'd have to hack into main controller firmware.

notsolowki, 2016-12-19:
Thabks for explaining all that. When you say main controller firmware do you mean the the handheld controller. Or do you mean the device controller in the drone. So bassicly if i wanted to add a little bit of pitch or yaw i would need to get into the handheld controller. But if i wanted to strobe the leds faster i could turn them on and of with the amberella firmware side of things? Please excuse my english. Thankyou im so glad to see a modding community for these drones

mefistotelis, 2016-12-20:
LEDs are not always controlled by the Ambarella module. I don't know how modules "share" access to LEDs. Definitely LEDs will work even if you will remove the camera completely.

By "main controller" I meant flight controller. If you want to increase max velocity or turning speed, these are controlled by the flight controller. The handheld radio/pilot only sends a number from a set range, ie. 0 for central position and 1000/-1000 for bounding positions of the stick. I believe it is up to the flight controller to convert this number into specific pitch or yaw.

Flight controller firmware is encrypted, and the encryption algorithm isn't known; but - there is one workaround for that; it looks like in firmware P3X_FW_V01.07.0060 a DJI employee had a bad day and forgot to encrypt this component - a moddable version can be taken from this specific firmware.

probonopd, 2016-12-20:
Did you find any traces of Linux in P3X_FW_V01.07.0060?

notsolowki, 2016-12-20:
it appears that dji took the download off their website. where can i get a copy of that version of the fw

mefistotelis, 2016-12-20:
@notsolowki I don't think that's right; there are no links to it, but the file is there. Note that you have DL links to several firmwares in "tests/test_all.sh".

@probonopd The firmware which I'm assuming is the flight controller (please note, I'm not really 100% sure) is a monolythic ARM binary. It does not contain any license or copyright strings.

It contains strings which are often found in flight logs, this is why I assume it's for the flight controller. For example, strings from the log posted by someone here are within that module:
http://forum.dev.dji.com/thread-32164-1-1.html

I just tried searching for a OS or library strings which I would recognize, but found nothing. I never looked at any open-source flight controllers, so I wouldn't recognize them - someone else should take a look.

notsolowki, 2016-12-20:
are the components for the flight controller in the ubifs or are they the electronic speed controler 0-4. thankyou

mefistotelis, 2016-12-20:
Very few? I'm seeing 10/16 unencrypted (9/16 really, because video encoder is encrypted in a different way):

$ grep -r -A2 "target" P3X_FW_V01.07.0060-split/*.ini
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi00.ini:target=m0305
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi00.ini-version=34.02.0009
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi00.ini-encrypt_type=1
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi01.ini:target=m0306
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi01.ini-version=02.04.3328
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi01.ini-encrypt_type=0
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi02.ini:target=m0400
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi02.ini-version=01.41.0000
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi02.ini-encrypt_type=1
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi03.ini:target=m1100
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi03.ini-version=01.07.3841
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi03.ini-encrypt_type=0
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi04.ini:target=m1200
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi04.ini-version=01.10.0000
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi04.ini-encrypt_type=1
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi05.ini:target=m1201
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi05.ini-version=01.10.0000
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi05.ini-encrypt_type=1
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi06.ini:target=m1202
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi06.ini-version=01.10.0000
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi06.ini-encrypt_type=1
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi07.ini:target=m1203
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi07.ini-version=01.10.0000
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi07.ini-encrypt_type=1
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi08.ini:target=m1500
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi08.ini-version=01.01.0512
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi08.ini-encrypt_type=0
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi09.ini:target=m1700
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi09.ini-version=01.01.0263
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi09.ini-encrypt_type=0
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi10.ini:target=m1701
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi10.ini-version=01.00.0519
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi10.ini-encrypt_type=0
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi11.ini:target=m1900
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi11.ini-version=01.00.2144
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi11.ini-encrypt_type=0
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi12.ini:target=m0100
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi12.ini-version=01.29.4920
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi12.ini-encrypt_type=0
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi13.ini:target=m0101
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi13.ini-version=01.29.4920
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi13.ini-encrypt_type=0
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi14.ini:target=m0800
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi14.ini-version=00.13.0007
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi14.ini-encrypt_type=0
--
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi15.ini:target=m0900
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi15.ini-version=02.13.0000
P3X_FW_V01.07.0060-split/P3X_FW_V01.07.0060_mi15.ini-encrypt_type=0
$

The module which is encrypted in every other firmware but not encrypted in this one is P3X_FW_V01.07.0060_mi01.

mefistotelis, 2016-12-20:
If you want to disassemble the P3X_FW_V01.07.0060_mi01 binary - it loads to address 0x08020000.

notsolowki, 2016-12-20:
Thank you. i've never emulated a file system or a binary file on ubuntu before. what are my options as far a disassembling the mi01.bin i have not tried to run binwalk or anything against that one in particular. i wouldn't know what to do about the header. or where to start disassembling it. again thank you for all your help

mefistotelis, 2016-12-20:
Disassembling a binary is a general subject, not related to DJI firmware. It is also something which takes a lot of time to master. If you'd like to get into it, search the net.

Personally, I'm using IDA Pro as my disassembler; but it is possible to do even with the standard linux objdump.

notsolowki, 2016-12-20:
this is great thanks hopefully i can make sense of this binary file. i will try to see what i can do with it. maybe one day there will be a whole kit of scripts for modding this copter. i am not good with assembly so i probably wont figure out much

mefistotelis, 2016-12-20:
IDA Pro contains a Hex-Rays plugin, which allows to generate pseudo-C code from each assembly function.

The generated code usually requires a rewrite to be used for compilation, but is good enough to try to understand functions.

notsolowki, 2016-12-20:
i was just trying to run objdump against it and i got architecture unknown. im going to try IDA and check out that plugin. but im sure rewriting the whole thing with all those functions would be so far out of my league. to bad theres no way to get straight to the parts that control the pitch and yaw tat would be priceless.

mefistotelis, 2016-12-20:
If you mean "segment" as from 16-bit architecture - division to such segments is long gone, addresses in 32-bit architectures (both x86 and ARM) are not divided into seg:offs.

If you mean segment of the executable - the firmware is a memory dump of a process, it is not divided into segments. It is a linear dump, created most likely with use of "objcopy -O binary". If a tool asks you explicitly for segment, then it probably assumes the whole binary is one segment - segment 0.

If/when I'll be writing a tool to wrap that binary into ELF header again, I might try to find a method of splitting it back into code segment and data segment; but that's not really needed.

Btw, there is a tool for Ambarella firmware which allows to re-create ELF header around one of Ambarella firmware partitions. There is a partition which is also an ARM exec converted to memory image. The tool is "amba_sys2elf.py", and there is also a test which demonstrates how to get to ELF and back - "tests/test_amba_sys2elf_rebin1.sh".

notsolowki, 2016-12-22:
well, i do not know enough to disassemble the flight controller binary. thats too bad because it would be really cool to change the top speed. i wish more people knew about these tools it might encourage some more development. i know you put in some work. thank for all your help .

notsolowki, 2016-12-23:
have there been any successful flashes with a repacked binary. was everything okay?

mefistotelis, 2016-12-23:
After un-packing and re-packing a firmware without modification, the file is binary identical to the original - so it will update exactly like original.

You can use "test/test_all.sh" to verify whether a specific version produces identical files with extraction and re-packing.

notsolowki, 2016-12-24:
hmm i just wonder if the handheld will still communicate properly.

mefistotelis, 2016-12-24:
Handheld communication:
Why are you worried about that specific component? Did you modified the pilot firmware?

Flashing same version twice:
Even if it will prove to be non-trivial, I'm sure we can solve that. If only you'll be able to provide update logs, we can check the exact message which was logged when the update was cancelled, and check which conditions can override it.

I already have a few ideas which I'd like to test - for example, some of the version checks are skipped if there is a file "P3X_FW_DEBUG" in the root directory of SD card; what's inside this file is irrelevant, its existence sets a flag which causes some of version checking to be disabled.

notsolowki, 2016-12-24:
do you think it would be possible to temporarily change the flight controller settings to make it go faster than 38 mph from a shell connection at the cameras usb port. maybe that would be easier than disassembling the flight controller binary and re-writing the whole thing. i wouldn't want to to always go that fist so it would be great when i'm places where going really fast is safe. then when the drones restarted its back to normal besides shell access of course.. i know some people looking to go back in firmware versions i think from 1.9 to 1.7. i could ask then to place that file in there root sdcard directory and see if its successful if you'd like to test your theory.

mefistotelis, 2016-12-25:
For downgrade with "P3X_FW_DEBUG":
If anyone wants to check it, it would be interesting to see the results.

For extending max speed by a command from camera module:
It's hard to tell if it's possible to increase max speed without modifying the flight controller firmware. It is possible that max speed is treated as some kind of "config value" which can be changed by sending a proper command to the flight controller. But to check whether there is such command, we still have to disassemble the flight controller firmware and get an understanding of where the max speed is written and what affects it.

notsolowki, 2016-12-25:
i asked some people to try it out and report back with some results. thankyou!!@!@!

notsolowki, 2016-12-27:
okay so after someone tried palcing the debug file on their sdcard this is the result.

1.9->1.7 failed. It went alot longer than usual .. it tried! Firmware result only shows this:

========== 2014.01.01 00:00:11 remo-con disconnect======
Packet: P3X_FW_V01.07.0060.bin
Upgrading ...

nothing else. The other log shows this

[00012527]========== remo-con disconnect. boot(15) ============
[00012580]Packet [C:\P3X_FW_V01.07.0060.bin] detected, card sn [0x2984f3d3].
[00012636]Packet upgrade start...

[00012689]Packet checking...
[00012747]Packet vlink 01.07.0060 <-> 01.06.0040.
[00012801]Done.

[00012854]Version checking[1]...
[00012940][03 05][00] v34.2.0.9 -> v34.2.0.9 need upgrade.
[00013040][03 06][00] v2.4.20.18 -> v2.4.13.0 need upgrade.
[00013099][04 00][00] v1.44.0.0 -> v1.41.0.0 need upgrade.
[00013239][11 00][00] v1.8.0.0 -> v1.7.15.1 need upgrade.
[00013354][12 00][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013449][12 01][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013544][12 02][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013644][12 03][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013800][15 00][00] v1.1.2.0 -> v1.1.2.0 need upgrade.
[00026356][17 00][00] device not detected.
[00038905][17 01][00] device not detected.
[00038979][19 00][00] v1.0.8.96 -> v1.0.8.96 need upgrade.
[00039030][01 00][00] v1.30.5036 -> v1.29.4920 need upgrade
[00039083][01 01][00] v1.30.5036 -> v1.29.4920 need upgrade
[00039152][08 00][00] v0.13.0.7 -> v0.13.0.7 need upgrade.
[00039220][09 00][00] v3.0.0.10 -> v2.13.0.0 need upgrade.
[00039275]Done.

[00041329]Waiting for user confirm...
[00051382]Timeout, start upgrade automatically.

[00051489]Firmware upgrading[1]...
[00064110][01 00] Firmware upgrade start...
[00012158][01 00] Firmware upgrade finished successfully.
[00012213]Done.

[00012269]Packet [C:\P3X_FW_V01.07.0060.bin] resumed, card sn [0x2984f3d3].
[00012334]Packet vlink 01.07.0060 <-> 01.06.0040.
[00012393]Version checking[2]...
[00012489][03 05][00] v34.2.0.9 -> v34.2.0.9 need upgrade.
[00012590][03 06][00] v2.4.20.18 -> v2.4.13.0 need upgrade.
[00012647][04 00][00] v1.44.0.0 -> v1.41.0.0 need upgrade.
[00012787][11 00][00] v1.8.0.0 -> v1.7.15.1 need upgrade.
[00012897][12 00][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00012996][12 01][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013236][12 02][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013337][12 03][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013486][15 00][00] v1.1.2.0 -> v1.1.2.0 need upgrade.
[00026045][17 00][00] device not detected.
[00038593][17 01][00] device not detected.
[00038668][19 00][00] v1.0.8.96 -> v1.0.8.96 need upgrade.
[00038727][01 00][05] v1.29.4920 -> v1.29.4920 need upgrade
[00038784][01 01][00] v1.30.5036 -> v1.29.4920 need upgrade
[00038867][08 00][00] v0.13.0.7 -> v0.13.0.7 need upgrade.
[00038949][09 00][00] v3.0.0.10 -> v2.13.0.0 need upgrade.
[00039001]Done.

[00039061]Firmware upgrading[2]...
[00039224][01 01] Firmware upgrade start...
[00069536][01 01] Firmware upgrade finished successfully.
[00070425][08 00] Firmware upgrade start...
[00583850][08 00] Firmware upgrade finished successfully.
[00583916][12 00] Firmware upgrade start...
[00644967][12 00] Firmware upgrade finished successfully.
[00645040][12 01] Firmware upgrade start...
[00704601][12 01] Firmware upgrade finished successfully.
[00704668][12 02] Firmware upgrade start...
[00764219][12 02] Firmware upgrade finished successfully.
[00764291][12 03] Firmware upgrade start...
[00822108][12 03] Firmware upgrade finished successfully.
[00822177][11 00] Firmware upgrade start...
[00870470][11 00] Firmware upgrade finished successfully.
[00870541][03 05] Firmware upgrade start...
[00936315][03 05] Firmware upgrade finished successfully.
[00936566][03 06] Firmware upgrade start...
[01248942][03 06] Firmware upgrade finished successfully.
[01249001][15 00] Firmware upgrade start...
[01287636][15 00] Firmware upgrade finished successfully.
[01290516][19 00] Firmware upgrade start...
[01460598][19 00] Firmware upgrade finished successfully.
[01460678][09 00] Firmware upgrade start...
[01514316][09 00] Firmware upgrade finished successfully.
[01514401][04 00] Firmware upgrade start...
[01571104][04 00] Firmware upgrade finished successfully.
[01571162]Done.

[01571316]Version checking[3]...
[01571406][03 05][05] v34.2.0.9 -> v34.2.0.9 need upgrade.
[01571506][03 06][05] v2.4.13.0 -> v2.4.13.0 need upgrade.
[01571571][04 00][05] v1.41.0.0 -> v1.41.0.0 need upgrade.
[01571707][11 00][05] v1.7.15.1 -> v1.7.15.1 need upgrade.
[01571820][12 00][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
[01571920][12 01][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
[01572053][12 02][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
[01572156][12 03][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
[01572318][15 00][05] v1.1.2.0 -> v1.1.2.0 need upgrade.
[01584872][17 00][00] device not detected.
[01597423][17 01][00] device not detected.
[01597492][19 00][05] v1.0.8.96 -> v1.0.8.96 need upgrade.
[01597549][01 00][05] v1.29.4920 -> v1.29.4920 need upgrade
[01597602][01 01][05] v1.29.4920 -> v1.29.4920 need upgrade
[01597680][08 00][05] v0.13.0.7 -> v0.13.0.7 need upgrade.
[01597759][09 00][05] v2.13.0.0 -> v2.13.0.0 need upgrade.
[01597819]Packet upgrade failed at version checking.

notsolowki, 2016-12-27:
1.9->1.6 also failed. it is getting further than it should without that file though

========== 2014.01.01 00:00:11 remo-con disconnect======
Packet: P3X_FW_V01.06.0040.bin
Upgrading ...

and detaild

[00012646]========== remo-con disconnect. boot(15) ============
[00012697]Packet [C:\P3X_FW_V01.06.0040.bin] detected, card sn [0x2984f3d3].
[00012758]Packet upgrade start...

[00012816]Packet checking...
[00012874]Packet vlink 01.06.0040 <-> 01.05.0030.
[00012932]Done.

[00012990]Version checking[1]...
[00013084][03 05][00] v34.2.0.9 -> v34.2.0.9 need upgrade.
[00013184][03 06][00] v2.4.13.0 -> v2.4.10.7 need upgrade.
[00013251][04 00][00] v1.41.0.0 -> v1.41.0.0 need upgrade.
[00013384][11 00][00] v1.7.15.1 -> v1.7.15.1 need upgrade.
[00013504][12 00][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013613][12 01][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013717][12 02][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013834][12 03][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00014003][15 00][00] v1.1.2.0 -> v1.1.2.0 need upgrade.
[00026563][17 00][00] device not detected.
[00039113][17 01][00] device not detected.
[00039200][19 00][00] v1.0.8.96 -> v1.0.8.96 need upgrade.
[00039368][01 00][00] v1.29.4920 -> v1.26.4315 need upgrade
[00039429][01 01][00] v1.29.4920 -> v1.26.4315 need upgrade
[00039497][08 00][00] v0.13.0.7 -> v0.13.0.7 need upgrade.
[00039581][09 00][00] v2.13.0.0 -> v2.13.0.0 need upgrade.
[00039637]Done.

[00041691]Waiting for user confirm...
[00051754]Timeout, start upgrade automatically.

[00051859]Firmware upgrading[1]...
[00063751][01 00] Firmware upgrade start...
[00012152][01 00] Firmware upgrade finished successfully.
[00012209]Done.

[00012262]Packet [C:\P3X_FW_V01.06.0040.bin] resumed, card sn [0x2984f3d3].
[00012339]Packet vlink 01.06.0040 <-> 01.05.0030.
[00012392]Version checking[2]...
[00012477][03 05][00] v34.2.0.9 -> v34.2.0.9 need upgrade.
[00012574][03 06][00] v2.4.13.0 -> v2.4.10.7 need upgrade.
[00012644][04 00][00] v1.41.0.0 -> v1.41.0.0 need upgrade.
[00012799][11 00][00] v1.7.15.1 -> v1.7.15.1 need upgrade.
[00012882][12 00][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013008][12 01][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013108][12 02][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013253][12 03][00] v1.10.0.0 -> v1.10.0.0 need upgrade.
[00013427][15 00][00] v1.1.2.0 -> v1.1.2.0 need upgrade.
[00025990][17 00][00] device not detected.
[00038542][17 01][00] device not detected.
[00038608][19 00][00] v1.0.8.96 -> v1.0.8.96 need upgrade.
[00038669][01 00][05] v1.26.4315 -> v1.26.4315 need upgrade
[00038722][01 01][00] v1.29.4920 -> v1.26.4315 need upgrade
[00038796][08 00][00] v0.13.0.7 -> v0.13.0.7 need upgrade.
[00038869][09 00][00] v2.13.0.0 -> v2.13.0.0 need upgrade.
[00038921]Done.

[00038979]Firmware upgrading[2]...
[00039140][01 01] Firmware upgrade start...
[00069457][01 01] Firmware upgrade finished successfully.
[00070321][08 00] Firmware upgrade start...
[00578190][08 00] Firmware upgrade finished successfully.
[00578263][12 00] Firmware upgrade start...
[00639789][12 00] Firmware upgrade finished successfully.
[00639860][12 01] Firmware upgrade start...
[00703137][12 01] Firmware upgrade finished successfully.
[00703209][12 02] Firmware upgrade start...
[00764649][12 02] Firmware upgrade finished successfully.
[00764717][12 03] Firmware upgrade start...
[00822845][12 03] Firmware upgrade finished successfully.
[00822910][11 00] Firmware upgrade start...
[00870303][11 00] Firmware upgrade finished successfully.
[00870375][03 05] Firmware upgrade start...
[00933347][03 05] Firmware upgrade finished successfully.
[00933599][03 06] Firmware upgrade start...
[01256124][03 06] Firmware upgrade finished successfully.
[01256184][15 00] Firmware upgrade start...
[01294449][15 00] Firmware upgrade finished successfully.
[01297326][19 00] Firmware upgrade start...
[01467405][19 00] Firmware upgrade finished successfully.
[01467486][09 00] Firmware upgrade start...
[01521139][09 00] Firmware upgrade finished successfully.
[01521225][04 00] Firmware upgrade start...
[01579082][04 00] Firmware upgrade finished successfully.
[01579144]Done.

[01579196]Version checking[3]...
[01579286][03 05][05] v34.2.0.9 -> v34.2.0.9 need upgrade.
[01579391][03 06][05] v2.4.10.7 -> v2.4.10.7 need upgrade.
[01579455][04 00][05] v1.41.0.0 -> v1.41.0.0 need upgrade.
[01579586][11 00][05] v1.7.15.1 -> v1.7.15.1 need upgrade.
[01579700][12 00][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
[01579826][12 01][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
[01579934][12 02][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
[01580035][12 03][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
[01580190][15 00][05] v1.1.2.0 -> v1.1.2.0 need upgrade.
[01592748][17 00][00] device not detected.
[01605304][17 01][00] device not detected.
[01605391][19 00][05] v1.0.8.96 -> v1.0.8.96 need upgrade.
[01605447][01 00][05] v1.26.4315 -> v1.26.4315 need upgrade
[01605509][01 01][05] v1.26.4315 -> v1.26.4315 need upgrade
[01605586][08 00][05] v0.13.0.7 -> v0.13.0.7 need upgrade.
[01605671][09 00][05] v2.13.0.0 -> v2.13.0.0 need upgrade.
[01605726]Packet upgrade failed at version checking.

notsolowki, 2016-12-27:
i found someone on the phantompilots page to test it. i hope thats okay. http://www.phantompilots.com/threads/people-that-want-to-downgrade-their-fw-on-their-phantom-3s.99609/#post-1029159

digdat0, 2016-12-27:
I tested using the DEBUG file on the SD card. This step definitely gets further along in the upgrade process than without. Without, the upgrade fails pretty fast, within 60 seconds. With the P3X_FW_DEBUG file on the root of the SD card the upgrade does attempt and gets pretty far down, but i hit the "Packet upgrade failed at version checking." message noted. I've tried this with 1.7 and 1.6. I'll try a few older ones as well and see what the results are.

mefistotelis, 2016-12-27:
This issue is beginning to be a mix of everything. Let's try to discuss subjects in separate issues in the future.

For the downgrade - are you sure it failed? Because the version numbers suggest it was a success.

alirz1, 2016-12-27:
Yeh you might have more luck in trying to flash a much older FW, e.6 1.3 or something.. Maybe they were more lenient.

notsolowki, 2017-01-05:
okay since ive extracted the m12_rfs ubifs, mounted it, made my changes. can you give me a little insight how to put the modified ubifs back into m12.bin. please and thankyou

notsolowki, 2017-01-05:
here is what i get 

~/Desktop/phantom-firmware-tools-master$ sudo python3 ./amba_romfs.py -vv -a -d p7_mi12_part_rfs.a9s
: Opening for creation
Traceback (most recent call last):
  File "./amba_romfs.py", line 344, in <module>
    main(sys.argv[1:])
  File "./amba_romfs.py", line 333, in main
    fwpartfile = open(po.fwpartfile, "wb")
FileNotFoundError: [Errno 2] No such file or directory: ''

ferraript, 2017-01-21:
@mefistotelis, I've got one question about dji_fwcon.py
when used to extract modules from older firmwares (1.2.8 and older), the firmware numbers in head.ini file have strange values
for example, in P3S_FW_V01.02.0008_head.ini, there is `ver_latest=04.166.39090`
is it normal? couldn't it be some kind of bug in dji_fwcon.py?

mefistotelis, 2017-01-21:
The dji_fwcon.py tool is written based mostly on FW V01.08.0080. If the firmware format is a bit different in older versions, it is possible that fields are interpreted incorrectly.

MrBurnsAT, 2017-01-21:
@ferraript I think thats why this Old Firmware Versions was downgradet By the MVOM0FW.BIN Trick   ;-)

GlovePuppet, 2017-03-17:
is P3X_FW_V01.07.0060 fw available to download anywhere? A quick gargle didn't turn it up and I really want to have a look at the FC binary. I posted elsewhere but the FC is an STM32F417 (at least on the 2 I have examined). That is supported by the "address 0x08020000", STM32F4 flash is at address 0x08000000 and I guess that means the first 128K is bootloader :)

mefistotelis, 2017-03-17:
@GlovePuppet You can find download links to all firmwares in supported_firmwares.csv.

KennethMcNutt, 2017-05-26:
is P3X_FW_V01.07.0060 fw available to download anywhere?
Her are the most relevant F/w for all P3 
https://drive.google.com/drive/folders/0B-0LxjETJ1DnUXFGMGpEVlZ4c1U

mefistotelis, 2017-05-27:
> is P3X_FW_V01.07.0060 fw available to download anywhere?

The link within supported_firmwares.csv should work. Have you tried it?

MAVProxyUser, 2018-01-17:
@mefistotelis "Flight controller firmware is encrypted, and the encryption algorithm isn't known; but - there is one workaround for that; it looks like in firmware P3X_FW_V01.07.0060 a DJI employee had a bad day and forgot to encrypt this component - a moddable version can be taken from this specific firmware". A little bird told me about some forked GitHub repos and a DMCA attempt... you uh... may want to fork this. https://github.com/MAVProxyUser/spray-system/blob/master/app/aes/aes.c#L38

nnup-qert, 2023-11-13:
my phantom 3s drone is connected to the controller but the light on the controller stays red but on the drone the light is blue and I can control the camera up down and start the motor but when I wifi connection and when I go to the dji go app it says the remote control signal is weak, on the phone screen the image and information of the aircraft cannot be connected, it only shows the controller settings. please help me