o-gs/dji-firmware-tools #4: fw_debug

notsolowki, 2016-12-28:
so digdat0 says the aircraft shows the version he tried to drowngrade to. he says now he has version mismatch and compass error. i figured id start this new issue in case he comes back. heres a link to the thread where he posts screen shots but im sure you probably already know about the thread if not here you go http://www.phantompilots.com/threads/people-that-want-to-downgrade-their-fw-on-their-phantom-3s.99609/#post-1029840

mefistotelis, 2016-12-28:
It is likely that one of the modules is not being up/down-graded by this specific firmware package.

Below is a list of target components in every firmware version which I know. You can check which target does not exist in that firmware, but exist in one of the newer upgrades. Then we can repack the firmware to contain all necessary packages for downgrade.

<code>fw/P3X_FW_V01.01.0006.bin: Header:
  'entry_count': 21,
    'target': 'm0305',
    'target': 'm0306',
    'target': 'm0400',
    'target': 'm0900',
    'target': 'm1100',
    'target': 'm1200',
    'target': 'm1201',
    'target': 'm1202',
    'target': 'm1203',
    'target': 'm1300',
    'target': 'm1400',
    'target': 'm1500',
    'target': 'm1600',
    'target': 'm1700',
    'target': 'm1701',
    'target': 'm1900',
    'target': 'm2000',
    'target': 'm2001',
    'target': 'm0100',
    'target': 'm0101',
    'target': 'm0800',
fw/P3X_FW_V01.01.0008.bin: Header:
  'entry_count': 21,
    'target': 'm0305',
    'target': 'm0306',
    'target': 'm0400',
    'target': 'm0900',
    'target': 'm1100',
    'target': 'm1200',
    'target': 'm1201',
    'target': 'm1202',
    'target': 'm1203',
    'target': 'm1300',
    'target': 'm1400',
    'target': 'm1500',
    'target': 'm1600',
    'target': 'm1700',
    'target': 'm1701',
    'target': 'm1900',
    'target': 'm2000',
    'target': 'm2001',
    'target': 'm0100',
    'target': 'm0101',
    'target': 'm0800',
fw/P3X_FW_V01.01.0009.bin: Header:
  'entry_count': 21,
    'target': 'm0305',
    'target': 'm0306',
    'target': 'm0400',
    'target': 'm0900',
    'target': 'm1100',
    'target': 'm1200',
    'target': 'm1201',
    'target': 'm1202',
    'target': 'm1203',
    'target': 'm1300',
    'target': 'm1400',
    'target': 'm1500',
    'target': 'm1600',
    'target': 'm1700',
    'target': 'm1701',
    'target': 'm1900',
    'target': 'm2000',
    'target': 'm2001',
[... 244 lines cut ...]
    'target': 'm1900',
    'target': 'm0100',
    'target': 'm0101',
    'target': 'm0800',
    'target': 'm0900',
fw/P3X_FW_V01.10.0090.bin: Header:
  'entry_count': 17,
    'target': 'm0305',
    'target': 'm0306',
    'target': 'm0400',
    'target': 'm1100',
    'target': 'm1101',
    'target': 'm1200',
    'target': 'm1201',
    'target': 'm1202',
    'target': 'm1203',
    'target': 'm1500',
    'target': 'm1700',
    'target': 'm1701',
    'target': 'm1900',
    'target': 'm0100',
    'target': 'm0101',
    'target': 'm0800',
    'target': 'm0900',
</code>

notsolowki, 2016-12-29:
i have a theory. when he used the p3 debug file he started the flash process and it skipped some of the modules and left their firmware versions intact. but the ones that it did install are now a different version resulting in his firmware mismatch and the errors hes getting. so i wonder if he could reverse the process back up to the point where all the versions match..?!?!? possibly even with the debug file

mefistotelis, 2016-12-29:
Yes, it looks like no module is missing in firmware file (at least in V01.07.0060/V01.08.0080), something just wasn't updated. So the solution which I proposed is not the correct one.

It looks like the communication protocol between components is somehow being updated across firmwares. This may lead to camera module not being able to communicate with the rest of the drone - resulting in bad initialization, and inability to upgrade the firmware again.

It is very likely that this is what happened - the drone reacts to upgrades on the card, but is not able to send the upgrade to any component.

Looks like "long jumps" in firmware versions are dangerous. Also, there may be a version below which the changes in protocol are severe enough to no longer allow camera to communicate with main controller.

What to do now - it's a tricky situation. I'd check the component versions displayed on the tablet, and select firmware package just one step above that version. Then try to update to it, both by the tablet and by the camera, trying to force update both by the special file and by the button on VPS module.

If that won't work:
- Maybe the older camera firmware has different update procedure? If we knew the camera firmware version, we could check if the updating code is somewhat similar.
- We could prepare a firmware package which updates only camera. If no other components are mentioned, maybe the camera will update by itself.
- The camera FW is based on Ambarella SDK, which has its own update mechanism. We could check if that code is not removed by DJI, and maybe use it. It probably involves putting only the extracted camera firmware on SD card under some specific name.
- Maybe when camera is disconnected, updating via tablet will work?
- We could connect the camera to another drone and to the update.

For "version mismatch" - I now think it was most likely a mismatch between the drone and the handheld radio, at lease at first (when the drone was downgraded to V01.07.0060/V01.08.0080). Further downgrades probably made a mismatch between the drone components as well.

notsolowki, 2016-12-29:
#We could prepare a firmware package which updates only camera. If no other components are mentioned, maybe the camera will update by itself.

that sounds like a good idea. i think he mentioned that he still gets the upgrade process indicators from the aircraft as if it was trying to update. so the big question, is m12 the module in the fw package that he would need to make stand alone for the camera?

notsolowki, 2017-01-02:
what is it gona take to make a binary to update his gimbal to an appropriate firmware version?!?

mefistotelis, 2017-01-02:
I don't know of any firmware which would be released by DJI and only have 1-3 modules; this means I did not compared the way my tools are creating such firmware package with any reference - I'm not sure it it is done correctly.

The list of firmwares to "pack" is taken from "*_head.ini". To pack less of them, you have to edit this file and place "0" in place of module names which are supposed to be skipped.

The index of each module has to remain the same. For example, to up/down-grade only camera, we need camera firmware itself (target=m0100) and loader (target=m0101). The modules other than those two have to be changed from "mi??" to just "0". It is important to place the correct amount of "0"s, only replacing each existing "mi??" with one "0".

For example, "P3X_FW_V01.10.0090_head.ini":
<code>modules=mi00 mi01 mi02 mi03 mi04 mi05 mi06 mi07 mi08 mi09 mi10 mi11 mi12 mi13 mi14 mi15 mi16</code>
Ambarella firmware and loader are mi13 and mi14 (we know that by checking "target" inside "P3X_FW_V01.10.0090_mi??.ini"), so we should make:
<code>modules=0 0 0 0 0 0 0 0 0 0 0 0 0 mi13 mi14 0 0</code>

mefistotelis, 2017-01-02:
Btw, the "mi??" index of the camera module was being changed over time. I wonder if this is the reason of "backward incompatibility".  Here is the index in every firmware version:

<code>P3X_FW_V01.01.0006.bin: Module index 18
P3X_FW_V01.01.0008.bin: Module index 18
P3X_FW_V01.01.0009.bin: Module index 18
P3X_FW_V01.01.1003.bin: Module index 18
P3X_FW_V01.01.1007.bin: Module index 20
P3X_FW_V01.02.0006.bin: Module index 20
P3X_FW_V01.03.0020.bin: Module index 20
P3X_FW_V01.04.0005.bin: Module index 13
P3X_FW_V01.04.0010.bin: Module index 13
P3X_FW_V01.05.0030.bin: Module index 12
P3X_FW_V01.06.0040.bin: Module index 12
P3X_FW_V01.07.0043_beta.bin: Module index 12
P3X_FW_V01.07.0060.bin: Module index 12
P3X_FW_V01.08.0080.bin: Module index 12
P3X_FW_V01.09.0060.bin: Module index 13
P3X_FW_V01.10.0090.bin: Module index 13</code>

aka1ceman, 2017-01-02:
How is it then that they allow to downgrade 1 firmware version but not 2?? Sorry, trying to understand.

mefistotelis, 2017-01-02:
The updater code "rollback" checks rollback possibility by the value of min rollback version which is saved on the "black box" sdcard below flight controller. In the extracted firmware, this version is stored as "ver_rollbk" in the "*_head.ini".

For example, version 01.10.0090 cannot be rolled back:
<pre>ver_latest=01.10.0090
ver_rollbk=01.10.0090</pre>

notsolowki, 2017-01-02:
fair enough thankyou for answering that. have you had any look at the amberella sdk to see if it might be possible to create its own firmware update through the sdk? i wonder whats going on when its sitting there looking like its going to update and does nothing, too bad theres no log.

mefistotelis, 2017-01-03:
I can see some file names bound to camera firmwares, but I'm not sure if they're really used, and I don't know if creating such files would do anything (these file names may be used, but only when you provide the drone with full firmware package).

File names are:
PFC300XFw3.bin - for "camera app" firmware, m0100
PCAMLDRFw3.bin - for "camera loader", m0101

notsolowki, 2017-01-03:
have you had any interest getting access to shell on the aircraft. id like to see what i can find out from there

notsolowki, 2017-01-03:
i wonder if there is a file to set a flag at boot to enable debugging on one of the usb ports.

notsolowki, 2017-01-03:
Thats what i get from the serial port on the drone

101 [FDI AHRS[1]]:ah___i+i+ begi+KÎU5h
                                                ÿÿ    102 [FDI AHRS[1]]:bia_ fdi ++_+ -+UU5h
            ÿÿ    102 [FDI AHRS[1]]:i+i+ fdi ++_+ -+Š-U:-
                                                         ÿÿ    102 [FDI AHRS[1]]:+ai+ f-_ _e+_-_ chec++ÙUHW
                           ÿÿ    111 [LED] cha+ged: +e+-e_a++_e +-+ _eady +he+ _+a_++-ûºU:-
           ÿÿ    121 Ba++e_y +a+e    :ATL  NVT  DJ005
ÙU2
    ÿÿ    121 +a++fac++_e Da+e:2016/3/3
Ó)U-ò
     ÿÿ    121 Se_ia+ +++be_   :354
KªU-ò
     ÿÿ+gi+ba+   -i+ch 0, _-++ 0, ya+ 0+-U<Ú
                                            ÿÿdfi++e___c_i+-++__-++, b+++e_+-_+h fi__+ -_de_Õ-U7ù
                 ÿÿd --fc: 10.000000 hz, de+ay: 15.915494 +_¢¶U=
                                                                ÿÿdfi++e___c_i+-++_-i+ch, b+++e_+-_+h fi__+ -_de_SU7ù
                                     ÿÿd --fc: 10.000000 hz, de+ay: 15.915494 +_¢¶U;´
     ÿÿdfi++e___c_i+-++_ya+, b+++e_+-_+h fi__+ -_de_2ÀU7ù
                                                         ÿÿd --fc: 10.000000 hz, de+ay: 15.915494 +_¢¶U@!
                         ÿÿdfi++e___c_i+-++_+h_-+++e, b+++e_+-_+h fi__+ -_de_+IU7ù
  ÿÿd --fc: 10.000000 hz, de+ay: 15.915494 +_¢¶UEÞ
                                                  ÿÿ    155 -+d.IOC _e_e+ i++e++ige+ce_UfJ
          ÿÿ    155 [_+a_+_ba++e_y]+hi_ fi_e+a_e ca+c g-h+e _-eed:7.800000 - +a+d _-eed:2.500000
 %AU7ù
      ÿÿ    155 CTRL _e_e+ a++ by _c +-de _+i+chæU=
                                                   ÿÿ    155 [C+_+<1>] REQ_RC_NORMAL ATTI c+_+_a++ióU0—
                       ÿÿ    161 Ee-_-+ +_i+e -ff_e+:1d0
'|U-ò
     ÿÿ    183 [LED] cha+ged: +- a++iTJU-ò
                                          ÿÿ+gi+ba+      -i+ch 0, _-++ 0, ya+ 0+-UK
   ÿÿ    233 c-+++e_: 0 +i+e-++_c-+++e_0 
 chec+_c-++__+a+e 0 
¦˜U>K
     ÿÿ    233 g-_ i+i+: +ac+:0 ac+:18 fa+_e_f+ag:0
 h/UD
     ÿÿ    243 [FDI AHRS[1]]:(+|yz)+ai+ f-_ _+a+ic +he+ i+i+•9U-ò
                                                                 ÿÿ+gi+ba+      -i+ch 0, _-++ 0, ya+ 0+-U-ò
                           ÿÿ+gi+ba+     -i+ch 0, _-++ 0, ya+ 0+-U-ò
                                                                    ÿÿ+gi+ba+   -i+ch 0, _-++ 0, ya+ 0+-U3Â
                           ÿÿ    450 [FDI AHRS[1]]:ah___i+i+ -a__ÙU;´
                                                                     ÿÿ    451 [FDI AHRS[1]]:e+a++a+i-+ fdi ++_+ -+§ÛU8á
                                        ÿÿ    463 [LED] cha+ged: _c c-+-+e+e+y +-_+yTU0—
        ÿÿ+gi+ba+        -i+ch 0, _-++ 0, ya+ -119h/U0—
                                                       ÿÿ+gi+

notsolowki, 2017-01-03:
0 [ha_dfa+++]:_8(0|406bfaef),_9(0|49cb7355),_10(0|00000000),_11(0|00000000)VKUD
                                                                               ÿÿ      0 [ha_dfa+++]:_3_+_(0|080349a7),-_-(0|2002be48)ÌHUM¨
                                                           ÿÿ      0 [ha_dfa+++]:----------+a_+ "+a_+_c" +cb_i+f-----------ã{UO9
                                                ÿÿ      0 [ha_dfa+++]:S++Ba_eP+_(0|2002d708),S++E+dP+_(0|2002df08)åUK
                                     ÿÿ      0 [hUO9
                                                    ÿÿ      0 [ha_dfa+++]:S++Ba_eP+_(0|2002df08),S++E+dP+_(0|2002e308)1&UK
                                          ÿÿ      0 [ha_dfa+++]:S++P+_(0|2002e20c),Ta_+S+a+e(0|00000000)&ÀU^
                            ÿÿ      0 [ha_dfa+++]:_4(0|080349a6),_5(0|01000000),_6(0|00000000),_7(0|3ea31f9f)üÅU`à
                                  ÿÿ      0 [ha_dfa+++]:_8(0|3a0b32e0),_9(0|413e79e8),_10(0|00000000),_11(0|00000000)Ý˜UD
                                         ÿÿ      0 [ha_dfa+++]:_3_+_(0|080349a7),-_-(0|2002e2a8)âªUM¨
                     ÿÿ      0 [ha_dfa+++]:----------+a_+ "+a_+_b" +cb_i+f-----------¾ÒUO9
          ÿÿ      0 [hUD
                        ÿÿ      0 [ha_dfa+++]:_3_+_(0|080349a7),-_-(0|2002d6a8)ÙƒU_Õ
    ÿÿ      0 [ha_dfa+++]:-----i_-_+_ace i_ ++++,add_(0|400244d0),f+g(0|00000000)-----ºÁU+
          ÿÿ      0 [ha_dfa+++]:----------+a_+_+_ace i_ ++++,add_(0|400244d0),f+g(0|00000000)----------5UX»
                           ÿÿ      0 [ha_dfa+++]:******************chec+ +a_+ +_ace ******************gùU_Õ
                           ÿÿ      0 [ha_dfa+++]:-----i_-_+_ace i_ ++++,add_(0|40024790),f+g(0|00000000)-----+ùU+
                                 ÿÿ      0 [ha_dfa+++]:----------+a_+_+_ace i_ +++U*œ
     ÿÿ      0 _e_e+ by +a+chd-g
ÑU*œ
    ÿÿ      0 _+a_++-:21.428645
Î'U&Ñ
     ÿÿ      0 B-a_d:"++320+2"¯U)É
                                  ÿÿ[> ai_--_+ +i+i+ i+i+ed [1]üšU.§
                                                                    ÿÿ      1 ai_--_+ +i+i+ i+i+ed[1]/{U'
                         ÿÿ[[0][0][1][804][1][10086]¹

mefistotelis, 2017-01-03:
Please make another issue for serial port enabling. This is unrelated to the subject discussed here.

mefistotelis, 2017-01-03:
I looked a bit more at the fw update code, and it looks like placing the PFC300XFw3.bin and PCAMLDRFw3.bin on the sd card will work as good as placing the complete bin package. Maybe even better, because there are no version checks when updating from single binaries.

Every module has its own file name, so it is possible to update all modules by placing individual firmwares on the card.

MrBuensAT, 2017-01-07:
Hi

I will tell u my experience!

I did an downgrade from 1.6 (which was downgradet from 1.7 the oficial way) with the P3X_FW_DEBUG file to 1.3.2 (step by step) without any problems at the end.

Two test flights shows that all gone good.
Max Altitude with RTH trick 806m!

At the first time i tryed the downgrade to 1.5 i did a big mistake.
After about 20min beeping i turned off the Phantom,
because i was thinking it doesnt do anything than beeping.
Caused a Phantom which was only beeping and blinking all colours after turning it on.

But.
I was able to install 1.3 without the DEBUG file.
So i think i caused a compleatly deleted firmware on all components.
Which caused a "hard reset", so i could install 1.5 without the DEBUG file,
what normaly should not be possible because it was downgrade from 1.7 to 1.6 the normal way.
(Maybe i had been able to install every Version there!!??)

After that i looked at the versions in the Go APP.
All perfect 1.5.0030!

Than a try to downgrade to 1.4 without the DEBUG file.
No success. Log file says not possible.
Than with the DEBUG file.
Success!

Than a downgrade to 1.3.2 the normal way without the DEBUG file.

notsolowki, 2017-01-09:
MrBurnsAT,  just out of curiosity was your top speed any higher with the 1.3?

MrBurnsAT, 2017-01-09:
http://www.gerhard-weinberger.at/phpBB2/download/file.php?id=132

notsolowki, 2017-01-09:
thats cool. too bad its not that easy to make it faster. i must ask how did u manage to get that far away with it lol

aka1ceman, 2017-01-10:
MrBurnsAT, Just curious, what version of FW is DJI Go sowing that your Remote has now that you rolled back??

Reason Im asking is usually, the RC will update FW with the Quad keeping both on same (or close to 1 FW away). 
Last year I was able to roll back the Remote multiple FWs but not the quad, so I put it on hold. I rolled back the remote back to 1.1.9 and quad on 1.5 and the 2 would no longer link properly. So I updated the remote back to 1.5 and once again it was ok.

I noticed in the Go app picks Digdat0 posted it showed that his aircraft was on 1.1.9 and Remote was 1.6 (LINK BELOW)
Is it possible that there might not be a camera issue with Digdat0 ? That it may be more that the 2 FWs use different code, etc to where DJI Go is showing a bad camera when it just may not be communicating properly? If I remember correctly from Litchi, DJI made a SDK change somewhere around 1.3 -1.4.

Just hoping to help 
See pick Post #42
http://www.phantompilots.com/threads/people-that-want-to-downgrade-their-fw-on-their-phantom-3s.99609/page-3

aka1ceman, 2017-01-10:
mefistotelis,
To clarify you post above,
are you saying that the P3X_FW_DEBUG works on everything except  01.10.0090?

notsolowki, 2017-01-10:
he was using ver_latest=01.10.0090 ver_rollbk=01.10.0090 as an example of the code within the update script. i think the ps3 debug file skips this check..
because he did say 
"The "P3X_FW_DEBUG" allows updating to any version, without checking if it's already current version or whether it meets rollback restriction."

MrBurnsAT, 2017-01-10:
The newest Firmware which i had was 1.7!!!

Only tested this one day. But After many "No Signal" Problems ich downgradet to 1.6!
(And had to stay there until ich found Ur solution with the P3X_FW_DEBUG last week)

So ist dont know if it works with 1.10!

MrBurnsAT, 2017-01-10:
Maybe they (dji) knows that we knows the Trick. And it will not be long before more people come.

So maybe they changed something in the 1.10 Firmware that the P3X_FW_DEBUG will Not work anymore

MrBurnsAT, 2017-01-10:
Ich tried some things to downgrade the rc. Without success

mefistotelis, 2017-01-10:
About DJI disabling this function:
I first published info about this functionality after 01.10.0090 was out.
Also, I don't think DJI will care - the P3 drones are now no longer produced, and will be supported for maybe a year. It would be nice if the last firmware they release would have all modules unencrypted.

MrBurnsAT, 2017-01-10:
How did u finden out the P3X_FW_DEBUG Trick?

Was there something in the Code of the firmware that u come on this?

MrBurnsAT, 2017-01-10:
I home u know that there always 3 Versions of one Firmware Version. One for Advanced  (P3S), one for Pro  (P3X), and one for Inspire 1 (WM610)

Maybe there is one where they forgott to encrypt

MrBurnsAT, 2017-01-10:
DJI says that 1.10 cant be downgradet

Some people at some groups on fb told that they was Not able to downgrade to 1.9 (the normal way)
But Some was able to go back to 1.9

So i dont know if 1.10 is really Not possible to downgrade

aka1ceman, 2017-01-10:
..yes but DJI also says you can't roll back more than 1 prior to 1.10. ;)

mefistotelis, 2017-01-10:
Source of tricks:
I learned everything I know about the drone by looking at the firmware code. In case of the version check disabling, I looked at function which is located at address E8A6608 in V01.08.0080 firmware.

Other FW versions:
Sooner or later I will make a script which downloads and checks every possible DJI firmware (at least from these which keep the same file format); for now I was focused on P3X.

MrBurnsAT, 2017-01-10:
Firmware Archive UPDATED!!!

For now i think most of all firmwares for the Pro (P3X) are there now.

Added now the Firmware Files for the RC.
Maybe u find a Way to downgade the RC.

At the moment the RC is only updated(and downgradet) via the GoApp
the only possible downgrade is 1.5.8 in the app.

Up to firmware 1.3.2 this was done by an usb stick or via the gimbals usb port (ext. drive)

i know dji go downloads this RC firmware files, renames it, 
like from C1_FW_V01.05.0070.bin (original) to RC_FW_V1.5.70.bin,
and saves it under "Samsung Galaxy S7\Phone\DJI\dji.pilot\Package"
 
from there it will be installed

MrBurnsAT, 2017-01-12:
I saw this code at a thread on phantompilots

MrBurnsAT, 2017-01-12:
What would happen if i would copy only 

P3X_FW_V01.07.0060_mi01.bin and
P3X_FW_V01.07.0060_mi01.ini

on the sd card? would it update???

Maybe with P3X_FW_DEBUG ????

mefistotelis, 2017-01-12:
The name "P3X_FW_V01.07.0060_mi01.bin" is invented for my tool, the drone does not understand it. But if you'd give the firmware a proper name - it would be accepted and used. the INI file is completely unneeded - it's also something specific to my tool.

Proper file names for each module are:
<pre> "PFC300XFw3.bin"	m0100
   "PESC0Fw3.bin"	m1200
   "PESC1Fw3.bin"	m1201
   "PESC2Fw3.bin"	m1202
   "PESC3Fw3.bin"	m1203
  "PMCAPPFw3.bin"	m0306
  "PMCLDRFw3.bin"	m0305
  "PMVOM4Fw3.bin"	m1700
  "PMVOM0Fw3.bin"	m1701
 "PCENTERFw3.bin"	m0500
   "P1765Fw3.bin"	m0900
   "PFPGAFw3.bin"	m1900
  "P68013Fw3.bin"	m1500
  "PDM368Fw3.bin"	m0800
 "PGIMBALFw3.bin"	m0400
"PBATTERYFw3.bin"	m1100
 "PCAMLDRFw3.bin"	m0101
 "PCAMBSTFw3.bin"	m0102
"PCAMBCPUFw3.bin"	m0104
"PCAMLCPUFw3.bin"	m0105
   "PWIFIFw3.bin"	m0700
"PBATTERY2Fw3.bin"	m1101
  "PTEST3Fw3.bin"	m3002
  "PTEST4Fw3.bin"	m3003
  "PTEST5Fw3.bin"	m3004
  "PTEST6Fw3.bin"	m3005
  "PTEST7Fw3.bin"	m3006
  "PTEST8Fw3.bin"	m3007
  "PTEST9Fw3.bin"	m3100
 "PTEST10Fw3.bin"	m3101
 "PTEST11Fw3.bin"	m3102
 "PTEST12Fw3.bin"	m3103</pre>

Note: I would be very careful with updating any of encrypted modules using this method - from my educated guess, this method will probably not decrypt the firmware, but try to use it directly.

MrBurnsAT, 2017-01-12:
But that should be able to solve the Problem of  Digdat0 and his Cam. I think so he should be able to bring his Cam back to a matching firmware.

MrBurnsAT, 2017-01-12:
PS: downgrade forced By P3X_FW_DEBUG also works only Firmware 1.10!!!!

Someone tried it for me
http://www.gerhard-weinberger.at/phpBB2/viewtopic.php?f=5&t=137

MrBurnsAT, 2017-01-13:
MVOM0Fw.bin was an Old downgrade Firmware up to 1.2.6 to be able to downgrade.

A copy of the Old Version can be found in my Firmware Archive

mefistotelis, 2017-01-13:
@MrBurnsAT :
> that should be able to solve the Problem of Digdat0 and his Cam
I agree. He should be able to upgrade his camera with CAMLDRFw3.bin and CAMBSTFw3.bin.

Looks like the "P" which is before each file name in not part of file name. I will edit the post and remove all starting "P"s.

MrBurnsAT, 2017-01-13:
I think the "3" in each filename also isnt Part of the File Name. 

Look at my 7th Post bevore this

mefistotelis, 2017-01-13:
Well I'm not sure about that. It is in the middle of the name. Maybe the "3" was added, and this is why the old "MVOM0Fw.bin" no longer works?

aka1ceman, 2017-01-16:
I was able to downgrade from 1.5, to 1.4 ok. I did download to 1.3.2, but I get an ESC status error. I go back to 1.4 and it goes away so I dont think its a physical issue. Did you MrBurns have this issue??

MrBurnsAT, 2017-01-16:
No i dont have this issue!

What was the newest Firmware which was installed?
How did u downgrade?
Step By Step? 
Or dirctly?

I recommend Step By Step!

notsolowki, 2017-01-16:
have you either of you tried reinstalling the same fw twice?

notsolowki, 2017-01-16:
with the new code thats being created im going to need to reinstall like 100 times lol

MrBurnsAT, 2017-01-16:
Hi 

Someone asked this at Phantompilots maybe u have an answer.

 I also have question for you guys who created those firmware tools
if I
1. unpack modules from FW's bin
2. change ver_rollbk in head.ini to 01.01.0006
3. pack modules back into bin

will I be able to install this modified firmware?
and, most importantly, will I be then able to install any original formware newer than and equal 1.1.6?

MrBurnsAT, 2017-01-16:
And i saw this

    I was searching around in one of the FW bin files with IDA and I saw where it said P3X_FW_DEBUG and I found two other possible file names…P3X_FW_DEBUG_2 and P3X_FW_LOOP…I don’t really know what I’m looking at or if they would do anything but thought it was interesting

notsolowki, 2017-01-16:
i think the ver rollback= file your looking for is on the sdcard right above the battery. ALL credit goes to mefistotelis hes the creator. this guys moving mountains. check out his new scripts. i cant wait!! and can i ask why you want to be on such a low firmware version.

notsolowki, 2017-01-16:
but to answer your question, yes it would probably take the update.

MrBurnsAT, 2017-01-16:
I dont want to go down so far! Only saw the Post on Phantompilots and posten it here.

Im back to 1.3.2, thats enough for me!

mefistotelis, 2017-01-16:
>  if I
1. unpack modules from FW's bin
2. change ver_rollbk in head.ini to 01.01.0006
3. pack modules back into bin
will I be able to install this modified firmware?

Yes. The tools allow you to both extract and re-pack the binaries, and while re-packing they're setting all checksums correctly, so the final firmware is useable. You only can't modify the encrypted files, because the tools have no way of computing their checksum.

notsolowki, 2017-01-16:
so you can only "spoof" the versions of the modules with non-encrypted format?

mefistotelis, 2017-01-16:
You can modify version names in all INI files, that will work even for encrypted firmwares. But please note that version number is also stored within each binary file, so only the updater will be "fooled" by version from INI file.

notsolowki, 2017-01-16:
hey mrburns. im on 1.5+ right now. when i upgrade to 1.7 whats the story with the controller. will it need to be updated from 1.5.7 to 1.6 and if so whats the method of upgrading the controller only?!?

notsolowki, 2017-01-16:
well im installing the firmware now wish me luck lol

notsolowki, 2017-01-16:
it took the update, motors still start.

MrBurnsAT, 2017-01-16:
No und dont must upgedate the rc!

Only firmwares 1.8 and newer needs RC Firmware 1.6.
Because Transmission Protocoll is changed in 1.8 and newer

MrBurnsAT, 2017-01-16:
Notsolowki WoW 

Thats a Huge step forward!!!!

Im thinking about 
Take the best Version of each Module from the firmware Versions and repack a New firmware with that

Maybe 1.4.0010 FC
1.10 Cam and Gimbal
1.3.2 for the esc's 

Would that be possible????

[347 more replies]