o-gs/dji-firmware-tools #5: Enabling Serial Connection

notsolowki, 2017-01-03:
anything you can think of

mefistotelis, 2017-01-03:
>     Thats what i get from the serial port on the drone
>
>101 [FDI AHRS[1]]:ah___i+i+ begi+KÎU5h
>                                                ÿÿ    102 [FDI AHRS[1]]:bia_ fdi ++_+ -+UU5h
>            ÿÿ    102 [FDI AHRS[1]]:i+i+ fdi ++_+ -+Š-U:-
>                                                         ÿÿ    102 [FDI AHRS[1]]:+ai+ f-_ _e+_-_ chec++ÙUHW
>                           ÿÿ    111 [LED] cha+ged: +e+-e_a++_e +-+ _eady +he+ _+a_++-ûºU:-
>           ÿÿ    121 Ba++e_y +a+e    :ATL  NVT  DJ005
>ÙU2
>    ÿÿ    121 +a++fac++_e Da+e:2016/3/3
>Ó)U-ò
>     ÿÿ    121 Se_ia+ +++be_   :354
>KªU-ò
>     ÿÿ+gi+ba+   -i+ch 0, _-++ 0, ya+ 0+-U<Ú
>                                            ÿÿdfi++e___c_i+-++__-++, b+++e_+-_+h fi__+ -_de_Õ-U7ù
>                 ÿÿd --fc: 10.000000 hz, de+ay: 15.915494 +_¢¶U=
>                                                                ÿÿdfi++e___c_i+-++_-i+ch, b+++e_+-_+h fi__+ -_de_SU7ù
>                                     ÿÿd --fc: 10.000000 hz, de+ay: 15.915494 +_¢¶U;´
>     ÿÿdfi++e___c_i+-++_ya+, b+++e_+-_+h fi__+ -_de_2ÀU7ù
>                                                         ÿÿd --fc: 10.000000 hz, de+ay: 15.915494 +_¢¶U@!
>                         ÿÿdfi++e___c_i+-++_+h_-+++e, b+++e_+-_+h fi__+ -_de_+IU7ù
>  ÿÿd --fc: 10.000000 hz, de+ay: 15.915494 +_¢¶UEÞ
>                                                  ÿÿ    155 -+d.IOC _e_e+ i++e++ige+ce_UfJ
>          ÿÿ    155 [_+a_+_ba++e_y]+hi_ fi_e+a_e ca+c g-h+e _-eed:7.800000 - +a+d _-eed:2.500000
> %AU7ù
>      ÿÿ    155 CTRL _e_e+ a++ by _c +-de _+i+chæU=
>                                                   ÿÿ    155 [C+_+<1>] REQ_RC_NORMAL ATTI c+_+_a++ióU0—
>                       ÿÿ    161 Ee-_-+ +_i+e -ff_e+:1d0
>'|U-ò
>     ÿÿ    183 [LED] cha+ged: +- a++iTJU-ò
>                                          ÿÿ+gi+ba+      -i+ch 0, _-++ 0, ya+ 0+-UK
>   ÿÿ    233 c-+++e_: 0 +i+e-++_c-+++e_0 
> chec+_c-++__+a+e 0 
>¦˜U>K
>     ÿÿ    233 g-_ i+i+: +ac+:0 ac+:18 fa+_e_f+ag:0
> h/UD
>     ÿÿ    243 [FDI AHRS[1]]:(+|yz)+ai+ f-_ _+a+ic +he+ i+i+•9U-ò
>                                                                 ÿÿ+gi+ba+      -i+ch 0, _-++ 0, ya+ 0+-U-ò
>                           ÿÿ+gi+ba+     -i+ch 0, _-++ 0, ya+ 0+-U-ò
>                                                                    ÿÿ+gi+ba+   -i+ch 0, _-++ 0, ya+ 0+-U3Â
>                           ÿÿ    450 [FDI AHRS[1]]:ah___i+i+ -a__ÙU;´
>                                                                     ÿÿ    451 [FDI AHRS[1]]:e+a++a+i-+ fdi ++_+ -+§ÛU8á
>                                        ÿÿ    463 [LED] cha+ged: _c c-+-+e+e+y +-_+yTU0—
>        ÿÿ+gi+ba+        -i+ch 0, _-++ 0, ya+ -119h/U0—
>                                                       ÿÿ+gi+

The string you've inserted looks unencrypted, just garbled a bit by not fully matched transmission parameters.

The strings match to messages within the flight controller firmware. Looks like all the messages which go to flight log are also sent to the serial port.

notsolowki, 2017-01-03:
ive tried alot of different parameters only thing i could get any english on was 115200 8n1 what do you suppose i do. i see in the ubifs drivers for a pl2303. i cant send any break commands or any commadn for that matter thought that usb port.

mefistotelis, 2017-01-03:
Maybe there's parity in the transmission. But first we need to gen an ungarbled output, then we may look into input.

Maybe try 7n2 ?

notsolowki, 2017-01-03:
7n2

U4¬
   ÿÿ      0 ee-_-+ +-ad  4  128  132  136ð`U4¬
                                               ÿÿ      0 ee-_-+ +-ad  5  264  128  136‚U4¬
          ÿÿ      0 ee-_-+ +-ad  6  400   12   16¯U4¬
                                                     ÿÿ      0 ee-_-+ +-ad  7  416   40   48·U4¬
                ÿÿ      0 ee-_-+ +-ad  8  464    9   16(KU4¬
                                                            ÿÿ      0 ee-_-+ +-ad  9  480    1    8bõU4¬
                        ÿÿ      0 ee-_-+ +-ad 10  488    6   16ÐãU4¬
                                                                    ÿÿ      0 ee-_-+ +-ad 11  504    4    8%ÝU4¬
                                ÿÿ      0 ee-_-+ +-ad 12  512   52   56¸]U4¬
                                                                            ÿÿ      0 ee-_-+ +-ad 13  568    8   16Œ4¬
                                      ÿÿ      0 ee-_-+ +-ad 14  584    7   16IMU4¬
  ÿÿ      0 ee-_-+ +-ad 15  600   12   16ÜU4¬
                                             ÿÿ      0 ee-_-+ +-ad 16  616   38   48æ¯U4¬
         ÿÿ      0 ee-_-+ +-ad 17  664   12   16„äU4¬
                                                     ÿÿ      0 ee-_-+ +-ad 18  680   56   64<çU4¬
                 ÿÿ      0 ee-_-+ +-ad 19  744   32   40+U4¬
                                                            ÿÿ      0 ee-_-+ +-aìU4¬  784    1    8
    ÿÿ      0 ee-_-+ +-ad 21  792   16   240zU4¬
                                                ÿÿ      0 ee-_-+ +-ad 22  816   16   24ÛýU4¬
            ÿÿ      0 ee-_-+ +-ad 23  840    1    8R}U4¬
                                                        ÿÿ      0 ee-_-+ +-ad 24  848    1    8“U4¬
                   ÿÿ      0 ee-_-+ +-ad 25  856   10   16[U4¬
                                                              ÿÿ      0 ee-_-+ +-ad 26  872   36   40©U4¬
                         ÿÿ      0 ee-_-+ +-ad 27  912   52   56iU4¬
                                                                    ÿÿ      0 ee-_-+ +-ad 28  968   44   48–éU4¬
                                ÿÿ      0 ee-_-+ +-ad 29 1016   28   32âJU4¬
                                                                            ÿÿ      0 ee-_-+ +-ad 30 1048   18   24–¸U4¬
                                        ÿÿ      0 ee-_-+ +-ad 31 1072   16   24(U4¬
   ÿÿ      0 ee-_-+ +-ad 32 1096   72   80$KU4¬
                                               ÿÿ      0 ee-_-+ +-ad 33 1176   24   32ýÀU4¬
           ÿÿ      0 ee-_-+ +-ad 34 1208   64   723SU4¬
                                                       ÿÿ      0 ee-_-+ +-ad 35 1280   24   32U4¬
                 ÿÿ      0 ee-_-+ +-ad 36 1312   96  104ûU8á
                                                             ÿÿdfi++e_-a+g++a__+e+, b+++e_+-_+h 2+d -_de_çŸU7ù
                              ÿÿd --fc: 20.000000 hz, de+ay: 11.252254 +_!–UD
                                                                             ÿÿdfi++e_-a+g++a__acc, +_ac+i+g diffe_e++ia+-_ 2+d -_de_î¿UK
                                                         ÿÿd --fc1: 15.000000 hz, fc2: 30.000000 hz, de+ay: 15.003007 +_ç+U7ù
                                             ÿÿdfi++e_-+i+ea__acc, b+++e_+-_+h 2+d -_de_$U7ù
            ÿÿd --fc: 15.000000 hz, de+ay: 15.003007 +_×½UŠ
                                                           ÿÿd+-+-_ i+i+ c-ef0TUI“
  ÿÿd0 _-++:-0.707107 -i+ch:0.707107 ya+:1.000000 +if+:1.000000­úUI“
                                                                    ÿÿd1 _-++:0.707107 -i+ch:0.707107 yUK
                         ÿÿd3 _-++:-0.707107 -i+ch:-0.707107 ya+:-1.000000 +if+:1.000000IU/c
            ÿÿdc_af+ c-+fig+_a+i-+ +-+-_ c-ef
›+UI“
     ÿÿd0 _-++:-0.707107 -i+ch:0.707107 ya+:0.000000 +if+:1.000000d_UI“
                                                                       ÿÿd1 _-++:0.707107 -i+ch:0.707107 ya+:-0.000000 +if+:1.000000î;UI“
                                                         ÿÿd2 _-++:0.707107 -i+ch:-0.707107 ya+:0.000000 +if+:1.000000{UK
                                         ÿÿd3 _-++:-0.707107 -i+ch:-0.707107 ya+:-0.000000 +if+:1.000000€ŠUHW
                             ÿÿd4 _-++:0.000000 -i+ch:0.000000 ya+:0.000000 +if+:0.000000ÎUHW
             ÿÿd5 _-++:0.00UI“
                              ÿÿd2 _-++:0.353553 -i+ch:-0.353553 ya+:0.000000 +if+:0.2500003JUK
               ÿÿd3 _-++:-0.353553 -i+ch:-0.353553 ya+:-0.000000 +if+:0.250000¢ýUHW
   ÿÿd4 _-++:0.000000 -i+ch:0.000000 ya+:0.000000 +if+:0.000000ÎUHW
                                                                   ÿÿd5 _-++:0.000000 -i+ch:0.000000 ya+:0.000000 +if+:0.000000„ãUHW
                                                    ÿÿd6 _-++:0.000000 -i+ch:0.000000 ya+:0.000000 +if+:0.000000KôUHW
                                     ÿÿd7 _-++:0.000000 -i+ch:0.000000 ya+:0.000000 +if+:0.000000UN
                   ÿÿdc_af+ +-de+ i+f-®ªU2
                                          ÿÿdI|:0.005591 Iy:0.005591 Iz:0.011U5h
ÿÿ      0 +i_ ca+i 55 55 674d fff0 c++:0        ,U#.
                                                    ÿÿ      0 i++ _+a++_:0-U\€
                                                                              ÿÿ      0 [ha_dfa+++]:******************chec+ fa+++ i+f- a+d +_ace ************ì¥U]D
  ÿÿ      0 [ha_dfa+++]:-----fa+++ i_ ++++: add_(0|40024000),f+ag(0|00723b69)-----ÖÃU]D
       ÿÿ      0 [ha_dfa+++]:-----fa+++ i_ ++++: add_(0|400241a0),f+ag(0|da2e493b)-----æÒUb-
[... cut ...]
     ÿÿ    183 [LED] cha+ged: +- a++iTJUD
                                         ÿÿ    198 [FDI AHRS[1]]:(+|yz)+ai+ f-_ _+a+ic +he+ i+i+CU-ò
                    ÿÿ+gi+ba+    -i+ch 0, _-++ 0, ya+ 0+-U:-
                                                            ÿÿ    228 Ba++e_y +a+e    :ATL  NVT  DJ005
U2
  ÿÿ    228 +a++fac++_e Da+e:2016/3/3
ŸU-ò
    ÿÿ    228 Se_ia+ +++be_   :354
0{UK
    ÿÿ    235 c-+++e_: 0 +i+e-++_c-+++e_0 
 chec+_c-++__+a+e 0 
ú*U>K
     ÿÿ    235 g-_ i+i+: +ac+:0 ac+:18 fa+_e_f+ag:0
 ³øU-ò

notsolowki, 2017-01-03:
this is 57600

U4¬
   ÿÿ      0 ee-_-+ +-ad  5  264  128  136‚U4¬
                                              ÿÿ      0 ee-_-+ +-ad  6  400   12   16¯U4¬
         ÿÿ      0 ee-_-+ +-ad  7  416   40   48·U4¬
                                                    ÿÿ      0 ee-_-+ +-ad  8  464    9   16(KU4¬
                ÿÿ      0 ee-_-+ +-ad  9  480    1    8bõU4¬
                                                            ÿÿ      0 ee-_-+ +-ad 10  488    6   16ÐãU4¬
                        ÿÿ      0 ee-_-+ +-ad 11  504    4    8%ÝU4¬
                                                                    ÿÿ      0 ee-_-+ +-ad 12  512   52   56¸]U4¬
                                ÿÿ      0 ee-_-+ +-ad 13  568    8   16Œ4¬
                                                                          ÿÿ      0 ee-_-+ +-ad 14  584    7   16IMU4¬
                                      ÿÿ      0 ee-_-+ +-ad 15  600   12   16ÜU4¬
 ÿÿ      0 ee-_-+ +-ad 16  616   38   48æ¯U4¬
                                             ÿÿ      0 ee-_-+ +-ad 17  664   12   16„äU4¬
         ÿÿ      0 ee-_-+ +-ad 18  680   56   64<çU4¬
                                                     ÿÿ      0 ee-_-+ +-ad 19  744   32   40+U4¬
ìU4¬            ÿÿ      0 ee-_-+ +-ad 20  784    1    8
    ÿÿ      0 ee-_-+ +-ad 21  792   16   240zU4¬
                                                ÿÿ      0 ee-_-+ +-ad 22  816   16   24ÛýU4¬
            ÿÿ      0 ee-_-+ +-ad 23  840    1    8R}U4¬
                                                        ÿÿ      0 ee-_-+ +-ad 24  848    1    8“U4¬
                   ÿÿ      0 ee-_-+ +-ad 25  856   10   16[U4¬
                                                              ÿÿ      0 ee-_-+ +-ad 26  872   36   40©U4¬
                         ÿÿ      0 ee-_-+ +-ad 27  912   52   56iU4¬
                                                                    ÿÿ      0 ee-_-+ +-ad 28  968   44   48–éU4¬
                                ÿÿ      0 ee-_-+ +-ad 29 1016   28   32âJU4¬
                                                                            ÿÿ      0 ee-_-+ +-ad 30 1048   18   24–¸U4¬
                                        ÿÿ      0 ee-_-+ +-ad 31 1072   16   24(U4¬
   ÿÿ      0 ee-_-+ +-ad 32 1096   72   80$KU4¬
                                               ÿÿ      0 ee-_-+ +-ad 33 1176   24   32ýÀU4¬
           ÿÿ      0 ee-_-+ +-ad 34 1208   64   723SU4¬
                                                       ÿÿ      0 ee-_-+ +-ad 35 1280   24   32U4¬
                 ÿÿ      0 ee-_-+ +-ad 36 1312   96  104ûU8á
                                                             ÿÿdfi++e_-a+g++a__+e+, b+++e_+-_+h 2+d -_de_çŸU7ù
                              ÿÿd --fc: 20.000000 hz, de+ay: 11.252254 +_!–UD
                                                                             ÿÿdfi++e_-a+g++a__acc, +_ac+i+g diffe_e++ia+-_ 2+d -_de_î¿UK
                                                         ÿÿd --fc1: 15.000000 hz, fc2: 30.000000 hz, de+ay: 15.003007 +_ç+U7ù
                                             ÿÿdfi++e_-+i+ea__acc, b+++e_+-_+h 2+d -_de_$U7ù
            ÿÿd --fc: 15.000000 hz, de+ay: 15.003007 +_×½UŠ
                                                           ÿÿd+-+-_ i+i+ c-ef0TUI“
  ÿÿd0 _-++:-0.707107 -i+ch:0.707107 ya+:1.000000 +if+:1.000000­úUI“
                                                                    ÿÿd1 _-++:0.707107 -i+ch:0.707107 yUK
                         ÿÿd3 _-++:-0.707107 -i+ch:-0.707107 ya+:-1.000000 +if+:1.000000IU/c
            ÿÿdc_af+ c-+fig+_a+i-+ +-+-_ c-ef
›+UI“
     ÿÿd0 _-++:-0.707107 -i+ch:0.707107 ya+:0.000000 +if+:1.000000d_UI“
                                                                       ÿÿd1 _-++:0.707107 -i+ch:0.707107 ya+:-0.000000 +if+:1.000000î;UI“
                                                         ÿÿd2 _-++:0.707107 -i+ch:-0.707107 ya+:0.000000 +if+:1.000000{UK
                                         ÿÿd3 _-++:-0.707107 -i+ch:-0.707107 ya+:-0.000000 +if+:1.000000€ŠUHW
                             ÿÿd4 _-++:0.000000 -i+ch:0.000000 ya+:0.000000 +if+:0.000000ÎUHW
             ÿÿd5 _-++:0.00UI“
                              ÿÿd2 _-++:0.353553 -i+ch:-0.353553 ya+:0.000000 +if+:0.2500003JUK
               ÿÿd3 _-++:-0.353553 -i+ch:-0.353553 ya+:-0.000000 +if+:0.250000¢ýUHW
   ÿÿd4 _-++:0.000000 -i+ch:0.000000 ya+:0.000000 +if+:0.000000ÎUHW
                                                                   ÿÿd5 _-++:0.000000 -i+ch:0.000000 ya+:0.000000 +if+:0.000000„ãUHW
                                                    ÿÿd6 _-++:0.000000 -i+ch:0.000000 ya+:0.000000 +if+:0.000000KôUHW
                                     ÿÿd7 _-++:0.000000 -i+ch:0.000000 ya+:0.000000 +if+:0.000000UN
                   ÿÿdc_af+ +-de+ i+f-®ªU2
                                          ÿÿdI|:0.005591 Iy:0.005591 Iz:0.011U5h
ÿÿ      0 +i_ ca+i 55 55 674d fff0 c++:0        ,U#.
                                                    ÿÿ      0 i++ _+a++_:0-U\€
                                                                              ÿÿ      0 [ha_dfa+++]:******************chec+ fa+++ i+f- a+d +_ace ************ì¥U]D
  ÿÿ      0 [ha_dfa+++]:-----fa+++ i_ ++++: add_(0|40024000),f+ag(0|00523b69)-----ŒÈU]D
       ÿÿ      0 [ha_dfa+++]:-----fa+++ i_ ++++: add_(0|400241a0),f+ag(0|da2e493b)-----æÒUb-
            ÿÿ      0 [ha_dfa+++]:-----+a_+ i+f-  i_ ++++: add_(0|40024340),f+ag(0|cd8ed971)-----?¯Ucµ
[... cut ...]
                                              ÿÿ    155 [BATTERY]:_e_e+ defa+++ _+a_+ cfg - L1:1  L2:2Ï.UfJ
                           ÿÿ    155 [_+aU7ù
                                            ÿÿ    155 CTRL _e_e+ a++ by _c +-de _+i+chæU=
         ÿÿ    155 [C+_+<1>] REQ_RC_NORMAL ATTI c+_+_a++ióU0—
                                                             ÿÿ    161 Ee-_-+ +_i+e -ff_e+:1d0
'|U-ò
     ÿÿ    183 [LED] cha+ged: +- a++iTJU-ò
                                          ÿÿ+gi+ba+      -i+ch 0, _-++ 0, ya+ 0+-

notsolowki, 2017-01-03:
could i look into the ubifs and check what the settings are for the console. or do you have any way to verify them?

mefistotelis, 2017-01-03:
Most upper case chars (codes 65-90) are OK. Same with numbers (48-57).
Lower case (97-122) - there are issues with k, l, n, o, r, t, u; but no problem with a, b, c, d, e, f, g, h, i, z.
if not the "z", I would say there are issues with ASCII codes above 106.
Example - string:
"[ha_dfa+++]:******************chec+ fa+++ i+f- a+d +ace ************"
Should be:
"[hardfault]:******************check fault info and trace *******"

notsolowki, 2017-01-03:
how did you do that. im using secureCrt. your very good observing the details

notsolowki, 2017-01-03:
so i take it the flight controller has been disassembled?

mefistotelis, 2017-01-03:
You can disassemble it from P3X_FW_V01.07.0060_mi01.bin - the only unencrypted one. It can be converted to ELF by arm_bin2elf.py, the it's easy to open it with IDA Pro.

notsolowki, 2017-01-03:
this si what ig et from putty U0▒
   ▒▒tgimbal     pitch 0, roll 0, yaw -179▒▒U0▒
                                               ▒▒tgimbal         pitch 0, roll 0, yaw -179▒▒U0▒
               ▒▒tgimbal         pitch 0, roll 0, yaw -179▒▒U0▒
                                                               ▒▒tgimbal        pitch 0, roll 0, yaw -179▒▒U0▒
                              ▒▒tgimbal  pitch 0, roll 0, yaw -179▒▒U0▒
                                                                       ▒▒tgimbal pitch 0, roll 0, yaw -179▒▒U0▒
                               ▒▒tgimbal         pitch 0, roll 0, yaw -179▒▒U0▒
                                                                               ▒▒tgimbal         pitch 0, roll 0, yaw -179▒▒U0▒
                                               ▒▒tgimbal         pitch 0, roll 0, yaw -179▒▒U0▒
               ▒▒tgimbal         pitch 0, roll 0, yaw -179▒▒U0▒
                                                               ▒▒tgimbal        pitch 0, roll 0, yaw -179▒▒U0▒
                              ▒▒tgimbal  pitch 0, roll 0, yaw -179▒▒U0▒

notsolowki, 2017-01-03:
▒      0 eeprom load  5  264  128  136▒U4▒
                                             ▒      0 eeprom load  6  400   12   16▒U4▒
       ▒      0 eeprom load  7  416   40   48▒U4▒
                                                 ▒      0 eeprom load  8  464    9   16(KU4▒
            ▒      0 eeprom load  9  480    1    8b▒U4▒
                                                       ▒      0 eeprom load 10  488    6   16▒▒U4▒
                  ▒      0 eeprom load 11  504    4    8%▒U4▒
                                                             ▒      0 eeprom load 12  512   52   56▒]U4▒
                        ▒      0 eeprom load 13  568    8   16▒4▒
                                                                 ▒      0 eeprom load 14  584    7   16IMU4▒
                            ▒      0 eeprom load 15  600   12   16▒U4▒
                                                                      ▒      0 eeprom load 16  616   38   48▒U4▒
                                ▒      0 eeprom load 17  664   12   16▒▒U4▒
                                                                           ▒      0 eeprom load 18  680   56   64<▒U4▒
                                      ▒      0 eeprom load 19  744   32   40lU4▒
                                                                               ▒▒U4▒  0 eeprom load 20  784    1    8
    ▒      0 eeprom load 21  792   16   240zU4▒
                                               ▒      0 eeprom load 22  816   16   24▒▒U4▒
          ▒      0 eeprom load 23  840    1    8R}U4▒
                                                     ▒      0 eeprom load 24  848    1    8▒U4▒
               ▒      0 eeprom load 25  856   10   16[U4▒
                                                         ▒      0 eeprom load 26  872   36   40▒U4▒
                   ▒      0 eeprom load 27  912   52   56iU4▒
                                                             ▒      0 eeprom load 28  968   44   48▒▒U4▒
                        ▒      0 eeprom load 29 1016   28   32▒JU4▒
                                                                   ▒      0 eeprom load 30 1048   18   24▒▒U4▒
                              ▒      0 eeprom load 31 1072   16   24(U4▒
                                                                        ▒      0 eeprom load 32 1096   72   80$KU4▒
                                   ▒      0 eeprom load 33 1176   24   32▒▒U4▒
                                                                              ▒      0 eeprom load 34 1208   64   723SU4▒
                                         ▒      0 eeprom load 35 1280   24   32U4▒
  ▒      0 eeprom load 36 1312   96  104▒U8▒
                                            ▒▒dfitler-angular_vel, butterworth 2nd order▒U7▒
            ▒▒d --fc: 20.000000 hz, delay: 11.252254 ms!▒UD
                                                           ▒▒dfitler-angular_acc, tracking differentiator 2nd order▒UK
                                      ▒▒d --fc1: 15.000000 hz, fc2: 30.000000 hz, delay: 15.003007 ms▒lU7▒
                          ▒▒dfitler-linear_acc, butterworth 2nd order$U7▒
                                                                         ▒▒d --fc: 15.000000 hz, delay: 15.003007 ms׽U▒
                                       ▒▒dmotor init coef0TUI▒
                                                              ▒▒d0 roll:-0.707107 pitch:0.707107 yaw:1.000000 lift:1.000000▒▒UI▒
                                                ▒▒d1 roll:0.707107 pitch:0.707107 yaw:-1.000000 lift:1.000000'▒UI▒
                                  ▒▒d2 roll:0.707107 pitch:-0.707107 yaw:1.000000 lift:1.000000▒▒UK
                   ▒▒d3 roll:-0.707107 pitch:-0.707107 yaw:-1.000000 lift:1.000000IU/c
      ▒▒dcraft configuration motor coef
▒kUI▒
     ▒▒d0 roll:-0.707107 pitch:0.707107 yaw:0.000000 lift:1.000000dsUI▒
                                                                       ▒▒d1 roll:0.707107 pitch:0.707107 yaw:-0.000000 lift:1.000000▒;UI▒
                                                         ▒▒d2 roll:0.707UHW
                                                                           ▒▒d6 roll:0.000000 pitch:0.000000 yaw:0.000000 lift:0.000000K▒UHW
                                                            ▒▒d7 roll:0.000000 pitch:0.000000 yaw:0.000000 lift:0.000000U.▒
                                           ▒▒dcraft configuration ccpm coef
▒QUI▒
     ▒▒d0 roll:-0.353553 pitch:0.353553 yaw:0.000000 lift:0.250000▒OUI▒
                                                                       ▒▒d1 roll:0.353553 pitch:0.353553 yaw:-0.000000 lift:0.250000▒CUI▒
                                                         ▒▒d2 roll:0.353553 pitch:-0.353553 yaw:0.000000 lift:0.2500003JUK
                                          ▒▒d3 roll:-0.353553 pitch:-0.353553 yaw:-0.000000 lift:0.250000▒▒UHW
                              ▒▒d4 roll:0.000000U*▒
                                                   ▒▒iw: 0.000000, lift: 0.000000w▒U,6
      ▒▒iw: 198.975159, lift: 0.356320▒cU,6
                                           ▒▒iw: 385.479095, lift: 1.337347n{U,6
                                                                               ▒▒iw: 561.598694, lift: 2.838538▒▒U,6
                                    ▒▒iw: 728.896790, lift: 4.7816159GU,6
                                                                         ▒▒iw: 888.579285, lift: 7.106159▒▒U5h
                              ▒      0 mis cali 55 55 674d fff0 cnt:0   ,U#.
[... cut ...]
                                                                          query hash fail 6ad4c516|▒UN
                      ▒▒timezone set:-60GAU1S
                                             ▒     21 received on/off 2 from app▒U@!
    ▒     21 CTRL reset all by APP.mission.request.off▒▒U*▒
                                                           ▒     21 ack on/off 0 to app1yU%▒
            ▒     34 ESC0 link up
h▒U%▒
     ▒     35 ESC1 link up
2NU%▒
     ▒     35 ESC2 link up
▒▒U%▒
     ▒     36 ESC3 link up
▒|U-▒

notsolowki, 2017-01-03:
▒      0 [hardfault]:-----task info: addr(0x40024340),flag(0xabcdef02)-----fJ
   ▒      0 [hardfault]:tick_stamp(83596),timebase_stamp(0x00000002bb5f9b97), taskUK
    ▒      0 [hardfault]:StkPtr(0x2002bdac),TaskState(0x00000000)▒▒U^
                                                                     ▒      0 [hardfault]:r4(0x080349a6),r5(0x01000000),r6(0x449f7429),r7(0x448a8b09)▒▒U`▒
                                                                          ▒      0 [hardfault]:r8(0x406bfaef),r9(0x49cb7355),r10(0x00000000),r11(0x00000000)VKUD
                                                                               ▒      0 [hardfault]:r3_lr(0x080349a7),psp(0x2002be48)▒HUM▒
                                                          ▒      0 [hardfault]:----------task "task_c" tcb_info----------▒{UO9
                                              ▒      0 [hardfault]:StkBasePtr(0x2002d708),StkEndPtr(0x2002df08)▒UK
                                  ▒      0 [hUO9
                                                ▒      0 [hardfault]:StkBasePtr(0x2002df08),StkEndPtr(0x2002e308)1&UK
                                     ▒      0 [hardfault]:StkPtr(0x2002e20c),TaskState(0x00000000)&▒U^
                      ▒      0 [hardfault]:r4(0x080349a6),r5(0x01000000),r6(0x00000000),r7(0x3ea216f0)▒U`▒
                          ▒      0 [hardfault]:r8(0x3a0a5103),r9(0x413e79e8),r10(0x00000000),r11(0x00000000)
                            ▒UD
                               ▒      0 [hardfault]:r3_lr(0x080349a7),psp(0x2002e2a8)▒UM▒
         ▒      0 [hardfault]:----------task "task_a" tcb_info----------H!UO9
                                                                             ▒      0 [hU_▒
           ▒      0 [hardfault]:-----irq_trace is null,addr(0x400244d0),flg(0x00000000)-----▒▒Uj
                ▒      0 [hardfault]:----------task_trace is null,addr(0x400244d0),flg(0x00000000)----------5UX▒
                                ▒      0 [hardfault]:******************check last trace ******************g▒U_▒
                               ▒      0 [hardfault]:-----irq_trace is null,addr(0x40024790),flg(0x00000000)-----u▒Uj
                                    ▒      0 [hardfault]:----------task_trace is null,addr(0x40024790),flg(0x00000000)----------▒▒U]D
                                                     ▒      0 [U*▒
                                                                  ▒      0 startup:11.647679
g▒U&▒
     ▒      0 Board:"wm320v2"▒U)▒
                                 ▒▒[> airport limit inited [1]▒U.▒
                                                                  ▒      1 airport limit inited[1]/{U'
                      ▒▒[[0][0][1][804][1][10086]▒
                                                  U▒
                                                    ▒▒[>end of db:[1]▒▒U0▒
                                                                          ▒      1 compass calibration init!8GU;▒
                                 ▒      1 [LED] changed: test led when startupuUB▒
  ▒      1 app connect changed:last(255) != current(0)7▒UHW
                                                           ▒      1 assistant connect changed:last(255) != current(0)XRU8▒
                                          ▒▒dfilter_motor_pwm, butterworth first order▒CU7▒
           ▒▒d --fc: 100.000000 hz, delay: 1.591549 ms▒▒U2
                                                          ▒      1 [FDI MAGN[1]] event:turn on▒U2
                 ▒      1 [FDI GYRO[1]] event:turn on▒▒U1S
                                                          ▒      1 [FDI ACC[1]] event:turn on▒\U2
                 ▒      1 [FDI BARO[1]] event:turn on▒▒U,6
                                                          ▒      1 [FDI AHRS[1]]:turn onB▒U0▒
             ▒      1 [FDI CTRL] event: turn onJ▒U?▒
                                                    ▒      1 temp cali (0.000000,0.000000) 0 fw:4 4
n▒U`▒
     ▒      1 temp cali 0 bw:0.000000 0.000000 0.000000 ba:0.000000 0.000000 0.000000
▒tUF▒
     ▒      1 app temp cali (22.000000,65.000000) aa fw:6 6
'▒Uh▒
     ▒      1 app temp cali aa bw:0.000212 -0.000181 0.000096 ba:-0.000084 0.000284 -0.002604
▒JU%▒
     ▒      1 ESC0 link up
▒|U5h
     ▒      2 [Err] OFDM_TX State:0x00000004y▒U%▒
                                                 ▒      2 ESC1 link up
▒▒U5h
     ▒      3 [Err] OFDM_TX State:0x00000000z▒U%▒
                                                 ▒      3 ESC2 link up
I^U%▒
     ▒      3 ESC3 link up
▒U-▒
    ▒      3 esc alive info = 0xf
▒▒U6=
     ▒      4 Battery barcode:6171160823259
Q▒U]D
     ▒      4 [esc_is_stall] status changed: last(0xffffffff) != current(0x00000000)▒eU]D
         ▒      4 [esc_is_empty] status changed: last(0xffffffff) != current(0x00000000)▒▒UQ
                ▒      4 ESC0 version: Protocol = [V1.0] Hardware = "WM320_ESC_V9" ▒U1S
       ▒      4 Loader   = [V01.00.02.02] *U3▒
                                              ▒      4 Firmware = [V01.10.00.00]
[... 31 lines cut ...]
K▒U-▒
     ▒▒tgimbal   pitch 0, roll 0, yaw 0loU<▒
                                            ▒▒dfilter_rc_input_roll, butterworth first order▒-U7▒
                 ▒▒d --fc: 10.000000 hz, delay: 15.915494 ms▒▒U=
                                                                ▒▒dfilter_rc_input_pitch, butterworth first orderSU7▒
                                     ▒▒d --fc: 10.000000 hz, delay: 15.915494 ms▒▒U;▒
     ▒▒dfilter_rc_input_yaw, butterworth first order2▒U7▒
[... cut ...]
         ▒    155 [smart_battery]this fireware calc gohme speed:7.800000 - land speed:2.500000
 %AU7▒
      ▒    155 CTRL reset all by rc mode switch▒U=
                                                  ▒    155 [Ctrl<1>] REQ_RC_NORMAL ATTI ctrl_atti▒U0▒
                     ▒    162 Eeprom write offset:1d0
&U-▒
    ▒    183 [LED] changed: no attiTJUK
                                       ▒    190 counter: 0 timeout_counter0
 check_comm_state 1
WU>K
    ▒    190 gps init: nack:0 ack:18 false_flag:0
 ▒▒U-▒
      ▒▒tgimbal  pitch 0, roll 0, yaw 0loU-▒

notsolowki, 2017-01-03:
what am i susposed to do about error importing elftools

mefistotelis, 2017-01-03:
Looks like text messages are wrapped into some some kind of binary container which starts with "U".

They are still readable, but it seem likely that since output is binary, the input also needs a specific binary format to be accepted (assuming that there is support for input).

elftools - the error message should explain what to do.

notsolowki, 2017-01-03:
i assume im looking at it right but i still have no idea how to disassemble somthing like this. i wouldnt mind the flight data usb port being turned into a serial port to the flight controller. have you tried anything like this. 

.text:0802149A ; ---------------------------------------------------------------------------
.text:0802149C dword_802149C   DCD 0x2001C4C8          ; DATA XREF: sub_8021122+E6r
.text:0802149C                                         ; sub_80212BE+26r ...
.text:080214A0 aIst8303D0xXIdE DCB "IST8303[%d:0x%x]:ID error %x[exp=%x or %x]",0
.text:080214A0                                         ; DATA XREF: sub_8021122:loc_802121Eo
.text:080214CB                 DCB 0
.text:080214CC aIst8303D0xXCom DCB "IST8303[%d:0x%x]:compass_type = COMPASS_IST8303",0
.text:080214CC                                         ; DATA XREF: sub_8021122+10Eo
.text:080214FC aIst8303D0xXC_0 DCB "IST8303[%d:0x%x]:compass_type = COMPASS_IST8310",0
.text:080214FC                                         ; DATA XREF: sub_8021122+11Ao
.text:0802152C aIst8303D0xXS_2 DCB "IST8303[%d:0x%x]:set mode step 1 failed",0
.text:0802152C                                         ; DATA XREF: sub_8021246+20o
.text:08021554 aIst8303D0xXS_3 DCB "IST8303[%d:0x%x]:set mode step 1 ok",0
.text:08021554                                         ; DATA XREF: sub_8021246:loc_802126Ao
.text:08021578 aIst8303D0xXS_4 DCB "IST8303[%d:0x%x]:set mode step 2 failed",0
.text:08021578                                         ; DATA XREF: sub_8021246+42o
.text:080215A0 aIst8303D0xXS_5 DCB "IST8303[%d:0x%x]:set mode step 2  ok",0
.text:080215A0                                         ; DATA XREF: sub_8021246:loc_802128Co
.text:080215C5                 DCB 0, 0, 0
.text:080215C8 aIst8303D0xXSet DCB "IST8303[%d:0x%x]:set mode step 3  failed",0
.text:080215C8                                         ; DATA XREF: sub_8021246+64o
.text:080215F1                 DCB 0, 0, 0
.text:080215F4 aIst8303D0xXS_0 DCB "IST8303[%d:0x%x]:set mode step 3 ok",0

notsolowki, 2017-01-03:
What is it gona take. i think my overall goal is to make this thing faster

notsolowki, 2017-01-03:
i see from the flight controller the speed settings are just numeric values .

mefistotelis, 2017-01-03:
This no longer seem related to serial port.

For reference - could you describe what you did to achieve the serial transmission in PuTTY?

notsolowki, 2017-01-03:
in putty i jsut used 8n1 no flow controll and 115200. putty had encoding set to utf8 somthing i couldnt change with securecrt.

notsolowki, 2017-01-03:
i still want to enable the serial connection i just dont know where to start. can the components be communicated with through the gimbal. or do you have a know way to communicate with the flight controller through the usb on the front.

notsolowki, 2017-01-03:
id like to add that i notice the drone dont even start booting untill i open the console connection, it sits there and the gimbal makes some noises and then when i hit enter to start the connection the noise stop and it start with all the booting process???!?!?

notsolowki, 2017-01-03:
not sure what to make of this. this is digdat0s log 

U;▒
▒ 0 [BAT]read barcode data success num:1▒rU<▒
▒ 0 [BAT]read begin:12211232 end:12345678▒xU4▒
▒U4▒
▒ 0 eeprom load 0 0 22 32;XU4▒
▒ 0 eeprom load 1 32 28 32▒▒U4▒
▒ 0 eeprom load 2 64 34 40▒\U4▒
▒ 0 eeprom load 3 104 20 24▒U4▒
▒ 0 eeprom load 4 128 132 136▒`U4▒
▒ 0 eeprom load 5 264 128 136▒U4▒
▒ 0 eeprom load 6 400 12 16▒U4▒
▒ 0 eeprom load 7 416 40 48▒U4▒
▒ 0 eeprom load 8 464 9 16(KU4▒
▒ 0 eeprom load 9 480 1 8b▒U4▒
▒ 0 eeprom load 10 488 6 16▒▒U4▒
▒ 0 eeprom load 11 504 4 8%▒U4▒
▒ 0 eeprom load 12 512 52 56▒]U4▒
▒ 0 eeprom load 13 568 8 16▒4▒
▒ 0 eeprom load 14 584 7 16IMU4▒
▒ 0 eeprom load 15 600 12 16▒U4▒
▒ 0 eeprom load 16 616 38 48▒U4▒
▒ 0 eeprom load 17 664 12 16▒▒U4▒
▒ 0 eeprom load 18 680 56 64<▒U4▒
▒ 0 eeprom load 19 744 32 40lU4▒
▒▒U4▒ 0 eeprom load 20 784 1 8
▒ 0 eeprom load 21 792 16 240zU4▒
▒ 0 eeprom load 22 816 16 24▒▒U4▒
▒ 0 eeprom load 23 840 1 8R}U4▒
▒ 0 eeprom load 24 848 1 8▒U4▒
▒ 0 eeprom load 25 856 10 16[U4▒
▒ 0 eeprom load 26 872 36 40▒U4▒
▒ 0 eeprom load 27 912 52 56iU4▒
▒ 0 eeprom load 28 968 44 48▒▒U4▒
▒ 0 eeprom load 29 1016 28 32▒JU4▒
▒ 0 eeprom load 30 1048 18 24▒▒U4▒
▒ 0 eeprom load 31 1072 16 24(U4▒
▒ 0 eeprom load 32 1096 72 80$KU4▒
▒ 0 eeprom load 33 1176 24 32▒▒U4▒
▒ 0 eeprom load 34 1208 64 723SU4▒
▒ 0 eeprom load 35 1280 24 32U4▒
▒ 0 eeprom load 36 1312 96 104▒U8▒
▒▒dfitler-angular_vel, butterworth 2nd order▒U7▒
▒▒d --fc: 20.000000 hz, delay: 11.252254 ms!▒UD
▒▒dfitler-angular_acc, tracking differentiator 2nd order▒UK
▒▒d --fc1: 15.000000 hz, fc2: 30.000000 hz, delay: 15.003007 ms▒lU7▒
▒▒dfitler-linear_acc, butterworth 2nd order$U7▒
▒▒d --fc: 15.000000 hz, delay: 15.003007 ms׽U▒
▒▒dmotor init coef0TUI▒
▒▒d0 roll:-0.707107 pitch:0.707107 yaw:1.000000 lift:1.000000▒▒UI▒
▒▒d1 roll:0.707107 pitch:0.707107 yUI▒
▒▒d2 roll:0.707107 pitch:-0.707107 yaw:1.000000 lift:1.000000▒▒UK
▒▒d3 roll:-0.707107 pitch:-0.707107 yaw:-1.000000 lift:1.000000IU/c
▒▒dcraft configuration motor coef
▒kUI▒
▒▒d0 roll:-0.707107 pitch:0.707107 yaw:0.000000 lift:1.000000dsUI▒
▒▒d1 roll:0.707107 pitch:0.707107 yaw:-0.000000 lift:1.000000▒;UI▒
▒▒d2 roll:0.707107 pitch:-0.707107 yaw:0.000000 lift:1.000000{UK
▒▒d3 roll:-0.707107 pitch:-0.707107 yaw:-0.000000 lift:1.000000▒▒UHW
▒▒d4 roll:0.0UI▒
▒▒d0 roll:-0.353553 pitch:0.353553 yaw:0.000000 lift:0.250000▒OUI▒
▒▒d1 roll:0.353553 pitch:0.353553 yaw:-0.000000 lift:0.250000▒CUI▒
▒▒d2 roll:0.353553 pitch:-0.353553 yaw:0.000000 lift:0.2500003JUK
▒▒d3 roll:-0.353553 pitch:-0.353553 yaw:-0.000000 lift:0.250000▒▒UHW
▒▒d4 roll:0.000000 pitch:0.000000 yaw:0.000000 lift:0.000000▒UHW
▒▒d5 roll:0.000000 pitch:0.000000 yaw:0.000000 lift:0.000000▒▒UHW
▒▒d6 roll:0.000000 pitch:0.000000 yaw:0.000000 lift:0.000000K▒UU,6
▒▒iw: 888.579285, lift: 7.106159▒▒U5h
▒ 0 mis cali 55 55 6503 fff0 cnt:0&nU#.
[... 72 lines cut ...]
▒ 120 Battery name :ATL NVT DJ005
UsU2
▒ 120 manufacture Date:2015/9/6
▒▒U.▒
▒ 120 Serial number :1369
U<▒
▒▒dfilter_rc_input_roll, butterworth first order▒-U7▒
▒▒d --fc: 10.000000 hz, delay: 15.915494 ms▒▒U=
▒▒dfilter_rc_input_pitch, butterworth first orderSU7▒
▒▒d --fc: 10.000000 hz, delay: 15.915494 ms▒▒U;▒
▒▒dfilter_rc_input_yaw, butterworth first order2▒U7▒
▒▒d --fc: 10.000000 hz, delay: 15.915494 ms▒▒U@!
▒▒dfilter_rc_input_throttle, butterworth first ordervIU7▒
▒▒d --fc: 10.000000 hz, delay: 15.915494 ms▒▒UE▒
▒ 155 old.IOC reset intelligence_U7▒
▒ 155 CTRL reset all by rc mode switch▒U=
▒ 155 [Ctrl<1>] REQ_RC_NORMAL ATTI ctrl_atti▒U0▒
▒ 161 Eeprom write offset:1d0
'|U-▒
▒ 183 [LED] changed: no attiTJU;▒
▒ 198 [FDI AHRS[1]]:(mxyz)compass stuck!!!A▒UK
▒ 230 counter: 0 timeout_counter0
check_comm_state 0
▒▒U>K
▒ 230 gps init: nack:0 ack:18 false_flag:0

notsolowki, 2017-01-03:
[image]
[image]
[image]

mefistotelis, 2017-01-07:
Regarding the log:
<code>> invalid flash key![0]</code>
This is harmless. I do not know what the flash key is, but when it's invalid, "airport limit" is disabled - so this seem to just inform that NFZ control is off. Might be interesting for some people, but not the reason of the drone not working.

<code>▒ 116 [Err] OFDM_TX State:0x00000008▒▒U:p</code>
This looks like the reason of the failure.

coptersafe, 2017-03-26:
> not sure what to make of this. this is digdat0s log

>serial log - Click to expand

last four symbols is a CRC , just del them
also for clear output use DJI PC Simulator