o-gs/dji-firmware-tools #6: Flight controller configuration

notsolowki, 2017-01-03:
okay so the big question, now that we have a known non-encrypted flight controller. what are the options as far as changing some angles of the aircraft. like more speed. thanks

notsolowki, 2017-01-03:
minimum id like to change my rth speed for when it runs into 15 mph wind and runs out of battery

notsolowki, 2017-01-03:
can you tell me what this means to you. 

# Startup the system
null::sysinit:/bin/mount -t proc proc /proc
null::sysinit:/bin/mkdir -p /dev/pts
null::sysinit:/bin/mkdir -p /dev/shm
null::sysinit:/bin/mount -a
null::sysinit:/bin/hostname -F /etc/hostname
# now run any rc scripts
::sysinit:/etc/init.d/rcS

# Put a getty on the serial port
ttyS2::respawn:-/bin/sh

# ttyUSB0 on usb host.
# ttyUSB0::respawn:-/sbin/getty  115200 ttyUSB0

# ttyGS0 on usb device
# ttyGS0::respawn:-/bin/sh

# Stuff to do for the 3-finger salute
::ctrlaltdel:/sbin/reboot

# Stuff to do before rebooting
null::shutdown:/etc/init.d/rcK
null::shutdown:/bin/umount -a -r
null::shutdown:/sbin/swapoff -a

whats going to happen if i change
# ttyUSB0 on usb host.
# ttyUSB0::respawn:-/sbin/getty  115200 ttyUSB0  to
# ttyUSB0 on usb host.
 ttyUSB0::respawn:-/sbin/getty  115200 ttyUSB0

do you think it will work if i repack it.

mefistotelis, 2017-01-03:
This is the content of /etc/inittab on the Ambarella Linux partition.

You skipped the header, which exlains what it is:
<pre># Note: BusyBox init doesn't support runlevels.  The runlevels field is
# completely ignored by BusyBox init. If you want runlevels, use
# sysvinit.
#
# Format for each entry: <id>:<runlevels>:<action>:<process>
#
# id        == tty to run on, or empty for /dev/console
# runlevels == ignored
# action    == one of sysinit, respawn, askfirst, wait, and once
# process   == program to run
</pre>

mefistotelis, 2017-01-03:
All Linuxes are made of components; usually for each component there are several alternatives.

One of the components is so-called "init".  It is [explained on wiki](https://en.wikipedia.org/wiki/Init) what it is. Depending on the init used, there are different formats of init scripts which configure the initialization process. The script above comes from a simple init included in [BusyBox package](https://en.wikipedia.org/wiki/BusyBox).

And yes, if you will uncomment these lines, it will most likely work. It will [start getty](https://en.wikipedia.org/wiki/Getty_(Unix)) on the tty (serial terminal) connected to usb.

What I'm not sure about is why are you getting any output on the termnial even without these lines.

notsolowki, 2017-01-03:
i found this in the flight controller binary 

.data:080D4A63 aIwdg_set_swdg_ DCB "iwdg_set_swdg_time_out",0 ; DATA XREF: sub_80222A4+3Eo
.data:080D4A63                                         ; sub_80222F0+2o ...
.data:080D4A7A aIwdg_set_time_ DCB "iwdg_set_time_out",0
.data:080D4A8C aIwdg_set_defau DCB "iwdg_set_default_timeout",0
.data:080D4AA5 aIwdg_set_max_t DCB "iwdg_set_max_timeout",0

im getting that output from the usb on the drone and not the gimbal

notsolowki, 2017-01-03:
i also found this which relates to digdat0's compass error 

.text:0803EEB5                 DCB 0, 0, 0
.text:0803EEB8 aMxyzCompassStu DCB "(mxyz)compass stuck!!!",0
.text:0803EEB8                                         ; DATA XREF: sub_803ECCA:loc_803ECFCo
.text:0803EECF                 DCB 0

notsolowki, 2017-01-03:
knowing some of the variables in the flight controller binary, do you think ill be able to change them through tty on the usb port?? then not being permanent changes don't matter to me

notsolowki, 2017-01-04:
i wonder what dji uses as a recovery method when they get "bricked" or messed up flashes

mefistotelis, 2017-01-04:
How to change the flight controller parameters - this is what we'll be trying to figure out in #7 .
It is evident that there is some kind of interface to these parameters. But how to access it, or is it read-only - I don't know.

MrBurnsAT, 2017-01-11:
Like i sayed, there are always more versions out. Advanced Firmware 1.7.0060 has the same leak as the pro firmware

C:\Program Files (x86)\GnuWin32\bin>grep -r -A2 "target" D:\AdvFW/*.ini
D:\AdvFW/P3S_FW_V01.07.0060_mi00.ini:target=m0305
D:\AdvFW/P3S_FW_V01.07.0060_mi00.ini-version=34.02.0009
D:\AdvFW/P3S_FW_V01.07.0060_mi00.ini-encrypt_type=1
--
D:\AdvFW/P3S_FW_V01.07.0060_mi01.ini:target=m0306
D:\AdvFW/P3S_FW_V01.07.0060_mi01.ini-version=02.04.3328
D:\AdvFW/P3S_FW_V01.07.0060_mi01.ini-encrypt_type=0
--
D:\AdvFW/P3S_FW_V01.07.0060_mi02.ini:target=m0400
D:\AdvFW/P3S_FW_V01.07.0060_mi02.ini-version=01.41.0000
D:\AdvFW/P3S_FW_V01.07.0060_mi02.ini-encrypt_type=1
--
D:\AdvFW/P3S_FW_V01.07.0060_mi03.ini:target=m1100
D:\AdvFW/P3S_FW_V01.07.0060_mi03.ini-version=01.07.3841
D:\AdvFW/P3S_FW_V01.07.0060_mi03.ini-encrypt_type=0
--
D:\AdvFW/P3S_FW_V01.07.0060_mi04.ini:target=m1200
D:\AdvFW/P3S_FW_V01.07.0060_mi04.ini-version=01.10.0000
D:\AdvFW/P3S_FW_V01.07.0060_mi04.ini-encrypt_type=1
--
D:\AdvFW/P3S_FW_V01.07.0060_mi05.ini:target=m1201
D:\AdvFW/P3S_FW_V01.07.0060_mi05.ini-version=01.10.0000
D:\AdvFW/P3S_FW_V01.07.0060_mi05.ini-encrypt_type=1
--
D:\AdvFW/P3S_FW_V01.07.0060_mi06.ini:target=m1202
D:\AdvFW/P3S_FW_V01.07.0060_mi06.ini-version=01.10.0000
D:\AdvFW/P3S_FW_V01.07.0060_mi06.ini-encrypt_type=1
--
D:\AdvFW/P3S_FW_V01.07.0060_mi07.ini:target=m1203
D:\AdvFW/P3S_FW_V01.07.0060_mi07.ini-version=01.10.0000
D:\AdvFW/P3S_FW_V01.07.0060_mi07.ini-encrypt_type=1
--
D:\AdvFW/P3S_FW_V01.07.0060_mi08.ini:target=m1500
D:\AdvFW/P3S_FW_V01.07.0060_mi08.ini-version=01.01.0512
D:\AdvFW/P3S_FW_V01.07.0060_mi08.ini-encrypt_type=0
--
D:\AdvFW/P3S_FW_V01.07.0060_mi09.ini:target=m1700
D:\AdvFW/P3S_FW_V01.07.0060_mi09.ini-version=01.01.0263
D:\AdvFW/P3S_FW_V01.07.0060_mi09.ini-encrypt_type=0
--
D:\AdvFW/P3S_FW_V01.07.0060_mi10.ini:target=m1701
D:\AdvFW/P3S_FW_V01.07.0060_mi10.ini-version=01.00.0519
D:\AdvFW/P3S_FW_V01.07.0060_mi10.ini-encrypt_type=0
--
D:\AdvFW/P3S_FW_V01.07.0060_mi11.ini:target=m1900
D:\AdvFW/P3S_FW_V01.07.0060_mi11.ini-version=01.00.2144
D:\AdvFW/P3S_FW_V01.07.0060_mi11.ini-encrypt_type=0
--
D:\AdvFW/P3S_FW_V01.07.0060_mi12.ini:target=m0100
D:\AdvFW/P3S_FW_V01.07.0060_mi12.ini-version=01.23.4920
D:\AdvFW/P3S_FW_V01.07.0060_mi12.ini-encrypt_type=0
--
D:\AdvFW/P3S_FW_V01.07.0060_mi13.ini:target=m0101
D:\AdvFW/P3S_FW_V01.07.0060_mi13.ini-version=01.23.4920
D:\AdvFW/P3S_FW_V01.07.0060_mi13.ini-encrypt_type=0
--
D:\AdvFW/P3S_FW_V01.07.0060_mi14.ini:target=m0900
D:\AdvFW/P3S_FW_V01.07.0060_mi14.ini-version=02.13.0000
D:\AdvFW/P3S_FW_V01.07.0060_mi14.ini-encrypt_type=0

MrBurnsAT, 2017-01-11:
I tested

P3X_FW_V01.01.0008
P3X_FW_V01.01.0009
P3X_FW_V01.01.1003
P3X_FW_V01.01.1007
P3X_FW_V01.02.0006
P3X_FW_V01.04.0001
P3X_FW_V01.04.0005
P3X_FW_V01.05.0011
P3X_FW_V01.08.0080
P3X_FW_V01.09.0060
P3X_FW_V01.10.0090

P3S_FW_V01.04.0001

WM610_FW_V01.03.00.00
WM610_FW_V01.04.00.10
WM610_FW_V01.05.00.30
WM610_FW_V01.06.00.40
WM610_FW_V01.07.00.90

But only
P3S_FW_V01.07.0060 and P3X_FW_V01.07.0060
has unencrypted main controller firmware

notsolowki, 2017-01-12:
has anyone found out where that annoying startup sound is programmed in at

notsolowki, 2017-01-13:
If you start the drone with the csc command and before the motors do the 2 spins ups you flip the mode switch it skips the spin up and seem to operate fine until you press donw left of the right stick it shuts off.

notsolowki, 2017-01-20:
you should read about the new dji sdk4.0 that just came out. it says something about a bridge app for development on the rc controller. it says the bridge app can be used for debugging and cl access

ferraript, 2017-01-20:
I created Excel file containing all RC- and P3A-firmwares and their modules' version numbers, so it's easy to see when each module was upgraded
is anybody interested?

MrBurnsAT, 2017-01-20:
Yes pls

U can Post it on my hp
http://www.gerhard-weinberger.at/phpBB2/viewforum.php?f=5

aka1ceman, 2017-01-20:
Yes please. 
 I have been collecting firmware for the last two years. It was until recent that I found there was a few I was missing. I have been saving them looking forward to a day like this where we could roll back and shoes which one we want to use.

MrBurnsAT, 2017-01-20:
@aka1ceman

Could u have a look at my HP?

There is my Firmware Archive. 
And could u upload missing Firmware Versions?

That would be genious

ferraript, 2017-01-20:
@mefistotelis: I'd like to upload [xls file](https://github.com/mefistotelis/phantom-firmware-tools/issues/6#issuecomment-274174899) to your repository
any chance of giving me one-time upload access?

@MrBurnsAT: there is supported_firmwares.csv in repository and it contains links too
aren't they working?
or are you looking for another firmware versions?

mefistotelis, 2017-01-20:
> I'd like to upload xls file to your repository

Such things are usually done by "pull requests". Clone the repo, push the change and you'll be given such option.

But I don't want XLS format in the repo; convert it to CSV first.
Also, similar file is already in the repo; it would be better to update existing one.

MrBurnsAT, 2017-01-21:
I ll make a Place where u can get any firmware Version. Not one form there, one from there and so on

Thats why i collect them

aka1ceman, 2017-01-21:
Mr Burns
yes, I have 1.1.1003
But whats funny is when I register on your site, it states UR Banned 

Please note that you must provide a valid e-mail address before your user account is activated. You will receive an e-mail to the specified address, which contains an activation key.

No email yet.
Wow, didnt know you had an issue with me....lol

mefistotelis, 2017-01-21:
The more copies are there, the better.

For the supported_firmwares.csv - please note that there's a script "tests/test_all.sh" in the repo, which can be used to download and try extract all the firmwares in CSV file automatically. It normally tests only selected firmwares, but you may change it to set EXEC_FLAG=0x02 and it will download all which are downloadable.

MrBurnsAT, 2017-01-21:
@aka1ceman which email Provider do u use

Maybe ive Banned it because of some spaming

MrBurnsAT, 2017-01-21:
@mefistotelis ill try that

aka1ceman, 2017-01-21:
gmail....lol

MrBurnsAT, 2017-01-21:
Gmail was blocked

Try again now.

Cant open ur link. Get Virus/Spam warning

mefistotelis, 2017-01-21:
Please note that there is a DL link for P3X_FW_V01.01.1003.bin within supported_firmwares.csv.

ferraript, 2017-01-21:
I uploaded that xls file to my [Google Drive](https://drive.google.com/open?id=0BzjAFKwV8KMYdENKRTNTbHZFVVU)
it contains two sheets, one is for P3A and one for RC firmwares
green color means that it's new version of the module
if somebody of you has P3P firmwares downloaded, maybe you could fill those module versions for P3P firmwares too and share that file again

MrBurnsAT, 2017-01-21:
@ferraript all Pro Firmwares are at my HP

http://www.gerhard-weinberger.at/phpBB2/viewtopic.php?f=5&t=9

notsolowki, 2017-01-25:
honestly im surprised more people are not interested in this.?!!?

ferraript, 2017-01-25:
not everybody needs those hacks
and if somebody wants, maybe he is afraid of damaging the drone, or that hacking is complicated
and as you can see in another thread, people are not even capable of repeating the steps you wrote

notsolowki, 2017-01-25:
but at the same time its like why are they even here then

MrBurnsAT, 2017-01-26:
I think many people would be Interested in this.

But not many people know of this here!
I did about 20 downgrades with different people in the last 2 weeks. 
But no one had known of this here

I have hitherto been restrained to pass this on.
Because I think too many people would make the whole messy.

Should I spread it further?

notsolowki, 2017-01-26:
yea

notsolowki, 2017-01-26:
we need people in here that know what they are doing. seems that no one here knows how or is willing to test the parameters.

MrBurnsAT, 2017-01-26:
Ill do if i have time for it ;-)

But at the moment. Every day about 5 people ask me 
if i can help them with downgrading their Firmwares.

And ill give them step By Step Instructions (live)
Because if something goes wrong, i can help emediatly 

But i will really test Ur Firmware 
And i will do that. 

Ok ill spread it further.

notsolowki, 2017-01-26:
@MrBurnsAT : i have a huge favor to ask you... is there any way you can unplug your ofdm board from the gimbal and tell my if it still yaws. im having issue with the motor on the very top motor on the gimabl. but i think some wire shorted but idk what fried yet.

MrBurnsAT, 2017-01-26:
Ill have a look After work.

notsolowki, 2017-01-26:
that would be awesome thankyou!!!!!!

notsolowki, 2017-01-26:
and let me know if you have a circuit board from a gimbal for sale thanks

notsolowki, 2017-01-26:
i bought digdats bricked gimbal hopefully it fixes my problem lol

MrBurnsAT, 2017-01-26:
I think the way i did the Recover of the Phantom 
Would work for this Gimbal too

MrBurnsAT, 2017-01-26:
Sorry had a look but dont find a way to check if its the esc or Gimbal Motor

notsolowki, 2017-01-26:
Its a shot in the dark but you never know

ferraript, 2017-01-29:
Android users still can't change aircraft's name
and in mi01.bin there is text "choose a name for your inspire"
@mefistotelis, any chance of finding out what needs to be done so AC will prompt for its name?

notsolowki, 2017-01-30:
what exactly is the parameter that your talking about?

notsolowki, 2017-01-30:
if your talking about a parameter thats not in flyc_param your probably going to learn how to byte patch

ferraript, 2017-01-31:
I found a place in the app, where naming the AC is executed:
dji\pilot2\usercenter\a\h.java
there is method e(), that reads name from EditText and sends it to the AC via class DataFlycSetPlaneName

but I haven't found an easy way to execute it "manually"

notsolowki, 2017-01-31:
@mefistotelis : what could be soo much different about the 1.6 developer firmware? i wonder if the parameters already have a higher limit in this firmware

mefistotelis, 2017-02-01:
@ferraript The most versatile way would be to create custom mobile app for this too. Another way would be to modify and re-compile a single .class file to do the change - but I never tried such java hacks and don't know a best way to make it happen.

@notsolowki no idea what the question is referring to.

notsolowki, 2017-02-01:
i mean why did they even make a developer firmware

mefistotelis, 2017-02-01:
While usually "developer version" of a software has some additional debugging code, I don't think it is the case with DJI. I'm pretty sure they call "developer version" something which other companies would call "release canditate version" - a version which needs broader testing, but is planned to become another release (if the tests will not reveal major issues).

notsolowki, 2017-02-01:
I found alot of flies that relate to parameters and setup of the aircraft in the apk. If you look in , dji/midware/data/model/p3. I even see a flag to enable disable debughing

notsolowki, 2017-02-01:
I cant find exactly what file ot was in. But i cann tell you ot clearly said , debug enable = flase

notsolowki, 2017-02-01:
I even seen some things realted to output power of the remote. I dont know java so i cant tell you any specifics but it had somthing to do with sending and setting parameters. There must be a thousand files for tge parameters i ntgat location.  I followed the files that were being imported and found all the settings
 Then i passed out from sleep deprevation and forgot lol

ferraript, 2017-02-06:
@mefistotelis , I don't know where to ask, so I chose this general FW topic
could you please give me some advice how to correctly import modules' bin files into IDA and do some research and changes just like you do?
for example, how did you find [those filenames](https://github.com/mefistotelis/phantom-firmware-tools/issues/4#issuecomment-272307530)?
or, what do I have to do in IDA to change just one flyc param? (for example I noticed that when I changed just one number in those params through your dji_flyc_param_ed.py, resulting bin file differs in 3 bytes from original version)

mefistotelis, 2017-02-06:
> how to correctly import modules' bin files into IDA

Convert BIN to ELF using one of example commands in the README. Then open the ELF in IDA.
Symbols:
If you have .MAP file for the executable - use ".loadmap" plug-in in IDA.
If you have .IDC script - disable "analysis" when loading the ELF and then use "execute IDC" command; re-enable "analysis" after this it's loaded.

> how to [..] do some research and changes

Check any IDA tutorial. Learn how to define structs, modify properties of functions, declare variables, use hex-rays. I have years of practice in IDA - it's not an easy path, but it is rewarding.

> how did you find those filenames?

Defined a struct, then exported it.

> what do I have to do in IDA to change just one flyc param

Define a struct, apply it to the params area, figure out what each parameter mean. I've notices the values are written in 3 variants, so updating 3 places.

ferraript, 2017-02-06:
well, I had suspected that BIN needs to be converted to ELF
but I didn't succeed
I ran that arm_bin2elf.py script, it required pyelftools
so I downloaded and installed it, but the script still ends with the same ImportError

mefistotelis, 2017-02-06:
> maybe I didn't correctly understand this request "clone to upper level folder, '../pyelftools'"

Yes, I'm pretty sure that's the cause. Do as the message says.

[36 more replies]