o-gs/dji-firmware-tools #7: Flight controller config

mefistotelis, 2017-01-03:
The flight controller contains a list of parameters, probably used in some kind of external API.

This issue is to discuss that API, and maybe to figure out how to use it.

Each of the parameters has a string name, and address of a variable where it is stored. It also has additional properties, not all known.

notsolowki, 2017-01-03:
i found this in the flight controller binary

.data:080D4A63 aIwdg_set_swdg_ DCB "iwdg_set_swdg_time_out",0 ; DATA XREF: sub_80222A4+3E�o
.data:080D4A63 ; sub_80222F0+2�o ...
.data:080D4A7A aIwdg_set_time_ DCB "iwdg_set_time_out",0
.data:080D4A8C aIwdg_set_defau DCB "iwdg_set_default_timeout",0
.data:080D4AA5 aIwdg_set_max_t DCB "iwdg_set_max_timeout",0

maybe something that relates downgrading

mefistotelis, 2017-01-03:
I was actually referring to these variables:
<pre>AttiData_200Hz.press_0
IMU_Data.gyro_tempX_0
IMU_Data.gyro_tempY_0
IMU_Data.gyro_tempZ_0
cfg_var_table_size_0
config_parameter.assistant_timeout_0
config_parameter.motor_stop_lock_0
default_gps_svn_0
fix_send[0].send_frequency_0
fix_send[0].send_size_0
fix_send[0].send_start_index_0
...
fix_send[3].send_frequency_0
fix_send[3].send_size_0
fix_send[3].send_start_index_0
fix_send_index[ 0]_0
...
fix_send_index[47]_0
g_cfg_debug.force_flash_led_0
g_cfg_debug.simulator_check_motor_0
g_cfg_debug.simulator_gps_hdop_0
g_cfg_debug.simulator_initial_lati_0
g_cfg_debug.simulator_initial_longti_0
g_cfg_debug.simulator_mvo_flag_0
g_config.advanced_function.avoid_ground_enabled_0
g_config.advanced_function.fail_safe_protection_enabled_0
g_config.advanced_function.height_limit_enabled_0
g_config.advanced_function.intelligence_gear_enabled_0
g_config.advanced_function.intelligence_orientation_enabled_0
g_config.advanced_function.one_key_go_home_enabled_0
g_config.advanced_function.pantilt_enabled_0
g_config.advanced_function.radius_limit_enabled_0
g_config.advanced_function.voltage_protection_enabled_0
g_config.aircraft.aircraft_type_0
g_config.aircraft.custom_mix[0][0]_0
g_config.aircraft.custom_mix[0][1]_0
g_config.aircraft.custom_mix[0][2]_0
g_config.aircraft.custom_mix[0][3]_0
...
g_config.aircraft.custom_mix[7][0]_0
g_config.aircraft.custom_mix[7][1]_0
g_config.aircraft.custom_mix[7][2]_0
g_config.aircraft.custom_mix[7][3]_0
g_config.aircraft.custom_num_of_motor_0
g_config.airport_limit_cfg.cfg_angle[FLY_LIMIT_TYPE_AIRPORT]_0
g_config.airport_limit_cfg.cfg_angle[FLY_LIMIT_TYPE_SPECIAL]_0
g_config.airport_limit_cfg.cfg_debug_airport_enable_0
g_config.airport_limit_cfg.cfg_disable_airport_fly_limit_0
g_config.airport_limit_cfg.cfg_enable[FLY_LIMIT_TYPE_AIRPORT]_0
g_config.airport_limit_cfg.cfg_enable[FLY_LIMIT_TYPE_SPECIAL]_0
g_config.airport_limit_cfg.cfg_h1[FLY_LIMIT_TYPE_AIRPORT]_0
g_config.airport_limit_cfg.cfg_h1[FLY_LIMIT_TYPE_SPECIAL]_0
g_config.airport_limit_cfg.cfg_limit_data_0
g_config.airport_limit_cfg.cfg_r1[FLY_LIMIT_TYPE_AIRPORT]_0
g_config.airport_limit_cfg.cfg_r1[FLY_LIMIT_TYPE_SPECIAL]_0
g_config.airport_limit_cfg.cfg_search_radius_0
g_config.airport_limit_cfg.cfg_sim_disable_limit_0
g_config.api_entry_cfg.abs_value_0
g_config.api_entry_cfg.acc_data_type_0
g_config.api_entry_cfg.alti_data_type_0
g_config.api_entry_cfg.api_authority_group[0]_0
g_config.api_entry_cfg.api_authority_group[1]_0
g_config.api_entry_cfg.api_authority_group[2]_0
g_config.api_entry_cfg.api_authority_group[3]_0
g_config.api_entry_cfg.authority_level_0
g_config.api_entry_cfg.baud_rate_0
g_config.api_entry_cfg.channel_0
g_config.api_entry_cfg.cheat_backdoor_0
g_config.api_entry_cfg.comp_sign_0
[... 547 lines cut ...]
range.control.control_mode[2]
range.go_home.standby
range.go_home.start
range.ioc.mode[0]
range.ioc.mode[1]
range.ioc.mode[2]
run_mode_0
standard_range_max_0
standard_range_min_0
test_deformation_0
wall_dis_back_0
wall_dis_front_0
wall_dis_left_0
wall_dis_right_0
wind_speed[0]_0
wind_speed[1]_0
wind_speed[2]_0
wp_setting.action_on_finish_0
wp_setting.action_on_rc_lost_0
wp_setting.idle_vel_0
wp_setting.length_0
wp_setting.mission_exec_num_0
wp_setting.trace_mode_0
wp_setting.vel_cmd_range_0
wp_setting.yaw_mode_0</pre>

notsolowki, 2017-01-03:
im not sure if it mattered or not but i have had p3x_fw_debug on the sdcard while i was connected.

notsolowki, 2017-01-03:
where did you find those variables?!?

notsolowki, 2017-01-03:
i seen them all in the flight controller binary but i can barley make sense of what going on. disassembly is not something im real good at.

mefistotelis, 2017-01-03:
In firmware V01.07.0060, the names start at .data:080D7B22; the names are then used in a huge struct array that is at .data:080A6900.  The struct item looks like this:
<pre>struc_unkn05    struc ; (sizeof=0x3C)
00000000 field_0         DCD ?                   ; offset (00000000)
00000004 field_4         DCD ?                   ; offset (00000000)
00000008 field_8         DCD 3 dup(?)
00000014 field_14        DCFS 2 dup(?)
0000001C field_1C        DCD 7 dup(?)
00000038 field_38        DCD ?                   ; offset (00000000)
struc_unkn05    ends
</pre>

notsolowki, 2017-01-03:
i wish i knew what that meant? i cant even begin to visualize this.,

notsolowki, 2017-01-04:
i wonder if these parameters are accessible from the gimbals usb port, assuming that enabling getty on USB0 works.

mefistotelis, 2017-01-04:
We dont know how, or from where these can be accessed.

To get further, we have two ways:
- investigate all the functions which reference array at .data:080A6900, understanding what they do and and where do they come from. Then either set new values from outside, or modify the function which sets initial values to have them modified from inside the firmware.
- try searching for some kind of SDK which is distributed by DJI. I think I've heard about DJI having mobile SDK for their drones; if this is the case, maybe the SDK is able to use (set/get) these parameters.

For me this is not a priority, what I want to achieve is somewhere else; maybe I'll get to it later.

notsolowki, 2017-01-04:
i read somewhere where someone modified the android app to remove some limitations but they didnt stick

notsolowki, 2017-01-04:
can i ask what would you like to achieve?

notsolowki, 2017-01-05:
why do i see a bunch of this           % 1
.bss:08170A8F                 % 1
.bss:08170A90                 % 1
.bss:08170A91                 % 1
.bss:08170A92                 % 1
.bss:08170A93                 % 1
.bss:08170A94                 % 1
.bss:08170A95                 % 1
.bss:08170A96                 % 1
.bss:08170A97                 % 1
.bss:08170A98                 % 1
.bss:08170A99                 % 1
.bss:08170A9A                 % 1
.bss:08170A9B                 % 1
.bss:08170A9C                 % 1
.bss:08170A9D                 % 1
.bss:08170A9E                 % 1
.bss:08170A9F                 % 1
.bss:08170AA0                 % 1
.bss:08170AA1                 % 1
.bss:08170AA2                 % 1
.bss:08170AA3                 % 1
.bss:08170AA4                 % 1
.bss:08170AA5                 % 1

mefistotelis, 2017-01-05:
The ".bss" segment of ELF executable contains uninitialized data. This is how IDA displays such data.

MrBuensAT, 2017-01-08:
g_config.advanced_function.height_limit_enabled_0

I think this would be Interesting for me ;-)

mefistotelis, 2017-01-08:
It was suggested [here](https://github.com/mefistotelis/phantom-firmware-tools/issues/8#issuecomment-270605415) that the API to change values of these parameters might be in [Guidance SDK](https://github.com/dji-sdk/Guidance-SDK).

mefistotelis, 2017-01-08:
I just checked these SDKs; Guidance SDK has no use example, but I found interesting code in one of the demos of Monile SDK (so code is in Java):
<pre>  public void setGohomeAltitude(float paramFloat, DJIExecuteBooleanResultCallback paramDJIExecuteBooleanResultCallback)
  {
    if (!checkLevel1IsValid()) {
      paramDJIExecuteBooleanResultCallback.onResult(false);
    }
    for (;;)
    {
      return;
      if (!DataOsdGetPushCheckStatus.getInstance().a())
      {
        paramDJIExecuteBooleanResultCallback.onResult(false);
      }
      else
      {
        a locala = DJIFlycParamInfoManager.read("g_config.go_home.fixed_go_home_altitude_0");
        if (locala.a(Float.valueOf(paramFloat))) {
          new DataFlycSetParams().a(locala.i, Float.valueOf(paramFloat)).a(new DJIInspireMainController.8(this, paramDJIExecuteBooleanResultCallback));
        } else {
          paramDJIExecuteBooleanResultCallback.onResult(false);
        }
      }
    }
  }
</pre>

MrBuensAT, 2017-01-08:
I think thats the Point in the Firmware where the Return to Home Height which can be choosen in the GoApp or Others.

I speak of the maximum height of 500m
Which is limited By the firmware. 

Up to Firmware 1.4.0010 
U could use a Trick to go higher
Only click RTH at thr RC nit the Apo and give full Throttle than it will go up higher than 500m

Since firmware 1.5 thats compleately blocked.

MrBuensAT, 2017-01-08:
What about other/okder SDK Versions? Are there any online? Did u try Firmware Version 1.4.0010(official) if there is something unencrypted?

Maybe in Version 1.4.0003 (developer)

MrBuensAT, 2017-01-08:
Or 1.5.0011 (developer)

mefistotelis, 2017-01-08:
I don't have much of beta/developer firmwares. The full list of firmwares I have is at top of "tests/test_all.sh".
If you have any versions not mentioned there, I'll be happy to get them.

I analysed the content of all firmwares which I have. There were no other FWs with unencrypted flight controller, or other normally encrypted module.

Thank you for explaining the trick - it is interesting.

It looks like many parameters can be modified by the flyc_params - looks like this is how DJI calls these parameters, as a file containing its list and information is named "flyc_param_infos". But there are limits, which would have to be removed, and very likely the only way to do this is to modify the flight controller FW.

So I'd say we will need two separate sub-projects:
1. Mobile app which allows to list and change all the parameters (would be best if it had the ability to save/load presets with custom sets of parameters).
2. Flight controller FW patch (currently ossible only for the one unencrypted firmware we have)

MrBuensAT, 2017-01-09:
Here is my little firmware archive

http://www.gerhard-weinberger.at/phpBB2/viewtopic.php?f=5&t=9

mefistotelis, 2017-01-09:
Thanks @MrBurnsAT , I'll check these.

notsolowki, 2017-01-10:
now that theres the possibility of lifting the restrictions in the android app. what is the next step to modifying the binary module to accept the new values.

notsolowki, 2017-01-11:
What program can i use to try to modify the binary.

mefistotelis, 2017-01-11:
I'm usually just using hex editor.

Also, note that you can edit bytes in IDA hex view, to test if the struct looks correctly. But I don't know whether you can "export" the modified ELF from IDA.

notsolowki, 2017-01-11:
When are you going to make a modified flight controller binary for us lol. I wish i knew how to do it. Should i edit the elf or the original bin file. I was look with hex edit but i didnt know how to look at the structure

mefistotelis, 2017-01-11:
It's up to you how you do it. You can modify ELF, and then generate BIN from it using objcopy.

See tests/test_amba_sys2elf_rebin1.sh for an example how to make BIN from ELF.

notsolowki, 2017-01-11:
Okay the elf to bin is 50% of what i needed. The other part is i dont know how to edit a value in the structure in hex editor. When i open it with hex edit i dont see much. How could i locate for example .data 080A670

notsolowki, 2017-01-11:
lets say i was looking for the pitch gain parameters. on the app the value is between 80 and 120. lets say for example we have this ".
.data:080A6900                 ExportParam <aG_config_co_30, unk_200011E8, 2, 5, <80.0, 120.0, 100.0>,\
.data:080A6900                              <80, 120, 100>, <80, 120, 100>, 0xB, 0>

what would this mean to you.

mefistotelis, 2017-01-11:
Computing position:
.data:080A6900 - this is the address which the structure would have if it was loaded into memory of the drone. This address consists of offset in file, and executable base address:
position_in_bin_file = address_in_memory - base_address
How to get base address - it is the address where file data starts; so - in IDA, go to the very first thing visible in the view and look at its address - this is base address.
Now you can open the bin file in hex editor and apply the changes at the correct position.

Note that you can also use "edit" option in the hex view within IDA Pro (right click on hex view to see the option), and then "export" the modified binary from IDA using a plugin (search the net for details).

Struct content:
aG_config_co_30 - this is a label assigned to an address where string which stores property name is.
unk_200011E8 - this is the place in drone RAM where the property value will be stored.
 2, 5, - some attributes, compare it to parameters in "flyc_param_infos" to understand what they are,
<80.0, 120.0, 100.0> - min, max, default value in case float values are used
<80, 120, 100>, - min, max, default value in case integer values are used
 <80, 120, 100> - min, max, default value in case integer values are used (probably a bit different integers than the previous ones, maybe a matter of sign or amount of bits per value)
0xB, 0 - some attributes, again compare to "flyc_param_infos"

notsolowki, 2017-01-13:
Could these be the parameters i have been looking for to add some more pitch? i notice each of there values are 100

.text:08078F98 sub_8078DAE                 ADR             R1, aDRollFPitchFYa ; "%d roll:%f pitch:%f yaw:%f lift:%f"

.text:08078D90 sub_8078C6C                 ADR             R1, aDRollFPitchFYa ; "%d roll:%f pitch:%f yaw:%f lift:%f"

.text:080830E6 sub_8082FDC                 ADR             R1, aDRollFPitchF_0 ; "%d roll:%f pitch:%f yaw:%f lift:%f"

.text:08083094 sub_8082FDC                 ADR             R1, aDRollFPitchF_0 ; "%d roll:%f pitch:%f yaw:%f lift:%f"

notsolowki, 2017-01-13:
i Also was wondering about this,

mefistotelis, 2017-01-13:
If you want to know what things do, then check the code.

And for the code which you provided - I usually search google specific instructions like "arm <instruction>:, ie. "arm adds".
Besides specific instructions, you may also want to read about calling conventions.

notsolowki, 2017-01-16:
have you or anyone found descriptions for the "flyc_param" parameters?

mefistotelis, 2017-01-16:
I don't think we'll find any; it does not look like any official API.

MrBurnsAT, 2017-01-16:
Maybe it would help look on older DJI flight Controllers like Naza, Wookong or A2.

Many of the Parameters in flyc_param i know from my wookong.

So maybe we can get some Informations from there.

mefistotelis, 2017-01-16:
Yes, that should be a good source of information.

A few parameters can also be described after checking them in DJI Panorama Demo and Dji Go.

mefistotelis, 2017-01-19:
I imagine an app which you can start before Dji Go and load a preset containing settings which you previously set at home. After the settings are applied to the drone, you can close it and use either DJI Go or Litchi to do the flying.

The app would show a list of categorized settings, allowing to change each of these and applying the changes to the drone instantly. Application should keep track of which parameters were changed and allow to save a couple of modifications under custom name as preset, ie. "Allow alt to 5000m" or "Double ascend/descend speed".

For a first version, it would be enough if the app would display a list and allow to read/write each of them when tapped.

notsolowki, 2017-02-09:
@mefistotelis will pmcappfw3.bin from the pro work on the advanced?

notsolowki, 2017-02-14:
could python be used to write a script that could extract the remainder of the parameters in the flight controller? there seems to be alot of things in the binary that are not in the json file.  thanks

notsolowki, 2017-02-14:
there are a few things i would like to see change most of them are
.Go home speed
.waypoint/mission velocity
.motor startup sounds
.max ascend speeds.

im sure all of these things except startup sounds are in the flight controller firmware. i'm no good at byte patching unless someone could walk me through it with this firmware.

notsolowki, 2017-02-14:
maybe i can get you to break this down for me

  sed 's/\(9a99f940\)\(2de9f041\)/cdcc7441\2/' | \

  sed 's/0000004033331f40/9a99999999992e40/' | \
  
i would like to know what exactly is going on right here. and how did you come up with the values exactly.
maybe it would give me a better idea how to do this myself. i would greatly appreciate it. thankyou

mefistotelis, 2017-02-14:
To learn regex in general, find a tutorial - there is a lot of both written and video ones.

Here I am using replace clause. The 2nd one is most basic - just replace hex values on the left with hex values on the right. The hex values on the left I got from IDA Pro - just clicked on the double I found, then switched to hex view and copied directly (minding char case). Values on the right I got from an online double to hex converter I found in Google; I used one without endianness support, so I written byte values the converter shown from right to left.

The first I built in the same way, I just used a few bytes after the searched value as context - so that I will not end up finding the same bytes which have different meaning. on the right I replaced 1st group with hexes from online float to hex converter, and 2ns group I just copied without change (\2 does that).

notsolowki, 2017-02-15:
im having trouble converting the hex value. i used numerous converters online. for example i have found this in the binary 00 00 00 40 33 33 1F 40. i cant figure out how to convert this back to 7.80000019. which makes me thing if i convert any other value to hex its going to be wrong. i eve looked from left to right but no match.

mefistotelis, 2017-02-15:
Little Endian is left-to-right, but by bytes, not by hex value. note that one hex digit is not one byte but 4 bits. Your value in big-endian is: 40 1F 33 33 40 00 00 00.

GlovePuppet, 2017-04-05:
I made a bit of progress here. Command F703 allows us to query the FC for some information about each entry in the table at 0x080A6900

This command causes the FC to search for the 4 byte payload in tables at 0x080B2DE0 (0x347 entries) and 0x080B4818 (0x15 entries)

F703 [02 -> 03] 11 : 55 11 04 92 02 03 5D 01 40 03 F7 **3D 7A 5C 4D** 96 D7

Each table entry is a pair of 32 bit numbers, the first is the 'hash' (I will call it a hash since DJI printfs call it a hash) the second is the index into the table at 0x080A6900. In this case the index is 0 and the table entry looks like this:

name                    DCD 0x80DBD8  -->  "cfg_var_table_size_0",0
storage                 DCD 0x200200D8 
unknown              DCD 4
param_type                    DCD 2  <----This tells which set of min/max/default to use
float_min_max_def         DCD 0, 0x4F800000, 0
int_min_max_def            DCD 0, 0xFFFFFFFF, 0
int2_min_max_def          DCD 0, 0xFFFFFFFF, 0
unknown2                       DCD 0 <----- Not always 'B'
unknown3                       DCD 0

The FC's response:

F703 [03 -> 02] 35 : 55 35 04 68 03 02 5D 01 80 03 F7 00 02 00 04 00 00 00 00 00 00 00 FF FF FF FF 00 00 00 00 63 66 67 5F 76 61 72 5F 74 61 62 6C 65 5F 73 69 7A 65 5F 30 00 E1 D9

00 <--- padding
02 00 <--- param_type
04 00 <--- unknown
00 00 <--- unknown2
00 00 00 00 FF FF FF FF 00 00 00 00   <--- int2_min_max_def (depends on param_type)
63 66 67 5F 76 61 72 5F 74 61 62 6C 65 5F 73 69 7A 65 5F 30 00  <--- "cfg_var_table_size_0",0

There's a whole family of Fx03 commands here that use the same functions so I expect there's a get and set too

One more (index 0xF1)

F703 [02 -> 03] 11 : 55 11 04 92 02 03 5D 01 40 03 F7 94 0A 5C 42 E7 F4

F703 [03 -> 02] 43 : 55 43 04 74 03 02 5D 01 80 03 F7 00 01 00 02 00 0B 00 0F 00 00 00 F4 01 00 00 1E 00 00 00 67 5F 63 6F 6E 66 69 67 2E 66 6C 79 69 6E 67 5F 6C 69 6D 69 74 2E 6D 61 78 5F 72 61 64 69 75 73 5F 30 00 8C 6B

GlovePuppet, 2017-04-05:
F803 looks to be read, command has the same format as F703

F803 [02 -> 03] 11 : 55 11 04 92 02 03 5D 01 40 03 F8 94 0A 5C 42 1B 9E
F803 [03 -> 02] 14 : 55 14 04 6D 03 02 5D 01 80 03 F8 00 94 0A 5C 42 F4 01 EC A5

Probably F903 is write

MAVProxyUser, 2018-01-17:
@mefistotelis "here were no other FWs with unencrypted flight controller, or other normally encrypted module." hmm... that is a shame... A little bird told me about some forked GitHub repos and a DMCA attempt... you uh... may want to fork this. https://github.com/MAVProxyUser/spray-system/blob/master/app/aes/aes.c#L38