o-gs/dji-firmware-tools #8: Changing tilt of aircraft

notsolowki, 2017-01-05:
What is it going to take to achieve such a thing. im not sure that the mobile SDK will be able to accomplish this task. I'm sitting here looking at a binary with ida and have no clue what to do.

mefistotelis, 2017-01-05:
Try searching for the names of parameters we've found in flight controller. If any SDK supports reading and writing these, it is likely that it will contains a few names.

notsolowki, 2017-01-05:
i did find a few of the parameters in the guidance sdk. i think there is a maximum value set within the firmware somewhere that you cant override.

notsolowki, 2017-01-06:
i was looking at g_config.control.atti_tilt_w_rate_0, although i really have no idea how to use ida. when i select the elf file it asks me what processor so i change it from metapc to arm little endian. i assume i have it right. do i have to build this structure array before i can see whats in it? thank you for all this help

mefistotelis, 2017-01-06:
Opening ELF:
The code is ARM little-endian; it will probably switch to it even if you won't select it, as ELF file incorporates that information.

Structs support:
After the auto-analysis finishes, it is up to you to define everything which cannot be automated. This includes structs (unless the file has debug info which includes struct definitions).

notsolowki, 2017-01-06:
so for example is, .text:080A357C aTooLargeTilt_0 DCB "too large tilt angle",0 ; DATA XREF: .text:080A34A8o

would i be looking to go to text:080A34A8o to start investigating its data

notsolowki, 2017-01-06:
.text:08068FB0                 LDR             R0, =aSmart_batteryT ; "[smart_battery]this fireware calc gohme"...
.text:08068FB2                 VSTR            D0, [SP,#0x10+var_10]
.text:08068FB6                 VLDR            D0, =7.80000019
.text:08068FBA                 VMOV            R2, R3, D0
.text:08068FBE                 BL              sub_806E08A

i think i remember seeing 7.800000 as the speed in a log once

notsolowki, 2017-01-07:
can you please tell me how you got to field_14. im trying to figure out what to do and im not really getting anywhere. and if i knew how to link these strings and functions i would be doing it. thanks

mefistotelis, 2017-01-07:
You should just learn how to use IDA Pro. You should be able to find a tutorial, either written one or on a video. I learned IDA by myself, so I don't have any specific links.

First create the struct in "Structures" tab, then give type to the data in "IDA View" tab. Finally after looking at the data, you may give proper types to fields in "Structures", for example set field_14 as "float".

notsolowki, 2017-01-07:
Thankyou.

mefistotelis, 2017-01-08:
Reverse engineering is complex, and learning IDA requires time. There's no shortcut. But there's also no better tool for the job.

If you want to check something simpler, try looking at Mobile SDK Panorama Demo - there's a file "flyc_param_infos" which is probably generated from the firmware and lists these parameters with limits.

The same demo contains a compiled Java code, which is able to modify these parameters.

notsolowki, 2017-01-09:
when i extracted the apk i found flyc_param_infos  .. you dont think its that easy???

notsolowki, 2017-01-09:
i also see 	"index" : 363,
		"typeID" : 1,
		"size" : 2,
		"attribute" : 11,
		"minValue" : 0,
		"maxValue" : 6000,
		"defaultValue" : 3500,
		"name" : "g_config.serial_api_cfg.input_pitch_limit_

which might explain how it communicates

mefistotelis, 2017-01-09:
The values may be already at the limit, we don't know.

The code (Java code which I put in #7) definitely suggests that values of all parameters can be read on mobile device, and at least some of them (but maybe all) can be set to new value.

We need to write a mobile app to read/write these parameters. Then we will see whether that's enough, or we need FW modification too.

notsolowki, 2017-01-09:
well i jsut modified the dji go app and set some valuse for height,pitch,roll,yaw. only thing i noticed is it says net height limit i put at 1000. JUST FOR TESTING. but when i put in 1000 it goes back to 500. and as far as roll pitch etc no change

notsolowki, 2017-01-09:
what the modify parameter all about?   "index" : 1400,
        "typeID" : 0,
        "size" : 1,
        "attribute" : 11,
        "minValue" : 0,
        "maxValue" : 255,
        "defaultValue" : 0,
        "name" : "g_config.gear_cfg.hide_gear_en_0",
        "modify" : true

mefistotelis, 2017-01-09:
I don't know what the dji app does. For the "modify" - you probably have the same ideas as me.

If you want to try FW modification:

Here's an IDC script to set up your IDA to the definitions I made:
https://mega.nz/#!BI8kyL7J!kBjmu-RtcaBjP6WfPvsfultOxPnT-RydsAFvvvd1yuY (I will remove the link in a few days)

To use it:
1. Load the ELF into IDA, but UNCHECK the field "Analysis: enabled".
2. Load the IDC script from "File" menu
3. Re-enable "Analysis" in "Options -> General".

Now you have the structs defined.

notsolowki, 2017-01-09:
thankyou, i got warning 80664CC local label are allowed online within functions. if that matters. when i first select the elf file from the processor drop down menu meta pc is selected do i need to change that to arm little endian

notsolowki, 2017-01-09:
if i havn't already annoyed you yet. how can i make changes to the structure parameters. would this be done with a hex editor? how would i get to a parameter in the structure. thankyou and thankyou again

notsolowki, 2017-01-09:
have you had a look in the dji go apk?

notsolowki, 2017-01-09:
to modify the FW would it have to be completely re-written or could the structure be changed with hex-edit.

mefistotelis, 2017-01-09:
I'm not planning any work on Dji Go.

Binary data can be changed.

notsolowki, 2017-01-09:
What are my options for changing the binary data of the structure parameters. I think making the app use the new parameters would be a simple task.

mefistotelis, 2017-01-10:
Then go for it, do the app. There's no point of changing the limits in firmware until we have any means to test them.

notsolowki, 2017-01-10:
in the app if you open it with winrar and navigate to res/raw/ theres a file flyc_param_infos, it looks like it hold all of the variables.

notsolowki, 2017-01-16:
I have changed every parameter in the fw that relates to tilt, angle, pitch, max and min to 1 with no effect???

notsolowki, 2017-01-16:
the only thing i could change that had any effect was the altitude limit. and even then i have not tested that in flight. i wonder whats going on

notsolowki, 2017-01-16:
one thing i noticed in the actual mi01 binary is they had the go home speed set at 7.8000. i just wonder if the values im looking for are not in the structure array. and the go home speed wasn't a pitch or angular parameter it simply said go home speed. ive changed lots of parameters and not one of them effected the way the drone fly's normally. i didn't test waypoint or anything.

mefistotelis, 2017-01-16:
It is possible that tilt is bound by multiple parameters.

MrBurnsAT, 2017-01-17:
notsolowki how did u test the firmware?

Did u do it with connected mobile device and djigo APP running?

I would try it without! 
Only Craft and RC! 
No Mobile
No GoApp

My thought is that the parameters in flyc_conf are only place holders without any used information in there.

When u start up ur Phantom and ur RC, i think then the real informations for the flyc_conf are transmitted/synced from the RC to the craft.

Maybe some informations Mord if a mobile device is connected and go APP is running.

Could that be possible?

If yes. 
We should really take a look in the RC Firmware
If there are maybe the real informations stored

notsolowki, 2017-01-17:
mrburns, i was thinking that. so i did try that and the results were the same. i changed alot of parameters. none of them effected the flight. i update the apps flyc_param_infos along with the flight controller. is it possible that the parameters im looking for ar not in the structure array? it seems very likely that they are in the flyc_param file but i dont know??!!

mefistotelis, 2017-01-17:
Before we assume that we need anything more, let's try with a simple parameter, like height limit. We know that changing it in DJI Go's flyc_param_infos influences the GUI, and we have the same parameter in flyc_param_infos from flight controller. Modifying both of these (and updating firmware + replacing file on tablet) should lift the limit.

notsolowki, 2017-01-17:
yea the hight limit was lifted no problems even without update the flyc_ in the app. it still populated it regardless

mefistotelis, 2017-01-17:
> yea the hight limit was lifted no problems even without update the flyc_ in the app. it still populated it regardless
Hm, interesting. This means the direction of update goes in the other way that the one MrBurnsAT suggested - flyc_param_infos are transmitted from the craft, and the mobiledevice only contains placeholders.

mefistotelis, 2017-01-17:
@notsolowki which of these have you tried?:

<pre># grep "\(speed\|tilt\)" flyc_param_infos
                "name" : "g_config.control.tilt_atti_gain_0",
                "name" : "g_config.control.tilt_gyro_gain_0",
                "name" : "g_config.control.tilt_gyro_acc_gain_0",
                "name" : "g_config.advanced_function.pantilt_enabled_0",
                "name" : "g_config.pantilt.output_frequency_0",
                "name" : "g_config.pantilt.roll_travel.travel_min_0",
                "name" : "g_config.pantilt.roll_travel.travel_max_0",
                "name" : "g_config.pantilt.roll_travel.travel_center_0",
                "name" : "g_config.pantilt.pitch_travel.travel_min_0",
                "name" : "g_config.pantilt.pitch_travel.travel_max_0",
                "name" : "g_config.pantilt.pitch_travel.travel_center_0",
                "name" : "g_config.pantilt.yaw_travel.travel_min_0",
                "name" : "g_config.pantilt.yaw_travel.travel_max_0",
                "name" : "g_config.pantilt.yaw_travel.travel_center_0",
                "name" : "g_config.pantilt.roll_gain_0",
                "name" : "g_config.pantilt.pitch_gain_0",
                "name" : "g_config.pantilt.yaw_gain_0",
                "name" : "g_config.pantilt.reverse_output_0",
                "name" : "g_config.pantilt.roll_speed_0",
                "name" : "g_config.pantilt.pitch_speed_0",
                "name" : "g_config.pantilt.yaw_speed_0",
                "name" : "g_config.craft_model.motor_tilt_angle_0",
                "name" : "g_config.control.horiz_emergency_brake_tilt_max_0",
                "name" : "g_config.control.atti_tilt_w_rate_0",
                "name" : "g_config.control.manual_tilt_w_rate_0",
                "name" : "g_config.control.tilt_exp_mid_point_0",
                "name" : "g_config.control.dyn_tilt_min_0",
                "name" : "g_config.control.rc_tilt_sensitivity_0",
                "name" : "g_config.imu_para_cfg.imu_adv_func.vo_vel_with_large_tilt_deny_0",
                "name" : "g_config.novice_cfg.atti_tilt_w_rate_0",
                "name" : "g_config.gear_cfg.gear_speed_0",
                "name" : "wind_speed[0]_0"
                "name" : "wind_speed[1]_0"
                "name" : "wind_speed[2]_0"</pre>

notsolowki, 2017-01-17:
all of them except
                   "name" : "g_config.imu_para_cfg.imu_adv_func.vo_vel_with_large_tilt_deny_0"
                "name" : "wind_speed[0]_0"
                "name" : "wind_speed[1]_0"
                "name" : "wind_speed[2]_0"
                "name" : "g_config.control.tilt_atti_gain_0",
                "name" : "g_config.control.tilt_gyro_gain_0",
                "name" : "g_config.control.tilt_gyro_acc_gain_0",
                "name" : "g_config.advanced_function.pantilt_enabled_0",

notsolowki, 2017-01-17:
i thought these two were borderline simple variables that had no effect either

	"index" : 323,
		"typeID" : 5,
		"size" : 2,
		"attribute" : 11,
		"minValue" : 1,
		"maxValue" : 20,
		"defaultValue" : 20,
		"name" : "g_config.control.vert_up_vel_0",
		"modify" : true
	},
	{
		"index" : 324,
		"typeID" : 5,
		"size" : 2,
		"attribute" : 11,
		"minValue" : 1,
		"maxValue" : 15,
		"defaultValue" : 10,
		"name" : "g_config.control.vert_down_vel_0",
		"modify" : true
	},

and the low battery version of those variables did nothing either

MrBurnsAT, 2017-01-17:
That is plausible for me!
The real informations are stored at the craft and will be transmitted to the RC.
So the most regulations are done by the rc and not the craft.

But there must also be also most of this informations at the craft, because if u loose signal.
Craft is retuning to home a specified speed, same with landing speed

Wind speed is Wind speed and has nothing to do with tilt.

If u load ur logs up to healthydrones u will See how fast wind was!

notsolowki, 2017-01-17:
i think they set the rth speed slow so that you will always loose the drone if windows are above 10 mph

MrBurnsAT, 2017-01-17:
"g_config.control.vert_up_vel_0", 
looks for me if thats ascending speed

And
"g_config.control.vert_down_vel_0"
Should be descending speed

notsolowki, 2017-01-17:
yea no doubt but they didnt effect anything

mefistotelis, 2017-01-17:
Look at their limiting values - vert_up_vel_0 has 1..10, default=5. I think the craft goes up about 6 or 6 mps. So min/max only set limits, and the "real" value is either in "default" or is controlled in a different way.

notsolowki, 2017-01-17:
can someone confirm what rate it ascends. i was ascending at 11 mps till about 250 ft. and i had mine set at 10. then i reflashed it with them set at 20 no change.

MrBurnsAT, 2017-01-17:
11mps thats really fast

notsolowki, 2017-01-17:
im sorry 11 MPH

mefistotelis, 2017-01-17:
Yeah, meters vs. miles - the non-metric system always causes problems.

Anyway [in the spec DJI says](https://www.dji.com/phantom-3-pro/info):
<pre>Max Ascent Speed	5 m/s</pre>

To modify "real" limits and noy limits of limits, we need a custom made mobile app.

notsolowki, 2017-01-17:
so the things that need changed are not in the flyc_param side of the app?

notsolowki, 2017-01-17:
this sure would look a lot more appealing lol

Weight (Battery & Propellers Included) 1280 g
Diagonal Size (Propellers Excluded) 350 mm
Max Ascent Speed 500 m/s
Max Descent Speed 300 m/s
Max Speed 1600 m/s (ATTI mode)
Max Service Ceiling Above Sea Level 19685 feet (6000 m)
Max Flight Time Approx. 23 hours
Operating Temperature Range 32° to 104°F (0° to 40°C)
Satellite Positioning Systems GPS/GLONASS
Hover Accuracy Range Vertical:
±0.1 m (when Vision Positioning is active)
±0.5 m
Horizontal:
±1.5 m
GIMBAL

mefistotelis, 2017-01-17:
When you're changing max height - flyc_params are used to increase limit of the limit, and then you can set the "real" limit value within the mobile app.

So with velocity, tilt or anything else - it's basically the same - you modified limits of limits and that's good, but now you need to set "real" value - the problem is, DJI Go doesn't have such function. Therefore you need a custom app.

MrBurnsAT, 2017-01-17:
Maybe it will help to look at litchi!

I know up to Litchi Version 2.3.1 it was possible to break the height Limit with only choose it in the settings of the App  (5m(or10m)-6000m)

With Firmware Versions up to max 1.4.0010

notsolowki, 2017-01-17:
i will note that i didnt have to update the app for the new altitude limit to show. example. i set it at 5000 int he app it still says 20-500 but the box is populated with 5000 until i try to change it then it goes to 500. do you think addind the parameters to the apps flyc_param would do any good?

notsolowki, 2017-01-17:
so then were are these real values stored at in the first place?

mefistotelis, 2017-01-17:
In the place referenced by the 2nd pointer in ExportParam struct, ofc.

notsolowki, 2017-01-17:
would you be mad if i requested that in english

notsolowki, 2017-01-17:
can you explain this a little more please " 2nd pointer in ExportParam struct, ofc."

notsolowki, 2017-01-17:
i get ExportParam as the "Structure" and i could probably find the second pointer. but im not sure what "ofc" is or where ExportParam structure exists at.

notsolowki, 2017-01-17:
no i mean in the app. how does it send the parameters where else would the be beside flyc_param_infos.

notsolowki, 2017-01-17:
i just tested some of the followme and waypoint parameters it had no effect

notsolowki, 2017-01-17:
yesterday for testing i set the max height to 5000 meters, today i set it to 9999 in both phone and drone. the phone lets me choose between 20-9999 but when 9999 is entered it reverts back to 5000...

notsolowki, 2017-01-17:
also. i see go home speed before and after the structure array 

.text:08068F90                 PUSH            {R2-R4,LR}
.text:08068F92                 LDR             R4, =0x200216C4
.text:08068F94                 SUBS            R4, #0x6C
.text:08068F96                 LDRB            R0, [R4,#4]
.text:08068F98                 CBNZ            R0, loc_8068FC2
.text:08068F9A                 MOVS            R0, #1
.text:08068F9C                 STRB            R0, [R4,#4]
.text:08068F9E                 BL              sub_8068C14
.text:08068FA2                 LDRB            R2, [R4,#2]
.text:08068FA4                 LDRB            R1, [R4,#1]
.text:08068FA6                 ADR             R0, aBatteryResetDe ; "[BATTERY]:reset default smart cfg - L1:"...
.text:08068FA8                 BL              sub_806E08A
.text:08068FAC                 VLDR            D0, =2.5
.text:08068FB0                 LDR             R0, =aSmart_batteryT ; "[smart_battery]this fireware calc gohme"...
.text:08068FB2                 VSTR            D0, [SP,#0x10+var_10]
.text:08068FB6                 VLDR            D0, =7.80000019
.text:08068FBA                 VMOV            R2, R3, D0
.text:08068FBE                 BL              sub_806E08A
.text:08068FC2
.text:08068FC2 loc_8068FC2                             ; CODE XREF: sub_8068F90+
.text:08068FC2                 BL              sub_8069288
.text:08068FC6                 BL              sub_80629EC
.text:08068FCA                 CMP             R0, #0x13
.text:08068FCC                 BEQ             loc_806909C
.text:08068FCE                 BL              sub_80629EC
.text:08068FD2                 CMP             R0, #0x11
.text:08068FD4                 BEQ             loc_8068FE2
.text:08068FD6                 BL              sub_80629EC
.text:08068FDA                 CMP             R0, #0x12
.text:08068FDC                 BEQ             loc_8068FE2
.text:08068FDE                 BL              sub_8068C14
.text:08068FE2
.text:08068FE2 loc_8068FE2                             ; CODE XREF: sub_8068F90+44
.text:08068FE2                                         ; sub_8068F90+4C
.text:08068FE2                 LDR             R0, =0x200197BC
.text:08068FE4                 SUBS            R0, #0x64
.text:08068FE6                 BL              sub_8068A88
.text:08068FEA                 LDR             R0, =0x200197BC
.text:08068FEC                 SUBS            R0, #0x64
.text:08068FEE                 BL              sub_8068C32
.text:08068FF2                 LDR             R3, =0x200197BC
.text:08068FF4                 SUBS            R3, #0x28
.text:08068FF6                 LDR.W           R0, [R3,#-0xE]
.text:08068FFA                 UBFX.W          R0, R0, #0x18, #1
.text:08068FFE                 B               loc_8069074
.text:08068FFE ; ---------------------------------------------------------------------------

.text:08069000 dword_8069000   DCD 0x200216C4          ; DATA XREF: sub_8068A88:loc_8068C0
.text:08069000                                         ; sub_8068C14
.text:08069004 dword_8069004   DCD 0x20001678          ; DATA XREF: sub_8068C32+A
.text:08069008 dword_8069008   DCD 0x20001A78          ; DATA XREF: sub_8068C32+C
.text:0806900C flt_806900C     DCFS 50.0               ; DATA XREF: sub_8068C32+28
.text:08069010 flt_8069010     DCFS 0.0                ; DATA XREF: sub_8068C32+2C
.text:08069014 dword_8069014   DCD 0x200197BC          ; DATA XREF: sub_8068C32+30
.text:08069014                                         ; sub_8068EFA
.text:08069018 dword_8069018   DCD 0x200199F0          ; DATA XREF: sub_8068C32+B8
.text:0806901C flt_806901C     DCFS 1000000.0          ; DATA XREF: sub_8068C32+126
.text:08069020 dbl_8069020     DCFD 1000000.0          ; DATA XREF: sub_8068C32+13E
.text:08069028 dword_8069028   DCD 0x3F19999A          ; DATA XREF: sub_8068C32+29C
.text:0806902C aBatteryResetDe DCB "[BATTERY]:reset default smart cfg - L1:%d  L2:%d",0
.text:0806902C                                         ; DATA XREF: sub_8068F90+1
.text:0806905D                 DCB 0, 0, 0
.text:08069060 dbl_8069060     DCFD 2.5                ; DATA XREF: sub_8068F90+1
.text:08069068 dbl_8069068     DCFD 7.80000019         ; DATA XREF: sub_8068F90+2
.text:08069070 off_8069070     DCD aSmart_batteryT     ; DATA XREF: sub_8068F90+2
.text:08069070                                         ; "[smart_battery]this fireware calc gohme"..

notsolowki, 2017-01-17:
is it that  ((.text:08069068 dbl_8069068       DCFD 7.80000019         ; DATA XREF: sub_8068F90+26r)) is pointing to ((.text:08068F90                 PUSH            {R2-R4,LR})) which ends up  ((.text:08068FBA                 VMOV            R2, R3, D0))
((.text:08068FB6                 VLDR            D0, =7.80000019))

[81 more replies]