o-gs/dji-firmware-tools #9: amba_rfs.py

notsolowki, 2017-01-05:
heres wha ti get when attempting to repack the rfs 

laptop@laptop-VPCEH14FM:~/Desktop/phantom-firmware-tools-master$ python3 ./amba_romfs.py -vv -a -p p7_mi12_part_romfs.a9s
p7_mi12_part_romfs.a9s: Opening for creation
Traceback (most recent call last):
  File "./amba_romfs.py", line 344, in <module>
    main(sys.argv[1:])
  File "./amba_romfs.py", line 335, in main
    romfs_create(po,fwpartfile)
  File "./amba_romfs.py", line 255, in romfs_create
    fshead, fsentries = romfs_read_filesystem_head(po)
  File "./amba_romfs.py", line 95, in romfs_read_filesystem_head
    with open(fname, "r") as lines:
FileNotFoundError: [Errno 2] No such file or directory: 'p7_mi12_part_romfs/_header.a9t'
laptop@laptop-VPCEH14FM:~/Desktop/phantom-firmware-tools-master$

notsolowki, 2017-01-05:
oay i figured out what i was doing wrong. but now i cant seem to make the changes to the ubifs stick. i changed the permissions and everything  and it said it saved but if i remount it they are back to default??

notsolowki, 2017-01-05:
how can i find the instructions in looking for to save the changes. i need to find an example

notsolowki, 2017-01-05:
okay i did dd if=/dev/mtd0 of=/home/ubifs    will this work? it make a 67MB file. should i just rename this to the correct name? or do i need to run  bs=$((1024*1024))  with the command. thanks

mefistotelis, 2017-01-05:
"bs=" just sets bigger buffer so that the copying goes faster.

If the resulting file is too large, you may need to copy only part of it, or trim it after copying.

Look at the original size in various firmwares - if it's always the same, then your file should have the exact same size.

closed 2021-06-14 by mefistotelis