o-gs/dji-firmware-tools #10: RC Firmware config

MrBurnsAT, 2017-01-12:
RC Firmware must be changeable too.

Some limitations was added with RC Firmwares 1.6 and newer.
Output Power is reduced from 25mW to 10mW in CE Mode.

And
It should be possible to change CE to FCC Mode
(RC Checks via GPS location and Sets than FCC or CE Mode, USA = FCC, Europe = CE

mefistotelis, 2017-01-12:
Actually, the power output mode needs to be changed in both transmitter in radio controller and in the transmitter on OFDM board.

It's good to know the change in power was made with 1.6 - it should allow to find the change by comparing two firmwares. Please make sure that this information is correct, if you can.

MrBurnsAT, 2017-01-12:
Yes the change was with RC Firmware 1.6 (C1_FW_v01.06.0000.bin)

Craft and RC Firmwares are not the same since 1.3.2 which was the last Craft and RC update which was done by USB drive. Since there RC Firmware is only updated via DJI GoApp.

And not every craft firmware Version there was a RC Firmware Update.
For P3 actual RC Firmware is 1.6 (where encryption of the Transmission also has been changed)

RC Firmware 1.6 is only compatible with craft firmware 1.8 or newer!!!

mefistotelis, 2017-01-12:
Thank you, I didn't know many of these facts. I will take a look at RC firmware.

MrBurnsAT, 2017-01-12:
I now over exactly one year searched for a solution to Downgrade more than only one Version.

So i really read many about the Phantom and all firmware out there.
And a RC models are my hobby since many years. Cars, Planes, Helis, and Copter

But i have not Programmen for a long time.
I really need time to come back in.

For you, this is a gimmick

aka1ceman, 2017-01-12:
1. Keep in mind the change between A and B remotes.
2. I have downgraded the RC in the past.(my B remote that per DJI can not be upgraded with USB) It may not have been the best way to do it but maybe it will shed some light to help you move forward. 
I did an update with DJI Go which downloaded 1.5 for my RC. (do NOT install) I closed the Go app, went into my android file structure and swapped the file with a 1.3 FW renamed what DJI named the 1.5 download. Restarted the Go app, and finished install. Afterwards, I restarted Go which showed 1.3 FW on the RC.

MrBurnsAT, 2017-01-12:
Cool thanks for the Tip

But I tried this too, and it didnt work for me.

Which GoApp Version did u use?

Ill try it one more time

aka1ceman, 2017-01-13:
It was around 2.3 or 2.4. I did get it to work on 2 occasions last year, but reverted back due to no ability to roll back the quad.

MrBurnsAT, 2017-01-13:
Which 1.3 RC Firmware did u Took?

Or did u Took the Craft Firmware 1.3.2?

aka1ceman, 2017-01-13:
it was 1.3.2 but not sure which one.

aka1ceman, 2017-01-16:
When DJI Go does a RC update, it downloads a file from their server (RC_FW_V1.6.0.bin) into a folder Android/device storage/DJI/dji.pilot/package. It DJI GO installs it from that location.

ferraript, 2017-01-17:
regarding FCC vs CE mode, I believe [this video](https://www.youtube.com/watch?v=Ys-svvng5TU) instruction should help

mefistotelis, 2017-01-18:
In P3 Pro/Adv the related hardware is completely different.

It would seem that transmission is configured by:
m1500 radio transmitter 'IG810 LB2_68013_TX'
and
m1600 radio receiver 'IG810 LB2_68013_RX ground'

I don't know whether these transmitters are visible from any other firmware as network interfaces.

I also don't know if the tx power is configurable within these firmwares directly, or sent to it from a more complex module.

mefistotelis, 2017-01-18:
> Yes the change was with RC Firmware 1.6 (C1_FW_v01.06.0000.bin)
(DJI support confirmed me this! They must do this because of local laws)

I compared C1_FW_V01.05.0080 and C1_FW_V01.06.0000. The only modules which are different between these versions are m1400 and m1401, which are MCUs for A and B remotes.

Will look in detail at this module.

aka1ceman, 2017-01-18:
Ok, I figured out what I did ......

Turn on RC, Plug in Tablet, start DJI Go (I used 2.4.2) 
Touch and hold top right corner of screen for 5-10 seconds. (tutorial button thanks to Inspire guys)
New screen opens offering to downgrade RC. (I was offered 1.6 and 1.5.8)
Download but do not install.
Using file explorer replace and rename file
I used  RC file for 1.3.2 and renamed, and replaced the existing 1.5 file.
Go back to DJI Go app and finish install. It will finish installing. I did install for me although renaming may not be the safest way. Hope this helps.

Now how can we get DJI GO to allow other options of files?

ferraript, 2017-01-18:
aka1ceman, that looks very easy, are you sure it was done this way?
because I would be afraid of two things:
1. bin file downloaded by DJI GO will be cached in phone's RAM (and DJI GO will use this one), so replacing the one stored in DJI/dji.pilot/package will have no effect
2. right after click on "Install", DJI Go will either check if checksum is the same, maybe it will also check if header inside the bin file has correct version

you didn't wrote this, so I'll ask: you only renamed the bin filename? or did you also change version number in header inside the bin file?

mefistotelis, 2017-01-18:
> Now how can we get DJI GO to allow other options of files?

Since the functionality id already there, it shouldn't be hard to unlock it.  We just need someone with experience in analysing and recompiling Java code. I'm certain everything is within the Dji Go app.

notsolowki, 2017-01-18:
Is it possible that the dji go just updates the controller with the parameters.

aka1ceman, 2017-01-18:
ferraript,
1. I verified that was the location of the downloaded file. It is not installed or in phones Ram at that point. 
2. The normal way "requires updates" button  at top of DJI Go will NOT install once the file is renamed. It gives error once name is modified. Once the modified file is removed from that folder, it offers to download again. It sees a file there. Only way I could bypass the error was this trick that Inspire users used to force downloads on RC in the past. Only thing I figure is this bypasses the checksum. NO I only changed the name.

mefistotelis, 2017-01-18:
Checksum is stored WITHIN the FW file; by replacing it with new one, you are also giving it a new checksum.

ferraript, 2017-01-20:
@aka1ceman, thanks for info
today I successfully downgraded my RC's FW from 1.5.7 to 1.4.3, using your instructions
in my case, after first closing of DJI GO and renaming the file, DJI GO redownloaded the FW again
after second closing and reopening DJI GO, it didn't redownload and allowed me to upgrade (actually downgrade) the firmware

MrBurnsAT, 2017-01-20:
@ferraript which GoApp Version did u use?

2.4.2???

notsolowki, 2017-01-20:
@mefistotelis : Have you found anything of interest in the rc firmware. is it encrypted?

mefistotelis, 2017-01-20:
Yes; but I'm not sure what is the correct base address. Without it, some pointers may lead to invalid locations. Maybe it's 0? Not sure, will figure it out later.

The ".bss" section should probably start at 0x10000000. Its size seem to extend to 0x50000000, but so much data would slow down any disassembler - I will try to select narrowed area.

ferraript, 2017-01-20:
@MrBurnsAT: I used 2.4.3

MrBurnsAT, 2017-01-20:
Great. Thx 

IT worked!

Litchi 2.3.1 is running

MrBurnsAT, 2017-01-20:
@ferraript 

I think about the Standard and 4K Phantoms

Do u think they might can be downgradet like this?
Because they where updated via GoApp and Not sd card

ferraript, 2017-01-20:
I'd say there will be no difference
but the best way to find out is to try it :)

MrBurnsAT, 2017-01-20:
So now we need a Phantom Standard owner 

Who would try this for us.

ferraript, 2017-01-24:
I just found out in DJI GO 2.4.3 sources interesting class regarding FCC and CE mode: DataRcSetPowerMode.java
it's used in one another class only, \dji\a\d\a.java
here is interesting method, onEventBackgroundThread, that checks, if ProductType is P3S or P34K and if HomePoint coordinates are set
if so, there is link to the most interesting class, \dji\a\f\a.java, where is an array of GPS coordinates and radiuses of CE areas (there is Australia, Japan, India, South Africa and Europe)
if HomePoint is inside any CE area, mode is set to CE, otherwise mode is set to FCC

so it would be interesting to know, that if we add P3X and P3A to that first condition and change the CE areas coordinates, if really Phantom uses FCC mode then

mefistotelis, 2017-01-24:
P3S is completely different from P3X.

What we can get from this code is how continents are defined - this code is very likely just copied into P3X, so the numbers should allow us to find the same code in P3X FW.

ferraript, 2017-01-26:
I think I finally found links that DJI GO uses for RC upgrade:
http://upgrade.dj2006.net/redirect/links/pilot_br
http://upgrade.dj2006.net/redirect/links/pliot

rulppa, 2017-02-01:
`RC Firmware must be changeable too.

Some limitations was added with RC Firmwares 1.6 and newer.
Output Power is reduced from 25mW to 10mW in CE Mode.`

I have few modded remotes, one with 1.5.8 and other one with latest FW, i will measure the output with immersion rc RF-meter to confirm both remotes connected to 1.7.6 AC.

MrBurnsAT, 2017-02-01:
U can downgrade Ur RC to 1.5.8, 1.5.7 or 1.4.3 ;-)

So u should have full 25mW power

I downgraded Mine to 1.4.3 because um using Old Litchi Version 2.3.1

With newer Firmwares than 1.4.3 this Old Litchi Versions wont work

mefistotelis, 2017-02-01:
RC hacking update:

We don't know what information the RC receives to set output power:
- if it receives GPS coordinates, then the firmware must have definition of CE area
- if it receives simple "set CE mode" instruction, then CE area definition must be in the aircraft FW

Either way, since we know how the CE area is defined in the mobile app (for P3 Standard), we can expect one of the firmware modules will have similar float/double numbers within. I believe this is the best starting point for getting the CE/FCC code - search for coordinates.

I tried looking at the RC firmware code directly, but this isn't easy - there are almost no debug messages which would allow to locate points of interests.

ferraript, 2017-02-01:
@rulppa: that would be great
it will finally reveal if those talks were true
by the way, where are you from? I'm thinking about you could test if my DJI GO hack for FCC/CE works

@mefistotelis: thanks for info
who knows, maybe P3X and P3A obey that _SetPowerMode_ command too :)

rulppa, 2017-02-01:
@ferraript: i will measure tomorrow, got new ESC-board for new motor p3p today, swapping now and it's getting late. I'm from europe. I guess you are from italy? Forza ferrari, go kimi go lol. Did you notice any difference with your hack? If i test it, i will test it with stock gl300b, other remotes are modded and stock one is more sensitive for these kind of tests.

ferraript, 2017-02-01:
@rulppa, I'm just Ferrari fan from Slovakia
so you are in CE area too, fine
I can send you my DJI GO apk then for testing

rulppa, 2017-02-01:
@ferraript: Can't test .apk, i have only iOS devices, sorry.

MrBurnsAT, 2017-02-01:
@ferraript I can Test it!

Austria  ;-)

And i only have Andoids

ferraript, 2017-02-01:
@rulppa: that's pity :(

@MrBurnsAT: greetings to Austria :)
yeah, you can test it, but do you also have some sort of RF meter just like rulppa does?
because I don't believe that the CE/FCC setting is easily distinguishable
for example, I never had problems with my P3A to go to distance 5 km (in CE mode)

notsolowki, 2017-02-02:
what changes did you make to the app ferrapit

ferraript, 2017-02-02:
@notsolowki: I made changes I [wrote few posts ago](https://github.com/mefistotelis/phantom-firmware-tools/issues/10#issuecomment-274774903)
as I am from Europe and I have P3A, I changed european radius to few meters and I replaced P3S for P3A in the condition
if you look into the code, I believe you'll understand

notsolowki, 2017-02-02:
good work, i hope you can get it tested. i wish i could test it for you but im fcc,

notsolowki, 2017-02-02:
i can tell you i had a fly away at a really high altitude and i was 9,000ft away before i was loosing signal but i never totally lost it

notsolowki, 2017-02-02:
could someone use a mltimeter to measure the mw from the antenna output?

rulppa, 2017-02-02:
Measured GL300A and B model radios, without GPS-lock, both 0.72-0.76mW. I will measure again during weekend, with GPS lock. 1.5.8 only, 1.7.6 doesnt seem to work with 1.6 radio.

ferraript, 2017-02-02:
>1.7.6 doesnt seem to work with 1.6 radio

yeah, that's right
RC FW 1.6 (and newer) is compatible only with AC FW 1.8 (and newer)
and since you said you don't have Android phone, I doubt you'll be able to downgrade RC to 1.4.3 :-(

notsolowki, 2017-02-03:
ferraript i bet you cant test the output power at the antennas connector on the circuitbaord

ferraript, 2017-02-03:
sorry, I don't intend to dismantle RC
and after all, I have no tools for measuring something like that

rulppa, 2017-02-03:
I'm testing directly from the RC circuit board with SMA-connectors, i will do the test with GPS lock today. IF it's full power with 1.5.8 (also in CE mode) i should see the same 0.72-0.76mW. (according to OP it should be)

notsolowki, 2017-02-03:
thats how i was going to do it minus the sma connector.

notsolowki, 2017-02-03:
hey ferraript, could you make it so that dji go 3 could send the settings to the drone?

ferraript, 2017-02-03:
>could you make it so that dji go 3 could send the settings to the drone?

I thought you were using 2.4.2 :)
I looked into 3.0.1 sources and it looks like it works the same way like in 2.4.3 (which I use)
so, do you want some specific version or you'll be fine with 3.0.1?

notsolowki, 2017-02-04:
3.0.1 or 3.1.1 would be fine

mefistotelis, 2017-02-04:
I prepared a list of places in C1_FW_V01.06.0000 which could store Europe coordinates. Now each of the matches needs to be checked.

How I made the list:
<pre># Europe is around 71,-18 .. 27,50; match -19..0 and 0..69 with no exponent
ls -1 C1_FW_*.bin | xargs -I {} sh -c 'od -tfF -w4 {} | sed "s/^\(.*\)/{}: \1/g"' | grep '^.*: [0-9]*[ ]*\([1-6]\?[0-9]\|[-][1]\?[0-9]\)\([.][0-9]*\)\?$' | grep -v "^.*: [0-9]*[ ]*[-]\?0\([.][0-9]*\|\)$" > _match_f1.txt
ls -1 C1_FW_*.bin | xargs -I {} sh -c 'od -tfD -w8 -j0 {} | sed "s/^\(.*\)/{}: \1/g"' | grep '^.*: [0-9]*[ ]*\([1-6]\?[0-9]\|[-][1]\?[0-9]\)\([.][0-9]*\)\?$' | grep -v "^.*: [0-9]*[ ]*[-]\?0\([.][0-9]*\|\)$" > _match_d1.txt
ls -1 C1_FW_*.bin | xargs -I {} sh -c 'od -tfD -w8 -j4 {} | sed "s/^\(.*\)/{}: \1/g"' | grep '^.*: [0-9]*[ ]*\([1-6]\?[0-9]\|[-][1]\?[0-9]\)\([.][0-9]*\)\?$' | grep -v "^.*: [0-9]*[ ]*[-]\?0\([.][0-9]*\|\)$" > _match_d2.txt
# Now assuming lon lat coords are adjacent, let's find 71..27 and get one before and one after
grep -B1 -A1 '^.*: [0-9]*[ ]*[2-7][0-9]\([.][0-9]*\)\?$' _match_f1.txt > _match_f1b.txt
grep -B1 -A1 '^.*: [0-9]*[ ]*[2-7][0-9]\([.][0-9]*\)\?$' _match_d1.txt > _match_d1b.txt
grep -B1 -A1 '^.*: [0-9]*[ ]*[2-7][0-9]\([.][0-9]*\)\?$' _match_d2.txt > _match_d2b.txt</pre>

Files with list of possible coords:
[_match_d1b.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/752313/_match_d1b.txt)
[_match_d2b.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/752314/_match_d2b.txt)
[_match_f1b.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/752315/_match_f1b.txt)

ferraript, 2017-02-04:
interesting (and crazy linux commands :D )

just remember, that coordinates may be saved in different ways
in DJI GO, Europe is stored as (55.529627,15.702531,2824707), where coordinates are double and radius is int
but, for example, coordinates may be stored as long (and they are then divided by 1000000)
or, they can be stored in radian format, so 55.529627° = 0.969174823554

KennethMcNutt, 2017-02-05:
This is how they modded The p3S... Through the wifi telnet.. 

Read this from bottom to top..
01:10 Log Filename: log201701201904.txt
01:10 
01:10 ============================================
01:10 Restart your RC & Aircraft and Enjoy New Mod
01:10 ============================================
01:10 
01:10 Credit to: Ojcze Nasz, SingLag, Argonaise & www.phantompilots.com
Disconnected.
Disconnecting.
01:10 c
01:10
BusyBox v1.22.1 (2015-11-16 16:28:58 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.
/ # 
01:10 chmod 775 /sbin/watchlog.sh
Connected.
Connecting to 192.168.1.2.
Disconnected.
Disconnecting.
01:09 wlan0 ESSID: "KenP3S"
Access Point: 60:60:1F:05:E0:BD
Mode: Master Channel: 10 (2.457 GHz)
Tx-Power: 24 dBm Link Quality: 70/70
Signal: -27 dBm Noise: -89 dBm
Bit Rate: 55.2 MBit/s
Encryption: WPA2 PSK (CCMP)
Type: nl80211 HW Mode(s): 802.11abgn
Hardware: unknown [Generic MAC80211]
TX power offset: unknown
Frequency offset: unknown
Supports VAPs: yes PHY name: phy0
wlan0.sta1 ESSID: unknown
Access Point: 00:00:00:00:00:00
Mode: Master (VLAN) Channel: 10 (2.457 GHz)
Tx-Power: 24 dBm Link Quality: 70/70
Signal: -10 dBm Noise: -89 dBm
Bit Rate: 52.0 MBit/s
Encryption: none
Type: nl80211 HW Mode(s): 802.11abgn
Hardware: unknown [Generic MAC80211]
TX power offset: unknown
Frequency offset: unknown
Supports VAPs: yes PHY name: phy0
/ # 
01:09 iwinfo
01:09
BusyBox v1.22.1 (2015-11-16 16:28:58 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.
/ # iw reg set BO
/ # iw dev wlan0 set txpower limit 2400
/ # 
01:09 iwinfo
01:09 iw dev wlan0 set txpower limit 2400
01:09 iw reg set BO
Connected.
Connecting to 192.168.1.1.
Disconnected.
Disconnecting.
01:09 ==================
01:08 File upload OKE
01:08 Checking file ... 
01:07 Update djiled.sh on Aircraft
01:06 Final process ...
01:06 hmod 775 /sbin/watchlog.sh
/ # 
01:06 File upload OKE
01:06 Checking file ... 
[... 52 lines cut ...]
01:03 RC Ready ...
Disconnected.
Disconnecting.
01:40 
01:40 Please proceed to STEP 2
01:40 
01:39 File upload OKE
01:39 Checking file ... 
01:39 Upload watchlog.sh to aircraft
01:38 File upload OKE
01:38 Checking file ... 
01:38 Upload watchlog.sh to RC
01:37 File upload OKE
01:37 Checking file ... 
01:37 Configure Telnet to Aircraft
01:36 File upload OKE
01:36 Checking file ... 
01:36 Configure Telnet to RC
01:35 Aircraft Ready ...
01:35 RC Ready ...

I found this in the 1.8.8 Pro bin. : (the pro has no visible wifi)
I am looking for a way to Make this type of mod for the PRo/ Avd :) 

Wifi configuration file ##########################################.## Empty lines and lines starting with # are ignored.# ap: SoftAP mode.# sta: Station mode.# p2p: Enable Wifi Direct Support for peer-to-peer connectibity.WIFI_MODE=ap.# GPIO pin to enable or disable WiFi.#WIFI_EN_GPIO=95..##### STA mode configuration ##########################################.# SSID.ESSID=amba_boss.# Passphrase. Leave empty at no security mode; please edit wpa_supplicant.conf when your AP cannot be detected.PASSWORD=1234567890.# Device Name for AMBA Discovery Protocol (optional).STA_DEVICE_NAME=amba-1..##### Wifi Direct configuration ##########################################.# Find devices with correct name prefix and automatically connect at startup.P2P_AUTO_CONNECT=no.# Auto-connect with devices if the name prefix matches.P2P_CONNECT_PREFIX=amba.## Do not enable this optional field unless you are certain.# please provide a unique name amoung multiple devices to prevent confusion.#P2P_DEVICE_NAME=amba-1.## Do not enable this optional field unless you are certain.# Set the default P2P GO Intent.#P2P_GO_INTENT=0.## Do not enable this optional field unless you are certain.# Specify P2P operating channel.#P2P_OPER_CHANNEL=1.## Do not enable this optional field unless you are certain.# Default mode for HT40 enable when operating as GO.#P2P_GO_HT40=0..##### SoftAP configuration ##########################################.# SSID (1 ~ 32 characters).AP_SSID=amba_boss.# IP address.LOCAL_IP=174.168.42.1.# IP subnet mask.LOCAL_NETMASK=255.255.255.0.# IP pool starting address of DHCP server.DHCP_IP_START=174.168.42.2.# IP pool end address of DHCP server.DHCP_IP_END=174.168.42.6.# Wifi channel number, set 0 to use Auto Channel Selection.AP_CHANNEL=0.# Maximum number of stations allowed in station table.AP_MAXSTA=5.# If you say yes here, all WPA/WEP settings will be ignored.AP_PUBLIC=no.# WPA Passphrase (8 ~ 63 characters).AP_PASSWD=1234567890..##### Advanced WPA configuration ##########################################.## Do not modify this section unless you are certain.# This is a bit field, bit0 = WPA, bit1 = WPA2.WPA_VERSION=2.# This is a space separated list of encryption algorithms, TKIP CCMP.WPA_PAIRWISE=CCMP..##### Advanced WEP configuration ##########################################.## Do not modify this section unless you are certain.# If you say yes here, AP_PASSWD, WPA_VERSION, WPA_PAIRWISE will be ignored.AP_WEP=no.# Bit fields of allowed authentication algorithms, bit 0 = Open System, bit 1 = Shared Key.AUTH_ALGS=1.# The key number to use when transmitting. It must be between 0 and 3, and the corresponding key must be set..WEP_DEFAULT_KEY=0.# A key may be a quoted string (5 or 13 characters) or unquoted hexadecimal digits (10 or 26 digits), depending on whether 64-bit or 128-bit WEP is used.WEP_KEY0=1234567890.WEP_KEY1="vwxyz".WEP_KEY2=0102030405060708090a0b0c0d.WEP_KEY3=".2.4.6.8.0.23".ÿÿÿÿÿÿ1...rª-ª........ .......G.......................r.......s×€V....

KennethMcNutt, 2017-02-05:
Also thought this might be useful..
DJI PHANTOM 3 PROFESSIONAL HASHES AND PASSWORDS

UPDATE: ftp password is cracked: admin999

Some hashes and passwords:

1. System access:

root:$6$zi2k1pqQ$aYoxWoM9suJzq4xcIz0Uh/sMBQxIrM7QzqpNH.UMrX6TAmBx37jN0ygKlnpmHkgilWV5YzpfikkaylTWWo8RU0:16184:0:99999:7:::
bin:*:10933:0:99999:7:::
daemon:*:10933:0:99999:7:::
adm:*:10933:0:99999:7:::
lp:*:10933:0:99999:7:::
sync:*:10933:0:99999:7:::
shutdown:*:10933:0:99999:7:::
halt:*:10933:0:99999:7:::
uucp:*:10933:0:99999:7:::
operator:*:10933:0:99999:7:::
ftp:$6$Kt6U5MHk$aCy81r9Wz49TlfDwSPHkx8bEouNFdt0khJg7Pj1HOJtECe5.t9KfNWOKKQXnyVqjd5whliLQGTQkXfB8p3rBX/:10933:0:99999:7:::
nobody:*:10933:0:99999:7:::
default::10933:0:99999:7:::
a8sdk:HHTiH8SGMjHoM:0:0:99999:7:::
root, ftp, default, a8sdk has an access to /bin/sh

2. Wireless networks:

ESSID=Amba_AP
PASSWORD=PWFORAMBA

ESSID=amba_boss
PASSWORD=1234567890
P.S.

Wi-Fi Channel is number 5

ifconfig wlan0 192.168.42.1
…dhcp-range=192.168.42.2,192.168.42.6

Have a safe flight! ;-)

notsolowki, 2017-02-06:
@FERRARIPT what would be involved in making ht elatest dji go app send all the parameters like 2.4

ferraript, 2017-02-06:
@notsolowki: I don't know if I would be able to hack DJI GO 3 in such a way
in 2.4.3, there is one class (dji\pilot\fpv\stage\cs) with onClick method
and according to clicked button, it does some actions
if "Reset All Settings" is clicked, there are some checks and in the end DataFlycResetParams class sends command to AC

however, in 3.0.1, every button has its own class and onClick method
(fortunately?) class for "Reset All Settings" button exists (dji\setting\ui\flyc\ResetView), but it's not used
maybe I could redirect actions from here to another button - and the most similar button (I mean with similar class structure) is ClearFlightLogsButton

but I can't guarantee it will work

I still think the easiest way is to use DJI GO 2 with existing "Reset All Settings" button

[436 more replies]