o-gs/dji-firmware-tools #11: dji_flyc_param_ed.py

notsolowki, 2017-01-15:
So this will read the structure array and export the parameters and variables?!?!

mefistotelis, 2017-01-15:
Yes. I will soon make a description of it in README.

notsolowki, 2017-01-15:
Do you know anything about whats going on right here.

notsolowki, 2017-01-15:
WOW!. thankyou!!! when trying to repack it ends with

Traceback (most recent call last):
  File "dji_flyc_param_ed.py", line 413, in <module>
    main(sys.argv[1:])
  File "dji_flyc_param_ed.py", line 404, in main
    flyc_update(po,fwmdlfile)
  File "dji_flyc_param_ed.py", line 344, in flyc_update
    raise NotImplementedError('Function unfininshed.')
NotImplementedError: Function unfininshed.

notsolowki, 2017-01-16:
Is it ready to use?!?!?

MrBurnsAT, 2017-01-16:
I think it should work.

notsolowki, 2017-01-16:
hey mrburns. im on 1.5+ right now. when i upgrade to 1.7 whats the story with the controller. will it need to be updated from 1.5.7 to 1.6 and if so whats the method of upgrading the controller only?!?

notsolowki, 2017-01-16:
it works!!!!!!! holly s**** thank you soo much!!!@!@

MrBurnsAT, 2017-01-16:
Mefistotelis is my Hero!!!!!!!

He is genius!!!!

notsolowki, 2017-01-16:
I cant take all this excitement!@##!@

chazzzzz, 2017-01-17:
Hey this is pretty cool. I think I found two things I'll try changing...I think the index 264 is when you get that annoying low battery warning and 256 is also as annoying, the 10% battery auto landing thing. Maybe I'll change them both to 0. What do you guys think?

"index" : 264,
		"typeID" : 0,
		"size" : 1,
		"attribute" : 11,
		"minValue" : 0,
		"maxValue" : 100,
		"defaultValue" : 15,
		"name" : "g_config.voltage2.level1_smart_battert_gohome_0",
		"modify" : true

"index" : 265,
		"typeID" : 0,
		"size" : 1,
		"attribute" : 11,
		"minValue" : 0,
		"maxValue" : 100,
		"defaultValue" : 10,
		"name" : "g_config.voltage2.level2_smart_battert_land_0",
		"modify" : true

any guesses what index 338 is??? Speed perhaps?
"index" : 338,
		"typeID" : 5,
		"size" : 2,
		"attribute" : 11,
		"minValue" : 0,
		"maxValue" : 100,
		"defaultValue" : 90,
		"name" : "g_config.control.throttle_output_max_0",
		"modify" : true

mefistotelis, 2017-01-18:
For me the warnings is useful; but if you want it gone, I'd change it to 1% and 2% rather than 0 - it is safer not to use zeros in places where there were no zeros before.

Also, don't touch min and max values, change only the "default".

chazzzzz, 2017-01-18:
Yeah I already did it...Default values yes, thanks...I figures as long as I stayed at the min value it would be okay...It did act a little differently during updating and took a long time, about 30 to 40 min...It's probably because it updates everything with that debug file, but it DID seem to take...The Dji go app still displays the normal values for these settings which I did expect...I will have to test it...Thanks bud...

notsolowki, 2017-01-18:
changing the battery values have no effect.

ferraript, 2017-01-20:
for clarification, has anybody had any success of changing any parameter, so it was effective?

notsolowki, 2017-01-20:
Yea the height

notsolowki, 2017-01-20:
if some tools ever turn up to pull the rest of the variables in the filght controller i have no problems testing them so descriptions can be made :)

ferraript, 2017-01-20:
>changing the battery values have no effect

maybe AC is pulling those from battery

notsolowki, 2017-01-21:
Hey mefistotelis, are you working on any more formware projects?

mefistotelis, 2017-01-21:
> are you working on any more firmware projects?

Besides the Phantom tools? no.

notsolowki, 2017-01-21:
Sorry, i mean and more firmware tools for phantom tools.

notsolowki, 2017-01-22:
are you still working on the phantom firmware tools?

notsolowki, 2017-01-22:
also i found if you tap flight controller in dji go under about, 12 times it gives you a uuid

notsolowki, 2017-01-22:
i noticed that if you go to dji go and look at the gains under advanced settings, their values are 8-12-,100 default. if you go back one menu and retore default settings then 130 gets populated into the boxes. weird

mefistotelis, 2017-01-22:
> are you still working on the phantom firmware tools?

I'm working on the remote controller fw.

There is also some work to do with firmwares support  - if you use test_all.sh, you'll notice that some tests fail. I'm not sure if II'll be working on that.

notsolowki, 2017-01-22:
do you think the ESC firmware monitors the RPM of the motors. i Re-calibrated the RC but instead of pushing the sticks all the way i only pushed them to about 20 percent. when i restarted the motors and pushed the sticks all the way i got a maximum motor speed reached

mefistotelis, 2017-01-22:
> do you think the ESC firmware monitors the RPM of the motors.

Of course it does.

ESC gets instructions from the flight controller; There is nothing interesting there for hacking. But it does contain some proprietary algorithms, ie. the active breaking implementation, which other companies (and open-source software) didn't had for many months after DJI released it. This is why ESC firmware is encrypted.

notsolowki, 2017-01-22:
have you found anything in the RC firmware thats interesting.

notsolowki, 2017-01-22:
i see alot of this in the app 

  Ldji/midware/data/model/P3/DataGimbalGetPushParams
  dji/midware/data/model/P3/DataGimbalGetPushParams 
   Ldji/midware/data/model/P3/DataRcGetPushParams
   dji/midware/data/model/P3/DataRcGetPushParams 
  Ldji/midware/data/model/P3/DataRcGetPushParams

notsolowki, 2017-01-22:
Some how my controls ended up backwards,

notsolowki, 2017-01-22:
the csc is up and left now

mefistotelis, 2017-01-22:
> can you tell me how to use the test scripts i cant seem to figure out the cml

for test_all.sh - you just run it. The others are used within test_all.sh, so to just look inside to get it.

These shell scripts are usable in any bash-compatible shell; I'm using msys2 to run them on my Windows machine.

notsolowki, 2017-01-22:
thankyou, can you tell me how can i downgrade my rc. some how i got the sticks screwed up and i think the firmware needs to be reflashed. i cant reset them ive tried everything i can think of

notsolowki, 2017-01-22:
okay i dont understan. i reinstall the rc and the aircraft and the app and still my controls are backwards???

ferraript, 2017-01-22:
another recalibration doesn't help?

notsolowki, 2017-01-22:
nope

MrBurnsAT, 2017-01-22:
Have u maybe Set custom Stick Mode???

notsolowki, 2017-01-22:
no. i tried all modes. when i setup custom they were backwards too

notsolowki, 2017-01-22:
this started when i installed dji go 2.4.2 and reset the default settings in the aircraft menu

notsolowki, 2017-01-22:
@mefistotelis : can you shed some light on this maybe

notsolowki, 2017-01-22:
i figured it out. i reinstalled the old app and upgraded my rc and ac. then in the old app i reset all to default. and all was back to normal. its weird. in dji go 2.4 if you hit reset it says its going to reset the flight controlelr parameters to default. if ou hit reset in the new dji go apk it says its going to reset the gain/expo....

notsolowki, 2017-01-22:
it might have happened when i changed flyc_param_infos in the old app then hit restore default settings.

ferraript, 2017-01-22:
there is reverse_input_0 in both DJI GO and AC firmware
maybe you had this set to wrong value

you can check it:
do a flight now, when you have controls right
then get the DAT flight logs, one from this last flight and one where the controls were reversed
then generate config file using DatCon and check that reverse_input_0 value

notsolowki, 2017-01-22:
i compared the values edited with flyc_param_ed.py to the new binary generated and they seem to change successfully within the structure

notsolowki, 2017-01-22:
i didnt upgrade past 1.7.6 that i was on. i upgraded the rc to 1.60 and the drone to 1.8. i noticed that when 1.8 is installed it updates alot differently

notsolowki, 2017-01-22:
okay before all this happened , i flashed a modified binary with all the pantilt-"pitch" settings really low. but the dji go app 3.1.1 wouldnt let me set them any lower or higher still. but when i would hit restore default it would set them to 130, which is out of their range according to the app. and the 130 made a difference in the pitch gain because i could set it back to 120 and notice a difference. but now that i have a non-modified FW  setting them to default only sets them abck to 100.. weird?!?!?

notsolowki, 2017-01-22:
When i tried to downgrade back to 1.7.60 after i had to update to 1.8 to fix my controls, it wouldnt talk to my rc. After i did the normal method to downgrade it said i was on 1.7.6+ and i got the no signal error. Then i tried to downgrade again to 1.6 but it wouldnt even start the process saying it cant be downgraded. Knowing this isnt true because ive done it 100 times. I decided to put the p3x_fw_debug on the sd card with 1.7.6 firmware file and started tge process. After it finished it said i was on 1.7.6 instead of 1.7.6+ and my controller issues went away

notsolowki, 2017-01-22:
So dji go 2.4 updated some parameters in flyc when i did a default reset. It even says its going to reset the flight controller parameters when 3.1.1 says it will reset the gain/expo only

notsolowki, 2017-01-23:
i figured out how to use the edited parameters

notsolowki, 2017-01-24:
@mefistotelis I'm sorry, i should have said can you add the go home speed to flyc_param_ed.py? thankyou again!!

ferraript, 2017-01-24:
>can you please add the go home speed to flyc_param_ed.py?

I don't think that's possible
1. flyc_param_ed extracts all params from firmware and as you can see, there is no such a parameter there
2. I believe that this parameter comes from battery, because in each of my flight logs I can see:
`155 [smart_battery]this fireware calc gohme speed:7.800000 - land speed:2.500000`

mefistotelis, 2017-01-24:
Yes, flyc_param_ed extracts all parameters from the array.

notsolowki, 2017-01-24:
the speed is not in the battery its in the flight controller binary. 
@mefistotelis is the go home speed in the array. i dont hink i seen it., but i did see it in the flight controller

mefistotelis, 2017-01-24:
Ok, since you seem determined for this, I looked at the code.

Go home speed is hard-coded into firmware. It is hard to find all places which would have to be modified to change it. Definitely it appears at:
<pre>sub_80680F4, sub_8068294; used value:
.text:08068290 flt_8068290     DCFS 7.8
sub_8068F90; used value:
.text:08069068 dbl_8069068     DCFD 7.80000019
</pre>

To modify it, try changing these two values. If this won't help, you'll have to search further.

ferraript, 2017-01-24:
in which module binary you found it, @mefistotelis , m0305 or m0306?

mefistotelis, 2017-01-24:
We don't have unencrypted m0305. Also, it contains a loader, not a real flyc code. It would help in hacking encryption, but not in changing any parameters.

notsolowki, 2017-01-24:
i wish i knew how to byte patch. do you know any easier way for us to do it?

notsolowki, 2017-01-25:
how can i allign part of the instruction to the hex view in ida. say i highlight 7.8 and align it with hex view it selects a whole lot of data in hex view. how can i change the instruction and view what the changes would look like in hex view

mefistotelis, 2017-01-25:
You can find the correct byte values in [any converted](http://lmgtfy.com/?q=float+to+hex+online).

At :08068290 you have float, at :08069068 there's double.

notsolowki, 2017-01-26:
im going to make a list of know parameters, ill update this as i figure them out. if anyone know of any add them in a comment and ill add them to the list

"name" : "g_config.control.vert_up_vel_0" ascend speed
"name" : "g_config.control.vert_down_vel_0" descend speed
"name" : "g_config.engine.upper_bound_0" motor speed limit
"name" : "g_config.engine.idle_level_0" controls idle speed on ground
"name" : "g_config.flying_limit.max_height_0" controls maximum altitude 
"name" : "g_config.novice_cfg.max_height_0" maximum height in beginner mode

mefistotelis, 2017-02-09:
Here's a little present - a command which changes the gohome speed value from 7.8 to 15.3:
<pre>xxd  -p -c16 P3X_FW_V01.07.0060_mi01.bin | \
  sed 's/\(9a99f940\)\(2de9f041\)/cdcc7441\2/' | \
  sed 's/0000004033331f40/9a99999999992e40/' | \
  xxd -r -p -c16 > P3X_FW_V01.07.0060_mi01-mod.bin</pre>

To verify which bytes were changes, you can use:
<pre>cmp -l P3X_FW_V01.07.0060_mi01-mod.bin P3X_FW_V01.07.0060_mi01.bin</pre>

[62 more replies]