o-gs/dji-firmware-tools #13: Allowed flight in DJI No Fly

aka1ceman, 2017-01-19:
Have you considered the possibility of attempting to remove DJI go no fly zone? I have a situation where I have legal permission to fly by control tower but DJI Go will not let me unless I upgrade to the newest firmware then register with them... Which I am not trying to do. I stopped upgrading at 1.6.

notsolowki, 2017-01-19:
The nfz stuff was in tge mobile app. When i was digging around i found a
file with all the names and latitudes and longitudes of the no fly zones

notsolowki, 2017-01-19:
dont hold me to the location of the file. all i know is its in the app. and its full of nothing but latitudes/longitudes/location names

notsolowki, 2017-01-19:
okay try going to /data/app/dji.pilot and extracting base.apk then use android apk editor pro to open baseapk and look for flyforbid.json, then see if you can find the name of your airfield and remove it. i cant promise that it will work but IF you have permission i dont see the harm

ferraript, 2017-01-19:
I already tried editing DJI GO in this way, but it had no effect (besides the fact that red circle wasn't displayed on the map), because no-fly zones are stored in aircraft's firmware too

aka1ceman, 2017-01-20:
On the older FW, I don't get red circles at all. Only a No Fly Zone and won't allow the motors to arm. Do you know the location of the file you found? I'm looking still

notsolowki, 2017-01-20:
i wouldn't loose any sleep over it. there are other ways you could approach this problem.

aka1ceman, 2017-01-20:
I have rolled back to 1.3.2 and have a working old Litchi and still not possible to bypass yet. 
I'm not trying to place aluminum foil over it....lol

mefistotelis, 2017-01-20:
In my voyages through firmware, I wasn't able to find a list on NFZ areas. I'm sure the craft stores it somewhere, but it probably isn't a part of firmware update. I think it is updated separately, when the DJI Go app sends it.

And Dji Go gets it simply via HTTP:
https://flysafe-api.dji.com/api/release_limitarea.json/?updated_at=

notsolowki, 2017-01-20:
i found one in the base.apk its called flyforbid.json, i wonder if disabling airport mode in the flight controller would help.

aka1ceman, 2017-01-20:
Could we redirect Go to look elsewhere ? Would that help?

mefistotelis, 2017-01-20:
It would be interesting to try. The page address can be modified, but I'm pretty sure the app will use previous configuration if the answer it gets from HTTP request is invalid. Meaning, we need to set up a fake server which would answer with modified (or empty) but still valid list.

ferraript, 2017-01-20:
yeah, I can see that webpage in DJI GO sources too
for start, I would just change the latitude and longitude of nearest zone to something else and also I'd change updated_at parameter to newer date
and save this file somewhere to web and change the address in DJI GO
who wants to try? :)

notsolowki, 2017-01-20:
where at in the dji source files does it point to the website?

ferraript, 2017-01-20:
>where at in the dji source files does it point to the website?

it depends on the version
for example, in 2.4.3, it's in \dji\pilot\flyforbid\FlyforbidUpdateService.java
in 3.1.1, it's in \dji\pilot\flyforbid\c.java

notsolowki, 2017-01-20:
what heppens if you edit out the website adn change the json file to remove your lat,lng

notsolowki, 2017-01-20:
well if it can fly without the phone i dont see hwo the mobile app could handle anything other than waypoints,mission

notsolowki, 2017-01-20:
unless someone that specializes in java comes out of nowhere and develops an app that can send the new parameters we are stuck with what we have currently

mefistotelis, 2017-01-22:
Mobile programming is one of the easiest.

Anyway, since you discovered how to use "default" settings, you can use them to test new values.

notsolowki, 2017-01-23:
I didnt really test it I changed some pitch parameters, note i set them lower, To like 1 and 2 "same for mobile flyc" and when i restored default it set them higher. So idk what happened but i assume its not going to be that easy. Unless you have confirmed this?

notsolowki, 2017-01-23:
And besides all that, i think the esc is going to prevent any more rpm than they programmed.

notsolowki, 2017-01-23:
It had to be somthing with flyc param in the app because when i installed the non modified flyc and restored defaults , my controller was back to normal and the gain values were too

ferraript, 2017-01-24:
today I tried for myself, what I was suggesting
I downloaded that original [release_limitarea.json](https://flysafe-api.dji.com/api/release_limitarea.json/?updated_at=) file, I edited it in a way that I moved all restricted areas somewhere to the ocean and I set update time to actual time
I saved this edited file to my Google Drive and I put direct link into DJI GO instead their link
I installed this DJI GO and... strange thing happened
now it looks like DJI GO is using old GEO SYSTEM map, because there are two red circles now in my city, first one around the airport and second one around the whole city
I was able to start the motors but DJI GO keeps popuping message about restricted area
I'll give it another try...

ferraript, 2017-01-25:
I figured out I was idiot because I hadn't noticed that in DJI GO sources, timestamp is appended to that address and so entire address to my file was invalid

second try: I changed the address of json file in a way that it is correct even after appending that timestamp
result: DJI GO downloaded the file, but didn't import anything

then I noticed another thing, country is defiined as integer in DJI GO, but it is string in json file
third try: I replaced all countries' strings for numbers in json file
result: DJI GO downloaded and imported all areas from my json file

so all I need to do now is to check (in the field - somewhere near airport) if DJI GO really sends all those new nofly parameters to the AC too
but we have bad frosty and windy weather here, so it'll take some time

TadbirPaydar, 2017-02-02:
Hey ferraript 
It is very surprising results 
Let us help you if there is some problems to solve and let us know the results

ferraript, 2017-02-02:
hi @TadbirPaydar
thanks, well, until I try it in the field, it's too early to say
maybe all of this is useless

I'm using DJI GO 2.4.3 with link to [this flyforbid json file](https://drive.google.com/uc?export=download&id=0BzjAFKwV8KMYdjRjVTQ2MjV0Vjg)

TadbirPaydar, 2017-02-02:
Hi @ferraript 
Why not work on 3.1.2 which is the last version ?

I decompile .APK with dex2jar and open .jar with jd-gui.
in \dji\pilot\flyforbid\FlyforbidUpdateService.java file cannot find that json file address.

is my decompilation right ?

TadbirPaydar, 2017-02-02:
As i see on 3.1.2 in \assets\flyforbid_airmap\ there are 29 files named flyforbid_airmap1.json to flyforbid_airmap28.json which forbidden areas are stored there and md5 of those files stored in \dji\pilot\flyforbid\FlyforbidUpdateService.java

So i think in this version these files must change

I have a scenario which @ferraript and @mefistotelis should check it:

First DJI loads forbidden areas to phantom somewhere else from firmware

Second DJI GO check for newer update and download it from internet and uploads it to phantom and stores them somewhere else from firmware 

So there are 2 attack types :

First the method which @ferraript is working on and 

Second to path firmware code which check for forbidden area in control loop which @mefistotelis should check.

any Idea ?

ferraript, 2017-02-02:
I don't use newer versions of DJI GO, because they are more complicated and they contain things I don't need / I don't like
I don't use dex2jar + JD-GUI, because they cannot decompile everything and so it is not possible to recompile the code back completely
so at first I decompile apk using jadx - it decompiles most of the code into java - then I use it to search for the parts I am interested in
secondly, I decompile apk using APK Studio - it decompiles everything (but into smali code) - here I make my modifications and recompile it back to apk

TadbirPaydar, 2017-02-03:
Some sniffing results may help :
```
GET https://flysafe-api.dji.com/api/release_limitarea.json/?updated_at=0&version=1.3&lat=##.302#####625851&lng=-###.630#####96885&country=GB&os=android&appVersion=2.4.4&timestamp=1451669629813&signature=78####E72EABB459CACA537111#######EAC31BE4F8408E8E69AF166&datasource=airmap HTTP/1.1
Host: flysafe-api.dji.com
Connection: Keep-Alive
Accept-Encoding: gzip
X-API-Key: AB#####8-15F1-B##C-32##-4BD#######B73
```

and also :

```
GET https://api.airmap.io/data/v1/search/dji?key=A######8-1####-###C-3###-4B####B73&unique_id=123&latitude=####.3###3562###51&longitude=-###.630#######885&radius=20000 HTTP/1.1
Host: api.airmap.io
Connection: Keep-Alive
Accept-Encoding: gzip
X-API-Key: A######8-####1-####C-####8-4########B73
```

See this :

[https://www.rcgroups.com/forums/showpost.php?p=33600003&postcount=46797](https://www.rcgroups.com/forums/showpost.php?p=33600003&postcount=46797)

Any Idea ?

ferraript, 2017-02-03:
yeah, it's complicated like hell :D
just another reason for not using firmwares with GEO

TadbirPaydar, 2017-02-04:
Do you know exact version of firmware and Go which does not have NFZ ?
And how to downgrade to them ?

ferraript, 2017-02-04:
I can assure you that all versions of DJI GO have NFZ, even DJI GO's predecessor - DJI Pilot, has NFZ
so I'm afraid it's the same story with firmwares

ferraript, 2017-02-04:
by the way, today I finally took my downgraded P3A outside to test my NFZ hack in DJI GO
result is: it's not working
AC wasn't able to penetrate border of NFZ
and somehow (my tablet wasn't connected to internet), in DJI GO, there was once again large light-red circle over my whole city
so there are two possibilities:
1. DJI GO didn't send my new hacked NFZ coordinates into the AC, but instead, AC sent its firmware-stored coordinates back to DJI GO
2. I deliberately hadn't touched the file flyforbid.json stored in DJI GO apk (in \res\raw), because I wanted to see on the map then where the border of NFZ exactly is - and who knows, maybe DJI GO uses that file for NFZs setting too

I am preparing new hacked DJI GO, this time I am replacing that flyforbid.json file too
but I don't know when the good weather is to test it
today there was dense mist with maybe 100 m visibility, they forecast rain for tomorrow

ferraript, 2017-02-05:
>In my voyages through firmware, I wasn't able to find a list on NFZ areas

@mefistotelis: I found it!!!
in P3S_FW_V01.07.0060_mi01.bin, at address 0x0A1620, there is `85B0E6022F0F44017908`
85B0E602, in Little-endian, is 48672901
2F0F4401, in Little-endian, is 21237551
7908, in Little-endian, is 2169
and (48.672901,21.237551,2169) are coordinates+radius of airport in my city

TadbirPaydar, 2017-02-05:
Great news !!!

What about last version firmware?
And how to bypass ?

mefistotelis, 2017-02-05:
Intersting. This is within an array which consists of 2500 entries. Each entry looks like:
<pre>00000000 NoFlyArea       struc ; (sizeof=0x22, mappedto_9) ; XREF: .data:no_fly_areas/r
00000000 field_0         DCD ?
00000004 lat             DCD ?                   ; base 10
00000008 lon             DCD ?                   ; base 10
0000000C radius          DCW ?                   ; base 10
0000000E field_E         DCW 9 dup(?)
00000020 field_20        DCW ?
00000022 NoFlyArea       ends</pre>

ferraript, 2017-02-05:
>Each entry looks like

actually, it's written in bin file too: `----lat------lon-----radi-con-cla-id----sta-----end--` and it fits, so:
4-byte latitude
4-byte longitude
2-byte radius
2-byte country_code
1-byte class
2-byte area_id
1-byte begin_at
1-byte end_at

mefistotelis, 2017-02-05:
That would mean a single entry has 17 bytes; it's very possible - I might misinterpreted two entries as a single entry.

TadbirPaydar, 2017-02-05:
So actually it's enough to write a patch for lat and lon.

But remaining question is 

> What is relation between this hard-coded NFZ and updating from dji GO software ?

ferraript, 2017-02-05:
that's not clear at the moment

today I was out at our airport to test my latest DJI GO hack
this time I really deleted all real references to airport's NFZ settings
when I started DJI GO at the place, all was good - no NFZ
but as soon I started AC, DJI GO loaded NFZ info from it and boom, one dark-red circle with 2 km radius and one light-red circle with 4 km radius

conclusion is, NFZs must be deleted (or preferably coordinates changed) in AC's firmware first
and then we can determine, if something needs to be done with DJI GO too or not

TadbirPaydar, 2017-02-05:
@ferraript How did you find it ?
I Wanna work on last version.
Did you open P3S_FW_V01.07.0060_mi01.bin in IDA or any other way?

ferraript, 2017-02-05:
I took 48.672901, tried to convert it into float, double, signed int (multiplied by 1000000) and I looked for the corresponding HEX value with both possibilities, that it could be written in big-endian or little-endian

TadbirPaydar, 2017-02-05:
OK Nice work but it is needed a method to find start of NFZ srtcture.
So didnt you open in IDA ?
may be @mefistotelis can help

mefistotelis, 2017-02-06:
Before that array, there's another one which also stores airport coords. But this second array has 8-byte entries - just coods, no radius or other attributes.

TadbirPaydar, 2017-02-06:
Hi @mefistotelis 

> Before that array, there's another one which also stores airport coords. But this second array has 8-byte entries - just coods, no radius or other attributes.

I did it in IDA and searched for "82517331" and "4EB1D53" which is latitude for first coordinate.
but no result 
Any mistake ?
I think its better to find conditional jmp which decides about NFZ and patch it.
Any success ?

> .data:080BDF64
.data:080BDF64 loc_80BDF64                             ; DATA XREF: sub_80945FE+4o
.data:080BDF64                                         ; sub_80945FE+Ar ...
.data:080BDF64                 ANDEQ           R1, R0, R7,ASR R0
.data:080BDF64 ; ---------------------------------------------------------------------------
.data:080BDF68                 NoFlyZone <82517331, 4232682763, 921, 124, 0, 9074, 0, 0>
.data:080BDF79                 NoFlyZone <78246084, 15465563, 1338, 744, 0, 4100, 0, 0>
.data:080BDF8A                 NoFlyZone <76531111, 4226268017, 1872, 304, 0, 7399, 0, 0>
.data:080BDF9B                 NoFlyZone <74716084, 4200012100, 1601, 124, 0, 9183, 0, 0>
.data:080BDFAC                 NoFlyZone <72980787, 4210352750, 1084, 124, 0, 9216, 0, 0>
.data:080BDFBD                 NoFlyZone <71927778, 114080000, 1226, 643, 0, 10156, 0, 0>
.data:080BDFCE                 NoFlyZone <71698071, 128901473, 1666, 643, 0, 2850, 0, 0>
.data:080BDFDF                 NoFlyZone <71285333, 4138197171, 1271, 840, 0, 3465, 0, 0>
.data:080BDFF0                 NoFlyZone <70910745, 4141727656, 906, 840, 0, 3909, 0, 0>
.data:080BE001                 NoFlyZone <70638041, 4134964851, 1139, 840, 0, 972, 0, 0>
.data:080BE012                 NoFlyZone <70625569, 147899927, 1254, 643, 0, 1867, 0, 0>
.data:080BE023                 NoFlyZone <70467326, 4137531414, 866, 840, 0, 1211, 0, 0>
.data:080BE034                 NoFlyZone <70344278, 4144022552, 955, 840, 0, 694, 0, 0>
.data:080BE045                 NoFlyZone <70331394, 4145372471, 1110, 840, 0, 7028, 0, 0>
.data:080BE056                 NoFlyZone <70212744, 4143958993, 1012, 840, 0, 4390, 0, 0>
.data:080BE067                 NoFlyZone <70197502, 4146503417, 1459, 840, 0, 8155, 0, 0>
.data:080BE078                 NoFlyZone <70133889, 4151385629, 931, 840, 0, 3502, 0, 0>
.data:080BE089                 NoFlyZone <70068140, 24973178, 1488, 578, 0, 3852, 0, 0>
.data:080BE09A                 NoFlyZone <69784431, 170596213, 2022, 643, 0, 6165, 0, 0>
.data:080BE0AB                 NoFlyZone <69764592, 61550685, 1891, 643, 0, 1067, 0, 0>
.data:080BE0BC                 NoFlyZone <69722428, 29883596, 1265, 578, 0, 4936, 0, 0>

TadbirPaydar, 2017-02-06:
There is a mistake !
in "release_limitarea.json" we have 
`"lat":23.471698761,"lng":162.38971996307`
 which belongs to "City Airport Manchester"
but when you search for it in google map it say :
`53.4714422,-2.3919457` for coordinates

some kind of coordinate conversion is needed.

ferraript, 2017-02-06:
@TadbirPaydar 
you are looking into my patched release_limitarea.json file, that's why it doesn't work for you
original is [here](https://flysafe-api.dji.com/api/release_limitarea.json/?updated_at=)

TadbirPaydar, 2017-02-06:
@ferraript 
Thanks. I had such stupid mistake

There are some questions which can help to eliminate NFZ.

1- What is relation between this hard-coded NFZ and updating from dji GO software ?
Logically update must be one directional from GO to firmware because GO must download updated NFZ and GO uploads it to firmware.

But @ferraript 

> when I started DJI GO at the place, all was good - no NFZ
but as soon I started AC, DJI GO loaded NFZ info from it and boom, one dark-red circle with 2 km radius and one light-red circle with 4 km radius

Didn't you make mistake @ferraript  ?

What is logic behind updating GO from firmware ?

ferraript, 2017-02-06:
you have already asked this kind question and I have already answered it
I have no idea why AC doesn't update its NFZ list from DJI GO

but it's confirmed now that if there is specific NFZ stored in AC and it's not stored in DJI GO, DJI GO loads info about this NFZ from AC

TadbirPaydar, 2017-02-06:
Ok 
I think the answer of that question is :
No update applied to firmware. Because it just hard-coded 
New NFZ updates in new firmware version. 

But I think GO downloads NFZ from Internet or firmware just for showing.
GO sends fly signals to firmware even if it was in NFZ. But firmware itself decides whether to fly or not

TadbirPaydar, 2017-02-07:
@ferraript 

> 85B0E602, in Little-endian, is 48672901
2F0F4401, in Little-endian, is 21237551

Did you test on P3X_FW_V01.10.0090.bin 
i couldn't find

TadbirPaydar, 2017-02-07:
I reinstalled official firmware 1.10
After that I downgraded to my Patched version with 2 bytes changed
Copied my patched version and made a file with name P3X_FW_DEBUG on root of sdcard with zero byte size.
So I downgraded by this way.

After that phantom restarts and beep continues.

But I didn't correct any CRC in my patched version.

Do you know if phantom checks it's firmware CRC ?

If so how can I bypass it or how to correct CRC of my patched firmware version ?

mefistotelis, 2017-02-07:
> I downgraded to my Patched version with 2 bytes changed

How did you patch it? Which version did you patch?

> Do you know if phantom checks it's firmware CRC ?

Yes we know. Yes it does.

> If so how can I bypass it or how to correct CRC of my patched firmware version ?

This is what the firmware tools are for.

TadbirPaydar, 2017-02-07:
I patched 2 bytes which belong to radius of my city Airport from "D0 D1" to "01 00" which make the radius 1 meter.

I patched main bin file "P3X_FW_V01.07.0060.bin" with Winhex.

OK please tell me where CRC is stored in main bin file or modules .

@mefistotelis
In "P3X_FW_V01.07.0060.bin" if I search for 85B0E602 which is latitude of NFZ near @ferraript winhex can find it.

But in upper versions like 1.08 or 1.10 cannot find.

May be some kind of encoding used for NFZ coordinates
Please tell me how can I find and change coordinates in such firmwares ?

mefistotelis, 2017-02-07:
1. Do not downgrade and modify at the same time.
2. Use the tools to extract and re-pack firmware.

And if you want to know where CRC is, check it within the python code.

TadbirPaydar, 2017-02-07:
> Do not downgrade and modify at the same time

What is the concept behind it ?

What about other versions and newer ?

mefistotelis, 2017-02-07:
I believe there is a downgrading instruction in #4 .
Besides the reasons mentioned in that thread, you are merging two possibly dangerous operations.

Also, you can update one module only to increase safety.

ferraript, 2017-02-07:
>I patched 2 bytes which belong to radius of my city Airport from "D0 D1" to "01 00" which make the radius 1 meter

it doesn't work like this
no offence, but you know even less about byte patching than I do
so don't make changes you don't understand at all

and now just let's hope your AC isn't bricked for good

notsolowki, 2017-02-07:
No i think he did it right, besides the checksum of the file not matching.
If his "byte-patching" was correct then he did it right. One way to tell
would be to load the elf into ida and check his changes.

TadbirPaydar, 2017-02-09:
@mefistotelis 

> And if you want to know where CRC is, check it within the python code.

Thanks 
I found and problem solved !

[203 more replies]