o-gs/dji-firmware-tools #14: Fly parameters in mobile app

notsolowki, 2017-01-22:
[image]

notsolowki, 2017-01-22:
@mefistotelis do you think the esc are going to hold back on any extra throttle even if we get the app send the new values

mefistotelis, 2017-01-22:
I don't know.

notsolowki, 2017-01-22:
That max motor speed warning wasnt very promising

notsolowki, 2017-01-22:
More interesting code for the RC
[image]
[image]
[image]
[image]
[image]

notsolowki, 2017-01-22:
[image]

ferraript, 2017-01-23:
almost every picture you posted is from the part of the code that reads and writes TXT flight logs
I'm familiar with these classes as [I created converter](http://www.phantompilots.com/threads/tool-win-offline-txt-flightrecord-to-csv-converter.70428/) for TXT logs
but I lost track of what you are trying to achieve now

notsolowki, 2017-01-23:
we need an app to be able to modify the flight controller parameters.i mean what else is left to do?

notsolowki, 2017-01-23:
mefistotelis thinks that the flight controller variables are updated the the mobile app

closed 2017-01-23 by notsolowki