o-gs/dji-firmware-tools #15: How to change the flight behavior

notsolowki, 2017-01-23:
i figured it out

notsolowki, 2017-01-23:
i should have said there are 2 possibilities. 1 being the above message, 2 being that install i installed the fw with p3x debug. i maybe even leaning more towards the p3x debug. let me know!! now you cann all edit and crash your drones.

notsolowki, 2017-01-23:
@mefistotelis

notsolowki, 2017-01-23:
@ferraript you should make a GUI to edit flyc_param. once all the parameters are defined. would be cool. i imagine it would be able to laod both flc_params. one from the app and one from the firmware. then make changes to both accordingly

notsolowki, 2017-01-23:
More notes: i didnt have to change anything in the app. just had to flash with p3x_debug

notsolowki, 2017-01-23:
back to the app having nothing to do with it

ferraript, 2017-01-23:
I'm lost in your thoughts once again :smile:
So is it possible to push params from app to AC or not?

>you should make a GUI to edit flyc_param

unfortunately, I'm neither Android nor Java developer

notsolowki, 2017-01-23:
you dont have to push them from the app

notsolowki, 2017-01-23:
you just have to install the fw with the p3x debug. example. WITH 1.7.6 INSTALLED modify the 1.7.6 flight controller binary to your liking. then use the p3x_debug to OVERWRITE 1.7.6 with 1.7.6

notsolowki, 2017-01-23:
i might add that it take 23 minutes to flash the fw with this method. it dont give very many indicators but it will make the update complete sound when its done. maybe even you could edit airport mode this was idk

notsolowki, 2017-01-23:
im still testing the parameters. but i can tell you that basic.craft.thrust makes it go from idle to max motor speed right away. if you set it too high it dances in the air lol

notsolowki, 2017-01-23:
and @ferraript, didnt you wirte that program to convert the logs? if you could make a windows gui like that. one that can read the parameters in the flyc_param file and list them in the gui. they would have boxes for the max,min,default value. each would have its own description. then it could save the changes to the json file.. right now im trying to figure out how the parameters behave so we can get some descriptions. unless someone else has the descriptions.. that would be cool

notsolowki, 2017-01-23:
okay another update: the parameters do have to be restored in the app. so the apps flyc and the firmwares flc have to be updated

notsolowki, 2017-01-23:
i can confirm my takeoff speed is 27 mph which is 12 meters per second

mefistotelis, 2017-01-23:
> i might add that it take 23 minutes to flash the fw with this method

If you're updating the whole firmware - note that you could update only the flyc module, would be faster and safer.

notsolowki, 2017-01-23:
i was wondering that. could i just delete all the un-needed modules when repacking. and dont it upgrade the mi01 binary the normal way? i wonder why it only works when i use the debug file. maybe ill try it the normal way again just to confirm.

mefistotelis, 2017-01-23:
No, that's not how it works.

You do not need the packing operation at all, you just need to rename the module you want to update. Correct file names are listed in the downgrade discussuin #4 .

notsolowki, 2017-01-23:
so
modules=0 mi01 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  ??

mefistotelis, 2017-01-23:
No. File name.

notsolowki, 2017-01-23:
so i should rename mi01 to m0305.bin?

mefistotelis, 2017-01-23:
No. [correct file name for m0306](https://github.com/mefistotelis/phantom-firmware-tools/issues/4#issuecomment-272307530)

Why do you mention m03s05? it is encrypted, so see my comment above for encrypted modules.

notsolowki, 2017-01-23:
i figured since if was the first one in the list it would be the first module. i forgot theres mi00. so i can take mi01.bin and rename it m0305.bin and place it on the sdcard? would i use the debug file too

notsolowki, 2017-01-23:
would be great to save some time because at this rate ill be testing parameters for a month lol

mefistotelis, 2017-01-23:
Please read the post which I linked to.

notsolowki, 2017-01-23:
i know im sorry i just did that. so i can rename mi01 to PMCAPPFw3.bin?

notsolowki, 2017-01-23:
so i cant take mi01 and rename it to PMCLDRFw3.bin and place it on the sdcard?

mefistotelis, 2017-01-23:
Yes. Placing this file on SD card should initiate the update. I don't think the *_debug file is needed, but you have to check that.

notsolowki, 2017-01-23:
okay we got PMCLDRFw3 and PMCLDRAPP, Whats the APP for?

notsolowki, 2017-01-23:
and another question. how come it would update the height but nothing else before when doing the upgrade the normal way. but now if i use the debug file they all work? i might be working ahead of myself here

mefistotelis, 2017-01-23:
Do not update with both. read everything again.

notsolowki, 2017-01-23:
i know i was just asking whats the difference between the two. i didnt know what the ldr app was for

mefistotelis, 2017-01-23:
> can i rename mi01 to PMCLDRFw3.bin

No. This is not correct and will brick your drone.

> and it will only write mi01.bin

If you will not provide any other file on the card, then no other modules will be updated

>  will it have and update finished indicator

I don't know.

notsolowki, 2017-01-23:
then i dont get what your saying., can you just show me what you mean

notsolowki, 2017-01-23:
so instead if  PMCLDRFw3.bin its m0306.bin?

notsolowki, 2017-01-23:
i asked you this in an above post and you said it was the correct way. thats why im confused

mefistotelis, 2017-01-23:
> i asked you this in an above post and you said it was the correct way

Your post looked different when I commented ;)

You seem to be confusing modules. Be sure everything is right before you try to do the update.

notsolowki, 2017-01-23:
okay so  PMCLDRFw3.bin is what i need to rename my modified mi0.bin1 to? thank you

mefistotelis, 2017-01-23:
No. You're still confusing modules.

notsolowki, 2017-01-23:
wow im sorry lol, i mean PMCAPPFw3.bin is this correct. change mi01 to PMCAPPFw3.bin

mefistotelis, 2017-01-23:
Yes, P3X_FW_V01.07.0060_mi01.bin -> MCAPPFw3.bin.

notsolowki, 2017-01-23:
yes without t P thank you for reminding me that. thank you!!!!!

mefistotelis, 2017-01-23:
No, sorry, that's my mistake. With P.

notsolowki, 2017-01-23:
lol thankyou

ferraript, 2017-01-23:
why don't you just create yourself firmware pack containing m0306 module only and flash it?

notsolowki, 2017-01-23:
yea thats what we were discussing

notsolowki, 2017-01-23:
i wonder how fast this drone can descend without vrs

ferraript, 2017-01-23:
no, you are renaming P3X_FW_V01.07.0060_mi01.bin to MCAPPFw3.bin
and I'm suggesting:
1. extract all modules from firmware using dji_fwcon.py (if you haven't done it already)
2. delete all files except head.ini, mi01.ini and mi01.bin
3. update flight parameters using dji_flyc_param_ed.py
4. rename mi01.* to mi00.*
5. create firmware file using dji_fwcon.py and flash it

ferraript, 2017-01-23:
>didnt you wirte that program to convert the logs?

I did, but it's written in Delphi / Pascal

>i wonder how fast this drone can descend without vrs

exactly, changing descending speed may be dangerous
I was also thinking of changing those 3 m/s to something higher, but I just like my P3A too much to lose it

ferraript, 2017-01-23:
>i can confirm my takeoff speed is 27 mph which is 12 meters per second

1. as vertical ascending speed?
2. you achieved that by changing vert_up_vel_0 or something else?
3. I'd like to see DAT log containing this flight, any chance of uploading it somewhere?

notsolowki, 2017-01-23:
@mefistotelis PMCAPPFw3.bin  PERFECT :)

chazzzzz, 2017-01-23:
People following this thread will brick their drone...

notsolowki, 2017-01-23:
hows that

notsolowki, 2017-01-23:
@mefistotelis have you tried changing your ascend speed?

notsolowki, 2017-01-23:
i candescend at 12-13 mph with no issues so far

chazzzzz, 2017-01-23:
Because it boggled my mind and I somewhat understand what is going on... I just hope they read all the way through before attempting this..

We are modifying target m0306 through mi01.bin and renaming it to PMCAPPFw3.bin...Is this Correct?

notsolowki, 2017-01-23:
if you read everything im sure you will figure it out

notsolowki, 2017-01-23:
i wouldnt do this unless u got a p3 pro...

notsolowki, 2017-01-23:
okay 
1. get dji go 2.4.1
2. modify flyc params in mi01
3.repack mi01 with flyc_param_ed.py
4.flash mi01.bin
5. edit the apk and make same said changes to flyc_param
6. with everything powered on goto setting and restore defualt values....
there are you happy

notsolowki, 2017-01-23:
because i know i am

notsolowki, 2017-01-23:
dont change any _basic.craft thrust or control.thrust . they are gain settings setting them higer will make it really unstable

notsolowki, 2017-01-23:
the esc i think still governs the rpm. so you cant get too crazy. you should hold your drone down and test your settings as much as you can before loosing it in the air

[47 more replies]