o-gs/dji-firmware-tools #16: Flyc_param_infos

notsolowki, 2017-01-26:
im going to make a list of know flyc_param_infos parameters, ill update this as i figure them out. if anyone know of any add them in a comment and ill add them to the list

"name" : "g_config.control.vert_up_vel_0" ascend speed
"name" : "g_config.control.vert_down_vel_0" descend speed
"name" : "g_config.engine.upper_bound_0" motor speed limit
"name" : "g_config.engine.idle_level_0" controls idle speed on ground
"name" : "g_config.flying_limit.max_height_0" controls maximum altitude
"name" : "g_config.novice_cfg.max_height_0" maximum height in beginner mode
"name" : "g_config.output_servo.pwm_max_0" effects rev speed when motor starts

mefistotelis, 2017-01-26:
Maybe try to also include tested values.

notsolowki, 2017-01-26:
un-fortunately my drone is probably going to be down for a couple weeks. my yaw motor on my gimbal is not working after my first crash. i had to hit the csc about 20 ft in the air one day because i screwed up some settings. maybe i should have let it go a little higher so i had a chance to sprint to it. the wires got damaged so im trying to figure out if its the esc board or the gimabl thats messed up. but everything else works fine except for the cameras yaw. its completely un-responsive

notsolowki, 2017-01-26:
[image]

chazzzzz, 2017-01-29:
These changes MAY have had some effect on the horizontal speed…Hard to tell with a slight wind. I may try increasing index 194 on the Phantom and 190 on the App and taking the rest back to original default values…These setting changes didn’t cause issues while hovering… They need more testing..

**Phantom changes**
index" : 194 - g_config.control.horiz_vel_gain_0 - defaultValue" : 100 **changed to 110**
index" : 195 - g_config.control.horiz_vel_damping_gain_0 - defaultValue" : 105 **changed to 110**
index" : 196 - g_config.control.horiz_atti_damping_gain_0 - defaultValue" : 100 **changed to 110**

**App. changes**
index" : 190 - g_config.control.horiz_vel_gain_0 - defaultValue" : 90 **changed to 110**
index" : 191 - g_config.control.horiz_damping_gain_0 defaultValue" : **70 changed to 110**

chazzzzz, 2017-01-29:
I tried this and no change from my previous test... The bird peaked at 38mph and back at 33. I think the wind is effecting the results..The max setting for Index 194 and 190 is 500 so either I'm not aggressive enough in my changes or its not doing anything noticeable or at all...Neither change had an effect while hovering

**Phantom changes****
index" : 328, -  g_config.control.dyn_safe_thrust_0 - defaultValue : 80 **changed to 90** Max is 90
index" : 194 - g_config.control.horiz_vel_gain_0 - defaultValue" : 100 **changed to 130** 

**App. changes**
index : 322, - g_config.control.dyn_safe_thrust_0 - defaultValue : 80 **changed to 90** Max is 90
index" : 190 - g_config.control.horiz_vel_gain_0 - defaultValue" : **90 changed to 130**

notsolowki, 2017-01-30:
Tge thrust setting i tested a few days ago. The higer i would set tgem the
faster the the change in throttle speed were. When i set them too high it
would dance in the air.have you made any sucess changes in the flight
behavior just to confirm your making any changes at all. If you change the
command mapper settings you could end up with reversed controls like i did
so be carful.

chazzzzz, 2017-01-30:
Yeah I first changed the vertical speed for a test...It did changed how it reacted...It would go up at 22mph and taper back to 14mph so i know my changes had an effect but it seemed to have some type of current limiting thing going on but i really didn't care about that yet because i just wanted to see if the changes i made worked...So I started with the settings I stated above...
Can you tell me the exact index #'s you changed that affected the thrust? on both the bird and app...

notsolowki, 2017-01-30:
I think the esc has a rpm limiter. The settings had thrust in the name. My drone is all apart right now so i cant test anything for a few days.but you will know when you find it. It will go from idle to max rpm instantly amd vice versa but it will not be able to hove stable ifcits too high. And its probably not good for the esc to surge the power like that but idk. I set mine lower and it wouldnt come down and i had to csc from 20 ft lol.

notsolowki, 2017-01-30:
I think it was basic_craft_thrust. I had mine going up at 27mph at one point  idk if it was luck or somthing i had changed but i couldnt find it again

chazzzzz, 2017-01-30:
I was trying to see which of these you changed that caused the erratic behavior. Also maybe knowing what you changed the default value too... I don't really understand what is the difference between these two settings I pasted below. I was thinking maybe one of them (the slower one index 182) was the RTH thrust when the bird isn't receiving a signal from the RC...I noticed the roll, pitch and yaw index's have a very similar pattern (two like indexes for each)...I'm grasping at straws here trying to make sense of things...

                 "index" : 182,  
		"typeID" : 5,
		"size" : 2,
		"attribute" : 11,
		"minValue" : 20,
		"maxValue" : 500,
		"defaultValue" : 80,
		"name" : "g_config.control.basic_craft_thrust_0",
		"modify" : true

or 

                 index" : 186,
		"typeID" : 5,
		"size" : 2,
		"attribute" : 11,
		"minValue" : 20,
		"maxValue" : 500,
		"defaultValue" : 100,
		"name" : "g_config.control.basic_thrust_0",
		"modify" : true

notsolowki, 2017-01-30:
"index" : 182,  
	"typeID" : 5,
	"size" : 2,
	"attribute" : 11,
	"minValue" : 20,
	"maxValue" : 500,
	"defaultValue" : 80,
	"name" : "g_config.control.basic_craft_thrust_0",
	"modify" : true

anything above 90 cause erratic flight. i didnt even test it outside leting it hoverfor more than 5 seconds it would almost flip

notsolowki, 2017-01-30:
when it slows down the speed at which it reacts to movements is was make it hover smooth. otherwise it over-corrects and continues until it probably falls. it also had alot to do with the throttle response

notsolowki, 2017-01-30:
i think one way to set if 22.4 is its max ascend speed is to pull some weight off of the drone and send it back up. it should go up faster. if not then we know its probably limited in the firmware. id imagine the IMU would have alot to do as far as giving feedback to the esc about how the flight is going. and also im sure the motors have rpm limiters. but until i see max motor speed reached while in flight i want to push it more

ferraript, 2017-02-02:
I found out why (old) Litchi is able to set height limit to 6000 m:
it uses _DataFlycSetLimits_ class - thanks to _SetLimits_ command - I believe that AC is not checking the min/max values in flyc_param_infos and allows to set so high number
on the other hand, all versions of DJI GO don't use _DataFlycSetLimits_ class at all, they set height limit through _DataFlycSetParams_ class with _SetParamsByHash_ command

notsolowki, 2017-02-03:
chazz did you loose your drone? lol

chazzzzz, 2017-02-03:
Nah not yet. Hadn't had much time to test. Hopefully this weekend I have some decent weather. Ive made some parameter changes I'm anxious to test...How bout you, did you get yours up and running again?

notsolowki, 2017-02-04:
Nope it was going to be today but ups dealayed delivery to monday

notsolowki, 2017-02-06:
these settings had no effect
	        "name" : "g_config.pantilt.roll_speed_0",
		"name" : "g_config.pantilt.pitch_speed_0",
		"name" : "g_config.pantilt.yaw_speed_0",
		"name" : "g_config.craft_model.motor_tilt_angle_0",
                  g_config.control.basic_craft_roll_0"  limits,
                "g_config.control.basic_craft_pitch_0"  limits,
                   "g_config.control.basic_craft_yaw_0"  limits,
                 "g_config.control.basic_craft_thrust_0"  limits,
                  "name" : "g_config.craft_model.motor_tilt_angle_0",

ferraript, 2017-02-07:
some of the parameters can be revealed using DJI GO and config txt from DAT flight log
for example, changing expo values results in these 3 lines in config txt:
```
hash write index 325
hash write index 326
hash write index 327
```
and those numbers, of course, are indexes of parameters in flyc_param_infos
but watch out, DJI GO sometimes sends some params without user's knowledge, for example I've got a lot of 211s (height_limit_enabled_0) in my config txts

notsolowki, 2017-02-08:
these parameters right here seem to directly effect the motor speed

PMCAPPFw3.bin: Updating "g_config.output_servo.min_limit[0]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.min_limit[1]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.min_limit[2]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.min_limit[3]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.min_limit[4]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.min_limit[5]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.min_limit[6]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.min_limit[7]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.max_limit[0]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.max_limit[1]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.max_limit[2]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.max_limit[3]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.max_limit[4]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.max_limit[5]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.max_limit[6]_0"  limits,
PMCAPPFw3.bin: Updating "g_config.output_servo.max_limit[7]_0"  limits,

notsolowki, 2017-02-08:
setting them lower made the motors run full speed on startup !

notsolowki, 2017-02-08:
i notice that if the drone tries to tilt too much the motor speeds are corrected to fix the angle. i think it might have somthing to do with the imu/compass/gps feedback

notsolowki, 2017-02-08:
i just found out that you can test the new parameters with the flight simulator

notsolowki, 2017-02-08:
@mefistotelis i got it up to 50 mph left,right,forward,backward. this is in the simulator. i changed these

PMCAPPFw3.bin: Updating "g_config.control.tilt_atti_gain_0"  limits,
PMCAPPFw3.bin: Updating "g_config.control.torsion_atti_gain_0"  limits,
PMCAPPFw3.bin: Updating "g_config.control.horiz_atti_damping_gain_0"  limits,
PMCAPPFw3.bin: Updating "g_config.control.horiz_vel_atti_range_0"  limits,
PMCAPPFw3.bin: Updating "g_config.control.atti_limit_0"  limits,
PMCAPPFw3.bin: Updating "g_config.control.atti_range_0"  limits,
PMCAPPFw3.bin: Updating "g_config.control.atti_tilt_w_rate_0"  limits,
PMCAPPFw3.bin: Updating "g_config.control.atti_torsion_w_rate_0"  limits,
PMCAPPFw3.bin: Updating "g_config.control.atti_bat_limit_0"  limits,

and the result are promising

notsolowki, 2017-02-08:
narrowed it down to 

PMCAPPFw3.bin: Updating "g_config.control.horiz_vel_atti_range_0"  limits,
PMCAPPFw3.bin: Updating "g_config.control.horiz_emergency_brake_tilt_max_0"  limits,
PMCAPPFw3.bin: Updating "g_config.control.atti_limit_0"  limits,
PMCAPPFw3.bin: Updating "g_config.control.atti_range_0"  limits,
PMCAPPFw3.bin: Updating "g_config.control.vert_up_vel_0"  limits,
PMCAPPFw3.bin: Updating "g_config.control.vert_down_vel_0"  limits,

notsolowki, 2017-02-08:
this parameter result in instant takeoff at max speed, ignoring height limits. be very cautious with this parameter
PMCAPPFw3.bin: Updating "g_config.control.horiz_vel_atti_range_0" limits,

notsolowki, 2017-02-08:
i confirmed speed changes with
PMCAPPFw3.bin: Updating "g_config.control.atti_limit_0" limits,
PMCAPPFw3.bin: Updating "g_config.control.atti_range_0" limits,

i think i can get it up to 63 mph

notsolowki, 2017-02-08:
which means its also possible that it could ascend at 60 miles per hour

notsolowki, 2017-02-09:
anyone want to try the modified speed parameters

http://www.mediafire.com/file/5wnm21na2c189md/dji+fw+hacked.zip

notsolowki, 2017-02-09:
@mefistotelis  please bytepatch the rth speed so i can use it as my base firmware module, you are really experienced with binary this would be a great task for you i think

notsolowki, 2017-02-09:
if you can, set it to 16 that shoudl get it to 35 mph, which is right below its default top speed.

notsolowki, 2017-02-10:
@mefistotelis how can i compare both flyc_param_infos to each other and see what parameters are in the aircrafts flyc that are not in the apps flyc? thankyou

notsolowki, 2017-02-10:
ive tried 
GREP
FGREP
DIFF
VDIFF
COMM
and i still cant get it to work. maybe you could tell me what to do. it would like to what parameters are in the ac flight controller that are not in the apps flyc.thanks again

notsolowki, 2017-02-10:
i also notice after byte-patching with your commands the log shows 
[smart_battery]this fireware calc gohme speed:7.800000 - land speed:2.500000

mefistotelis, 2017-02-10:
> after byte-patching ...  gohme speed:7.800000

Well this is strange. The command should have modified that. Are you sure you updated the drone with patched firmware?

notsolowki, 2017-02-10:
im going to double check the go home speed in the morning. i will re-patch a binary to make sure

notsolowki, 2017-02-10:
onet hing i notcie is it says "gohme" speed and not gohome speed,

[smart_battery]this fireware calc gohme speed:7.800000 - land speed:2.500000
i dont know if its just ferraripts log converter but i copied and pasted this from the log

notsolowki, 2017-02-10:
thanks ill try it

notsolowki, 2017-02-13:
here is a list of parameters that are not in the andoird app

	"name" : "g_config.knob.control_channel[KNOB_BASIC_YAW]_0",
		"name" : "g_config.knob.control_channel[KNOB_BASIC_THRUST]_0",
		"name" : "g_config.knob.control_channel[KNOB_ATTI_GAIN]_0",
		"name" : "g_config.knob.control_channel[KNOB_GYRO_GAIN]_0",
		"name" : "g_config.voltage2.vol_ceof_0",
		"name" : "g_config.control.vel_braking_threshold_0",
		"name" : "g_config.control.horiz_emergency_brake_tilt_max_0",
		"name" : "g_config.control.atti_limit_0",
		"name" : "g_config.control.avoid_atti_range_0",
		"name" : "g_config.control.rotor_fault_test_type_0",
		"name" : "g_config.control.rotor_fault_detection_time_0",
		"name" : "g_config.control.rc_yaw_sensitivity_0",
		"name" : "g_config.control.rc_throttle_sensitivity_0",
		"name" : "g_config.control.torsion_ang_vel_cmd_slope_0",
		"name" : "g_config.control.torsion_ang_acc_cmd_slope_0",
		"name" : "g_config.control.throttle_output_max_0",
		"name" : "g_config.control.yaw_output_max_0",
		"name" : "g_config.control.atti_bat_limit_0",
		"name" : "g_config.control.torsion_w_rate_bat_limit_0",
		"name" : "g_config.control.vert_vel_up_bat_limit_0",
		"name" : "g_config.control.vert_vel_down_bat_limit_0",
		"name" : "g_config.imu_para_cfg.imu_adv_func.fuse_with_rtk_0",
		"name" : "g_config.gps_cfg.super_svn_0",
		"name" : "g_config.gps_cfg.fix_svn_0",
		"name" : "g_config.gps_cfg.good_svn_0",
		"name" : "g_config.gps_cfg.super_hdop_0",
		"name" : "g_config.gps_cfg.fix_hdop_0",
		"name" : "g_config.gps_cfg.good_hdop_0",
		"name" : "g_config.gear_cfg.pack_type_0",
		"name" : "g_config.api_entry_cfg.enable_time_stamp_0",
		"name" : "g_config.api_entry_cfg.acc_data_type_0",
		"name" : "g_config.api_entry_cfg.gyro_data_type_0",
		"name" : "g_config.api_entry_cfg.alti_data_type_0",
		"name" : "g_config.api_entry_cfg.height_data_type_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[0]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[1]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[2]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[3]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[4]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[5]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[6]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[7]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[8]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[9]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[10]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[11]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[12]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[13]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[14]_0",
		"name" : "g_config.api_entry_cfg.std_msg_frq[15]_0",
		"name" : "g_config.api_entry_cfg.authority_level_0",
		"name" : "g_config.api_entry_cfg.cheat_backdoor_0",
		"name" : "g_config.airport_limit_cfg.cfg_search_radius_0",
		"name" : "g_config.airport_limit_cfg.cfg_disable_airport_fly_limit_0",
		"name" : "g_config.airport_limit_cfg.cfg_debug_airport_enable_0",
		"name" : "g_config.airport_limit_cfg.cfg_limit_data_0",
		"name" : "g_config.airport_limit_cfg.cfg_sim_disable_limit_0",
		"name" : "g_config.airport_limit_cfg.cfg_enable[FLY_LIMIT_TYPE_AIRPORT]_0",
		"name" : "g_config.airport_limit_cfg.cfg_enable[FLY_LIMIT_TYPE_SPECIAL]_0",
		"name" : "g_config.airport_limit_cfg.cfg_r1[FLY_LIMIT_TYPE_AIRPORT]_0",
		"name" : "g_config.airport_limit_cfg.cfg_r1[FLY_LIMIT_TYPE_SPECIAL]_0",
		"name" : "g_config.airport_limit_cfg.cfg_h1[FLY_LIMIT_TYPE_AIRPORT]_0",
		"name" : "g_config.airport_limit_cfg.cfg_h1[FLY_LIMIT_TYPE_SPECIAL]_0",
		"name" : "g_config.airport_limit_cfg.cfg_angle[FLY_LIMIT_TYPE_AIRPORT]_0",
		"name" : "g_config.airport_limit_cfg.cfg_angle[FLY_LIMIT_TYPE_SPECIAL]_0",
		"name" : "g_config.avoid_obstacle_limit_cfg.a_dec_max_0",
		"name" : "g_config.avoid_obstacle_limit_cfg.v_fwd_max_0",
		"name" : "g_config.avoid_obstacle_limit_cfg.v_bck_max_0",
[... 189 lines cut ...]
		"name" : "wp_setting.action_on_rc_lost_0"
		"name" : "g_config.bat_limit.limit_delt_vol.min_0",
		"name" : "g_config.bat_limit.limit_delt_vol.max_0",
		"name" : "g_config.bat_limit.limit_delt_vol.weight_0",
		"name" : "g_config.bat_limit.limit_vol.min_0",
		"name" : "g_config.bat_limit.limit_vol.max_0",
		"name" : "g_config.bat_limit.limit_vol.weight_0",
		"name" : "g_config.bat_limit.limit_max_power.min_0",
		"name" : "g_config.bat_limit.limit_max_power.max_0",
		"name" : "g_config.bat_limit.limit_max_power.weight_0",
		"name" : "g_config.bat_limit.limit_temp.min_0",
		"name" : "g_config.bat_limit.limit_temp.max_0",
		"name" : "g_config.bat_limit.limit_temp.weight_0",
		"name" : "g_config.bat_limit.limit_scale.min_0",
		"name" : "g_config.bat_limit.limit_scale.max_0",
		"name" : "g_config.bat_limit.limit_scale.weight_0",
		"name" : "g_config.bat_limit.limit_power_rate_capacity.min_0",
		"name" : "g_config.bat_limit.limit_power_rate_capacity.max_0",
		"name" : "g_config.bat_limit.limit_power_rate_capacity.weight_0",
		"name" : "g_config.bat_limit.limit_power_rate_temp.min_0",
		"name" : "g_config.bat_limit.limit_power_rate_temp.max_0",
		"name" : "g_config.bat_limit.limit_power_rate_temp.weight_0",
		"name" : "g_config.bat_limit.limit_bat_current_after_overflow_0",
		"name" : "g_config.bat_limit.limit_up_rate_0",
		"name" : "g_config.bat_limit.limit_down_rate_0",

polarisax, 2017-02-17:
Does anyone have any idea what can do the following parameter?
{
		"index" : 211,
		"typeID" : 0,
		"size" : 1,
		"attribute" : 11,
		"minValue" : 1,
		"maxValue" : 2,
		"defaultValue" : 1,
		"name" : "g_config.advanced_function.height_limit_enabled_0",
		"modify" : true
	}

Is it possible that is sufficient change defaultValue to "2" to disable the maximum height reachable?

Thanks in advance

polarisax, 2017-04-14:
Hi @notsolowki , 
I saw in your old FW versions you have touched a lot of parameters.
Can you give us a comment for the purpose of these?

-g_config.voltage2.level1_smart_battert_gohome_0
-g_config.voltage2.level2_smart_battert_land_0
-g_config.control.horiz_emergency_brake_tilt_max_0
-g_config.control.vert_vel_up_bat_limit_0
-g_config.control.vert_vel_down_bat_limit_0

I notice that you made a lot of adjustment of these in your versions, but it's not very clear to me the impact they have on the copter! :(
Thanks in advance!

marck1991, 2017-07-11:
I extracted mi01 from P3P 1.7 firmware using dji_fwcon.py and then used dji_flyc_param_ed.py to read and write data from and to mi01. Then renamed edited mi01.bin to PMCAPPFw3.bin and put it on SD card. It flashed successfully and got renamed to PMCAPPFw3_updated.bin .

My problem: The up and down speed didn't change! :( 
But the flashing progress must be successful, because in GO3 app i can see the new max high applied.
I also could fly 80+ km/h in atti mode, so I think the degree adjustments also have been applied.

Does someone (maybe @notsolowki ) have an answer, why the ascend and descend speed from my P3P didn't change?
They are still 3m/s down and 5m/s up.

Ich changed this parameters:
g_config.control.vert_up_vel_0 --> 8
g_config.control.vert_down_vel_0 --> 5
g_config.flying_limit.max_height_0 --> 1195
g_config.control.atti_limit_0 --> 45
g_config.control.atti_range_0 --> 45

ferraript, 2017-07-11:
I suppose you changed defaultValue only
you need to change minValue too

marck1991, 2017-07-11:
Yes you are right, thanks so much for your answer!!
Should I set minValue to defaultValue for all of the five values?

For example I did this before your answer @ferraript :

"index" : 316,
		"typeID" : 5,
		"size" : 2,
		"attribute" : 11,
		"minValue" : 10,
		"maxValue" : 60,
		"defaultValue" : 45,
		"name" : "g_config.control.atti_limit_0",
		"modify" : true

"index" : 323,
		"typeID" : 5,
		"size" : 2,
		"attribute" : 11,
		"minValue" : 1,
		"maxValue" : 10,
		"defaultValue" : 8,
		"name" : "g_config.control.vert_up_vel_0",
		"modify" : true

ferraript, 2017-07-11:
problem is, that some of the parameters are present in the DJI GO too... it means that it will overwrite those values you did set in defaultValue in firmware
so if you don't intend to edit the apk (and don't want to care about those params within it) I suggest you change minValues the same way you changed defaultValues

marck1991, 2017-07-11:
@ferraript testet it with your suggestions and I got the results I expected! Thanks a lot!

ferraript, 2017-07-11:
glad to help, fly safe ;)

danieltroger, 2017-08-06:
Sorry for not having read everything regarding the topic, but would it theoretically be possible to implement something like mavic's tripod mode for the p3 by changing flight parameters just by the app side, without having to reflash the FC firmware?
I mean when there's enough understanding about the binary protocol DJI GO uses to communicate with the flight controler, if it then would be possible to change something that would result in decreaed speed, but higher level of precision and stability?

mefistotelis, 2017-08-25:
Recently someone asked me for this one, so since I investigated this - there's the info:

**Parameter:**
g_real.status.motor_status_0

**Definition:**
<pre>	{
		"index" : 733,
		"typeID" : 0,
		"size" : 1,
		"attribute" : 0,
		"minValue" : 0,
		"maxValue" : 1,
		"defaultValue" : 0,
		"name" : "g_real.status.motor_status_0"
	}</pre>

**Function:**
The parameter allows ignoring status error of various drone components while starting motors.

**Values:**
0 - Do not force starting motors when status of any component indicates error. This value makes the drone allow starting motors only if all components are returning "safe to fly" status. This is the default.
1 - Force starting motors when status of any component indicates error. This will cause the motors to start even if the drone is not ready to fly yet, or errors are indicated by components.

phhh3, 2017-08-31:
hey  guys  i have questions.... i  use   dji_fwcon.py -vv -x -p P3X_FW_V01.07.0060.bin  extract the files but i cant find file called   mi01.bin  what happend to me ?

mefistotelis, 2017-08-31:
I recently renamed the files created to use only the mXXYY naming instead om miNN. The mXXYY convention became kind of a standard way of naming modules in our community.

marck1991, 2017-08-31:
what @phhh3 is now looking for is P3X_FW_V01.07.0060_m0306.bin :) .

phhh3, 2017-08-31:
so i mean  if i wanna  get     flyc_param_infos   this  parameter file   what am i  supposed to do?  i  extracted the  P3X_FW_V01.07.0060.bin   then got m0100, m0101,m0306  something like this .............  should  i  use  this  pathon code ->    dji_flyc_param_ed.py -vv -x -m P3X_FW_V01.07.0060_m0306.bin     get flyc_param_infos?  thanks~

marck1991, 2017-08-31:
Yes thats the right command. After editing flyc_param_infos use dji_flyc_param_ed.py -vv -u -m P3X_FW_V01.07.0060_m0306.bin to write back.

notsolowki, 2017-08-31:
@mefistotelis, can you figure out what the cheat_backdoor parameter does.
Also is there a way to shut off the startup tone of the esc's and maybe you
have found a way to override the esc max rpm? I bet this drone could go
faster than 45. Also i noticed on the day of the eclipse i got my drone to
almost 7kft altitude before i lost signal and i thought that was strange
because i could never get it higher than 4kft.

phhh3, 2017-09-01:
@notsolowki  hey  man  can  u  send me  your  flyc_param_infos file? thanks~

polarisax, 2017-09-06:
@notsolowki hey man can u send me your flyc_param_infos file? thanks~

...me too, please!

mefistotelis, 2017-09-06:
> @mefistotelis, can you figure out what the cheat_backdoor parameter does.

Hard to tell; the related structure travels as reference to a lot of functions. Not easy to track.

notsolowki, 2017-09-06:
Heres a link to the flyc and app if you guys need it,

Flight controller binary
https://www.mediafire.com/file/d1jlss9dbakuwd5/fligh_control
ler_p3_pro_and_advanced.zip
2.4.2 app
https://www.mediafire.com/file/bow4mdbdle1vugi/2.4.2_modified.zip
3.7 app
https://www.mediafire.com/file/4dpacnqxo5soxo6/go_3.7.zip

notsolowki, 2017-09-06:
@mefistotelis, did you add the lala-haha-MA key lol. Somthing to do with dji-demo? Maybe its for factory resetting the firmware? I notice alot of other dji drones have the same parameter. 

if ( g_real__config.api_entry_cfg.cheat_backdoor == 1 && *(_DWORD *)a1 == 10086 ) { if ( *(_DWORD *)(a1 + 4) <= 2u ) { sub_805A964("DJI-DEMO AES256 KEY-lala-haha-MA", *(_DWORD *)a1 - 10086); return 0; } return 7; }

[2 more replies]