o-gs/dji-firmware-tools #19: dji_flyc_nofly_ed.py

notsolowki, 2017-02-13:
when i run the tool this is my output

laptop@laptop-VPCEH14FM:~/Desktop/phantom-firmware-tools-master (2)$ python3 dji_flyc_nofly_ed.py -vv -x -m p7_mi01.bin
p7_mi01.bin: Opening for extraction
p7_mi01.bin: Matching zones array at 0x0009df68: 5000 entries
p7_mi01.bin: Matching coords array at 0x00086678: 61 entries
p7_mi01.bin: Matching coords array at 0x00095cac: 4183 entries
p7_mi01.bin: Warning: multiple (2) matches found for fly coords array with alignment 0x02
p7_mi01.bin: Merging No Fly arrays...
laptop@laptop-VPCEH14FM:~/Desktop/phantom-firmware-tools-master (2)$ python3 dji_flyc_nofly_ed.py -vv -u -m p7_mi01.bin
p7_mi01.bin: Opening for update
p7_mi01.bin: Matching zones array at 0x0009df68: 5000 entries
p7_mi01.bin: Matching coords array at 0x00086678: 61 entries
p7_mi01.bin: Matching coords array at 0x00095cac: 4183 entries
p7_mi01.bin: Warning: multiple (2) matches found for fly coords array with alignment 0x02
Traceback (most recent call last):
  File "dji_flyc_nofly_ed.py", line 403, in <module>
    main(sys.argv[1:])
  File "dji_flyc_nofly_ed.py", line 394, in main
    flyc_nofly_update(po,fwmdlfile)
  File "dji_flyc_nofly_ed.py", line 334, in flyc_nofly_update
    raise NotImplementedError('Not implemented.')
NotImplementedError: Not implemented.

mefistotelis, 2017-02-13:
This is correct. Updating is not implemented.

rulppa, 2017-02-15:
My understanding is this workaround, when available. will reset the location to middle of nowhere? Do you think, it picks the FCC/CE mode at that time, or is there another check for that?

mefistotelis, 2017-02-15:
> this workaround, when available. will reset the location to middle of nowhere?

No idea what you mean by that. I'm not sure if I'll be making write support to the nofly_ed tool.

> Do you think, it picks the FCC/CE mode at that time

NFZ proximity is being checked all the time during flight, while FCC/CE power zone check is likely done only once. Most likely it is done while storing home point position, but as of now I wasn't able to identify any code which relates to that.

ferraript, 2017-02-28:
@mefistotelis, are you still working on this? any estimate of when updating of NFZ list will be possible?

mefistotelis, 2017-02-28:
Since the areas are reasonably small, I see no need of changing them. I stopped working on that.

The only thing which still needs to be figured out is how to change radius for the coordinates which don't have it stored per-entry. The functions which are using these entries have to be analyzed, and I'm sure this will result is finding some kind of constant value stored in the code.

After this is done, anyone with basic Python skills should be able to make the update support.