o-gs/dji-firmware-tools #20: Phantom 4 pro

notsolowki, 2017-02-18:
Do you think you can make your parser work with p4p firmware. would be nice to be able to extract the p4p params. thats the only thing really stopping me from buying one right now lol.

mefistotelis, 2017-02-18:
Currently I'm only interested in P3P/P3A.

coptersafe, 2017-03-09:
about p4p (p4,p4p+,mavic,inspire 2 ) you have 2 ways :

1.you can write parameters directly to external EEPROM on FC (but you need password for EEPROM because it is secured)

2. spoof command when configuring drone via Assistant., yes . it is worked. . but at now i need more and more tests.

coptersafe, 2017-03-15:
update:  i make full flash from NAND with android from P4(P4PRO) , find some partitions  ,lots of file - many logs , buts cant find system image.... if any need link for full flash from NAND - write me.

Also if any whant to help me with  Android on P4(P4PRO) i will be grateful

notsolowki, 2017-03-22:
you should just put the link on this page

coptersafe, 2017-03-22:
http://dropmefiles.com/5WE6M

turnersr, 2017-04-01:
@coptersafe do you mind posting that link again?

vishaldeyiiest, 2017-05-19:
Did anyone reverse engineered the firmware for DJI phantom 4 pro.?..The firmware seems pretty big around 1 GB. Is this the firmware GL300E_RC_v1130_20170307.bin given in DJI website?..

coptersafe, 2017-05-19:
This is a image for android based on Rockchip proc, also this for rc only.
All binaries for p4 encrypted and signed. So i think reverse possible but
not way for reflash p4 with new binary

coptersafe, 2017-05-19:
@turnersr ,no i dont have any more this file.i found that file not useful for me..sorry

vishaldeyiiest, 2017-05-19:
Thanks for the reply. Actually, I'm a researcher working on Phantom 4 pro.
I'm given a project to find security loopholes in Phantom 4. I can only
find GPS Spoofing to hijack the drone. I'm just a beginner in reverse
engineering. I would like to have some word with you. And where is the
image you referred to?

coptersafe, 2017-05-19:
I think no way for gps spoofing for p4 pro... Or you mean p4? For p4 we
have simple solution at my website.. but p4pro have secured ublox chip and
all packets signed

>
> Thanks for the reply. Actually, I'm a researcher working on Phantom 4 pro.
> I'm given a project to find security loopholes in Phantom 4. I can only
> find GPS Spoofing to hijack the drone. I'm just a beginner in reverse
> engineering. I would like to have some word with you. And where is the
> image you referred to?
>
>
> > @turnersr <https://github.com/turnersr> ,no i dont have any more this
> > file.i found that file not useful for me..sorry
> >
>
> > <
https://github.com/mefistotelis/phantom-firmware-tools/issues/20#issuecomment-302602240
https://github.com/

vishaldeyiiest, 2017-05-19:
> tools/issues/20#issuecomment-302602240
> C8UVDfsHHluX8n_ebKNtvks5r7RKOgaJpZM4MFR0D

Actually I looked into this
https://medium.com/@swalters/how-can-drones-be-hacked-the-updated-list-of-vulnerable-drones-attack-tools-dd2e006d6809
which says GPS spoofing for DJI p1/2/3/4, no mention about p4 pro. What do
you mean by packets signed? aren't the communications done via radio waves?

vishaldeyiiest, 2017-05-19:
> how-can-drones-be-hacked-the-updated-list-of-vulnerable-
> drones-attack-tools-dd2e006d6809 which says GPS spoofing for DJI
> ASRyDi60z1-C8UVDfsHHluX8n_ebKNtvks5r7RKOgaJpZM4MFR0D

Do you know whether the live video transmission is encrypted in P4 pro?

coptersafe, 2017-05-19:
@vishaldeyiiest ublox chipset send signed packets with coordinates inside drone to FC.. so no any way injecting new coordinates into traffic

vishaldeyiiest, 2017-05-19:
But I saw your site..you have designed a module for NFZ for DJI P4 Pro? How is it possible then..as reversing and flashing not possible and also GPS packets are signed

coptersafe, 2017-05-19:
Video encrypted at fpga chip , on older version Dji useful altera cyclon v , at new version useful remarked IV but i cant identify them

notsolowki, 2017-05-19:
Does anyone have a copy of the drone firmware

coptersafe, 2017-05-19:
You can them download from Dji.

notsolowki, 2017-05-19:
No it only the remote control firmware

notsolowki, 2017-05-19:
I like to be able to configure the flight controller on tge p4 pro and the mavric

coptersafe, 2017-05-19:
http://pro-dji-service-usa-cdn.aasky.net/paraconfig_file/7b9b2e04-4a8d-4810-bc17-246bee0c3610/wm331_0000_v01.02.0304_20170106.pro.cfg.sig

http://pro-dji-service-usa-cdn.aasky.net/paraconfig_file/7b9b2e04-4a8d-4810-bc17-246bee0c3610/wm620_0000_v01.00.0240_20170302.pro.cfg.sig

you just need right link

notsolowki, 2017-05-19:
What is wm331 is the sig file a binary container?

coptersafe, 2017-05-19:
wm331 - p4pro , wm331_0000 - it is xml container file

vishaldeyiiest, 2017-05-19:
@notsolowki how can you configure the flight controller?
@coptersafe how to make change to fly P4 Pro in a NFZ?

coptersafe, 2017-05-19:
@vishaldeyiiest  i just send serial number to DJI and  after make payment and approving from DJI , DJI  make new firmware for needed serial number , and when you update drone  you get new specific firmware  . so... but we get this options special for goverment forces drone

vishaldeyiiest, 2017-05-19:
@coptersafe I need to verify all the points I gathered today. Can I get official documentation from DJI. because in the documentation in the website nothing is said about secure GPS module or encrypted video feed.!..

vishaldeyiiest, 2017-05-19:
@coptersafe I dont understand..if video is encrypted, so how the video is decrypted if I download from the SD card?

coptersafe, 2017-05-19:
@vishaldeyiiest Amvarella make vide stream to Davinchi module via HDMI (or USB) and at the same time write video stream to microSD (H.264 or H.265) , then Davinchi send video to Encryption module Altera and  then encrypted stream goes ro RF transceiver (AD9364 or same) ..

coptersafe, 2017-05-19:
of course you can interrup video stream but you need pin password for RF module and password for decryption video

vishaldeyiiest, 2017-05-19:
@coptersafe can we get a documentation verifying the internal hardware and encryption? I haven't got any resource useful for hacking dji p4 pro...any help will be great!

coptersafe, 2017-05-19:
best way for hacking it is own drone ... pf course no any docs  - only some scraps of info

vishaldeyiiest, 2017-05-19:
I have the drone...I mean I need to find security loopholes that can be used to hijack the drone...I cannot find any loophole, so I said if anyone has cracked it?!

mefistotelis, 2017-05-19:
Documentation - you won't find much beyond what's on the wiki of this project. The wiki explains Phantom 3 HW and SW, but Ph4 is just an evolution of Ph3 Pro hardware. If you want more docs, you have to write them.

Security holes - your best bet would be to check older firmwares, maybe there is a version which has issues. We know the internal communication model to some extent, but I don't think anyone tried to crack Lightbridge.

Since we have the synthesized FPGA image, it would be worth checking if there are any tools for opening such image, and maybe even de-synthesizing it to some extent.

notsolowki, 2017-05-19:
So copter safe ypu figured out how to communicate with the the flight
crontroller thru a serial connection.?  How long is this password?

coptersafe, 2017-05-19:
@notsolowki password is13 symbol long, what you mean connect to fc?

notsolowki, 2017-05-19:
I was referring to you saying there are 2 ways to configure the flight
controller.

vishaldeyiiest, 2017-05-22:
@mefistotelis I wrote to them regarding the internal hardware, they said they cannot disclose, now the ponly option seems to disassemble the drone or do nothing.
can you share the FPGA image?

mefistotelis, 2017-05-22:
I don't have any modules for Phantom 4 pro. Phantom 4 and Mavic seem to have different firmware distribution system. I never cared to look at it in detail.

For Phantom 3, you can extract the modules (including fpga image) from a full firmware package using the tools developed in this project.
The tools also allow you to download all known Dji firmware packages (the tests can do that).

Description of specific modules of the firmware is here:
https://github.com/mefistotelis/phantom-firmware-tools/wiki/DJI-Firmwares

vishaldeyiiest, 2017-05-23:
@mefistotelis thanks, but my project is only on P4 Pro. Can you give me an idea how can I verify that both band of frequencies(2.4 GHz and 5.8 GHz) are encrypted as claimed by DJI support? Also the RC firmware(aircraft not available) they claim to be encrypted, I have reverse engineered to some extent.

mefistotelis, 2017-05-23:
> Can you give me an idea how can I verify that both band of frequencies(2.4 GHz and 5.8 GHz) are encrypted as claimed by DJI support?

It is possible to confirm that using high frequency SDR equipment (ie. HackRF board, or a lot more expensive but available at universities laboratory equipment, like Agilent devices), but that requires knowledge on how to figure out modulation and convert the signal to digital form. Reading datasheet of the transmitter used should allow you to figure out whether this will be easy or not.

> Also the RC firmware(aircraft not available) they claim to be encrypted, I have reverse engineered to some extent.

Yes, the firmware tools allow to extract C1 firmware package, and most modules are not encrypted. I think only the DaVinchi video decoder firmware module is encrypted.

vishaldeyiiest, 2017-05-23:
@mefistotelis thanks a lot...I was hoping for some kind of radio scanner could do...otherwise I have to disassemble the drone...the firmware after extracting with binwalk gave me a total of >500 GB. Obvoiusly the whole assembly code base cannot be understood..I cannot find the bootloader also..

mefistotelis, 2017-05-23:
> Obvoiusly the whole assembly code base cannot be understood..

If you extract the modules with dji_fwcon.py, and propely use arm_bin2elf.py, then you can load the ARM binary modules (ie. m1400) into IDA Pro.

> I cannot find the bootloader also..

Agreed. Most bootloaders are not included in the firmware package.

vishaldeyiiest, 2017-05-23:
@mefistotelis I tried running dji_fwcon.py but it does not work for P4 pro firmware..gets unexpected magic number...

mefistotelis, 2017-05-23:
Then it's probably not in the form of firmware package containing modules. Can't tell much more without having the binary.

vishaldeyiiest, 2017-05-23:
@mefistotelis I got the firmware from DJI website..if you have time to look into it..
https://drive.google.com/file/d/0B4aR1zUqD_Jmd0ZkUlNHZXlTMGM/view?usp=sharing

mefistotelis, 2017-05-23:
Wow, this thing is absurdly large.

Anyway, from the header, it looks like the normal firmware package. Probably some minor change was made to file format.

mefistotelis, 2017-05-23:
Added support. Only two modules are inside, but one of them is ARM binary. The second is another package with RKFW magic.

vishaldeyiiest, 2017-05-23:
@mefistotelis thanks for looking into it..yeah got two modules. looking into RKFW.

vishaldeyiiest, 2017-05-23:
@mefistotelis imgRePacker when executed on the second module cannot unpack, gives wrong file format..etc.. Even if reversing is possible, after some modification, is it possible to reflash with the modified firmware? There is a DJI Assistant...

mefistotelis, 2017-05-23:
> Even if reversing is possible, after some modification, is it possible to reflash with the modified firmware?

The dji_fwcon tool can re-pack the firmware after modifications, and it re-computes the checksums so that the firmware is accepted by the standard update process.

mefistotelis, 2017-05-23:
Yup, the rkflashtool works. I built a windows version using msys2, then:
<code>rkunpack.exe GL300E_RC_v1130_20170307_mi00.bin

rkunpack: info: RKFW signature detected
rkunpack: info: version: 5.0.0
rkunpack: info: date: 2017-03-07 15:46:43
rkunpack: info: family: rk32xx
rkunpack: info: 00000066-0005b1ec BOOT                       (size: 373127)
rkunpack: info: 0005b1ed-405d41f0 embedded-update.img        (size: 1079480324)
unpacked</code>

hankknah, 2022-02-13:
@mefistotelis I have downloaded the firmware which shared by @vishaldeyiiest  and tried rkunpack tool, but it said "invalid signature". Can you share me the rkflashtool you maked or the firmware you have unpacked. Thank you.

hankknah, 2022-02-13:
@mefistotelis Thank you. I have solved it.