o-gs/dji-firmware-tools #21: P3P wifi ?

KennethMcNutt, 2017-02-20:
I am simply  looking for a way to get into the Wifi parameters.. To increase the output level of the P3P...
I have Used the ftp method on the P3S.. To set it... But there is no method of doing this to the P3P.. It maybe just my ignorance that is stopping me?? I have found all of this info.. But I don't see a way into the settings like in the "PowerMagic" Mod for the P3S... I also found in the links below.. A reference to enabling the "wifi" it is from the 1.8.8 bin file.. But I have seen it found in the P3S controller .config.. By Enabling the P3P to have a "Visible" Wifi that you can connect to.. Does two things..
First it is a way into the system.. (ftp) 
[P3Ppasswords.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/788626/P3Ppasswords.txt)
and also away for it to be "repeated" like I have done with my P3S .. (with any "wireless extender") http://forum.dji.com/thread-58369-1-1.html <this is what i found in the 1.8.8 bin file..
[Powermagiclog.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/788619/Powermagiclog.txt) <this is a update log fro PowerMagic..
[1.8.8hexbindump.txt]< this is what I found in the 1.8.8 bin with a Hex editor.. 
[P3Ppasswords.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/788628/P3Ppasswords.txt) This is all the system passwords I could come up with..

So Again is there anyone that can Understand all this? Is there a way to mod a bin file to turn on the "WiFi"??

mefistotelis, 2017-02-20:
What gave you an idea that there is a wi-fi chip somewhere in P3P?
On which board have you seen it?

P3P/P3A is completely different hardware than P3S/P3-4k. There is no point in comparing these.

KennethMcNutt, 2017-02-21:
Only because I was using a Hex editor.. And found this in 1.8.8 pro firmware?? That is why I thought it was possible ... I have not looked at the board for a chip... I just saw no reason for the Info to be in PRO firmware? Right?
https://github.com/mefistotelis/phantom-firmware-tools/files/788636/1.8.8hexbindump.txt
That is from the 1.8.8 bin file..

KennethMcNutt, 2017-02-21:
Do you have a reason .. that explains why Pro firmware has that in it's bin? I don't see any reason the P3S and the P3P would share programing??

mefistotelis, 2017-02-21:
Have you ever looked at "firmware" of your PC? I mean, the operating system you're using. It has tons of drivers inside; do you have hardware to use all of them?

If you still want to dig deeper, start by checking which firmware module has the wifi config.

KennethMcNutt, 2017-02-21:
Yes that is understandable.. But PC'S are not hardware specific.. Like the p3... Assuming that.. It seems that that code it found is truly useless? There is no Plug and play LOL

mefistotelis, 2017-02-21:
Again:
If you still want to dig deeper, start by checking which firmware module has the wifi mention.

Otherwise we're doing a theoretical discussion.

KennethMcNutt, 2017-02-21:
I am truly ignorant.. I will have to learn much!! I will start with that... :) Thank you Sir!!!

KennethMcNutt, 2017-02-27:
Can you point me ?? I would look in the modules... But have not a clue how to start.. I would like to be involed with this (backward engineering...) But I have much to learn.. Please forgive me.. Linux , coding, programing are not my forte...I understand the theory, But I have no practical use experience :( this is also what I found in the 1.8.8 bin..
# Atheros..if [ -e /sys/module/ar6000 ]; then...return 0..fi...# Broadcom..if [ -e /sys/module/dhd ]; then...return 0..fi...return 1.}..hostapd_conf().{..if [ "${usecache}" != "" ] && [ -e /tmp/hostapd.conf ]; then...cat /tmp/hostapd.conf..else...#generate hostapd.conf...echo "interface=wlan0" > /tmp/hostapd.conf...echo "ctrl_interface=/var/run/hostapd" >> /tmp/hostapd.conf...echo "beacon_int=100" >> /tmp/hostapd.conf...echo "dtim_period=1" >> /tmp/hostapd.conf...echo "preamble=0" >> /tmp/hostapd.conf...#WPS support...echo "wps_state=2" >> /tmp/hostapd.conf...echo "eap_server=1" >> /tmp/hostapd.conf....#AP_SSID...AP_SSID=`echo "${conf}" | grep AP_SSID | cut -c 9-`...echo "AP_SSID=${AP_SSID}"...echo "ssid=${AP_SSID}" >> /tmp/hostapd.conf....#AP_MAXSTA...AP_MAXSTA=`echo "${conf}" | grep AP_MAXSTA | cut -c 11-`...echo "max_num_sta=${AP_MAXSTA}" >> /tmp/hostapd.conf....#AP_CHANNEL...AP_CHANNEL=`echo "${conf}" | grep AP_CHANNEL | cut -c 12-`...if [ ${AP_CHANNEL} -lt 0 ]; then....reset_conf....return 1...fi....# TODO: For 5G?!...if [ ! -e /sys/module/ar6000 ] && [ $AP_CHANNEL -eq 0 ]; then....#choose 1~10 for HT40....#RAND_CHANNEL=`echo $(( $RANDOM % 10 +1 ))`....RAND_CHANNEL=`echo $(( ($RANDOM % 3) * 5 + 1 ))`....echo "channel=${RAND_CHANNEL}" >> /tmp/hostapd.conf...fi...if [ $AP_CHANNEL -ne 0 ]; then....echo "channel=${AP_CHANNEL}" >> /tmp/hostapd.conf...else....AP_CHANNEL=${RAND_CHANNEL}....echo "AP_CHANNEL (randomly)=${AP_CHANNEL}"...fi....#WEP, WPA, No Security...AP_PUBLIC=`echo "${conf}" | grep AP_PUBLIC | cut -c 11- | tr '[:lower:]' '[:upper:]'`...echo "AP_PUBLIC=${AP_PUBLIC}"...if [ "${AP_PUBLIC}" != "YES" ]; then....#WPA....echo "wpa=2" >> /tmp/hostapd.conf....echo "wpa_pairwise=CCMP" >> /tmp/hostapd.conf....AP_PASSWD=`echo "${conf}" | grep AP_PASSWD | cut -c 11-`....echo "wpa_passphrase=${AP_PASSWD}" >> /tmp/hostapd.conf....echo "wpa_key_mgmt=WPA-PSK" >> /tmp/hostapd.conf...fi....# Check nl80211...chk_nl80211...rval=$?...if [ ${rval} -eq 1 ]; then....echo "driver=nl80211" >> /tmp/hostapd.conf....if [ ${AP_CHANNEL} -gt 14 ]; then.....echo "hw_mode=a" >> /tmp/hostapd.conf....else.....echo "hw_mode=g" >> /tmp/hostapd.conf....fi....echo "ieee80211n=1" >> /tmp/hostapd.conf....if [ ! -e /sys/module/bcmdhd ]; then.....# TODO: Support HT40 for 5G.....if [ ${AP_CHANNEL} -lt 6 ]; then......# HT40+ for 1-7 (1-9 in Europe/Japan)......echo "ht_capab=[SHORT-GI-20][SHORT-GI-40][HT40+]" >> /tmp/hostapd.conf.....else......# HT40- for 5-13......echo "ht_capab=[SHORT-GI-20][SHORT-GI-40][HT40-]" >> /tmp/hostapd.conf.....fi....fi....#echo "wme_enabled=1" >> /tmp/hostapd.conf....#echo "wpa_group_rekey=86400" >> /tmp/hostapd.conf...fi..fi.}..wpa_supplicant_conf().{..if [ -e /sys/module/ar6000 ]; then...driver=wext..else...driver=nl80211..fi...if [ "${usecache}" != "" ] && [ -e /tmp/wpa_supplicant.ap.conf ]; then...cat /tmp/wpa_supplicant.ap.conf..else...#generate /tmp/wpa_supplicant.ap.conf...echo "ctrl_interface=/var/run/wpa_supplicant" > /tmp/wpa_supplicant.ap.conf...echo "ap_scan=2" >> /tmp/wpa_supplicant.ap.conf....#AP_MAXSTA...AP_MAXSTA=`echo "${conf}" | grep AP_MAXSTA | cut -c 11-`...echo "max_num_sta=${AP_MAXSTA}" >> /tmp/wpa_supplicant.ap.conf....echo "network={" >> /tmp/wpa_supplicant.ap.conf...#AP_SSID...AP_SSID=`echo "${conf}" | grep AP_SSID | cut -c 9-`...echo "AP_SSID=${AP_SSID}"...echo "ssid=\"${AP_SSID}\"" >> /tmp/wpa_supplicant.ap.conf....#AP_CHANNEL...AP_CHANNEL=`echo "${conf}" | grep AP_CHANNEL | cut -c 12-`...echo "AP_CHANNEL=${AP_CHANNEL}"...if [ ${AP_CHANNEL} -lt 0 ]; then....reset_conf....return 1...fi....# TODO: for 5G...if [ $AP_CHANNEL -eq 0 ]; then....#choose 1~10 for HT40....#AP_CHANNEL=`echo $(( $RANDOM % 11 +1 ))`....AP_CHANNEL=`echo $(( ($RANDO1...èžmxæ.......0.......g...... ................M % 3) * 5 + 1 ))`...fi....# cf. http://en.wikipedia.org/wiki/List_of_WLAN_channels...if [ $AP_CHANNEL -lt 14 ]; then....# 2.4G: 2412 + (ch-1) * 5....echo "frequency=$((2412 + ($AP_CHANNEL - 1) * 5))" >> /tmp/wpa_supplicant.ap.conf...else....# 5G: 5000 + ch * 5....echo "frequency=$((5000 + $AP_CHANNEL * 5))" >> /tmp/wpa_supplicant.ap.conf...fi....#WEP, WPA, No Security...AP_PUBLIC=`echo "${conf}" | grep AP_PUBLIC | cut -c 11- | tr '[:lower:]' '[:upper:]'`...if [ "${AP_PUBLIC}" != "YES" ]; then....# proto defaults to: WPA RSN....echo "proto=WPA2" >> /tmp/wpa_supplicant.ap.conf....echo "pairwise=CCMP" >> /tmp/wpa_supplicant.ap.conf....echo "group=CCMP" >> /tmp/wpa_supplicant.ap.conf....AP_PASSWD=`echo "${conf}" | grep AP_PASSWD | cut -c 11-`....echo "psk=\"${AP_PASSWD}\"" >> /tmp/wpa_supplicant.ap.conf....echo "key_mgmt=WPA-PSK" >> /tmp/wpa_supplicant.ap.conf...else....echo "key_mgmt=NONE" >> /tmp/wpa_supplicant.ap.conf...fi...echo "mode=2" >> /tmp/wpa_supplicant.ap.conf...echo "}" >> /tmp/wpa_supplicant.ap.conf...if [ -e /sys/module/bcmdhd ]; then....echo "p2p_disabled=1" >> /tmp/wpa_supplicant.ap.conf...fi..fi.}..bcm_ap_start().{..#AP_SSID..AP_SSID=`echo "${conf}" | grep AP_SSID | cut -c 9-`..echo "AP_SSID=${AP_SSID}"...#AP_MAXSTA..AP_MAXSTA=`echo "${conf}" | grep AP_MAXSTA | cut -c 11-`..echo "AP_MAXSTA=${AP_MAXSTA}"...#AP_CHANNEL..AP_CHANNEL=`echo "${ [... cut ...] ......é.......................................................1...lwÝ–........@.......T...·õ®J........r...............load.sh.1...y.Àú................s...... ........Ò.......#!/bin/sh.sysl=`ps | grep syslogd | grep -v grep`.LOGGER ().{..if [ "${sysl}" == "" ]; then...echo "$@"..else...logger "$@"..fi.}..sleep 0.5.LOGGER "$@"..HCI_LE_Set_Advertising_Parameters ().{..LE_Controller_Commands_OGF=0x08..HCI_LE_Set_Advertising_Parameters_OCF=0x0006..#500 ms..Advertising_Interval_Min="20 03"..#4 sec..Ad

mefistotelis, 2017-02-27:
What you have here looks like a part of Linux partition from the Ambarella module. You should have written that.

Unless you haven't really extracted separate modules, in which case you should get the tools which are being developed in this GIT repository, and learn how to use them from README.

If you want to extract strings from a firmware, it is best to use "strings" command from binutils package.

mefistotelis, 2017-02-28:
One more thing - note that modifying the firmware without extracting modules will not work anyway, because the firmware contains checksums. Modify without extraction -> checksum becomes invalid.

closed 2021-06-14 by mefistotelis