o-gs/dji-firmware-tools #22: Phantom 3 boards photos request

mefistotelis, 2017-02-27:
It would help in hacking if we had sharp pictures of all the boards making up Phantom 3 Pro/Adv (both Aircraft and RC).

@notsolowki already prepared photos of camera top board - these are not perfect, but did helped me a lot:
https://github.com/mefistotelis/phantom-firmware-tools/issues/5#issuecomment-270053099

The photos should be in good enough quality so that the markings on chips are readable; if they're not - it is best to follow whole board photo by a close-up on parts of the board, and find an angle at which the markings are well visible.

We need photos of both sides of each board.

Photos of the main controller board are easy to find in google; but I wasn't able to find good photos of the RC internals.

MrBurnsAT, 2017-02-27:
Ill open my GL300B and make Photos the next Days!

neven385, 2017-02-27:
Count me in for GL300C. I'll take some pics today.

MrBurnsAT, 2017-02-27:
http://www.phantompilots.com/attachments/dsc03660_cr-jpg.28141/

This should be back side if 300B

neven385, 2017-02-27:
[image]

notsolowki, 2017-02-27:
Heres a couple of pics of the rc board. Therly are not the best i took then long ago.
[image]
[image]

mefistotelis, 2017-03-02:
I found this about GL300C:
http://www.sb-dji.com/?p=12751

But I don't think GL300A uses the same components. For example the ARM firmware which I worked on has striking resemblance to LPC17xx series, and there are major differences between it and the LPC1549 used in GL300C.

Anyone can share pictures of chips from A or B?

mefistotelis, 2017-03-03:
I've checked mine, but the USB board only:

[image]
[image]

coptersafe, 2017-03-09:
no any difference between A and B . only firmware and hdmi board different.

B model - make USB video on FPGA processor, but A - model just transmit raw signal from traseiver to HDMI board - that all differences.

GlovePuppet, 2017-03-17:
I can tell you the following:

P3S (and probably all the others) FC is STM32F417

P3S Gimbal has STM32F407, DSPIC32 and Artheros AR9342 

Model A RC TX has AR9342 (Running openwrt no less - I desoldered the SPI flash and dumped it). Also STM32F074 and the RF chip is Beken BK5811

When my wife gets back next weekend I will get you some high res pics of various PS main boards (also RC TX, Gimbal controller, OFDM etc). A lot of the chips numbers are obscured by the conformal coating they use and I have to remove it and squint to read the part numbers.

There's a gs_ofdm.bin file floating around people are using to unbrick their RC TX and that is targeted at  LPC17xx (they left the RTC library ASSERTS in the binary :S)

mefistotelis, 2017-03-17:
Great; it would help me a lot to have all boards photographed.

If you're looking at the code of modules, you may use the partially recovered symbols:
https://github.com/mefistotelis/phantom-firmware-tools/tree/master/symbols

There are many symbols named for C1_FW_V01.06.0000_mi06 (which is the LPC17xx firmware).

rylan01, 2017-03-19:
Ok, I have a few things I can photograph:
P3 Pro
OFDM Board v2
GL300A with original USB module
HDMI Module
Spare USB module after the original USB module in the GL300A stopped working after an update
P3 Advanced Camera, picked up on eBay because the original camera stopped transmitting live video after an update.
An additional P3 Pro Camera which also stopped working after an update (nice trend here.)

Willing to dismantle to get any pictures. If I dont get the camera live feed going, I am probably just going to sell all this stuff.

GlovePuppet, 2017-03-19:
I found a few pictures of my RC TX.

GL358wA Main PCB
[image]

GL358wA BK5811 PCB
[image]

GL358wA WIFI Front PCB
[image]

GL358wA WIFI Back PCB
[image]

GlovePuppet, 2017-03-22:
After a little bit of time with my DMM and a P3S main board:

[image]

The WiFi connector on the main PCB connects to the Gimbal PCB, there are 2 UARTs on the STM32F07 connected to the Gimbal's STM32407 and the Artheros. 

All the ESCs UART's are common-ed up and connected to the FC, I have only seen traffic being transmitted by the FC

I will provide an annotated picture and pinouts when the AV dept gets home at the weekend

notsolowki, 2017-03-22:
Maybe someone will figure out how to recover the bootloader on the
stm32f407. Would be nice to unbrick them.

GlovePuppet, 2017-03-22:
That's one of my goals - see my Hackaday blog for a dissection of the Naza M/P2V+

https://hackaday.io/project/19995-hacking-dji-naza-m/log/53751-big-dump

notsolowki, 2017-03-22:
great job. i have 2 gimbals that i bricked by trying to flash the encrypted module to the stm32f407. someone on here mentioned that the bootloader from the stm32f407 could not be recovered. but i think your on the right path to decrypting everything.

GlovePuppet, 2017-03-22:
My P3A OFDM receiver (V2) just arrived. Big chips:

STM32F103 (boring)
CY7C68013 (USB uController)
Aryston AR8001 & AR8003 (Can't find any data)
2 * PA5208 (Appears to be the RC RF RX)

GlovePuppet, 2017-03-22:
Thinking about the Naza M (and its AT88SC crypto memory) reminded me that there is a ref to an sha204p so there is likely on of these on the FC

http://www.atmel.com/Images/Atmel-8885-CryptoAuth-ATSHA204A-Datasheet.pdf

coptersafe, 2017-03-22:
i think no way to decrypt or hack or bruteforce AT88.. need another way.
also AT88 store configuration drone file

GlovePuppet, 2017-03-22:
You should read my "Big Dump" - link above. Here is what I learned about AT88SC in Naza M / P2V+

1) There is a secret seed that both AT88SC and Naza M share, every Naza M / P2V+ has a different seed
2) DJI try to hide the seed in the bootloader
3) DJI screwed up implementation (Atmel say use 8 byte nonce but DJI use 8 * 0). Maybe this makes a 'hack' much easier but who cares? Dump the bootloader, get the seed ;)
4) All (including serial number, device type and license key) config data is stored in AT88SC

I think there is an Atmel ATSHA204(A) in the P3

coptersafe, 2017-03-22:
hhmm.. sorry. seems i mistaked. later i make photo P3 FC - they use STM32F27 proc as main FC proccessor

GlovePuppet, 2017-03-22:
My samples use ST32F417 as FC, ST32F407 in Gimbal base, ST32F072 for RC receiver. I have OFDM RX that uses ST32F103 instead (I think older version uses Nuvoton M051 based on reversing binary)

GlovePuppet, 2017-03-26:
OFDM V2 top
[image]

OFDM V2 bottom
[image]

P3S Gimbal top
[image]

P3S Gimbal bottom
[image]

P3S Main top
[image]

P3S Main bottom
[image]

P3A/P Main top
[image]

P3A/P Main bottom
[image]

mefistotelis, 2017-03-26:
Thank you @GlovePuppet.

The hardware database is slowly taking shape:
https://github.com/mefistotelis/phantom-firmware-tools/wiki/DJI-Hardware
But there are still a few photos missing - if anyone can make a better quality photo than those currently on the wiki, or has an unobfuscated version of any usually painted chip - please share.

@rylan01, from your list I'm especially waiting for the OFDM v2 and GL300a main board.

GlovePuppet, 2017-03-26:
No problem, I will try and find some time to annotate some of the pictures with IC part numbers. I think I listed most of the major ICs on the boards I have above

mefistotelis, 2017-04-20:
I finished sharpening all the relevant photos I had; the wiki now contains info on all P3X, P3S and P3C boards.

If anyone has an additional photo, it is a good time to share.
I'm especially interested in:
- Flight controller (MC) photo with chip markings visible
- GL300(a/b/c) main board photo with main chip (Altera/Lattice(?)) markings visible
- VPS module chip markings
Less important but still missing markings:
- Gimbal pitch/roll control chip
- The compass chip
- older version of OFDM module (v1) main chip

Also, if you have more information on a specific board (ie. list of components, or service pins descriptions), please either make an issue with update request, or clone the wiki repository and prepare a patch.

GlovePuppet, 2017-04-27:
About the wiki: I would spend some time to document the FC commands so far. Do you want a patch or is this a possible approach:

http://stackoverflow.com/questions/10642928/how-to-pull-request-a-wiki-page-on-github#11481887

mefistotelis, 2017-04-27:
I would accept both ways. Do what is preferable to you.

We definitely need such "communication protocol" wiki, for all available serial interfaces.

mefistotelis, 2017-05-07:
Hey @notsolowki, are the pictures of RC board which you posted from GL300c? They does not show signs of BGA chip on the other side, so the board you posted definitely uses Artosyn SDR instead of FPGA.

notsolowki, 2017-05-07:
Yes they are from the gl300c, is that a good thing?

mefistotelis, 2017-05-07:
I marked it as GL300b for some reason, and got baffled after I was able to recognize any board by just looking at it. Thanks for clarification.

MrBurnsAT, 2017-07-03:
Here are photos of an non working HDMI Board from an GL658B Inspire Remote.
I replaced the HDMI Board with an new one. Now RC is back to Live.

I think it has the NAND problem.

Board is labelled "WM610_GRCP_MFI368_V1"

Second Board is an "WM610_GRCP_INTERFACE_V3"

[image]
[image]
[image]
[image]

mefistotelis, 2017-07-03:
Thank you @MrBurnsAT.

In the meantime I scrubbed resin and photographed by OFDMv1 board and gimbal roll driver. Also, @pawelski sent me his GL300b main board photos.

So we mostly miss VPS board info; but GL300a main board photo would also help, and we don't know the compass chip markings.

coptersafe, 2017-07-03:
compass is hmc5883l

mefistotelis, 2017-07-03:
> compass is hmc5883l

Updated [wiki](https://github.com/mefistotelis/phantom-firmware-tools/wiki/P3-Compass-board).

MrBurnsAT, 2017-07-03:
Like i sayed

Hooked up serial adapter
Here is what i get:

DM36x initialization passed!
UBL Product Vesion : DJI-GRC-SUPER-UBL-1.0-rc0(2014-09-19)
Dji UBL Version: 1.51(Sep 19 2014 - 16:13:09)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
Valid magicnum, 0xA1ACED66, found in block 0x0000001D.
No valid boot image found!
NAND Boot failed.
Aborting...

Do we have Firmware files for this?
Can we extract uboot image?
Maybe like lightbirdge hack

mefistotelis, 2017-07-03:
> MrBurnsAT: Like i sayed...

This doesn't seem related to photos. Please move the questions to proper thread.

MrBurnsAT, 2017-07-03:
Sorry na fault! Ill Open a New issue

MrBurnsAT, 2017-07-08:
Found some interesting Pics on
http://www.sb-dji.com/?p=20637

From LB2 RC (GL658) with SDI Out board

I think that could be the board for m1301 Module

[image]

[image]

MrBurnsAT, 2017-07-09:
So Now i brought a SDI Module 

[image]

Have to wait some days for arrive. 

Fotos will come as fast as i can

MrBurnsAT, 2017-07-09:
Next interesting Thing i found!

Channel Expansion Kit 
https://m.dji.com/de/product/matrice-600-series-remote-controller-channel-expansion-kit

Look at the Fotos!
Looks like an Standard HDMI Module where the extra switches are connected via CAN Bus

danieltroger, 2017-08-03:
This isn't phantom 3 pro/adv but phantom 3 standard, but I've got two "proposals"to  these [three](https://github.com/mefistotelis/phantom-firmware-tools/wiki/WM321-Gimbal-top-board) [wiki](https://github.com/mefistotelis/phantom-firmware-tools/wiki/WM321-Camera-encoder-board) [pages](https://github.com/mefistotelis/phantom-firmware-tools/wiki/WM321-Camera-sensor-board) (@mefistotelis):

1. On the wiki it says: "The top gimbal board is responsible for **processing**, **encoding** and transmitting video for live FPV feed. It also controls the gimbal.": Are you sure? Isn't the reason that the p3 standard only does 2.7k just that, that the Ambarella SoC has to encode the live feed *and* what is being recorded to the sd card simultaneously, and therefore only can do 2.7k to the sd card? Another hint would be that you [get the encoded h264 stream from 192.168.1.3](https://github.com/hardening/Cave/blob/master/phantom3-video.cpp) which clearly is the Ambarella SoC on the camera encoder board.
2. When I recently (last friday) recieved my p3s from DJI's repair department (they got it to replace the gimbal top board) the focus was bad, so I opened the camera housing to adjust the screws. I thought the Amba SoC looked new. Then, one screw on the sensor board was impossible to open. I damaged some resistors around it
[image]
So I thought I'd replace the camera housing + sensor board with another one I had laying around. I switched it on, but it kept saying "SD Card error". Then I noticed both boards in the camera housing where different and a new revision.
So here are some pictures of the camera encoder board, the sensor board is impossible to get out due to a broken screw :/
It's mainly a new (?) Amba SoC (A9SE) with different flash + ram. Also, the root user on telnet 192.168.1.3 is no longer password protected.

Angled view so you can read:
[image]

Crappy pics from a vertical POV:
[image]
[image]

Probably the gimbal top board is differen, too. But if nobody is interested in pics I don't wanna open it at the mo., 'cause it works.

MrBurnsAT, 2017-08-03:
I would try to drill of the head of the screw.

Than u should be able to get the board out of the housing. Most times there should be a little rest of the screw that u can turn it out witch calipers

If u have a second board, measure the 2 resistors on this which now are off at the new board and replace them.

MrBurnsAT, 2017-08-03:
PS Gimbal top Board of Pro should have a TI DaVinci DM368 for encoding which Standard Top Board should not have!

mefistotelis, 2017-08-03:
Thank you @danieltroger.

And you're right, the top board in Standard gets the signal already compressed/encoded, it just sends it via Wi-Fi.

danieltroger, 2017-08-03:
Those are capacitors? Uh, ok.
Yeah, I probably could take some off another board, but those resistors are a bit small to solder with my normal iron.
Oddly enough, everything still works. The gimbal only "goes crazy" when you switch it on (puts the camera in some random position after the initialization), but if I leave the drone for a minute it slowly goes back to the normal position and works fine in the air, so I'm going to leave it like this.
Can you add the pictures to the wiki, @mefistotelis ? I don't know how to fork a wiki so it'd be easier. Also, though I bet you already know it, here there are some fantastic pictures of the p3s's internals, if you need some good ones: https://www.scotttorborg.com/dji-phantom-3-standard-teardown

@MrBurnsAT yeah, I'll do that as soon as I get my hands on a good drill, thanks for the advice.

mefistotelis, 2017-08-03:
> here there are some fantastic pictures of the p3s's internals, if you need some good ones: https://www.scotttorborg.com/dji-phantom-3-standard-teardown

Yes, I already used these.

If you have enough determination, you could also help by reading markings on each chip from the photos and writing them as text; that would make it easier to prepare the "parts" chapter on the wiki.

MrBurnsAT, 2017-11-04:
do u still need photos?

teared down a crashed P3P without gimal
can make ever thing with this

MrBurnsAT, 2017-11-04:
here 2 snapshots with crapy lighting

[image]

[image]

mefistotelis, 2017-11-04:
I'm definitely missing VPS module, and GL300 main board.

Can you scrub the resin from VPS chips? I'm usually doing it with a blade taken out of segment knife.
Requires time and precision, but is doable. If you're not willing to risk damaging your boards, just make photos as they are.

MrBurnsAT, 2017-11-04:
[image]

MrBurnsAT, 2017-11-04:
ok vps module comes take a moment

MrBurnsAT, 2017-11-04:
pls

[image]

[image]

[image]

[image]

[image]

MrBurnsAT, 2017-11-05:
[image]

[image]

MrBurnsAT, 2017-11-05:
[image]

[image]

[image]

[image]

MrBurnsAT, 2017-11-05:
[image]

[image]

[image]

mefistotelis, 2017-11-07:
VPS wiki updated; thank you. I will use that IMU image too, sooner or later.

noiseBOY, 2018-05-17:
Hi.I want to communicate with the board.
please guide me.
And what's the output of the board?

noiseBOY, 2018-05-17:
Hi.I want to communicate with the board vps phantom(dji).
please guide me.
And what's the output of the board?

mefistotelis, 2018-05-17:
> I want to communicate with the board vps phantom(dji). please guide me.

Get USB-to-TTL, solder to service pads which look like a serial interface and go with it. When you have any findings, please report them so we can update the wiki. You can create a separate issue for your progress and related questions.

[8 more replies]

closed 2021-06-14 by mefistotelis