o-gs/dji-firmware-tools #23: new P3C firmware ..

coptersafe, 2017-03-09:
https://s3-ap-northeast-1.amazonaws.com/dji-firmware-test/P3C/P3C_FW_V01.09.0080.bin 

strange and huge firmware ( consists two module for camera FC3XX )

MrBurnsAT, 2017-03-15:
War is P3C???

Only know
P3X Pro
P3S Advanced

KennethMcNutt, 2017-06-14:
Yes, that is what I understand...
P3X_FW_DEBUG (Pro)
P3XW_FW_DEBUG (P3-4K)
P3S_FW_DEBUG (ADV)
P3C_FW_DEBUG (STA)
It was a surprise to me...
Everything I read is DJI Go4 and above??
The P3S I assumed would be Dropped?? Before this "New Update" because it is last of the P3...
So the P3S will live on?? :) 
But as of yet, there is no Law in the USA to abide by??  But the rest of the world has "laws" in place.. So... we will soon see...

mefistotelis, 2021-06-14:
DJIs AWS storages are now access controlled.
So not much useful info here, closing.

closed 2021-06-14 by mefistotelis