o-gs/dji-firmware-tools #26: Feature Licensing

GlovePuppet, 2017-03-25:
If you read my Naza M/P2V+ write up then you know that the 32 character license string can be calculated like this:

YYYY000000000000f4XXXXXXXXXX50

Where:

Y – May be be '1' or '0', setting it to '1' enables the feature. Only the 4 left hand bits can be set in a valid license file

XXXXXXXXXX – The 10 digit serial number of the FC, for example, 1000000123

On the Linux console a valid “serial number” with no feature flags can be generated for my example FC:

$ printf 0000000000000000f4100000012350 | md5sum

Turns out the same code is in the P3 FC, the only difference being that f450 -> m4a3

This, then, should calculate a license string for a serial number of 03Z0397973 

$ printf 0000000000000000m403Z0397973a3 | md5sum

I haven't actually _seen_ such a string in the GUI but I only have access to a P3S. Can someone with one of the more advanced models tell me is there anything that looks like a string of 32 random characters?

GlovePuppet, 2017-04-02:
I have now confirmed the licensing scheme. The P3S FC will answer to the following packet on its UART interface to the Artheros WIFI chip:

7003 [02 -> 03] 0D : 55 0D 04 33 02 03 0C 0D 40 03 70 02 5C
7003 [03 -> 02] 44 : 55 44 04 1A 03 02 0C 0D 80 03 70 01 65 30 61 38 64 64 36 33 63 61 61 31 65 32 36 36 64 61 62 65 36 36 33 33 66 33 34 36 66 36 38 31 00 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 00 1D 00 00 00 ED 33

MD5Sum:  65 30 61 38 64 64 36 33 63 61 61 31 65 32 36 36 64 61 62 65 36 36 33 33 66 33 34 36 66 36 38 31

License Bits: 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30

This FC has a serial number of: 03Z0510812

Then we can check that the licensing scheme is based on an MD5 checksum

 $ printf 0000000000000000m403Z0510812a3 | md5sum | hexdump -C
00000000  65 30 61 38 64 64 36 33  63 61 61 31 65 32 36 36  |e0a8dd63caa1e266|
00000010  64 61 62 65 36 36 33 33  66 33 34 36 66 36 38 31  |dabe6633f346f681|

This also confirms that the FC is at address 3 which matches with what I see in the binary

Mavic129, 2017-09-03:
hi @GlovePuppet 
i was testing some commands on Inspire 1 latest FW and remembered your post

55 0D 04 33 0A C3 10 27 40 00 01 76 5D

55 2C 04 36 C3 0A 10 27 80 00 01 00 00 30 34 31 
44 33 35 32 36 34 31 00 00 00 00 00 00 05 00 01 
22 32 14 04 02 09 00 00 00 01 D3 68

55 0D 04 33 0A 03 11 27 40 00 01 41 51

55 2C 04 36 03 0A 11 27 80 00 01 00 00 30 34 31 
44 33 35 32 36 34 31 00 00 00 00 00 00 05 00 01 
22 32 14 04 02 09 00 00 00 01 20 F5

both above messages returned SN: 041D352641

55 0D 04 33 0A 03 12 27 40 03 71 62 15

55 44 04 1A 03 0A 12 27 80 03 71 00 34 34 35 33
37 38 34 38 36 37 65 33 62 66 61 66 62 61 30 66
34 31 36 62 63 33 38 36 66 64 61 35 00 30 30 30
30 30 30 30 30 30 30 30 30 30 30 30 30 00 13 00
00 00 55 CA

55 0D 04 33 0A 03 13 27 40 03 70 AF 0F

55 44 04 1A 03 0A 13 27 80 03 70 01 34 34 35 33
37 38 34 38 36 37 65 33 62 66 61 66 62 61 30 66
34 31 36 62 63 33 38 36 66 64 61 35 00 30 30 30
30 30 30 30 30 30 30 30 30 30 30 30 30 00 12 00
00 00 2C 45

both above messages returned MD5: 
4453784867e3bfafba0f416bc386fda5.0000000000000000

$printf 0000000000000000f4041D35264150 | md5sum
4453784867e3bfafba0f416bc386fda5