o-gs/dji-firmware-tools #27: Stm32f407

notsolowki, 2017-03-28:
Can anyone find a solution to unbricking this chip.

notsolowki, 2017-03-28:
Maybe one of you wpuld be willing to sell me the gimbals mainboard cheap?

coptersafe, 2017-03-28:
take it free.. ) which version and model you need ?

coptersafe, 2017-03-28:
also i think you may reflash STM when he in a boot mode.. ..may be.

notsolowki, 2017-03-28:
Seriously? I need gimbal board for p3p. Any version should do. If your
serious my email is notsolucki@gmail.com

coptersafe, 2017-03-28:
ok. no problem.

notsolowki, 2017-03-28:
I flashed the encrypted module to the gimbal and this is where im at .
Thanks

mefistotelis, 2017-03-28:
We would need to have bootloader code to get further. I'm pretty sure there is a way to force it boot into bootloader by shorting specific pins. At least if I would be making such bootloader, I would definitely make several methods of entering the "flash mode".

notsolowki, 2017-03-28:
I belive there are pads on the board. Boot0? To force the chip into
bootloader mode. I would love to get my gimbal working agin. Would i need
the stm flasher? If your thinking abput possibly recovering or writing the
boooader then should i go ahead and get the flasher?. Thanks

coptersafe, 2017-03-28:
no you dont STM flasher - just flash via standart way. Also  when  I brick my Osmo - i found that after few bat starts ,Osmo self reflashed itself. may be with gimbal also this work.

solder at anoyher RX TX to view flashing log

coptersafe, 2017-03-28:
---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 132000000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 15428571
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x5acad81e 0x5acad81e 0x5acad81e
System COLD boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 54000000
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x1 0x789aedca
System WARM boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 27000000
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x2 0x789aedc9
System WARM boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 27000000
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x3 0x789aedc8
System WARM boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 27000000
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x4 0x789aedcf
System WARM boot.
Detecting done ......

[00000000][CA9_0] [DJI] Start WDT of 20 seconds for boot up......

[00000000][CA9_0] [Version 1.0.11824], Oct 16 2015, 14:06:46

[00000000][CA9_0] [DJI] SD Updater Version [v01.26.4315] Build Nov  5 2015 15:33:09

[00000333][CA9_0] [DJI] Stop WDT.

[00000333][CA9_0] [DJI] Firmware updating......

[00000333][CA9_0] Card In Slot.
[00000634][CA9_0] SD clock source, frequency = 216000000
[00000634][CA9_0] SDCalcClk for SD card initiation:: Actual Clk = 140625, pCLK = 128
[00000653][CA9_0] start to check media
[00000660][CA9_0] AmbaSD_InitCard - SD_CARD OK
[00000664][CA9_0] Packet Detected / c:\P3X_FW_V01.06.0040.bin /.

[00000672][CA9_0] [DJI] Try Ddr Param 0.

[00000673][CA9_0] Back up partion data ......
[00028949][CA9_0] Image "System Software" is found!
[00028949][CA9_0] Image "DSP uCode" is found!
[00028949][CA9_0] Image "System ROM Data" is found!
[00028949][CA9_0] Image "Linux Kernel" is found!
[00028949][CA9_0] Image "Linux Root FS" is found!
[00028949][CA9_0] Start firmware CRC check...

[00028949][CA9_0] Checking System Software
[00032599][CA9_0] Verifying image CRC ... done
[00032599][CA9_0]       length:         9195328
[00032599][CA9_0]       crc32:          0x6e93a81e
[00032599][CA9_0]       ver_num:        1.0
[00032599][CA9_0]       ver_date:       2015/11/5
[00032599][CA9_0]       img_len:        9195072
[00032599][CA9_0]       mem_addr:       0x0e600000

[00032599][CA9_0] Checking DSP uCode
[00034153][CA9_0] Verifying image CRC ... done
[00034153][CA9_0]       length:         3889408
[00034153][CA9_0]       crc32:          0xd97b99c4
[00034153][CA9_0]       ver_num:        1.0
[00034153][CA9_0]       ver_date:       2015/11/5
[00034153][CA9_0]       img_len:        3889152
[00034153][CA9_0]       mem_addr:       0x0ff00000

[00034154][CA9_0] Checking System ROM Data
[00038801][CA9_0] Verifying image CRC ... done
[00038801][CA9_0]       length:         11624704
[00038801][CA9_0]       crc32:          0xc6dfedab
[00038801][CA9_0]       ver_num:        1.0
[00038801][CA9_0]       ver_date:       2015/11/5
[00038801][CA9_0]       img_len:        11624448
[00038801][CA9_0]       mem_addr:       0x03c00000

[00038801][CA9_0] Checking Linux Kernel
[00040814][CA9_0] Verifying image CRC ... done
[00040814][CA9_0]       length:         5038864
[00040814][CA9_0]       crc32:          0xf6a51577
[00040814][CA9_0]       ver_num:        1.0
[00040814][CA9_0]       ver_date:       2015/11/5
[00040814][CA9_0]       img_len:        5038608
[00040814][CA9_0]       mem_addr:       0x00508000

[00040814][CA9_0] Checking Linux Root FS
[00051507][CA9_0] Verifying image CRC ... done
[00051507][CA9_0]       length:         26870016
[00051507][CA9_0]       crc32:          0xb7355472
[00051507][CA9_0]       ver_num:        1.0
[00051507][CA9_0]       ver_date:       2015/11/5
[00051507][CA9_0]       img_len:        26869760
[00051507][CA9_0]       mem_addr:       0x00001000

[00052085][CA9_0] Program "System Software" to NAND flash ...
[00054611][CA9_0] Done
[00054855][CA9_0] Program "DSP uCode" to NAND flash ...
[00055897][CA9_0] Done
[00056618][CA9_0] Program "System ROM Data" to NAND flash ...
[00059726][CA9_0] Done
[00060041][CA9_0] Program "Linux Kernel" to NAND flash ...
[00061402][CA9_0] Done
[00063066][CA9_0] Program "Linux Root FS" to NAND flash ...
[00070290][CA9_0] Done
[00070625][CA9_0] [DJI] Done!

[00070625][CA9_0] [DJI] Reboot......

notsolowki, 2017-03-28:
I have restarted it alot of times. It would be great to get it working
again. I have 2 of them with the exact same problem. When i turn the gimbal
on it dont move or anything.

notsolowki, 2017-03-28:
Im sorry which rx tx are you talking about. Thankyou

solder at anoyher RX TX to view flashing log

coptersafe, 2017-03-28:
also .. you can go to consol ambarella and type command manuall for reflash module 0400

coptersafe, 2017-03-28:
pins - U0_TX,  U0_RX

coptersafe, 2017-03-28:
Start WDT timer OK.
Magic: 0x87651234 0x4 0x789aedcf
System WARM boot.
Detecting done ......

[00000002][CA9_0] [DJI] Start WDT of 20 seconds for boot up......

[00000002][CA9_0] [Version 1.0.11824], May 29 2015, 18:59:46

[00000002][CA9_0] [DJI] SD Updater Version [v01.22.4095] Build Oct  9 2015 14:32:09

[00000502][CA9_0] [DJI] Stop WDT.

[00000502][CA9_0] [DJI] Firmware updating......

[00000502][CA9_0] Card In Slot.
[00000803][CA9_0] SD clock source, frequency = 216000000
[00000803][CA9_0] SDCalcClk for SD card initiation:: Actual Clk = 140625, pCLK = 128
[00000822][CA9_0] start to check media
[00000829][CA9_0] AmbaSD_InitCard - SD_CARD OK
[00000833][CA9_0] Can't detect firmware.
[00000900][CA9_0] Firmware  program fail--- ARM DATA ABORT EXCEPTION ---
Oops: CPU Exception!
pc : [<00011348>]    lr : [<00000000>]
sp : 0006f328  ip : 00000000  fp : 00000000
r10: 0006b400  r9 : 00000000  r8 : 00000000
r7 : 00000000  r6 : 00000000  r5 : 00000000
r4 : 0006f394  r3 : 00000000  r2 : 00000000
r1 : fff90c5b  r0 : ffffffa4
Flags: NzCv
IRQs on  FIQs on  Mode SVC_32
.

notsolowki, 2017-03-28:
. What baud and pairing settings shpuld i choose. Im going to wire it up
and post what i get. thanks again

coptersafe, 2017-03-28:
about baud . dont remember .. it was long time ago

notsolowki, 2017-03-28:
Which one are you talking about 
[image]
[image]

coptersafe, 2017-03-28:
1 - ok , 2 - console Davinchi

fo console Amba , you must go to console Davinchi , and then from DaVinchi go to Amba

coptersafe, 2017-03-28:
console Davinchi 115200

notsolowki, 2017-03-28:
Okay your loosing me lol. What is console davinchi?

notsolowki, 2017-03-28:
here is what i get from the standard console 8n1 115200

Êë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åë¿¿¿¿y!5¿{!!¥U¿Y995åëåëy!5¿3‹¿•Ÿ—ŸŸŸŸŸŸåë}[e¿¿¿¿3‹¿™›ŸŸŸŸŸŸåëww[¿¿¿¿3‹¿•›ŸŸŸŸŸŸåëy!¿¿¿3‹¿›“ŸŸŸŸŸŸåë-wY_¿¿¿3‹¿›“—ŸŸŸŸŸŸåë}Om¿¿¿¿3‹¿“ŸŸŸŸŸŸåë}o{¿¿¿¿3‹¿ŸŸŸŸŸŸŸåë}_{¿¿¿¿3‹¿•—ŸŸŸŸŸŸåëU}[W¿¿¿3‹¿—“•™åëYwŸ¿¿¿¿3‹¿•—›•‘åëYw¿¿¿¿3‹¿•—›•‘åë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥5%¿yagw¿;!!£åëy!#-#5¿!¿;!!-#1¿¿££££££åëë‹¿Ÿ•=9=75¿Ÿ•=9=75¿Ÿ•=9=75åëY

mefistotelis, 2017-03-28:
Looks quite good. The line:
<pre>¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥</pre>
is almost like original:
<pre>---------------------------------------------</pre>

The length matches, so it looks like you got the speed right.

notsolowki, 2017-03-28:
Do i need to use a max323 or can i get away with a standard usb to serial
adapter? It is printing the startup to the console. the console settings
must be wrong?

notsolowki, 2017-03-28:
[image]

notsolowki, 2017-03-28:
Ive tried a mixture of all the  #data and stop bits. Do you think parity needs adjusted too? Thanks

notsolowki, 2017-03-28:
this is what it looks like after about 1 minute of sitting there

”ë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åë¿¿¿¿y!5¿{!!¥U¿Y995åëåëy!5¿3‹¿•Ÿ—ŸŸŸŸŸŸåë}[e¿¿¿¿3‹¿™›ŸŸŸŸŸŸåëww[¿¿¿¿3‹¿•›ŸŸŸŸŸŸåëy!¿¿¿3‹¿›“ŸŸŸŸŸŸåë-wY_¿¿¿3‹¿›“—ŸŸŸŸŸŸåë}Om¿¿¿¿3‹¿“ŸŸŸŸŸŸåë}o{¿¿¿¿3‹¿ŸŸŸŸŸŸŸåë}_{¿¿¿¿3‹¿•—ŸŸŸŸŸŸåëU}[W¿¿¿3‹¿—“•™åëYwŸ¿¿¿¿3‹¿•—›•‘åëYw¿¿¿¿3‹¿•—›•‘åë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥5%¿yagw¿;!!£åëy!#-#5¿!¿;!!-#1¿¿££££££åëëÊë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åë¿¿¿¿y!5¿{!!¥U¿Y995åëåëy!5¿3‹¿•Ÿ—ŸŸŸŸŸŸåë}[e¿¿¿¿3‹¿‘“ŸŸŸŸŸåëww[¿¿¿¿3‹¿•›ŸŸŸŸŸŸåëy!¿¿¿3‹¿›“ŸŸŸŸŸŸåë-wY_¿¿¿3‹¿›“—ŸŸŸŸŸŸåë}Om¿¿¿¿3‹¿“ŸŸŸŸŸŸåë}o{¿¿¿¿3‹¿ŸŸŸŸŸŸŸåë}_{¿¿¿¿3‹¿•—ŸŸŸŸŸŸåëU}[W¿¿¿3‹¿—“•™åëYwŸ¿¿¿¿3‹¿•—›•‘åëYw¿¿¿¿3‹¿•—›•‘åë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åëåëVQ¿QwW¿-%5i£åëe=1-9‹5%¿Q}[e¿;!!£åëy!#-#5¿!¿;!!-#1¿¿££££££åëëÊë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åë¿¿¿¿y!5¿{!!¥U¿Y995åëåëy!5¿3‹¿•Ÿ—ŸŸŸŸŸŸåë}[e¿¿¿¿3‹¿‘“ŸŸŸŸŸåëww[¿¿¿¿3‹¿•›ŸŸŸŸŸŸåëy!¿¿¿3‹¿›“ŸŸŸŸŸŸåë-wY_¿¿¿3‹¿›“—ŸŸŸŸŸŸåë}Om¿¿¿¿3‹¿“ŸŸŸŸŸŸåë}o{¿¿¿¿3‹¿ŸŸŸŸŸŸŸåë}_{¿¿¿¿3‹¿•—ŸŸŸŸŸŸåëU}[W¿¿¿3‹¿—“•™åëYwŸ¿¿¿¿3‹¿•—›•‘åëYw¿¿¿¿3‹¿•—›•‘åë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åëåëVQ¿QwW¿-%5i£åëe=1-9‹5%¿Q}[e¿;!!£åëy!#-#5¿!¿;!!-#1¿¿££££££åëëÊë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åë¿¿¿¿y!5¿{!!¥U¿Y995åëåëy!5¿3‹¿•Ÿ—ŸŸŸŸŸŸåë}[e¿¿¿¿3‹¿‘“ŸŸŸŸŸåëww[¿¿¿¿3‹¿•›ŸŸŸŸŸŸåëy!¿¿¿3‹¿›“ŸŸŸŸŸŸåë-wY_¿¿¿3‹¿›“—ŸŸŸŸŸŸåë}Om¿¿¿¿3‹¿“ŸŸŸŸŸŸåë}o{¿¿¿¿3‹¿ŸŸŸŸŸŸŸåë}_{¿¿¿¿3‹¿•—ŸŸŸŸŸŸåëU}[W¿¿¿3‹¿—“•™åëYwŸ¿¿¿¿3‹¿•—›•‘åëYw¿¿¿¿3‹¿•—›•‘åë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åëåëVQ¿QwW¿-%5i£åëe=1-9‹5%¿Q}[e¿;!!£åëy!#-#5¿!¿;!!-#1¿¿££££££åëëÊë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åë¿¿¿¿y!5¿{!!¥U¿Y995åëåëy!5¿3‹¿•Ÿ—ŸŸŸŸŸŸåë}[e¿¿¿¿3‹¿‘“ŸŸŸŸŸåëww[¿¿¿¿3‹¿•›ŸŸŸŸŸŸåëy!¿¿¿3‹¿›“ŸŸŸŸŸŸåë-wY_¿¿¿3‹¿›“—ŸŸŸŸŸŸåë}Om¿¿¿¿3‹¿“ŸŸŸŸŸŸåë}o{¿¿¿¿3‹¿ŸŸŸŸŸŸŸåë}_{¿¿¿¿3‹¿•—ŸŸŸŸŸŸåëU}[W¿¿¿3‹¿—“•™åëYwŸ¿¿¿¿3‹¿•—›•‘åëYw¿¿¿¿3‹¿•—›•‘åë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åëåëVQ¿QwW¿-%5i£åëe=1-9‹5%¿Q}[e¿;!!£åëw559-#1»3åëY
                      ªúëërRVJ}ŸŸŸŸŸŸŸEIy}AŸE¿ÿIwkmE¿Y=¿QwW¿!3¿›Ÿ¿59!#7¿3!!!¿££££££ëÉIŸ%åëÉIŸ%IŸŸŸŸŸŸŸŸEIy}AŸE¿ÿIS5-!#¿£Ÿ£›—E§¿a9¿“¿›Ÿ•§¿—‹Ÿ“‹—“ëëÉIŸ%åëÉIŸ%IŸŸŸŸŸŸŸŸEIy}AŸE¿ÿIwkmE¿Yw¿U7=55-!#¿IŸ£›£—›ŸE¿{-'7¿e=Ÿ%åërRVJ}ŸŸŸŸ™›‘EIy}AŸE¿ÿIwkmE¿Y!¿QwW£ëÉIŸ%åëÉIŸ%IŸŸŸŸŸ™›‘EIy}AŸE¿ÿIwkmE¿s-=¿7=-#1££££££ëÉIŸ%åëÉIŸ%IŸŸŸŸŸ™›‘EIy}AŸE¿ÿy=¿7!5#±¿5-¿Ÿ½§¿9=#±¿7=5¿/5¿3-=£ÿÊë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åë¿¿¿¿y!5¿{!!¥U¿Y995åëåëy!5¿3‹¿•Ÿ—ŸŸŸŸŸŸåë}[e¿¿¿¿3‹¿‘“ŸŸŸŸŸåëww[¿¿¿¿3‹¿•›ŸŸŸŸŸŸåëy!¿¿¿3‹¿›“ŸŸŸŸŸŸåë-wY_¿¿¿3‹¿›“—ŸŸŸŸŸŸåë}Om¿¿¿¿3‹¿“ŸŸŸŸŸŸåë}o{¿¿¿¿3‹¿ŸŸŸŸŸŸŸåë}_{¿¿¿¿3‹¿•—ŸŸŸŸŸŸåëU}[W¿¿¿3‹¿—“•™åëYwŸ¿¿¿¿3‹¿•—›•‘åëYw¿¿¿¿3‹¿•—›•‘åë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åëåëVQ¿Qw5%¿Q}[e¿;!!£åëw559-#1»›™—¿Ÿ•¿Ÿ‘=5795åëY
                      ªúëërRVJ}ŸŸŸŸŸŸŸEIy}AŸE¿ÿIwkmE¿Y=¿QwW¿!3¿›Ÿ¿59!#7¿3!!!¿££££££ëÉIŸ%åëÉIŸ%IŸŸŸŸŸŸŸŸEIy}AŸE¿ÿIS5-!#¿£Ÿ£›—E§¿a9¿“¿›Ÿ•§¿—‹Ÿ“‹—“ëëÉIŸ%åëÉIŸ%IŸŸŸŸŸŸŸŸEIy}AŸE¿ÿIwkmE¿Yw¿U7=55-!#¿IŸ£›£—›ŸE¿{-'7¿e=Ÿ%åërRVJ}ŸŸŸŸ™›‘EIy}AŸE¿ÿIwkmE¿Y!¿QwW£ëÉIŸ%åëÉIŸ%IŸŸŸŸŸ™›‘EIy}AŸE¿ÿIwkmE¿s-=¿7=-#1££££££ëÉIŸ%åëÉIŸ%IŸŸŸŸŸ™›‘EIy}AŸE¿ÿy=¿7!5#±¿5-¿Ÿ½§¿9=#±¿7=5¿/5¿3-=£ÿ€ë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åë¿¿¿¿y!5¿{!!¥U¿Y995åëåëy!5¿3‹¿•Ÿ—ŸŸŸŸŸŸåë}[e¿¿¿¿3‹¿‘“ŸŸŸŸŸåëww[¿¿¿¿3‹¿•›ŸŸŸŸŸŸåëy!¿¿¿3‹¿›“ŸŸŸŸŸŸåë-wY_¿¿¿3‹¿›“—ŸŸŸŸŸŸåë}Om¿¿¿¿3‹¿“ŸŸŸŸŸŸåë}o{¿¿¿¿3‹¿ŸŸŸŸŸŸŸåë}_{¿¿¿¿3‹¿•—ŸŸŸŸŸŸåëU}[W¿¿¿3‹¿—“•™åëYwŸ¿¿¿¿3‹¿•—›•‘åëYw¿¿¿¿3‹¿•—›•‘åë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åëåëVQ¿Qw5%¿Q}[e¿;!!£åëw559-#1»›™—¿Ÿ“¿Ÿ‘=5797åëY
                      ªúëërRVJ}ŸŸŸŸŸŸŸEIy}AŸE¿ÿIwkmE¿Y=¿QwW¿!3¿›Ÿ¿59!#7¿3!!!¿££££££ëÉIŸ%åëÉIŸ%IŸŸŸŸŸŸŸŸEIy}AŸE¿ÿIS5-!#¿£Ÿ£›—E§¿a9¿“¿›Ÿ•§¿—‹Ÿ“‹—“ëëÉIŸ%åëÉIŸ%IŸŸŸŸŸŸŸŸEIy}AŸE¿ÿIwkmE¿Yw¿U7=55-!#¿IŸ£›£—›ŸE¿{-'7¿e=Ÿ%åërRVJ}ŸŸŸŸ™›‘EIy}AŸE¿ÿIwkmE¿Y!¿QwW£ëÉIŸ%åëÉIŸ%IŸŸŸŸŸ™›‘EIy}AŸE¿ÿIwkmE¿s-=¿7=-#1££££££ëÉIŸ%åëÉIŸ%IŸŸŸŸŸ™›‘EIy}AŸE¿ÿy=¿7!5#±¿5-¿Ÿ½§¿9=#±¿7=5¿/5¿3-=£ÿÊë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åë¿¿¿¿y!5¿{!!¥U¿Y995åëåëy!5¿3‹¿•Ÿ—ŸŸŸŸŸŸåë}[e¿¿¿¿3‹¿‘“ŸŸŸŸŸåëww[¿¿¿¿3‹¿•›ŸŸŸŸŸŸåëy!¿¿¿3‹¿›“ŸŸŸŸŸŸåë-wY_¿¿¿3‹¿›“—ŸŸŸŸŸŸåë}Om¿¿¿¿3‹¿“ŸŸŸŸŸŸåë}o{¿¿¿¿3‹¿ŸŸŸŸŸŸŸåë}_{¿¿¿¿3‹¿•—ŸŸŸŸŸŸåëU}[W¿¿¿3‹¿—“•™åëYwŸ¿¿¿¿3‹¿•—›•‘åëYw¿¿¿¿3‹¿•—›•‘åë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åëåëVQ¿Qw5%¿Q}[e¿;!!£åëw559-#1»›™—¿Ÿ‘¿Ÿ‘=5799åëY
                      ªúëërRVJ}ŸŸŸŸŸŸŸEIy}AŸE¿ÿIwkmE¿Y=¿QwW¿!3¿›Ÿ¿59!#7¿3!!!¿££££££ëÉIŸ%åëÉIŸ%IŸŸŸŸŸŸŸŸEIy}AŸE¿ÿIS5-!#¿£Ÿ£›—E§¿a9¿“¿›Ÿ•§¿—‹Ÿ“‹—“ëëÉIŸ%åëÉIŸ%IŸŸŸŸŸŸŸŸEIy}AŸE¿ÿIwkmE¿Yw¿U7=55-!#¿IŸ£›£—›ŸE¿{-'7¿e=Ÿ%åërRVJ}ŸŸŸŸ™›‘EIy}AŸE¿ÿIwkmE¿Y!¿QwW£ëÉIŸ%åëÉIŸ%IŸŸŸŸŸ™›‘EIy}AŸE¿ÿIwkmE¿s-=¿7=-#1££££££ëÉIŸ%åëÉIŸ%IŸŸŸŸŸ™›‘EIy}AŸE¿ÿy=¿7!5#±¿5-¿Ÿ½§¿9=#±¿7=5¿/5¿3-=ãÊë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åë¿¿¿¿y!5¿{!!¥U¿Y995åëåëy!5¿3‹¿•Ÿ—ŸŸŸŸŸŸåë}[e¿¿¿¿3‹¿‘“ŸŸŸŸŸåëww[¿¿¿¿3‹¿•›ŸŸŸŸŸŸåëy!¿¿¿3‹¿›“ŸŸŸŸŸŸåë-wY_¿¿¿3‹¿›“—ŸŸŸŸŸŸåë}Om¿¿¿¿3‹¿“ŸŸŸŸŸŸåë}o{¿¿¿¿3‹¿ŸŸŸŸŸŸŸåë}_{¿¿¿¿3‹¿•—ŸŸŸŸŸŸåëU}[W¿¿¿3‹¿—“•™åëYwŸ¿¿¿¿3‹¿•—›•‘åëYw¿¿¿¿3‹¿•—›•‘åë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åëåëVQ¿QwW5%¿Q}[e¿;!!£åëw559-#1»™—¿Ÿ¿Ÿ‘=579™åëY
                      ªúëërRVJ}ŸŸŸŸŸŸŸEIy}AŸE¿ÿIwkmE¿Y=¿QwW¿!3¿›Ÿ¿59!#7¿3!!!¿££££££ëÉIŸ%åëÉIŸ%IŸŸŸŸŸŸŸŸEIy}AŸE¿ÿIS5-!#¿£Ÿ£›—E§¿a9¿“¿›Ÿ•§¿—‹Ÿ“‹—“ëëÉIŸ%åëÉIŸ%IŸŸŸŸŸŸŸŸEIy}AŸE¿ÿIwkmE¿Yw¿U7=55-!#¿IŸ£›£—›ŸE¿{-'7¿e=Ÿ%åërRVJ}ŸŸŸŸ™›‘EIy}AŸE¿ÿIwkmE¿Y!¿QwW£ëÉIŸ%åëÉIŸ%IŸŸŸŸŸ™›‘EIy}AŸE¿ÿIwkmE¿s-=¿7=-#1££££££ëÉIŸ%åëÉIŸ%IŸŸŸŸŸ™›‘EIy}AŸE¿ÿy=¿7!5#±¿5-¿Ÿ½§¿9=#±¿7=5¿/5¿3-=£ÉÊë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åë¿¿¿¿y!5¿{!!¥U¿Y9 [... cut ...] ¥¥¥¥¥¥¥¥¥¥¥¥åëåëVQ¿QwW¿-%5i£åëe=1-9‹¿Ÿ‘“•›™5%¿Q}[e¿;!!£åë[!''-#1¿{=9)»
                           ªúëëÊë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åë¿¿¿¿y!5¿{!!¥U¿Y995åëåëy!5¿3‹¿•Ÿ—ŸŸŸŸŸŸåë}[e¿¿¿¿3‹¿‘“ŸŸŸŸŸåëww[¿¿¿¿3‹¿•›ŸŸŸŸŸŸåëy!¿¿¿3‹¿›“ŸŸŸŸŸŸåë-wY_¿¿¿3‹¿›“—ŸŸŸŸŸŸåë}Om¿¿¿¿3‹¿“ŸŸŸŸŸŸåë}o{¿¿¿¿3‹¿ŸŸŸŸŸŸŸåë}_{¿¿¿¿3‹¿•—ŸŸŸŸŸŸåëU}[W¿¿¿3‹¿—“•™åëYwŸ¿¿¿¿3‹¿•—›•‘åëYw¿¿¿¿3‹¿•—›•‘åë¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥¥åëåëVQ¿QwW¿-%5i£åëe=1-9‹¿Ÿ‘“•›™5%¿Q}[e¿;!!£åë[!''-#1¿{=9)»

notsolowki, 2017-03-28:
is there a specific console software i have to use?

mefistotelis, 2017-03-28:
While it's hard to tell for sure, this doesn't really look like incorrect parameters. Maybe your console utility doesn't interpret chars correctly? Or your UART doesn't like this input?

notsolowki, 2017-03-28:
this first console was secureCRT this next console is with putty same settings.
▒륥▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒뿿▒▒y!▒{!!▒U▒Y995▒▒▒▒y!▒3▒▒▒▒▒▒▒▒▒▒▒▒▒}[e▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒▒ww[▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒▒y!▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒▒-wY_▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒▒}Om▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒▒}o{▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒▒}_{▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒U}[W▒▒▒3▒▒▒▒▒▒▒▒▒▒▒Yw▒▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒Yw▒▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒륥▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒5%▒yagw▒;!!▒▒▒y!#-#5▒!▒;!!-#1▒▒▒▒▒▒▒▒▒▒▒▒륥▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒뿿▒▒y!▒{!!▒U▒Y995▒▒▒▒y!▒3▒▒▒▒▒▒▒▒▒▒▒▒▒}[e▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒ww[▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒▒y!▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒▒-wY_▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒▒}Om▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒▒}o{▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒▒}_{▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒U}[W▒▒▒3▒▒▒▒▒▒▒▒▒▒▒Yw▒▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒▒Yw▒▒▒▒▒3▒▒▒▒▒▒▒▒▒▒▒륥▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒VQQwW▒-%5ai▒▒▒e=1-9▒▒▒▒▒▒5%▒Q}[e▒;!!▒▒▒y!#-#5▒!▒;!!-#1▒▒▒▒▒▒▒▒▒▒▒

do you think this needs to been done on a linux machine. i now know coptersafe was talking about davinchi linux. maybe i can vmware ubuntu because my laptop is not with me atm so all i have right now is windows?

notsolowki, 2017-03-28:
what character set should i use. its set to UTF-8??

notsolowki, 2017-03-28:
i have tried ever character set putty has by default. im using a prolific rs232 adapter

mefistotelis, 2017-03-28:
Character set - the standard english/US charset should do. But using UTF-8 will not damage anything.

I don't have any more ideas about what the issue might be. You could definitely try another RS232 interface. You could also check broader range of the transmission params, including changing baud rate.

notsolowki, 2017-03-28:
i tried 8n1 8n1 7n2 etc. also with odd even mark and space.  is this connection going to be critical to flashing the bootloader?

notsolowki, 2017-03-28:
this is what happens if i change the data bits. 
k%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%ek????Z5?/
                                                        :%UtK995ekekZ5?3
                                                                        ?ek+V{{{5
 ?kWW+{{{5
          ?kZ5???3
                  ?k-uu{{5
                          ?k=5}{{{5
                                   ?ek=m}{{{5
                                             ?ek5}{{{5
                                                      ?ekU+U{{5
                                                               ?ek]????3
                                                                        ?k]????3
                                                                               ?k%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%ekek=]U;T)*Zt
+ku                                                        %
   -9
     ?=9=75?=9=75?=9=75ekU(5%?]}[

                                 :4+k:T*5?!?;!!-#1??######ekk

notsolowki, 2017-03-28:
Is there a way to confirm the settings by examining the code?

notsolowki, 2017-03-28:
this is what it loks like at 230400
0øžþfž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†žþþžžžžøf˜~`æ˜€þžþf˜€˜fžfæ`žxf†˜ø˜æ˜x†`†žþfžþ†f˜~`æ˜€þžxž`æ˜~žàþžfþx†þxxxxxx†žþææžfxþžžžžxž`æ˜~žàþž~†xþxxxxxx†žþ~æ~žfžžžžxž`æ˜~žàþžfžxàþxxxxxx†žþ†f˜f˜žžžxž`æ˜~žàþžæxþxxxxxx†žþæ†~fffžžžxž`æ˜~žàþžx†þxxxxxx†žþææ˜æžþžžžžxž`æ˜~žàþž~xàþxxxxxx†žþææžžæžžžžxž`æ˜~žàþž~þxàþxxxxxx†žþææfþæžžžžxž`æ˜~žàþžf~xxxxxxx†žþffþæžæfþžžžxž`æ˜~žàþž~þà~xæxf†x†žþ†f~þxžžžžxž`æ˜~žàþž~fx†žxàfx`æx†žþ†f~æxžžžžxž`æ˜~žàþž~fx†žxàfx`æx†žþfž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†ž†žþfžþxææ˜~`ž`æ~~fžæ†f˜æ˜þž€fžž†žþff˜à˜ž˜ø˜þžx€fxþ˜ø˜þ˜þàæxæ˜žx€fxþ˜ø˜þ˜þàæxæ˜žx€fxþ˜ø˜þ˜þàæxæ˜†žþ†f†`f˜†˜žøfæx~æž~˜€˜ž†žþ†f˜`~`ž˜`f`æ˜ž˜ž~˜€˜æ†˜à˜žfþ`ž`ž`ž`ž†žþžþ

notsolowki, 2017-03-28:
can you recommend a specific rs232 adapter?

notsolowki, 2017-03-28:
the output of tx is 3.3 volts

notsolowki, 2017-03-28:
do i need a ttl converter?

notsolowki, 2017-03-28:
i got it. it need ttl converter here is what im getting

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 132000000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 15428571
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x5acad81e 0x5bcad81e 0x5acad81e
System COLD boot.
Continue to booting up ......

notsolowki, 2017-03-28:
more,

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 132000000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 15428571
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x5acad81e 0x5bcad81e 0x5acad81e
System COLD boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 15428571
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x1 0x789aedca
System WARM boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 15428571
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x2 0x789aedc9
System WARM boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
[... 245 lines cut ...]
Start WDT timer OK.
Magic: 0x87651234 0xb 0x789aedc0
System WARM boot.
Rolling Back done ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 15428571
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0xc 0x789aedc7
System WARM boot.
Rolling Back done ......

notsolowki, 2017-03-28:
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 15428571
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x5acad81e 0x5bcad81e 0x5acad81e
System COLD boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 15428571
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x1 0x789aedca
System WARM boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 15428571
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x2 0x789aedc9
System WARM boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
[... 448 lines cut ...]
Start WDT timer OK.
Magic: 0x87651234 0x15 0x789aedde
System WARM boot.
Rolling Back done ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 15428571
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x16 0x789aeddd
System WARM boot.
Rolling Back done ......

notsolowki, 2017-03-28:
this is without the gimbal ribbon cable connected

notsolowki, 2017-03-28:
not this is all im getting,
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 132000000
DDR    freq: 528000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 15428571
SD1    freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x5a4ad81e 0x73cad81e 0x5acad81e
System COLD boot.
Continue to booting up ......

what is my next step?

notsolowki, 2017-03-28:
For the future this cpl chip did the trick. 

[image]

notsolowki, 2017-03-28:
should i pull boot0 low and try to flash with the single module now??

notsolowki, 2017-03-29:
why does it say rolling back when my gimbal arm is not connected? 

and @coptersafe i cant send any break commands. am i at the wrong console port?
"you can go to console of ambarella and type command manuall for reflash module 0400"

coptersafe, 2017-03-29:
i dont have more info .. because never brick gimbal and have only logs from worked modules .. sorry)

notsolowki, 2017-03-29:
Is there any way you can think of to recover the bootloader so i can
relfash it.?

coptersafe, 2017-03-29:
about console - connect CP2102 to DM368_RX DM_368_TX and go to inside console DaVinchi... next try telnet to Amba  , ip adress  dont remember (can look later)

notsolowki, 2017-03-29:
Ill connect it and see what it spits out. Do you think theres any chance of
aquiring a bootloader?

coptersafe, 2017-03-29:
show log what you see ? DM_368

notsolowki, 2017-03-29:
okay ehre it is,
DM36x initialization passed!
UBL Product Vesion : DJI-297M-UBL-1.0-rc0(2015-10-30)
Dji UBL Version: 1.51(Nov  2 2015 - 15:45:39)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
   DONE
Jumping to entry point at 0x81080000.

U-Boot Product Vesion : DJI-ENC-Uboot-1.0-rc0(2015-11-02)
U-Boot 2010.12-rc2-svn3205-Dji (Nov 02 2015 - 19:07:26)
Cores: ARM 297 MHz
DDR:   270 MHz
I2C:   ready
DRAM:  128 MiB
NAND:  128 MiB
MMC:   davinci: 0
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds

Loading from nand0, offset 0x4a0000
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2015-11-02   7:28:36 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4050848 Bytes = 3.9 MiB
   Load Address: 80008000
   Entry Point:  80008000
## Booting kernel from Legacy Image at 80700000 ...
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2015-11-02   7:28:36 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4050848 Bytes = 3.9 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Kernel Product Vesion : DJI-WIFI-WM610-Kernel-1.0-rc4(Nov  2 2015-15:24:51)
[    0.000000] Linux version 2.6.32.17-davinci1 (root@xueb-VirtualBox) (gcc version 4.3.3 (Sourcery G++ Lite 2009q1-203) ) #1 PREEMPT Mon Nov 2 15:28:34 CST 2015
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: DaVinci DM36x EVM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] DaVinci dm36x_rev1.2 variant 0x8
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 12192
[    0.000000] Kernel command line: console=ttyS0,115200n8 rw dm365_imp.oper_mode=0 video=davincifb:vid0=0,0:vid1=0,0:osd0=0,0:osd1=0,0 mem=48MB davinci_enc_mngr.ch0_output=COMPOSITE davinci_enc_mngr.ch0_mode=pal ubi.mtd=2,2048 root=ubi0:rootfs rootfstype=ubifs ip=off lpj=1077248
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 48MB = 48MB total
[    0.000000] Memory: 44432KB available (3652K code, 453K data, 120K init, 0K highmem)
[    0.000000] SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] Console: colour dummy device 80x30
[    0.000000] Calibrating delay loop (skipped) preset value.. 215.44 BogoMIPS (lpj=1077248)
[    0.000000] Mount-cache hash table entries: 512
[    0.000000] CPU: Testing write buffer coherency: ok
[    0.000000] DaVinci: 8 gpio irqs
[    0.000000] NET: Registered protocol family 16
[    0.130000] bio: create slab <bio-0> at 0
[... 232 lines cut ...]
hdmi status is 1
ARM Load: 26% Video fps: 30 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:09 Resolution: 1280x720 

GetFPGAstatus err!
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 17% Video fps: 30 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:10 Resolution: 1280x720 

GetFPGAstatus err!
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 21% Video fps: 30 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:11 Resolution: 1280x720 

GetFPGAstatus err!
GetFPGAstatus err!
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 14% Video fps: 31 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:12 Resolution: 1280x720 

GetFPGAstatus err!
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 14% Video fps: 30 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:13 Resolution: 1280x720 

GetFPGAstatus err!

notsolowki, 2017-03-29:
DM36x initialization passed!
UBL Product Vesion : DJI-297M-UBL-1.0-rc0(2015-10-30)
Dji UBL Version: 1.51(Nov  2 2015 - 15:45:39)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
   DONE
Jumping to entry point at 0x81080000.

U-Boot Product Vesion : DJI-ENC-Uboot-1.0-rc0(2015-11-02)
U-Boot 2010.12-rc2-svn3205-Dji (Nov 02 2015 - 19:07:26)
Cores: ARM 297 MHz
DDR:   270 MHz
I2C:   ready
DRAM:  128 MiB
NAND:  128 MiB
MMC:   davinci: 0
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds

Loading from nand0, offset 0x4a0000
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2015-11-02   7:28:36 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4050848 Bytes = 3.9 MiB
   Load Address: 80008000
   Entry Point:  80008000
## Booting kernel from Legacy Image at 80700000 ...
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2015-11-02   7:28:36 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4050848 Bytes = 3.9 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Kernel Product Vesion : DJI-WIFI-WM610-Kernel-1.0-rc4(Nov  2 2015-15:24:51)
[    0.000000] Linux version 2.6.32.17-davinci1 (root@xueb-VirtualBox) (gcc version 4.3.3 (Sourcery G++ Lite 2009q1-203) ) #1 PREEMPT Mon Nov 2 15:28:34 CST 2015
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: DaVinci DM36x EVM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] DaVinci dm36x_rev1.2 variant 0x8
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 12192
[    0.000000] Kernel command line: console=ttyS0,115200n8 rw dm365_imp.oper_mode=0 video=davincifb:vid0=0,0:vid1=0,0:osd0=0,0:osd1=0,0 mem=48MB davinci_enc_mngr.ch0_output=COMPOSITE davinci_enc_mngr.ch0_mode=pal ubi.mtd=2,2048 root=ubi0:rootfs rootfstype=ubifs ip=off lpj=1077248
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 48MB = 48MB total
[    0.000000] Memory: 44432KB available (3652K code, 453K data, 120K init, 0K highmem)
[    0.000000] SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] Console: colour dummy device 80x30
[    0.000000] Calibrating delay loop (skipped) preset value.. 215.44 BogoMIPS (lpj=1077248)
[    0.000000] Mount-cache hash table entries: 512
[    0.000000] CPU: Testing write buffer coherency: ok
[    0.000000] DaVinci: 8 gpio irqs
[    0.000000] NET: Registered protocol family 16
[    0.130000] bio: create slab <bio-0> at 0
[    0.140000] DM365 IPIPE initialized in Continuous mode
[... 353 lines cut ...]
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 12% Video fps: 30 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:33 Resolution: 1280x720 

GetFPGAstatus err!
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 13% Video fps: 30 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:34 Resolution: 1280x720 

GetFPGAstatus err!
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 11% Video fps: 30 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:35 Resolution: 1280x720 

GetFPGAstatus err!
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 16% Video fps: 30 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:36 Resolution: 1280x720 

GetFPGAstatus err!
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 13% Video fps: 30 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:37 Resolution: 1280x720 

[   47.à

notsolowki, 2017-03-29:
i dont know why its editing out part of the log on the forum??

coptersafe, 2017-03-29:
put crtl-c  and you will go to console  for enter commands

coptersafe, 2017-03-29:
GetFPGAstatus err! - this mean that you Gimbal PROCCESSOR not connected to LB unit

notsolowki, 2017-03-29:
about not being connected, everything is connected i just messed it up with a bad flash of the single module?!?

notsolowki, 2017-03-29:
i think its the smae. everything is plugged in correctly,
DM36x initialization passed!
UBL Product Vesion : DJI-297M-UBL-1.0-rc0(2015-10-30)
Dji UBL Version: 1.51(Nov  2 2015 - 15:45:39)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
   DONE
Jumping to entry point at 0x81080000.

U-Boot Product Vesion : DJI-ENC-Uboot-1.0-rc0(2015-11-02)
U-Boot 2010.12-rc2-svn3205-Dji (Nov 02 2015 - 19:07:26)
Cores: ARM 297 MHz
DDR:   270 MHz
I2C:   ready
DRAM:  128 MiB
NAND:  128 MiB
MMC:   davinci: 0
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds

Loading from nand0, offset 0x4a0000
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2015-11-02   7:28:36 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4050848 Bytes = 3.9 MiB
   Load Address: 80008000
   Entry Point:  80008000
## Booting kernel from Legacy Image at 80700000 ...
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2015-11-02   7:28:36 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4050848 Bytes = 3.9 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Kernel Product Vesion : DJI-WIFI-WM610-Kernel-1.0-rc4(Nov  2 2015-15:24:51)
[    0.000000] Linux version 2.6.32.17-davinci1 (root@xueb-VirtualBox) (gcc version 4.3.3 (Sourcery G++ Lite 2009q1-203) ) #1 PREEMPT Mon Nov 2 15:28:34 CST 2015
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: DaVinci DM36x EVM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] DaVinci dm36x_rev1.2 variant 0x8
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 12192
[    0.000000] Kernel command line: console=ttyS0,115200n8 rw dm365_imp.oper_mode=0 video=davincifb:vid0=0,0:vid1=0,0:osd0=0,0:osd1=0,0 mem=48MB davinci_enc_mngr.ch0_output=COMPOSITE davinci_enc_mngr.ch0_mode=pal ubi.mtd=2,2048 root=ubi0:rootfs rootfstype=ubifs ip=off lpj=1077248
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 48MB = 48MB total
[    0.000000] Memory: 44432KB available (3652K code, 453K data, 120K init, 0K highmem)
[    0.000000] SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] Console: colour dummy device 80x30
[    0.000000] Calibrating delay loop (skipped) preset value.. 215.44 BogoMIPS (lpj=1077248)
[    0.000000] Mount-cache hash table entries: 512
[    0.000000] CPU: Testing write buffer coherency: ok
[    0.000000] DaVinci: 8 gpio irqs
[    0.000000] NET: Registered protocol family 16
[    0.130000] bio: create slab <bio-0> at 0
[... 318 lines cut ...]
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 27% Video fps: 30 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:26 Resolution: 1280x720 

GetFPGAstatus err!
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 24% Video fps: 30 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:27 Resolution: 1280x720 

GetFPGAstatus err!
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 21% Video fps: 30 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:28 Resolution: 1280x720 

GetFPGAstatus err!
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 16% Video fps: 30 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:29 Resolution: 1280x720 

GetFPGAstatus err!
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 13% Video fps: 31 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:30 Resolution: 1280x720 

[   40.76

[40 more replies]

closed 2021-06-14 by mefistotelis