o-gs/dji-firmware-tools #28: forcing fw downgrade with P3C_FW_DEBUG (STA)

KennethMcNutt, 2017-04-17:
Just some background.... When some Drones were updated from 1.6.8 to 1.7.9 they got "sensor errors"... (IMU & compass) 
 DJI released 1.7.94 to fix them... 
So....I have just found a Solution??? For the "sensor error" When it is invoked by the downgrade implementation.
It was simple?? 
But it has to be done in this order... 
Assuming you're still at 1.8.1.. 
If you are 1.8.1 + debug it first... 
Then... Downgrade to 1.7.9.. Get the error... Restart... Still Sensor error... Rerun F/w 1.7.9 with debug... Still Sensor error...Cal the Imu (yep)... still get the error... :( 
Update with 1.7.94 No more error?? :) 
Recal Imu ;) 
This is the setups I just took to fix it for a member of my group... So I am not Sure if the first IMU cal was anything more... then a waste of time...  So,  
Revised P3s Debug downgrade How to...
 If you have v1.08.10 with a + 
Then start here v1.08.10 -> v1.08.10 + Debug 
v1.08.10 -> v1.07.9 
v1.07.9 -> v1.07.9 + Debug 
(v1.07.9 -> v1.7.94 If you get "Sensor Error", if not skip this step...( I am waiting for clear info on the next step from here for further downgrade.. ) v1.05.7 -> v1.05.7 + Debug 
v1.05.7 -> V1.5.75 If you get "Sensor Error", if not skip this step...( I am waiting for clear info on the next step from here for further downgrade.) and So on.. Here is a list..
 v1.0.14 
v1.0.17 
v1.0.2 
v1.01.3 
v1.02.4 
v1.03.5 
v1.04.6 
v1.05.7 
v1.5.75 
v1.06.8 
v1.07.8 
v1.07.9 
v1.7.94 
v1.08.10 
There are newer and older Firmware Versions that are not listed... Anyone Know them? Links are hard to find.. 
That all said,
If you get the Sensor error on 1.7.9...
then 1.7.94 will fix it...
But further downgrades it returns? 
and 1.5.75 does not fix it?? 
Should I advise 1.7.94 + debug?? 1.5.75???

KennethMcNutt, 2017-04-17:
http://www.phantompilots.com/threads/p3s-p3-4k-downgradable-all-the-way-back-confirmed-working-update-sensor-error-fixed.108575/page-5#post-1160499

KennethMcNutt, 2017-04-17:
I most cases people skip 1.6.8?? It does not seem to matter either way..

KennethMcNutt, 2017-04-19:
Should I advise 1.7.94 + debug?? 1.5.75 + debug???
How can We get past this 1.7.94 (1.7.9+) issue?
And downgrade further without error??

aka1ceman, 2017-10-12:
Is it true that we can not roll back 1.11 using tbis method without errors? Has it been done without errors? Can anyone confirm?

00Gery00, 2018-04-30:
Hey Ken,

i have written on the Thread at phantompilots, are you still active? ;)

KennethMcNutt, 2018-04-30:
I haven't checked that site in a while.. I'll take a look this afternoon