o-gs/dji-firmware-tools #29: Extrernal PWM and switches

notsolowki, 2017-05-11:
i was just thinking. if someone found a way to make a better antenna for a esp8266. they are so tiny you could put on in a drone a potentially have ALOT of switches all by using it as a access point for your phone controlling it by wifi via script/console

mefistotelis, 2017-05-30:
We can have a lot of switches via the original Lightbridge link. All we need is a mobile application which sends messages to the drone.

Making a separate RF link would introduce a lot of noise, and decrease the effective range.

coptersafe, 2017-05-30:
look at this. http://kopterkraft.com/index.php

notsolowki, 2017-05-30:
How can i use the lightbridge to pull a pin high or low. Where are the pins.

coptersafe, 2017-05-30:
Need raspberry zero for pwm outputs