o-gs/dji-firmware-tools #30: image sensor recognition

kazan417, 2017-05-12:
It seams phantom 3 standard uses sony imx377eqh5 sensor.
I am not sure but specification looks similar 
http://www.sony-semicon.co.jp/products_en/IS/sensor2/img/products/IMX377EQH5_ProductSummary_v1.5_20150414.pdf

kazan417, 2017-05-12:
maybe IMX378

kazan417, 2017-05-12:
also imx117 is another variant

kazan417, 2017-05-12:
IMX078CQK more one candidat

mefistotelis, 2017-05-30:
There's a list of the sensor models on wikipedia:
https://en.wikipedia.org/wiki/Exmor

Would be nice to know what kind of output the sensor has.

mefistotelis, 2017-05-30:
IMX378  -released in September 2016, so can't be it.

IMX377 - rel. in April 2015, so DJI would have to cooperate with Sony before release, unlikely. Also 12.2 MP effective, so lower than what DJI claims

IMX117  - only 12 MP, lower than what DJI claims

IMX078CQK - released in February 2011 and matches all DJIs parameters. I'd say it's most likely candidate.

notsolowki, 2017-07-31:
@mefistotelis it would be nice to hookup another camera and just use the gimbals motherboard to accept another cameras input. also to find a light enough camera with a lot of optical zoom for under a thousand dollars. then one wouldn't need a $30,000 drone setup to have some optical zoom.. would be cool to have a gopro with x100 zoom and light enough to attach to the phantom 3

mefistotelis, 2017-08-01:
I don't know enough about image sensor technologies to describe the protocol correctly.

Regarding switching camera, it might be possible to connect other cameras to OFDM - the image is transmitted over USB. Not sure whether it is any standard USB video transmission.

Definitely not easy to connect another camera to gimbal top board.