o-gs/dji-firmware-tools #32: Height limit discussion

KyokushinPL, 2017-06-07:
I tried to set g_config.flying_limit.max_height_0 and g_config.advanced_function.height_limit_enabled_0 "modify" variables to the "false" from a "true" but this is not possible.
Settings  g_config.flying_limit.max_height_0 min/max/def to 5000 is not removing a limit. I am thinking (maybe i am wrong) the app is overriding these settings so i would like to prevent overriding that block by app by changing a 'modify' variable.

Any ideas?

mefistotelis, 2017-06-07:
What I read from @notsolowki, you need to reset settings somewhere after the firmware update.

Also, check the settings changes made by @Coptersafe, ie.:
<pre>diff --git a/flyc_param_infos b/flyc_param_infos
index bf5d20d..c9578e5 100644
--- a/flyc_param_infos	Sat Jun  3 16:13:21 2017
+++ b/flyc_param_infos	Sat Jun  3 16:08:43 2017
@@ -2262,3 +2262,3 @@
 		"maxValue" : 2,
-		"defaultValue" : 1,
+		"defaultValue" : 2,
 		"name" : "g_config.advanced_function.height_limit_enabled_0",
@@ -2601,5 +2601,5 @@
 		"attribute" : 11,
-		"minValue" : 20,
-		"maxValue" : 500,
-		"defaultValue" : 120,
+		"minValue" : 5000,
+		"maxValue" : 5000,
+		"defaultValue" : 5000,
 		"name" : "g_config.flying_limit.max_height_0",
@@ -4395,3 +4395,3 @@
 		"maxValue" : 5,
-		"defaultValue" : 0,
+		"defaultValue" : 5,
 		"name" : "g_config.api_entry_cfg.height_data_type_0",
</pre>

KyokushinPL, 2017-06-07:
what is "g_config.api_entry_cfg.height_data_type_0" doing?

polarisax, 2017-06-07:
With correct modified-DJIGO app and properly mod-FW1.7.6 all work perfectly!
Pay attention on the autolanding system above 1300-1400m.

KyokushinPL, 2017-06-07:
What is happening with autolanding system above 1km?

MAVProxyUser, 2017-06-09:
Have you tried tampering with these variables via the Assistant for bird that allow it? 

[image]

notsolowki, 2017-06-09:
The settings you need to change are the smart battery level 1 and 2 settings

KyokushinPL, 2017-06-09:
Why smart battery have something to it? Autolanding should be released by low battery level, so if a copter will be at high altitude, but without critical battery level then autoland should not be released, am i wrong?

notsolowki, 2017-06-09:
Its the part of the spftware that calculates how long its going to take to
get from point a to b. It for some reason still thinks the ascend and
descent speeds are factory. When you get to 4kft it assumes you wont have
enough battery to get down at their slow factory speeds.

notsolowki, 2017-06-09:
changing these settings was a temporary fix. im not sure if there is another way 

		"index" : 256,
		"typeID" : 0,
		"size" : 1,
		"attribute" : 11,
		"minValue" : 0,
		"maxValue" : 1,
		"defaultValue" : 0,
		"name" : "g_config.voltage.level_1_protect_type_0",
		"modify" : true
	},
	{
		"index" : 257,
		"typeID" : 0,
		"size" : 1,
		"attribute" : 11,
		"minValue" : 0,
		"maxValue" : 2,
		"defaultValue" : 2,
		"name" : "g_config.voltage.level_2_protect_type_0",
		"modify" : true
	{
		"index" : 260,
		"typeID" : 0,
		"size" : 1,
		"attribute" : 11,
		"minValue" : 0,
		"maxValue" : 10,
		"defaultValue" : 0,
		"name" : "g_config.voltage2.level_1_function_0",
		"modify" : true
	{
		"index" : 262,
		"typeID" : 0,
		"size" : 1,
		"attribute" : 11,
		"minValue" : 0,
		"maxValue" : 10,
		"defaultValue" : 1,
		"name" : "g_config.voltage2.level_2_function_0",
		"modify" : true

compare then with your factory settings to see what ones changed

notsolowki, 2017-06-09:
i think someone said on the older app it would let them cancel the autoland

KyokushinPL, 2017-06-09:
I have factory speeds to not overheat motors. I would like only to remove height limit at now. So battery/autoland calculation should not have matter in that case. Or i am wrong?

polarisax, 2017-06-09:
It's possible to cancel the autoland.....but not the "critical autoland" which appear over about 4000ft.
...and if you lose radio-signal your AC will landing in that point! No RTH!

polarisax, 2017-06-09:
I do some parameters changes ...but no luck. 
From logs it seems that battery residual power is compare to the height & distance from home.
In far distance case the AC switch in RTH while in very height fly result in autolanding.

KyokushinPL, 2017-06-09:
OK, but i have factory speed of ascending, descending and others, so calculation for battery should be correct, right?

polarisax, 2017-06-09:
yes...on paper....but I have no success changing a lot of parameters...like ascend/descend speed,voltage level....etc...

notsolowki, 2017-06-09:
You wont overheat those motors.. i think with the older app you can cancell
the critical land

polarisax, 2017-06-09:
@notsolowki not in 2.4.2.
Only "normal autolanding" is cancelable not the "critical" until the bar on the upper side of screen return to yellow.

ferraript, 2017-06-09:
yeah, that's right
I also tried to modify multiple parameters, but no success, I wasn't able to override that "Critically low power; the aircraft will land now"

if anybody of you is using my [TXTlogToCSVtool](http://www.phantompilots.com/threads/tool-win-offline-txt-flightrecord-to-csv-converter.70428/), then in csv you can find columns SMART_BATTERY.usefulTime and SMART_BATTERY.landTime - aircraft is somehow calculating these values
and in the moment, when usefulTime < landTime, autolanding is iniciated

my maximum height is 1801.3 m, but above 1400 m it was really fighting with that autoland security feature

KyokushinPL, 2017-06-09:
@ferraript could you share your setup of parameters about heigh? i cant figure out how to set it up to working.

notsolowki, 2017-06-09:
I have no problem going above 4000 except for signal, i disable the
autoland and problem solved

polarisax, 2017-06-09:
@notsolowki If u can try to go higher...above 4600-4700ft

polarisax, 2017-06-09:
@notsolowki Have u disable g_config.flying_limit.auto_landing_enabled_0?
....and in case of radio-LOS can the AC initiate the RTH?

notsolowki, 2017-06-09:
I know what you mean. I dont get the critical landing anymore. It has
somthing to do with level2. I used to get that problem. But not anymore.
Play with those settings and you will figure it out

notsolowki, 2017-06-09:
It still rth, it didnt effect that

notsolowki, 2017-06-09:
Change level 2 to level 1

polarisax, 2017-06-09:
@notsolowki thanks a lot! I will try it ASAP

notsolowki, 2017-06-09:
I mean it would still try to autoland but it would let me cancel it

notsolowki, 2017-06-09:
I think "level 2" is critical so it wontblet you cancel. But, level 1 you
can cancel

polarisax, 2017-06-09:
but parameter voltage..._protect_type_0 or voltage2..._function_0?

notsolowki, 2017-06-09:
How do those parameters i poseted earlier compare to yours? I cant remember
100% what the parameters name was, it was level 2 somthing.. just be sure
you test your rth if you want to be safe. Make it land from 5 ft to be sure
it works. Also make sure to double check all your app settings before take
off. And idk if you using litchi or not but the last time i used litchi i
flipped from p-gps to gps and my drone freaked out flipped and landed in a
tree almost 75 dlft to the top soo.. i can just gove ypu a copy of my fw if
you want to try it and you can extract my flyc-params then upload it to the
factory firmware container and it should tell you what parameters were
modified. I happen to live right under where the big planes turn at very
low altitude to land. So i cant really test it for you.

polarisax, 2017-06-09:
Thanks @notsolowki
I know what u mean...same situation here. To try this mod I have to move by 30 miles into Alps!
I'll check these parameters soon!

notsolowki, 2017-06-09:
Make sure you chnage it in the app too. If i did it i would
Change it in the fw on the drone
Change it in 2.4.2 app flyc
Upload setting to drone.
Change latest app flyc if you wish
Use latest app if you wish
Enjoy..

aka1ceman, 2017-06-11:
I am willing to test  here in the US if you guys create a bin to patch.

ferraript, 2017-06-26:
yesterday I finally had time to test my latest hack of flight parameters in order to disable that autoland feature
I changed all parameters that could be related to battery life, for example I lowered minimum voltage, autoland percentage and so on...
result: no success, above 1300 m autoland was kicking in again and again
I also lowered landing speed to 1 m/s, so I wasn't losing so much height - thanks to that I broke my previous height record, I got to 1897.8 m
as I ran out of options, I'm done with testing
it's really very risky - planes, helicopters, battery, connection - something could get wrong and consequencies could be fatal

neven385, 2017-06-26:
Maybe try different approach? Can you trick FC to take barometer reading differently? Like 10m = 1m?

ferraript, 2017-06-26:
as much as I like to get to 2000 m (battery has enough power to get there), I rather stop now than when it's too late
and if somebody gets there, I will congratulate him

neven385, 2017-06-26:
Your contribution here and on phantompilots is priceless. No need to risk anything to prove it.

aka1ceman, 2017-06-26:
Ferript, is it possible to package increased height ,(even if it's only 1000m) with the allowed flight package you were able to figure out?

MrBurnsAT, 2017-06-26:
Ferraript i know someone who has done 2400m with P3P without this Crack! Only with Downgrade to 1.3.2 and using Litchi 2.2.1

ferraript, 2017-06-26:
@neven385, if you think so... :) thanks

@aka1ceman, you mean "no NFZ mod" + "no height limit mod"?

@MrBurnsAT, but I believe he had some kind of battery mod, right?
because with original battery and original firmware (where maximum ascending and descending speeds are only 5 m/s and 3 m/s) there is not enough power to get safely to 2400 m and back

aka1ceman, 2017-06-26:
Ferraript, yes Sir.

ferraript, 2017-06-26:
that was requested few weeks ago
nobody has released it?

MrBurnsAT, 2017-06-26:
Ferraript Yes with Battery Mod 

But he had Problems above 2400 
I Think same Problem like Urs 

I did 1500 with my P3P only with Downgradet Firmware and old Litchi without any Problems. Landet at shout 45-50%

viejoperro, 2017-06-26:
I agree with @neven385, THANK YOU Ferraript for all the support and testing, personal too, for me the NFZ was the most important, 500m is a plus even if it dosent go higher than 1500. The V2 of Novotoski is also fun, nice speeds. Thanks for sharing your findings.

MrBurnsAT, 2017-06-26:
Ferraript could u send me a Copy of ur Version pls

Notsolowki Urs too pls 

mr.burns@gmx.org

mefistotelis, 2017-06-26:
>  above 1300 m autoland was kicking in again and again

Interesting. @ferraript, would you share a flight log which demonstrates that? The one from "black box" sd-card ofc.

aka1ceman, 2017-06-26:
Ferraript,  nobody addressed it. I did receive your no NFZ file but it had height limits. Someone else attempted to release no NFZ and remove height but it did Not work. 

And once again, thank you for all you are doing.

ferraript, 2017-06-26:
guys, I'm just using what @mefistotelis invented, so thank him rather than me :)

@aka1ceman : ok then, I'll cook that bin when I get home

@mefistotelis : ok, I'll upload my logs if you think you can find something there

adamazad, 2017-06-26:
@ferraript, I would like a copy of the bin that has only NFZ mod, please. Thank you for your efforts.

ferraript, 2017-06-26:
@adamazad look [here](https://github.com/mefistotelis/phantom-firmware-tools/issues/13#issuecomment-296258465)

MAVProxyUser, 2017-06-26:
Stop on by if you all have anything to contribute to Mavic, Inspire2 (i2), Spark, or Phantom 4 discussion along the same lines of rooting, height bypass, distance limits, etc. 
https://forums.hak5.org/index.php?/topic/41304-mavicpilotscom-alternative-coptersafe-hack-mod-discussion

KyokushinPL, 2017-06-26:
I have a suggestion. @ferraript  could you prepare a bin with NFZ enabled, 1000m limit, but enabled all battery safety functions? That kind of version may be safe and add 500m extra.

thatdumbDronie, 2017-06-26:
@mavProxyUser

It's me off Hak5 
I'm on here with u now 
Inboxed u with my email
But spose we can talk here .

thatdumbDronie, 2017-06-26:
I have coptersafe installed . full pack. To my mav. 
I have the install programme and the updater programme. 

If there's a way I can sniff USB traffic or anything to help people out let me know. 
Need this unlocked and free for everyone

ferraript, 2017-06-26:
@mefistotelis: here you go, [flight logs](https://drive.google.com/open?id=0BzjAFKwV8KMYWW5MTEFmcWVQT0E) from my 1897.8 m flight

ferraript, 2017-06-26:
patched flight controller binaries for P3A/P3P with FW 1.7.60:

[NoFlyZones disabled](https://ulozto.net/!yBHNpQe6vYRV/pmcappfw3-bin)
[Max Height Limit at 2000 m](https://ulozto.net/!uNbthPrJvXFm/pmcappfw3-bin)
[NoFlyZones disabled + Max Height Limit at 2000 m](https://ulozto.net/!DyWuhSM2NiLh/pmcappfw3-bin)

thatdumbDronie, 2017-06-26:
how do u flash these to the craft?

RyanLee1119, 2017-06-26:
@thatdumbDronie I would like to try to unpack it, please send a copy to tomruth1235@gmail.com, Thanks!

thatdumbDronie, 2017-06-26:
@ryanlee1119 
email sent

ferraript, 2017-06-26:
>how do u flash these to the craft?

same way you flash the firmware
put the bin on the SD card and flash it into AC

[88 more replies]