o-gs/dji-firmware-tools #35: Dji assistant 2 developer mode

singlag, 2017-06-25:
does anyone know how to enter developer mode for dji assistant 2 ?
it seem we can set all the parameters / downgrade firmware without limitation and not necessary to flash a patched firmware

here is some screenshot
https://github.com/droner69/MavicPro/blob/master/DJI_Assistant_2_Dev_Pictures/009_Flight%20Restrictions.PNG

https://github.com/droner69/MavicPro/blob/master/DJI_Assistant_2_Dev_Pictures/005_Developer%20Tools.PNG

https://github.com/droner69/MavicPro/blob/master/DJI_Assistant_2_Dev_Pictures/006_FW%20Update%20List.PNG

notsolowki, 2017-06-28:
Run assistant 1.1.2" beta" with --debugger then hit ctrl shift f9 and f10

nezarbeidas, 2017-06-28:
@singlag sent you message on Hak5

WindyYam, 2017-07-16:
I just cant believe they put this developer mode into release version. I thought it's easy to modify the flyc param with the app way, then i found this, it's silly