o-gs/dji-firmware-tools #36: Mavic firmware/general

notsolowki, 2017-06-28:
How can we modify the parameters on the mavic drone. It seems like theres alot of info with missing parts all scattered everywhere?

nezarbeidas, 2017-06-28:
connect your drone to assistant
then open "simple WebSocket Client" on chrom and connect to 
ws://localhost:19870/general

fvantienen, 2017-06-28:
For newer firmwares this slack is getting active:
https://join.slack.com/dji-rev/shared_invite/MjA0NTE3MzM5NjM0LTE0OTg1OTc5MjUtNzE0NWM3ODI5OQ

notsolowki, 2017-06-28:
@mefisotelis
@mavic

Mavic pro firmware containrs, https://github.com/droner69/MavicPro?files=1

notsolowki, 2017-06-28:
#mefistotelis https://github.com/fvantienen/dji_rev/blob/master/tools/image.py

notsolowki, 2017-06-28:
@mefistotelis  you can update the firmware with assistant 1.1.2 beta in developer mode, ran
 with --debugger  in the shortcut. Then when in the program hit shift ctrl f10 do make int upgrade popup below the connected device name. From there it looks like you can manually flash, also if you type --help in the shortcut path it tells you more flags, and with never versions theres more options

notsolowki, 2017-06-28:
https://forums.hak5.org/index.php?/topic/39735-reversing-mavic-pro-firmware/

2008jiayu, 2017-06-28:
@@notsolowki where to download the dji assistant 1.1.2 beta

notsolowki, 2017-06-28:
https://github.com/MAVProxyUser/DJIAssistant2Binaries/blob/master/Win/DJI%20Assistant2%20Beta112.zip

notsolowki, 2017-06-28:
@glovepuppet
@mefistotelis
I wonder if it would be less time consuning to learn the serial data for the mavic rather than reverseing the firmware, do you think it would be a better approach to adress the parameters from the serial connection. I guess maybe someone could attemp to overide a max value this way and see what happens?

adamazad, 2017-06-28:
A crowdsourcing on Facebook made it possible to override figure out how to edit parameters for Mavic Pro using DJI Assistant 2. Eventually, they managed to remove the height limit. They also used help from #32. 
They bundled everything into one text file with full instructions. I attached it [here](https://github.com/mefistotelis/phantom-firmware-tools/files/1110458/Mavic.Full.Unlock.Instructions.and.Download.Links.txt)

Join the [group](https://web.facebook.com/groups/DjiJailbreak), people are constantly putting providing more user-friendly tutorials.

fvantienen, 2017-06-28:
@mefistotelis do you maybe know a way to talk to the ESC's on the P3 or where that is in the FC firmware? Since on the Mavic(which is similar in communication and stuff) I want to be able to control the motors in order to run custom firmware.

notsolowki, 2017-06-29:
If you can do this, 
set g_config.flying_limit.limit_height_abs 2500.000000
Then why does @mavproxyuser and all these other people say you cant set them outside their max range?

mefistotelis, 2017-06-29:
> do you maybe know a way to talk to the ESC's on the P3 or where that is in the FC firmware?

In P3, FC talks to ESC via serial interface, in the same manner all other components communicate.

To find it in P3 FC firmware, search for ESC related strings, ie:
<pre>printf_sub_804BE9C("ESC:BUG,esc send a package to esc\r\n");</pre>

MAVProxyUser, 2017-06-29:
@notsolowki I've only *seen* pictures of *certain* variables set beyond their implied max range, and seen the params that CopterSafe sends. Of course anything can be PhotoShopped. 

https://github.com/mefistotelis/phantom-firmware-tools/issues/32#issuecomment-311488395
[image]

Personally... I was doing the opposite and limiting my Spark to 20 on max height and 20 on max distance. I've honestly never played with, or tested the limit_height_abs variable. 

When I was playing around, I was only messing with fly_limit_height and I only tried to set it to 500 max, and my example (in the header) shows 111 which are of course valid values.

https://github.com/fvantienen/dji_rev/blob/master/tools/websocket_tool.py

Beyond that I've only verified that the airport NFZ settings can be altered by taking props off and starting the motors. I live inside an NFZ (and have to call the tower every time I go out), so this was very easy to test. 

I honestly do not know the theory behind exactly what parameters allow the "max" to be abused. I'm sure if CopterSafe wanted to share, he could. But I did want to clear up the record regarding MY suggestions vs. observations, and possible suspicions.

Mavic129, 2017-06-29:
dear @notsolowki 
to unlock all limits in any DJI drone you just need to do the following:
* connect your drone to DJI assistant 
from simple WebSocket client on Chrome connect to 
* ws://localhost:19870/general to get your drone param path like: ws://localhost:19870/controller/config/user/YOUR DRONE NUMBERS HERE
disconnect from general and connect to :
* ws://localhost:19870/controller/config/user/YOUR DRONE NUMBERS HERE
in the command write the below commands one by one:

to disable NFZ:
{"SEQ":"yb0au5sq","CMD":"write","INDEX":"g_config_airport_limit_cfg_cfg_disable_airport_fly_limit","VALUE":1}
{"SEQ":"jg4anshw","CMD":"write","INDEX":"g_config_airport_limit_cfg_cfg_limit_data","VALUE":20250910}
-----------------------------------------
to disabke height limit:

{"SEQ":"acbm54ja","CMD":"write","INDEX":"g_config_flying_limit_limit_height_abs_without_gps","VALUE":2500}
{"SEQ":"vc4km1vi","CMD":"write","INDEX":"g_config_flying_limit_limit_height_abs","VALUE":2500}
{"SEQ":"maza5cgv","CMD":"write","INDEX":"g_config_flying_limit_limit_height_rel","VALUE":2500}
{"SEQ":"bn0kjbwv","CMD":"write","INDEX":"g_config_flying_limit_height_limit_enabled","VALUE":2}
-----------------------------------------
to enable sport+

{"SEQ":"bsixzcic","CMD":"write","INDEX":"g_config_mode_sport_cfg_tilt_atti_range","VALUE":60}
{"SEQ":"04fbplxd","CMD":"write","INDEX":"g_config_mode_sport_cfg_vert_vel_up","VALUE":10}
{"SEQ":"fviwf4kz","CMD":"write","INDEX":"g_config_mode_sport_cfg_vert_vel_down","VALUE":-10}
{"SEQ":"uoo9h9u0","CMD":"write","INDEX":"g_config_fw_cfg_max_speed","VALUE":20}
-----------------------------------------
Enjoy

MrBurnsAT, 2017-06-29:
This will only Work for P4 and younger DJI Drones right!?

Or on P3 too?

MAVProxyUser, 2017-06-29:
@MrBurnsAT I think the best way to phrase it, is it will work on DJI equipment that is based on  A3 architecture. https://www.dji.com/a3

This includes the raw A3 itself, Spark, Mavic, Inspire2, and Phantom4. 

I believe p3 series is based on slimmed down A2 architecture. 
https://www.dji.com/a2

notsolowki, 2017-06-29:
Wow thanks everyone!!

MrBurnsAT, 2017-06-29:
For sure is P3 built Up in A2!

notsolowki, 2017-06-29:
How did you come up with the byte seq, did you use a usb sniffer to flush out the commands??

Mavic129, 2017-06-29:
i sniffed only Mavic 1.3.07, i believe commands will change based on drone type and FW

its more safe to use the WS as it is same for all drones

notsolowki, 2017-06-29:
Honestly, i dont know why i didnt think of that, but thankyou for taking the time to doit!!! Im sire it will help alot of people

MAVProxyUser, 2017-06-29:
@notsolowki based on the wording over at http://mavicunlocked.com I am going to assume the 2500 limit was specific to .400 firmware possibly (on Mavic, unknown analogs on other platforms)

[image]

notsolowki, 2017-06-29:
So what happens when you use the websocket to try to overide a parameters
max?

Mavic129, 2017-06-29:
it will give you error message that its failed to set the value exceeding the max limit

Mavic129, 2017-06-29:
BTW even coptersafe doesn't exceed the max limits

notsolowki, 2017-06-29:
If theres soemthing you dont want dij to see or you want to pm,
 notsoluckiATgmailDOTcom

notsolowki, 2017-06-29:
Really, so the only other options would be,
1. Firmware modification
2.root the drone?

The flight controller dont run android. Is it encrypted? @mefistotelis, do
you have ida running lol

mefistotelis, 2017-06-29:
I haven't checked the Mavic FW yet. I expect Ambarella runs Android, as new Ambarella SDK is based on it. During firmware update, I expect Ambarella to extract single modules from FW package and send them to specific components in encrypted form, same as it is in Ph3. Bootloaders of these components are tasked with decrypting.

notsolowki, 2017-06-29:
@mefistotelis, here are some tools that may help in extracting individual binaries . They appear to be encrypted though.

https://github.com/fvantienen/dji_rev

Thats if your willing to do any work with the mavic. I personally would like to use some tools similar to the ph3. Im sure theres lots of people willing to test it

fvantienen, 2017-06-30:
Both the LC and the amberella core are running android. And @mefistotelis is right about the part that the individual bootloaders decrypt the fw, but that is only for the FC and ESC's. I tried some known keys and ways to decrypt it but didn't work.
When you start looking at the Mavic you should contact me as I can provide you with a lot of information.

MAVProxyUser, 2017-06-30:
FWIW, I have shelled out on the Ambarella, and was not under the impression it was Android based. You can use the RNDIS access and a semi-known password to get in. It looks just like the Ambarella linux on the P3, the RTOS runs beneath it and there are weird IPC processes / services that are spoken to via "t" commands. 

Much of the info in GoPRO or GTFO applies indirectly. 
https://www.youtube.com/watch?v=nofJJoj5iys
https://www.defcon.org/images/defcon-21/dc-21-presentations/Manning-Lanier/DEFCON-21-Manning-Lanier-GoPro-or-GTFO-Updated.pdf

IIRC there are "t dji" commands that the SoC can execute... but again, it did not look very Androidish to me.

FLDataTeK, 2017-07-01:
Looks like it here..

https://github.com/droner69/MavicPro/tree/master/Firmware/Firmware_01.03.0400/_RC_wm220_1301_v01.04.17.03_20170120.pro.fw.sig.extracted/system/bin

..............................

# init adb device serial
if [ -f /data/dji/cfg/adb_serial ]; then
serial=`cat /data/dji/cfg/adb_serial`
busybox printf "$serial" > /sys/class/android_usb/android0/iSerial
fi

Mavic129, 2017-07-01:
@FLDataTeK I tested your method it didn't enable the ADB

dmanisgnarly, 2017-07-01:
When trying to send the command for any of the above modes on FW700 via DJI assitant 2 on Win10, I get the following "garbage" error. The example below is the message log output for sport mode. 
[image]

{"SEQ":"bsixzcic","CMD":"write","INDEX":"g_config_mode_sport_cfg_tilt_atti_range","VALUE":60}
{"SEQ":"04fbplxd","CMD":"write","INDEX":"g_config_mode_sport_cfg_vert_vel_up","VALUE":10}
{"SEQ":"04fbplxd","CMD":"write","INDEX":"g_config_mode_sport_cfg_vert_acc_up","VALUE":10}
{"SEQ":"fviwf4kz","CMD":"write","INDEX":"g_config_mode_sport_cfg_vert_vel_down","VALUE":-10}
{"SEQ":"fviwf4kz","CMD":"write","INDEX":"g_config_mode_sport_cfg_vert_acc_down","VALUE":-10}
{"SEQ":"uoo9h9u0","CMD":"write","INDEX":"g_config_fw_cfg_max_speed","VALUE":20}}
{
    "ERROR": "FAILURE",
    "ERROR_MESSAGE": "garbage at the end of the document"
}

Mavic129, 2017-07-01:
there is double }} ate the end of the command
i never tried to execute all commands at once 
usually i execute them one by one

dmanisgnarly, 2017-07-01:
Thanks!  All is well now.  I did not realize the strings had to be entered one at a time.  

Received an over current discharge error within 5 mins, backed off briefly and finished the battery out.  Will probably tweak some settings with a dev parameters spreadsheet I got ahold of last night.

MAVProxyUser, 2017-07-02:
I could use a few beta testers... I've finally figured out the P0V ftpd "dir traversal" bug and can share details of how it works. You need to be added as a collaborator. Hit me up. We can now write files as root anywhere *writable*. This was "patched" by making /system "ro" in newer firmware versions. On Mavic, I suspect only very early (shill / beta versions like Casey Neistat had) were exploitable. This does manifest on other *similar* hardware such as Spark in current firmware. 

https://github.com/MAVProxyUser/P0VsRedHerring

2008jiayu, 2017-07-03:
@Mavic129 I just update the LC1860 model using wrong firmware asisstant 2 beta, my drone cannt connect the computer now,how to recover it ,how can i go into the bootloader .😂
@notsolowki

coptersafe, 2017-07-03:
@2008jiayu  when connect drone to PC , have you any DJI COM port in device manager?

devkadji, 2017-07-03:
I made a similar stupid mistake like @2008jiayu by flashing earlier FW version for FlightCtrl device (which went ok according do DJI Assistant beta 112). @coptersafe - yes, there is a DJI COM port device in the device manager. It had some driver-related issue so I had to edit the Vision_(Interface_3).inf file a bit to install the driver.

2008jiayu, 2017-07-03:
@copersafe,yes  ,have there are two usb ports   ,the one is com port , the other is usb lan but it say the Cable not inserted
@everyone

2008jiayu, 2017-07-03:
help

MAVProxyUser, 2017-07-04:
The DJI ftpd directory transversal exploit technique by P0V is now public... 
https://github.com/MAVProxyUser/P0VsRedHerring

https://www.youtube.com/watch?v=BTQ_CTih1HM

skeimi, 2017-07-04:
Drone number here? Where i can see my "drone number"?

skeimi, 2017-07-05:
Got it! :)

Goof245, 2017-07-20:
So I'm guessing all this is why the steaming pile of crap that is the P4 2.00.0106 FW was rushed out.... Hoping I can get this thing downgraded....

MAVProxyUser, 2017-07-20:
@Goof245 stop on by...

MAVProxyUser, 2017-07-20:
@Goof245 stop by and see us... I'll repost this here, as referenced in the dji_system.bin repo, this is a good place to start: http://dji.retroroms.info

For those of you not in the loop, that want to help with the "retention" process regarding control of your DJI aircraft, please familiarize yourself with the following repos: 

This represents the front lines of the resistance as it were... "the movement" pretty well begins in all of these individual battle grounds. 

[image]

https://github.com/Bin4ry/deejayeye-modder - APK "tweaks" for settings & "mods" for additional / altered functionality 

https://github.com/hdnes/pyduml - Assistant-less firmware pushes and DUMLHacks referred to as DUMBHerring when used with "fireworks.tar" from RedHerring. DJI silently changes Assistant? great... we will just stop using it. 

https://github.com/MAVProxyUser/P0VsRedHerring - RedHerring,  aka "July 4th Independence Day exploit", "FTPD directory transversal 0day", etc. (Requires Assistant). We all needed a *public* root exploit... why not burn some 0day? 

https://github.com/MAVProxyUser/dji_system.bin - Current Archive of dji_system.bin files that compose firmware updates referenced by MD5 sum. These can be used to upgrade and downgrade, and root your I2, P4, Mavic, Spark, Goggles, and Mavic RC to your hearts content. (Use with pyduml or DUMLDore)

https://github.com/MAVProxyUser/firm_cache - Extracted contents of dji_system.bin, in the future will be used to mix and match pieces of firmware for custom upgrade files. This repo was previously private... it is now open. 

https://github.com/jezzab/DUMLdore - Even windows users need some love, so DUMLDore was created to help archive, and flash dji_system.bin files on windows platforms. 

So... that is all! Have fun folks, stop by Slack and see us if you get bored. #android_apk_patching, #archived_fw_flashing, #factory_mode_access, #firm_cache, #hardware, #mavic_rooting, #safety_shaming channels all have something for everyone. If not.. feel free to lurk in #general.

We are currently looking to archive as much firmware as possible if anyone wants to help... https://www.rcgroups.com/forums/showpost.php?p=37941901&postcount=1704

stephengardner, 2017-07-27:
Mavic on fw 0400 here, successfully switched cfg_cfg_disable_airport_fly_limit to 1 and the cfg_cffg_limit_data to 20250915 but I'm still unable to take-off in an NFZ.  GPS mode still disables this.  Thoughts?

ruckusman, 2017-09-05:
@MAVProxyUser  - You will find a lot of useful information on the T commands - which are executed directly by the Ambarella RTOS, amongst the Gopro hacks here - mot are dedicated to camera control and configuration functions

https://github.com/KonradIT

skeimi, 2017-10-21:
Does Mavic129 have any WebSocket Client commands for normal gps speed boost?

jmchugh02, 2017-12-24:
Sorry for the complete n00b comment, but I'm still trying to figure out how one obtains their drone # from simple websocket client chrome extension? Running a DJI Mavic Pro.

w/DJI Assistant running & connected, I'm seemingly able to connect to:

ws://localhost:19870/general... however all I see from there is what appears to be a large string of encrypted text. No distinct numbers that I can tell to use.

***YOUR DRONE NUMBERS HERE***

skeimi, 2018-02-21:
this web socket doesnt work anymore, all crypted.. any ideas?

MAVProxyUser, 2018-02-21:
We need to pull the app.asar apart and see what is new...

skeimi, 2018-02-21:
I was going to modify braking because i backflipped my mavic when braking+full up 😆 
I havent updated firm since, dunno whats changed now 😢

PreawR, 2018-12-14:
> https://github.com/MAVProxyUser/DJIAssistant2Binaries/blob/master/Win/DJI%20Assistant2%20Beta112.zip

Can you give me a DJI Assitant 1.1.2 download address please！！

PreawR, 2018-12-14:
> https://github.com/MAVProxyUser/DJIAssistant2Binaries/blob/master/Win/DJI%20Assistant2%20Beta112.zip

Please！

PreawR, 2018-12-14:
> @notsolowki基于http://mavicunlocked.com上的措辞我将假设2500限制特定于.400固件可能（在Mavic上，在其他平台上未知的模拟）
> 
> [image]
Do you know How to go debugger mode ? I can't do it .I've tried --debugger and ctrl+shift+i .

PreawR, 2018-12-14:
> @notsolowki基于http://mavicunlocked.com上的措辞我将假设2500限制特定于.400固件可能（在Mavic上，在其他平台上未知的模拟）
> 
> [image]

Do you know how to debugger mode ? like fo you this picture .  I tried --debugger and ctrl+shift +i ,all can't do it .

[2 more replies]