o-gs/dji-firmware-tools #37: P3P refuses to start upgrade process after forced downgrades

DannyBoyCJ, 2017-07-02:
Hello guys, I am new to this so go easy on me please...  I upgraded my P3P to version P3X_FW_V01.10.0090, following the DJI guide, and got a failed upgrade signal and the dreaded message in the log file:

[15 00][00] device not detected
[08 00][00] device not detected

After this, there was no video transmission. Then I googled the issue and found a guy who said he resolved this by downgrading all the way to the first version available to him (by the "debug" file method), and then upgrading again, step by step, up to version P3X_FW_V01.08.0080, at which time he got a successful upgrade process, and a functional drone. I proceeded to do the same, but when I tried to go from 1.2.3 to 1.1.9, it went wrong. The process didn't finish, and I found the drone with a limp gimbal, the camera just hanging at an angle underneath, beeping. The last thing written in the log file is: 

 [00055385]Firmware upgrading[1]...
 [00066342][01 00] Firmware upgrade start...

My problem is that now I can't get it to enter the upgrade mode no matter what firmware version  I place on the card, with or without the "Debug" file. When I try to force start the upgrade process by pressing the link button during startup, the gimbal immediately goes limp and starts beeping (but not the normal beeps it emits during upgrade). It actually resumes the state in which I found it after the last failed upgrade. It does nothing and writes no log, no matter how long I leave it at it.

This is the last log file, written at the time of that last failed downgrade:
[P3X_FW_LOG_AB.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1118079/P3X_FW_LOG_AB.txt)

So, if you have any ideas, or know of a procedure to make it upgrade again (via serial service pads, or any other method), please help.  Also, if you need more info, just let me know.
Thank you!

mefistotelis, 2017-07-03:
Connect to Ambarella serial console (U0 pins) and check what you see.
If there's nothing - you have to reflash MX30LF by hand. If there's something - maybe you will be able to get into bootloader and flash from sd-card.

DannyBoyCJ, 2017-07-03:
Thank you, I will try this and post back. Just to be clear, do I need to power the board outside the drone to have access to the pins, or do I do this with the drone (and gimbal) assembled normally, and just run thin wires from the U0 pins to the TTL-USB converter?  If necessary, I have a stabilized variable voltage source that I can use. I just want to make sure I understand the procedure correctly. Also, if there's a thread on this that you know of, please share the link... Thank you!

mefistotelis, 2017-07-03:
>  do I need to power the board outside the drone to have access to the pins, or do I do this with the drone (and gimbal) assembled normally

Whichever works best for you. You will not need communication with any other part of the drone to flash Ambarella FW.

>  if there's a thread on this that you know of, please share the link

There is. [Link](https://www.google.pl/search?q=phantom+gimbal+%22u0_tx%22+%22u0_rx%22).

DannyBoyCJ, 2017-07-03:
Thanks a lot, I'll try this tomorrow and get back with the results!

DannyBoyCJ, 2017-07-04:
Ok, so it looks alive... This is the log from Putty:

[Putty_log_1_U0_R_U0_T.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1122646/Putty_log_1_U0_R_U0_T.txt)

How do you think I should proceed based on this?

notsolowki, 2017-07-04:
Looks like the camera is bricked, it dont detect the sdcard

DannyBoyCJ, 2017-07-04:
The SD card wasn't inside, I'll try to do it again with it mounted and post back...

DannyBoyCJ, 2017-07-04:
Ok, this is weird, I think it just started updating from SD (I had firmware on it), still connected to the TTL adapter. This is what it's done so far:

[Putty_log_2_U0_R_U0_T.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1122736/Putty_log_2_U0_R_U0_T.txt)

Do I let it go on? Seems like it's in a loop to me.

DannyBoyCJ, 2017-07-04:
Ok, definitely a loop, I stopped it. Bad blocks... is that solvable?

mefistotelis, 2017-07-04:
You'd probably have to replace the NAND chip. I don't have experience in it.
@coptersafe, any advice for NAND with bad blocks?

MrBurnsAT, 2017-07-04:
Should Repair itself by powering on Time by Time.
Same Problem here with NAND run Full of Bad Blocks at an HDMI Output Module

DannyBoyCJ, 2017-07-04:
Does that mean powering it on and off repeatedly? Is that with or without the firmware on the SD? And for how long should I let it run each time?

notsolowki, 2017-07-05:
Maybe the chip itself became unseated from The pcb

DannyBoyCJ, 2017-07-05:
The chip looks completely flush with the board. Of course, that's no guarantee... but the drone never had a physical shock. And it all happened while it was sitting on my desk, updating. I could try to do a reflow (I've done it before on a laptop video-board chip). I am bit reluctant though, because there are a ton of very small components around this chip, and my heating equipment isn't exactly precise. Just to be clear, the nand chip is this one...

[image]

...right? The MXIC. I wouldn't want to work on the wrong one. In the meantime, does anyone have a spare gimbal board like this? I might have to buy another one.

DannyBoyCJ, 2017-07-06:
This is what I get when I connect to 368_U0 pins:

DM36x initialization passed!
UBL Product Vesion : DJI-NOHDMI-WM610-UBL-1.0-rc0(2015-01-13)
Dji UBL Version: 1.51(Jan 13 2015 - 16:42:11)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
Valid magicnum, 0xA1ACED66, found in block 0x0000001D.
No valid boot image found!
NAND Boot failed.
Aborting...

Is there a way to boot from another device instead of the nand memory?

mefistotelis, 2017-07-06:
> Is there a way to boot from another device instead of the nand memory?

Don't know.
But there is a way to update firmware via that serial connection:
https://github.com/mefistotelis/phantom-firmware-tools/wiki/Firmware-m0800#by-service-serial-port

MrBurnsAT, 2017-07-06:
Try flashing my Package.

[HDMI.Final.zip](https://github.com/mefistotelis/phantom-firmware-tools/files/1129375/HDMI.Final.zip)

With Tools from here
[LB_hack.zip](https://github.com/mefistotelis/phantom-firmware-tools/files/1129377/LB_hack.zip)

DannyBoyCJ, 2017-07-06:
Thanks guys, I'll try anything at this point.... I'll post back the results.

MrBurnsAT, 2017-07-07:
here are the right ubl1 and uboot images for ur board!!!

i decrypte the firmware file and edited the ubl1 and uboot file that u can flash it with the tool from LB Hack

[Uboot OFMD.zip](https://github.com/mefistotelis/phantom-firmware-tools/files/1130599/Uboot.OFMD.zip)

DannyBoyCJ, 2017-07-07:
Thanks a lot, I'll give it a try!

DannyBoyCJ, 2017-07-07:
One more question before I go at it: which one is the service pad for boot select on the p3p top gimbal board? I found BOOT0 ans B_S as possible candidates so far...

DannyBoyCJ, 2017-07-07:
I tried shorting BOOT0 to 3.3VD, but it doesn't say BOOTME. It tries to boot normally, from NAND:

DM36x initialization passed!
UBL Product Vesion : DJI-NOHDMI-WM610-UBL-1.0-rc0(2015-01-13)
Dji UBL Version: 1.51(Jan 13 2015 - 16:42:11)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
Valid magicnum, 0xA1ACED66, found in block 0x0000001D.
No valid boot image found!
NAND Boot failed.
Aborting...

 The same with B_S pad shorted to 3.3VD. Am I missing something?

MrBurnsAT, 2017-07-07:
Try Shorting BOOT0 with VCC3V3

MrBurnsAT, 2017-07-07:
Found this in ner

Try Shorting BT01, BT02 and 3.3

MrBurnsAT, 2017-07-07:
[image]

[image]

DannyBoyCJ, 2017-07-07:
Excellent advice, MrBurnsAT! It worked. I had to Ctr+C from the "BOOTME BOOTME" loop two times (and repeat the "-nandflash" command), but it finally took, and ended in "operation successfull". I guess I'm going to try an old fashion SD card FW upgrade next, I'm just not sure what version I should use. The last one I flashed was 1.1.9 (coming from 1.2.3, with debug), but it never finished upgrading. If you have any advice at this stage, it would be very welcome.

mefistotelis, 2017-07-07:
Good. Updated wiki with your findings.

MrBurnsAT, 2017-07-07:
Yesssssss

One with an Working Unit!
Im struggling with my HDMI Board which Think ist an Lightbridge 😂😂😂

DannyBoyCJ, 2017-07-07:
Oh man, your HDMI Board who thinks it's a Lightbridge is awesome! I read your thread and it's gold. And yes, thanks a lot, I do have a working(ish) unit now! I'm actually getting image on my tablet (albeit delayed and at low frame rate), without doing any  firmware upgrading. I will have to, though, because the app says there's a mismatch in the firmware. Also, my remote is now on a newer version, so I might have to downgrade that too. I will start by trying to get a clean 1.2.3 version on the aircraft and, if successful, work my way up from there. Also, a very big thumbs up to mefistotelis, for creating and maintaining this outstanding knowledge pool!

DannyBoyCJ, 2017-07-08:
Well, sadly, my drone still won't update. I still get this fom Ambarella:

[00057809][CA9_0] Program "System Software" to NAND flash ...
[00060922][CA9_0] Initial bad block. <block 173>
[00060967][CA9_0] Done
[00061171][CA9_0] Program "DSP uCode" to NAND flash ...
[00062498][CA9_0] Done
[00063108][CA9_0] Program "System ROM Data" to NAND flash ...
[00063108][CA9_0] Initial bad block. <block 309>
[00063241][CA9_0] Initial bad block. <block 313>
[00067047][CA9_0] Done
[00067330][CA9_0] Program "Linux Kernel" to NAND flash ...
[00069145][CA9_0] Done
[00070592][CA9_0] Program "Linux Root FS" to NAND flash ...
[00071974][CA9_0] Initial bad block. <block 541>
[00072018][CA9_0] Initial bad block. <block 543>
[00079742][CA9_0] Initial bad block. <block 886>
[00079791][CA9_0] Done
[00080123][CA9_0] [DJI] Done!
[00080123][CA9_0] [DJI] Reboot...... 

I wonder if I should erase the NAND before flashing it... Would that mark the bad blocks?

mefistotelis, 2017-07-08:
This log comes from Ambarella, not DM36x. It's an unrelated problem. Different NAND chip.

DannyBoyCJ, 2017-07-09:
Sorry, my  bad, I thought they were using the same chip... so here's the log from DM36x, with the SD card inside (with firmware on it):
[putty_09_07_2017_025755.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1133457/putty_09_07_2017_025755.txt)
The errors at the end are probably there because the camera is not connected to the top gimbal board. I tried the procedure with the drone fully assembled, still won't update. And the gimbal still goes limp when I try to force-update by link button, or try to run the service file (MVOM0Fw).

Mavic129, 2017-07-09:
this message means the DM36x is not connected to the LB
GetFPGAstatus err!
FPGA[1,0,1,0]-1.000000:0
hdmi status is 1
ARM Load: 7% Video fps: 8 fps Video bit rate: 0 kbps PHY Bandwidth: 0 kbps Time: 00:00:03 Resolution: 1280x720

mefistotelis, 2017-07-09:
> this message means the DM36x is not connected to the LB
...
and from the log i can tell the DM36x is functioning properly

Fully agree with @Mavic129. As I previously stated, you have an issue with Ambarella. DM36x is fixed.

DannyBoyCJ, 2017-07-09:
Ok, that is not good news. I don't know what to do next. From the Ambarella log, I take it that it is flashing it's own NAND. Maybe I should try flashing the LB? Is that possible on the P3P?

MrBurnsAT, 2017-07-09:
Loading from nand0, offset 0x4a0000
** Unknown image type
Wrong Image Format for bootm command
ERROR: can't get kernel image!

Same Problem as at my HDMI Board.

Im Thinking now:
We are only flashing ubl1 and u-boot File.

But in decodet Firmware there is a uImage File too. 
Which contains the Linux of DM368 

But i dont know how to Flash it.

Look at the decrypted Firmware in
/dji/bin/loadmodule-rc
And 
/dji/etc/updatefinish

MrBurnsAT, 2017-07-09:
thats in /dji/etc/updatefinish

```
#!/bin/sh
rm -f /opt/tmp/update.tar.gz

UBL1="/opt/dji/data/ubl1.img"
UBL2="/opt/dji/data/ubl2.img"
UBOOT="/opt/dji/data/u-boot.img"
KERNEL="/opt/dji/data/uImage"

RCS="/opt/dji/etc/rcS"

if [ -f $UBL1 ]
then
	echo "mtd write $UBL1"
	mtdwritefile -f $UBL1 -o 0x20000
	rm -rf $UBL1
fi

if [ -f $UBL2 ]
then
	echo "mtd write $UBL2"
	mtdwritefile -f $UBL2 -o 0x40000
	rm -rf $UBL2
fi

if [ -f $UBOOT ]
then
	echo "mtd write $UBOOT"
	mtdwritefile -f $UBOOT -o 0x320000
	rm -rf $UBOOT
	echo "delete uboot env"
	mtderase -o 0x420000 -l 0x40000
fi

if [ -f $KERNEL ]
then
	echo "mtd write $KERNEL"
	mtdwritefile -f $KERNEL -o 0x4a0000
	rm -rf $KERNEL
fi

if [ -f $RCS ]
then
	echo "replace rcS"
	mv $RCS /etc/init.d/
fi

sync
#reboot
```

DannyBoyCJ, 2017-07-09:
Hmm... I see it now... Maybe it can be flashed by the same method. MrBurnsAT, do you perhaps have this image for Linux (or is there a way to extract it)? You say you have the same problem with your HDMI board, have you tried flashing the Linux image by any method, even if not successful, yet? Also, how do you access the Linux system on the drone? (/dji/etc/updatefinish). Sorry for asking such questions, but as I said, I have no experience and I'm learning as I go here...

MrBurnsAT, 2017-07-09:
No Not in Linux System!!!

In the extracted Firmware
Decrypted.tar.gz

Into that File are all These folders

In Folder Data u have the ubl1, ubl2, u-boot and uImage Files

No i wasnt able to Flash the uImage File.

Because i dont know how

DannyBoyCJ, 2017-07-09:
Oh, ok, thanks. I'll research this, there must be a way to do it. If I find anything useful, I'll post it here.

MrBurnsAT, 2017-07-09:
If i find something Ill Post it too

DannyBoyCJ, 2017-07-09:
Updating a kernel from a linux prompt
If you want to be able to update your kernel or uboot image in NAND (I'm using a DM355 platform, but this should work for others) *while* your system is running and without the use of a JTAG device, these are the commands that can be used to do this:

   /usr/sbin/flash_eraseall -j /dev/mtd2
   /usr/sbin/nandwrite -p /dev/mtd2 uImage

I found this method here:
http://processors.wiki.ti.com/index.php/Writing_Kernel_Image_to_NAND_Flash#Writing_the_Kernel_Image_into_NAND_Flash

The page is for DM355 and does not apply to DM36x, but, as written above, this particular way of flashing "should work for other" platforms. So... how do  we access shell on our chip? And do you think this would work?

Another possibly useful link: http://processors.wiki.ti.com/index.php/UG:_DaVinci_PSP_Installation_on_DM36x_EVM

DannyBoyCJ, 2017-07-09:
Hey guys, apparently the GetFPGAstatus error was indeed caused by the camera not being attached to the main gimbal board. Here is the log with the camera attached (and no FPGA errors):

[putty_10_07_2017_012219.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1134132/putty_10_07_2017_012219.txt)

The link to LB works, but the drone still won't take an update from SD, though.
MrBurnsAT, regarding the kernel image, as you said, at first it reports:

--------------------------------
Loading from nand0, offset 0x4a0000
** Unknown image type
Wrong Image Format for bootm command
ERROR: can't get kernel image!
--------------------------------

I think the Linux image is not located at offset 0x4a0000. Only the "bootm" command is located there, and it is probably corrupted. But a duplicate of this command is located at offset 0x900000 (maybe a backup?), and it starts executing after the first one fails. The "Load Address" and name of the Linux image (80008000) is the same as the one in the sample here:
https://github.com/mefistotelis/phantom-firmware-tools/wiki/Firmware-m0800#debug-serial-interface-368_u0

-------------------------------
Loading from nand0, offset 0x900000
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2015-03-30   9:54:39 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4032832 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
## Booting kernel from Legacy Image at 80700000 ...
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2015-03-30   9:54:39 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4032832 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
------------------------------

If the uImage is corrupted, what is this image that it's loading? I can send commands to linux through the 368_U0 serial interface and it will execute them (I can list the processes and also kill them). Also, it appears to do it's job, apart from updating. This got me thinking that maybe your uImage is intact too (and at the same address), and only the command to load it got somehow corrupted. What do you think? If that is the case, maybe you could try launching it manually, from a uboot prompt. Also, if what I said 'till now is completely wrong, please disregard it, because I litterally don't know anything about Linux, apart from what I read these last two days.

Getting back at my issue, what else do you think could cause my "no updating" symptom? What other diagnostics can I run?

MrBurnsAT, 2017-07-10:
Do u know which firmware is in ur Bird?

If yes. I would try to Update m0800 and m1500 with 
PDM368Fw3.bin (m0800) and P68013Fw3.bin (m1500) from correct firmware on SD Card

DannyBoyCJ, 2017-07-10:
Truth is, I'm not quite sure. The last fully flashed firmware was "Phantom_3_Professional_Firmware_v1.2.6_en", which ended in failure. Then I attempted to flash "Phantom_3_Professional_Firmware_v1.1.9_en", but that process didn't finish, and broke my drone. So, maybe it's 1.2.6, or maybe a combination with 1.1.9.... I should probably extract them from version 1.2.6.
I never used the tools from this repository before, so I guess I'm about to learn some new skills :))

MrBurnsAT, 2017-07-10:
If u want i can extract it for u and Post it here.

Wich Update did u try?
1.1.9 and 1.2.6?

Try 1.3.2 or higher with debug File!

MrBurnsAT, 2017-07-10:
PS bevore u van Downgrade from 1.2.6 down to 1.1.9 u must use MVOM0Fw.bin 
After that u can go down

DannyBoyCJ, 2017-07-10:
That was probably my mistake, I tried going to 1.1.9 without using MVOM0Fw.bin first. I tried it now, but it doesn't work, either. If you could extract those files for me it would really help me.  I tried most of the versions until now, and none worked. It would be useful if I could "listen in" on the updating process through the serial connection, but I don't know which is the best interface for monitoring this, or if it is even possible. U0 and 368_U0 do not give signs of updating. On the U0 I can see an updating log, but only when the camera is disconnected from the main gimbal board, and it is a very short process. It looks like this:

[putty_05_07_2017_162816.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1135563/putty_05_07_2017_162816.txt)

So I will try to see if it works with the debug file now on version 1.3.2, as you suggested, and if it doesn't, I'll give it a try with the extracted firmware for m0800 and m1500 (probably from 1.2.6).

DannyBoyCJ, 2017-07-10:
Nope, firmware + debug doesn't work, and neither does  MVOM0Fw.bin. The drone acts like there's no firmware on card.

MrBurnsAT, 2017-07-10:
Why did u try an Update to V01.09.0060???

Ans Not 1.3.2???

Dont do so big Version jumps!

Now i Need the log File from sd Card After Update.

/misc/log/P3X_FW_LOG_AB.txt

DannyBoyCJ, 2017-07-10:
I didn't try 1.09.0060, but "Phantom_3_Professional_Firmware_v1.1.9_en", the one before 1.2.6. The last log my drone wrote is from 6 weeks ago, when it got screwed up during downgrading. It hasn't written a log since. This is it, the log from 6 weeks ago:
[P3X_FW_LOG_AB.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1136418/P3X_FW_LOG_AB.txt)

MrBurnsAT, 2017-07-10:
Ur Putty log says something different

[0m[00000820][CA9_0]  Packet Detected / c:\P3X_FW_V01.09.0060.bin /.

DannyBoyCJ, 2017-07-10:
You are right, I probably wanted to use 1.1.9 and used 1.09 instead... Anyway it does exactly the same thing no matter what version of firmware is on the card, but only if the camera is disconnected from the gimbal main board. If the drone is fully assembled, the serial log looks like this:

[putty_10_07_2017_U0.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1136637/putty_10_07_2017_U0.txt)

It's like it never detects there's a firmware file on the SD card. So, my guess is that when it can't detect the camera, it tries to update, because it thinks it's a hardware error. But when the camera is present, it has no reason to look for an upgrade. It would be useful to know which process is responsible for starting the upgrade process. Maybe it can be initiated manually, from the linux prompt. For reference, here is a list of active processes, displayed with the "top" command:

[7m  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND[0m
  428     1 root     S <  83256185.9   0 61.4 /opt/dji/bin/encode_usb -a -s
  437   431 root     R     3140  7.0   0  7.6 top
  431   430 root     S     3144  7.0   0  0.0 -sh
    1     0 root     S     3140  7.0   0  0.0 init
  430     1 root     S     3140  7.0   0  0.0 {autologin} /bin/sh /sbin/autologi
  427     1 root     S     1748  3.9   0  0.0 /opt/dji/bin/monitor
    5     2 root     SW       0  0.0   0  0.0 [khelper]
    4     2 root     SW       0  0.0   0  0.0 [events/0]
    3     2 root     SW       0  0.0   0  0.0 [ksoftirqd/0]
    2     0 root     SW       0  0.0   0  0.0 [kthreadd]
    8     2 root     SW       0  0.0   0  0.0 [async/mgr]
  137     2 root     SW       0  0.0   0  0.0 [sync_supers]
  139     2 root     SW       0  0.0   0  0.0 [bdi-default]
  141     2 root     SW       0  0.0   0  0.0 [kblockd/0]
  155     2 root     SW       0  0.0   0  0.0 [khubd]
  158     2 root     SW       0  0.0   0  0.0 [kseriod]
  190     2 root     SW       0  0.0   0  0.0 [rpciod/0]
  199     2 root     SW       0  0.0   0  0.0 [kswapd0]
  200     2 root     SW       0  0.0   0  0.0 [aio/0]
  201     2 root     SW       0  0.0   0  0.0 [nfsiod]

MrBurnsAT, 2017-07-10:
Here are the files from 1.02.0006

i added it 2 times! 

2 files with old filename (without P at start, and without 3 at end)
2 files with new filename (with P at start, and with 3 at end)

[1.02.0006.zip](https://github.com/mefistotelis/phantom-firmware-tools/files/1136869/1.02.0006.zip)

DannyBoyCJ, 2017-07-10:
Thank you! I'll give them a run, see what happens!

DannyBoyCJ, 2017-07-10:
Ok, so good news so far. I copied the files with P at start, and with 3 at end on the SD card, and my bird started updating. And it finished succesfully! However, it still won't take the normal 1.2.6 bin placed on SD. So I'm thinking: can we do the same thing for all the modules? Maybe flashing all of them this way will finally put a consistent (and fully working) firmware on the drone. Here is the log it wrote now:
[P3X_FW_LOG_AB.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1137173/P3X_FW_LOG_AB.txt)
The way I see it from the log file, last time it didn't finish upgrading module 0100. So maybe that's where my problem is. Is m0100 upgradeable in this way? Anyway, very good work, MrBurnsAT !!

MrBurnsAT, 2017-07-10:
pls here is m0100

[PFC300XFw3.zip](https://github.com/mefistotelis/phantom-firmware-tools/files/1137222/PFC300XFw3.zip)

DannyBoyCJ, 2017-07-10:
Awesome, let's see how this goes :)

DannyBoyCJ, 2017-07-10:
Ok, so it looks like it starts updating, but it actually doesn't. This is the log:
[P3X_FW_LOG_AB.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1137251/P3X_FW_LOG_AB.txt)
And this is the log from Ambarella console (U0 pads):
[putty_11_07_2017_PFC300XFw3.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1137260/putty_11_07_2017_PFC300XFw3.txt)
What I find strange is that although the putty log states that it can't find firmware on the SD, the AB log shows that it is trying to program m0100, but fails with the same error as last time. There's something obviously wrong with module 0100. From what I see, modules 0100 and 0101 are part of Ambarella firmware. Hmm... Should I try this with the files for all the modules on the SD? I know that they don't work if they are encrypted, so... can the encrypted ones be decrypted and prepared for flashing?

MrBurnsAT, 2017-07-10:
No they dont Need to be decrypted!

Only rename to right Name.
And put on sd

[37 more replies]