o-gs/dji-firmware-tools #38: HDMI Interface Board DaVinci DM368 firmware

MrBurnsAT, 2017-07-03:
Hello

I have an non working HDMI module from an early Inspire1 Remote
Which was long mit used and so NAND run full of bad blocks. 

Board is labeled "WM610_GRCP_MFI368_V1"
Interface Board is "WM610_GRCP_INTERFACE_V3"

[image]
[image]

I Hooked it Up to serial Adapter 

[image]

And got this

DM36x initialization passed!
UBL Product Vesion : DJI-GRC-SUPER-UBL-1.0-rc0(2014-09-19)
Dji UBL Version: 1.51(Sep 19 2014 - 16:13:09)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
Valid magicnum, 0xA1ACED66, found in block 0x0000001D.
No valid boot image found!
NAND Boot failed.
Aborting...

Now my question 

Can we extract Firmware for this Board?
Maybe something like uboot image like at Lightbridge hack

MrBurnsAT, 2017-07-03:
Found this interesting Site, with an interesting answer

https://phantompilots.com/threads/crowdfund-hacker-to-build-custom-firmware.64329/page-6

mefistotelis, 2017-07-04:
> Can we extract Firmware for this Board?

Yes, but we need a working one for that.

It looks like this is firmware m1300, which is "Salted" in C1 packages. We can't be sure it is really this one, and we don't know the password required for decryption - so extracting it from a working board is easier and safer.

MrBurnsAT, 2017-07-04:
I have New Board which Works.

Must look which Version the New one is.
But i Think they all have the same firmware because of same components.

If u can tell me how to do it.
Ill try it

mefistotelis, 2017-07-04:
You just have to send the content of the NAND device to your pc via serial.

Here's one article which lists a few ways:
http://k.japko.eu/serial-transfers.html

MrBurnsAT, 2017-07-04:
Tryed the Lightbridge hack with annexed files.

Now borard is running  BUT !
I cant connect via GoApp. 
Dont get the pop up message which program to start (GoApp, Go4,...)
But instead the USB Debugging Message like if i connect my Lightbridge 1 with my Android Tab

So now i thought.
Hmmm. 
Maybe thinks this thing now its an Lightbridge. 
hmmmm
Lets try the DJI Lightbridge APP
ok. App startet. Message on Screen "Connection Broken"
now usb cable connected and rc turned on
and see!?!?!
WOW "Connection Broken" dissappears

I have flashed the ubl Image of an Lightbridge. so this thing now is an Lightbridge. 

I think i need decrypted files (ubl and uboot) for module m1300 HDMI Video Decoder
(maybe from P3X_FW_V01.04.0001_mi09.bin, P3X_FW_V01.02.0006_mi09.bin, P3X_FW_V01.01.1003_mi09.bin)

MrBurnsAT, 2017-07-05:
So ok i Need original Firmware Files(ubl1 and u-boot). Dont know how to extract it from Firmware Module.

Here is my New Working Unit.
Ist exactly the same as my fault Unit which Thinks at the Moment that its an Lightbridge 1.

[image]

I Hooked it Up Now.
Lets see what we get. 

If all Works. I wanna do a NAND dump.
Hope i can restore the whole Firmware 

[image]

Wish me luck

MrBurnsAT, 2017-07-05:
juhu 

looks good

here is what i get

```
DM36x initialization passed!
UBL Product Vesion : DJI-486M-UBL-1.0-rc0(2015-10-30)
Dji UBL Version: 1.51(Nov  2 2015 - 15:46:28)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
   DONE
Jumping to entry point at 0x81080000.

U-Boot Product Vesion : DJI-DEC-Uboot-1.0-rc0(2015-11-02)
U-Boot 2010.12-rc2-svn3214-Dji (Nov 02 2015 - 19:39:56)
Cores: ARM 486 MHz
DDR:   360 MHz
I2C:   ready
DRAM:  128 MiB
NAND:  128 MiB
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds

Loading from nand0, offset 0x4a0000
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2015-06-23   9:37:05 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4065664 Bytes = 3.9 MiB
   Load Address: 80008000
   Entry Point:  80008000
## Booting kernel from Legacy Image at 80700000 ...
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2015-06-23   9:37:05 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4065664 Bytes = 3.9 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Kernel Product Vesion : DJI-GRC-Kernel-1.1-rc0(2015-06-23)
[    0.000000] Linux version 2.6.32.17-davinci1 (root@dji) (gcc version 4.3.3 (Sourcery G++ Lite 2009q1-203) ) #6 PREEMPT Tue Jun 23 17:37:03 HKT 2015
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: DaVinci DM36x EVM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] DaVinci dm36x_rev1.2 variant 0x8
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 12192
[    0.000000] Kernel command line: console=ttyS0,115200n8 rw dm365_imp.oper_mode=0 video=davincifb:vid0=0,10K:vid1=0,10K:osd0=1920x1080X16,8100K mem=48MB davinci_enc_mngr.ch0_output=COMPOSITE davinci_enc_mngr.ch0_mode=pal ubi.mtd=2,2048 root=ubi0:rootfs rootfstype=ubifs ip=off lpj=1077248
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 48MB = 48MB total
[    0.000000] Memory: 44424KB available (3672K code, 444K data, 116K init, 0K highmem)
[    0.000000] SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] Console: colour dummy device 80x30
[    0.000000] Calibrating delay loop (skipped) preset value.. 215.44 BogoMIPS (lpj=1077248)
[... 210 lines cut ...]
[   12.390000] HTotal = 0 -> 1650
[   12.390000] VTotal = 0 -> 750
[   12.410000] ScanMode = I -> P
[   12.410000] PCLK = 0,(1650x750) PROGRESS CHANGE MODE
[   12.410000] VIC = 4
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
encrypt chip exist
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
DummyRead68013 0.000000kb
```

MrBurnsAT, 2017-07-05:
dumped ubl1 image (3 times) and flashed these 
but only V2 works little bit. i think there are some characters wrong in .bin

annexed my dumped files

her is the output of flashed ubl1

```
▒▒0▒▒▒ initialization failed!
ed!
-486M-UBL-1.0-rc0(2015-10-30)
2015-10-30)
10-30)
-30)
BL Version: ersion: 1.5151atalog Boot Loader
BootMode = ta verification failed! Block:
```

[dump UBL 1 _001.zip](https://github.com/mefistotelis/phantom-firmware-tools/files/1126263/dump.UBL.1._001.zip)

mefistotelis, 2017-07-05:
Yes there are.

It is good you dumped 3 copies, but you should've compare these copies to find a correct code.

You're missing the beginning in two, and the end in one.

This file combines good parts of the 3:
[dump02.txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1126343/dump02.txt)

MrBurnsAT, 2017-07-06:
THX alot !!!

I managed it!!!
Could both recover!!!
ubl1 and u-boot

here is the response now from the old bricked board!

```
DM36x initialization passed!
UBL Product Vesion : DJI-486M-UBL-1.0-rc0(2015-10-30)
Dji UBL Version: 1.51(Nov  2 2015 - 15:46:28)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
   DONE
Jumping to entry point at 0x81080000.

U-Boot Product Vesion : DJI-DEC-Uboot-1.0-rc0(2015-11-02)
U-Boot 2010.12-rc2-svn3214-Dji (Nov 02 2015 - 19:39:56)
Cores: ARM 486 MHz
DDR:   360 MHz
I2C:   ready
DRAM:  128 MiB
NAND:  128 MiB
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds

Loading from nand0, offset 0x4a0000
** Unknown image type
Wrong Image Format for bootm command
ERROR: can't get kernel image!

Loading from nand0, offset 0x900000
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
## Booting kernel from Legacy Image at 80700000 ...
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Kernel Product Vesion : DJI-GRC-Kernel-1.0-rc8(2014-11-21)
[    0.000000] Linux version 2.6.32.17-davinci1 (root@dji) (gcc version 4.3.3 (Sourcery G++ Lite 2009q1-203) ) #3 PREEMPT Fri Nov 21 12:02:46 HKT 2014
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: DaVinci DM36x EVM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] DaVinci dm36x_rev1.2 variant 0x8
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 12192
[    0.000000] Kernel command line: console=ttyS0,115200n8 rw dm365_imp.oper_mode=0 video=davincifb:vid0=0,10K:vid1=0,10K:osd0=1920x1080X16,8100K mem=48MB davinci_enc_mngr.ch0_output=COMPOSITE davinci_enc_mngr.ch0_mode=pal ubi.mtd=2,2048 root=ubi0:rootfs rootfstype=ubifs ip=off lpj=1077248
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[... 230 lines cut ...]
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547

```

Here are all files which i extracted and bin files for flashing also

[HDMI Final.zip](https://github.com/mefistotelis/phantom-firmware-tools/files/1126429/HDMI.Final.zip)

MrBurnsAT, 2017-07-06:
It Boots the wrong Kernel

Hmmm

MrBurnsAT, 2017-07-06:
No Kernel is ok and the right

Something with USB

mefistotelis, 2017-07-06:
Firmware version doesn't match  the rest of the drone?

MrBurnsAT, 2017-07-06:
[ 3.840000] exFAT: Version 1.2.9 settime failed * daemon not running. starting it now on port 5037 * * daemon started successfully * sulogin: no password entry for root login[486]: root login on 'ttyS0' #Process /etc/profile... root@Dji-Pro:~# encrypt check ok init libusb successed!!! !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 usbsocket_init OK usbsocket_create_handle ok, port:22346, bCanRecv:0 watchdog timeout:5 usbsocket_create_handle ok, port:22345, bCanRecv:1 latency1:33332 latency2:33000 latency3:30000 displaybuffnum1:4 displaybuffnum2:9 start decode stream from 68013 decode init ok, main thread exit !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 Osd_Init_Font succeed [ 12.210000] pinmux1:0x00145555 [ 12.210000] pinmux1:0x00145555 [ 12.270000] pinmux1:0x00145555 [ 12.280000] pinmux1:0x00145555 [ 12.280000] pinmux1:0x00145555 !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 umount: can't umount /mnt/usb: No such file or directory rmdir: '/mnt/usb': No such file or directory !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 DummyRead68013 0.000000kb !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 DummyRead68013 0.000000kb !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 DummyRead68013 0.000000kb !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 [ 15.990000] PCLK = 0 -> 0 [ 15.990000] HTotal = 0 -> 1650 [ 16.000000] VTotal = 0 -> 750 [ 16.000000] ScanMode = I -> P [ 16.000000] PCLK = 0,(1650x750) PROGRESS CHANGE MODE [ 16.010000] VIC = 4 DummyRead68013 0.000000kb !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 DummyRead68013 0.000000kb !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 DummyRead68013 0.000000kb !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 DummyRead68013 0.000000kb !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 DummyRead68013 0.000000kb !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547 !!!!!!!!!!!!!!!!!!!!!!! open 68013 failed VENDOR_ID = 0x0547

MrBurnsAT, 2017-07-06:
Dont know.

Look at the log.

Here Starts the Problem 
[ 3.840000] exFAT: Version 1.2.9

I Think thats because that

 Loading from nand0, offset 0x4a0000
 ** Unknown image type
 Wrong Image Format for bootm command
 ERROR: can't get kernel image!

mefistotelis, 2017-07-06:
> I Think thats because that
ERROR: can't get kernel image!

Well, it finally got the kernel image, as kernel obviously booted.
I'd try to update the firmware via mobile app now. I'm pretty sure the version is now the only remaining issue.

MrBurnsAT, 2017-07-06:
Cant Connect GoApp! Dont get any Connection.

Only like bevore with App for Lightbridge it will Connect.

Thats missing at my Unit

[ 8.140000] usb aoa module init ok 
[ 8.190000] usbcore: registered new interface driver usb_68013 
[ 8.220000] usb 68013 module init ok sulogin: no password entry for root Dji-Pro login: encrypt check ok 
[ 8.850000] usbcore: registered new interface driver usb_applehid 
[ 8.870000] usb applehid module init ok init libusb successed!!! usbmuxd disabled usbsocket_init OK 
[ 8.900000] usbcore: registered new interface driver usb_aoa 
[ 8.920000] aoa start ok 
usbsocket_create_handle ok, port:22345, bCanRecv:1 usbsocket_create_handle ok, port:22346, bCanRecv:0

Mavic129, 2017-07-06:
@MrBurnsAT my friend you can extract the board firmware using @mefistotelis tools then decrypt the salted file with the well known password, then decompress the files and remove the first 800 bytes from the uboot and ubl files
after that you are good to flash these two files to your board then regular firmware update
note: latest firmware update makes the ssh to the DM368 requires password which i don't know

MrBurnsAT, 2017-07-06:
Hmmmm

I know how to extract the Board Firmware ist Module m1300. Have already extracted the bins.

But dont know how to decrypt and dont have this well known password. Saw it i Think in an Post of notsolowki.

Could u explain me how to decrypt?

Mavic129, 2017-07-07:
openssl des3 -d -k Dji123456 -in enc.bin -out decrypted.tar.gz

in the decrypted file you will find /dji/data/u-boot.img and /dji/data/ubl_xxx_m_x.img

extract the u-boot.img and ubl_297m_1.img

open then in hex editor and remove the first 0x800 bytes
then flash them using sfh flash tool
sfh_DM36x.exe -nandflash -v -p "COM5" ubl1.img u-boot.img

MrBurnsAT, 2017-07-07:
Many Thanks!!!

Ill give that a try 

[image]
[image]

MrBurnsAT, 2017-07-07:
Thanks could manage it with mi09.bin from 1.3.2

Thanks again for the help.

Next Problem.
Tried to decrypt with same Methode
mi17 from 1.3.2 but got wrong Magic Key Message

Do u know some others Passwords or Magic Values which i can try?

MrBurnsAT, 2017-07-07:
Mi17 is for m2000 fpga ground 

What i know would this Too be a DM368 
So i want this ubl1 and u-boot images too

mefistotelis, 2017-07-07:
> Mi17 is for m2000 fpga ground
What i know would this Too be a DM368

This makes little sense. We have wikis on all modules, so you should always read what we already know before trying to get further:
https://github.com/mefistotelis/phantom-firmware-tools/wiki/Firmware-m2000

Specifications of Altera bitstream container format was open at some point, but there were many modifications since. Still, it would be interesting to try get into it.

MrBurnsAT, 2017-07-07:
What i found out Now.

M1600 is for USB Controller on the rc.

Exactly this makes the Problem for me.

Do u Think is an Firmware Problem.
Or are only sent wrong identification from u-boot via the USB?

Because its Working. Only wrong.
Because it sends identification as it would be a Lightbridge 1.
Ans Not like P3 or P4 RC

MrBurnsAT, 2017-07-07:
Exported/decrypted now mi09.bin (m1300) and mi22.bin (m0800) from 1.3.2 firmware.

and edited the files (delete first 0x800)

here for all to test
[UBOOTs for P3.zip](https://github.com/mefistotelis/phantom-firmware-tools/files/1130554/UBOOTs.for.P3.zip)

MrBurnsAT, 2017-07-07:
Shit. Same Problem with ubl1 and u-boot images from Firmware.

So it must be a Problem with CY7C68013 (m1600)

MrBurnsAT, 2017-07-07:
I found out that i can send a "stop" Command via serial After Linux is booted. Than u LL come in Root prompt and Can Work with it.

But!!!!

In my New Unit which Works.
I can send the "Stop" Command.
But After that it ask me "Password:"

So on the newer Unit i cant Login in Linux Promt

MrBurnsAT, 2017-07-07:
extracted from RC Firmware V1.01.0004 the mi00 which is for m1301

thats NOT for DM365!!!
thats for "ti810x"

annexed the decrypted file
[decrypted_1301.tar.gz](https://github.com/mefistotelis/phantom-firmware-tools/files/1132560/decrypted_1301.tar.gz)

mefistotelis, 2017-07-07:
I checked inside, and it seem to support both DM8107 and DM385.

But yes, definitely not for DM365.

I have no idea which product uses it.

MrBurnsAT, 2017-07-08:
Cendence Remote maybe with CrystalSky Monitor?
(But at so old Firmwares????)

Or Maybe Lightbridge 2 with SDI out should can do this too i Think.

MrBurnsAT, 2017-07-08:
PS i tried to Flash it but without success.

MrBurnsAT, 2017-07-09:
Look at Update Version list for LB2

m1301 must be for LB2!!!

```
"versionlistlb2":
	[	
		{"date":"1465198573",
        "version":"1.1.30",
		"m2003":"01.00.09.18&0",
		"m1300":"02.26.00.00&0",
		**"m1301":"01.00.00.28&0",**
        "m1400":"04.10.00.15&0",
        "m1501":"01.01.02.00&0",
		"m1600":"01.03.01.00&0",
		"m1900":"01.00.09.18&0",
		"m0801":"01.00.00.24&0",
        "m0900":"02.01.00.02&0",
        "rcversion":"1.1.30",
		"rc1url":"https://dl.djicdn.com/downloads/lightbridge2/C1_FW_V01.01.0030.bin"
		},
		
		{"date":"1459935590",
        "version":"1.1.10",
		**"m1301":"01.00.00.26&0",**
        "m1400":"04.10.00.06&0",
        "m1501":"01.01.02.00&0",
		"m1600":"01.03.01.00&0",
		"m1900":"01.00.09.18&0",
		"m0801":"01.00.00.20&0",
        "m0900":"02.00.00.03&0",
        "rcversion":"1.1.0",
		"rc1url":"https://dl.djicdn.com/downloads/lightbridge2/C1_FW_V01.01.0000.bin"
		},
	
	
		{"date":"1449498540",
        "version":"1.0.10",
        "m2000":"01.00.09.18&0",
		**"m1301":"01.00.00.18&0",**
        "m1400":"04.09.03.26&0",
        "m1501":"01.01.02.00&0",
		"m1600":"01.03.01.00&0",
		"m1900":"01.00.09.18&0",
		"m0801":"01.00.00.18&0",
        "m0900":"01.03.00.08&0",
        "rcversion":"1.0.10",
		"rc1url":"http://dl.djicdn.com/downloads/lightbridge2/C1_FW_V01.00.0010.bin"
		},
	
		{"date":"1449497791",
        "version":"1.0.2",
        "m2000":"01.00.09.18&0",
		**"m1301":"01.00.00.18&0",**
        "m1400":"04.09.03.26&0",
        "m1501":"01.01.02.00&0",
		"m1600":"01.03.01.00&0",
		"m1900":"01.00.09.18&0",
		"m0801":"01.00.00.17&0",
        "m0900":"01.03.00.08&0",
        "rcversion":"1.0.10",
		"rc1url":"http://dl.djicdn.com/downloads/lightbridge2/C1_FW_V01.00.0010.bin"
		},
	
		{"date":"1444359306",
        "version":"1.0.1",
        "m2000":"01.00.09.17&0",
		**"m1301":"01.00.00.12&0",**
        "m1400":"04.09.03.16&0",
        "m1501":"01.01.02.00&0",
		"m1600":"01.03.01.00&0",
		"m1900":"01.00.09.17&0",
		"m0801":"01.00.00.17&0",
        "m0900":"01.03.00.08&0",
        "rcversion":"1.0.4",
		"rc1url":"http://download.dji-innovations.com/downloads/lightbridge2/C1_FW_V01.00.0004.bin"
		}	
	],
```

MrBurnsAT, 2017-07-09:
thats the real problem

**[    8.140000] usb aoa module init ok
[    8.190000] usbcore: registered new interface driver usb_68013
[    8.220000] usb 68013 module init ok
sulogin: no password entry for root

Dji-Pro login: encrypt check ok
[    8.850000] usbcore: registered new interface driver usb_applehid
[    8.870000] usb applehid module init ok
init  libusb successed!!!
usbmuxd disabled
usbsocket_init OK
[    8.900000] usbcore: registered new interface driver usb_aoa
[    8.920000] aoa start ok**

this would not be done on my HDMI Board after flashing

Mavic129, 2017-07-10:
My friend

you are loading very old version, try to extract u-boot and ubl from newer version like 1.6 or 1.5.8

MrBurnsAT, 2017-07-10:
Ill try it.

But i bet that wont Change anything

MrBurnsAT, 2017-07-10:
I tried this old versions because my RC is on 1.3.2

MrBurnsAT, 2017-07-10:
Exactly the same with extracted from C1_FW_V01.05.0070

UBL Product Vesion : DJI-486M-UBL-1.0-rc0(2015-10-30)
Dji UBL Version: 1.51(Nov  2 2015 - 15:46:28)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
   DONE
Jumping to entry point at 0x81080000.

```
U-Boot Product Vesion : DJI-DEC-Uboot-1.0-rc0(2015-11-02)
U-Boot 2010.12-rc2-svn3214-Dji (Nov 02 2015 - 19:39:56)
Cores: ARM 486 MHz
DDR:   360 MHz
I2C:   ready
DRAM:  128 MiB
NAND:  128 MiB
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds

Loading from nand0, offset 0x4a0000
** Unknown image type
Wrong Image Format for bootm command
ERROR: can't get kernel image!

Loading from nand0, offset 0x900000
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
## Booting kernel from Legacy Image at 80700000 ...
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Kernel Product Vesion : DJI-GRC-Kernel-1.0-rc8(2014-11-21)
[    0.000000] Linux version 2.6.32.17-davinci1 (root@dji) (gcc version 4.3.3 (Sourcery G++ Lite 2009q1-203) ) #3 PREEMPT Fri Nov 21 12:02:46 HKT 2014
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: DaVinci DM36x EVM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] DaVinci dm36x_rev1.2 variant 0x8
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 12192
[    0.000000] Kernel command line: console=ttyS0,115200n8 rw dm365_imp.oper_mode=0 video=davincifb:vid0=0,10K:vid1=0,10K:osd0=1920x1080X16,8100K mem=48MB davinci_enc_mngr.ch0_output=COMPOSITE davinci_enc_mngr.ch0_mode=pal ubi.mtd=2,2048 root=ubi0:rootfs rootfstype=ubifs ip=off lpj=1077248
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 48MB = 48MB total
[    0.000000] Memory: 44464KB available (3632K code, 444K data, 116K init, 0K highmem)
[    0.000000] SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] Console: colour dummy device 80x30
[    0.000000] Calibrating delay loop (skipped) preset value.. 215.44 BogoMIPS (lpj=1077248)
[... 212 lines cut ...]
[   16.050000] PCLK = 0 -> 0
[   16.050000] HTotal = 0 -> 1650
[   16.060000] VTotal = 0 -> 750
[   16.060000] ScanMode = I -> P
[   16.060000] PCLK = 0,(1650x750) PROGRESS CHANGE MODE
[   16.070000] VIC = 4
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547

```

MrBurnsAT, 2017-07-10:
Exactly the same with extracted from C1_FW_V01.05.0071

```
DM36x initialization passed!
UBL Product Vesion : DJI-486M-UBL-1.0-rc0(2015-10-30)
Dji UBL Version: 1.51(Nov  2 2015 - 15:46:28)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
   DONE
Jumping to entry point at 0x81080000.

U-Boot Product Vesion : DJI-DEC-Uboot-1.0-rc0(2015-11-02)
U-Boot 2010.12-rc2-svn3214-Dji (Nov 02 2015 - 19:39:56)
Cores: ARM 486 MHz
DDR:   360 MHz
I2C:   ready
DRAM:  128 MiB
NAND:  128 MiB
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds

Loading from nand0, offset 0x4a0000
** Unknown image type
Wrong Image Format for bootm command
ERROR: can't get kernel image!

Loading from nand0, offset 0x900000
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
## Booting kernel from Legacy Image at 80700000 ...
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Kernel Product Vesion : DJI-GRC-Kernel-1.0-rc8(2014-11-21)
[    0.000000] Linux version 2.6.32.17-davinci1 (root@dji) (gcc version 4.3.3 (Sourcery G++ Lite 2009q1-203) ) #3 PREEMPT Fri Nov 21 12:02:46 HKT 2014
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: DaVinci DM36x EVM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] DaVinci dm36x_rev1.2 variant 0x8
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 12192
[    0.000000] Kernel command line: console=ttyS0,115200n8 rw dm365_imp.oper_mode=0 video=davincifb:vid0=0,10K:vid1=0,10K:osd0=1920x1080X16,8100K mem=48MB davinci_enc_mngr.ch0_output=COMPOSITE davinci_enc_mngr.ch0_mode=pal ubi.mtd=2,2048 root=ubi0:rootfs rootfstype=ubifs ip=off lpj=1077248
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 48MB = 48MB total
[    0.000000] Memory: 44464KB available (3632K code, 444K data, 116K init, 0K highmem)
[    0.000000] SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] Console: colour dummy device 80x30
[... 210 lines cut ...]
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
[   16.030000] PCLK = 0 -> 0
[   16.030000] HTotal = 0 -> 1650
[   16.030000] VTotal = 0 -> 750
[   16.040000] ScanMode = I -> P
[   16.040000] PCLK = 0,(1650x750) PROGRESS CHANGE MODE
[   16.050000] VIC = 4
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547

```

MrBurnsAT, 2017-07-10:
Exactly the same with extracted from C1_FW_V01.05.0080

```
DM36x initialization passed!
UBL Product Vesion : DJI-486M-UBL-1.0-rc0(2015-10-30)
Dji UBL Version: 1.51(Nov  2 2015 - 15:46:28)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
   DONE
Jumping to entry point at 0x81080000.

U-Boot Product Vesion : DJI-DEC-Uboot-1.0-rc0(2015-11-02)
U-Boot 2010.12-rc2-svn3214-Dji (Nov 02 2015 - 19:39:56)
Cores: ARM 486 MHz
DDR:   360 MHz
I2C:   ready
DRAM:  128 MiB
NAND:  128 MiB
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds

Loading from nand0, offset 0x4a0000
** Unknown image type
Wrong Image Format for bootm command
ERROR: can't get kernel image!

Loading from nand0, offset 0x900000
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
## Booting kernel from Legacy Image at 80700000 ...
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Kernel Product Vesion : DJI-GRC-Kernel-1.0-rc8(2014-11-21)
[    0.000000] Linux version 2.6.32.17-davinci1 (root@dji) (gcc version 4.3.3 (Sourcery G++ Lite 2009q1-203) ) #3 PREEMPT Fri Nov 21 12:02:46 HKT 2014
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: DaVinci DM36x EVM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] DaVinci dm36x_rev1.2 variant 0x8
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 12192
[    0.000000] Kernel command line: console=ttyS0,115200n8 rw dm365_imp.oper_mode=0 video=davincifb:vid0=0,10K:vid1=0,10K:osd0=1920x1080X16,8100K mem=48MB davinci_enc_mngr.ch0_output=COMPOSITE davinci_enc_mngr.ch0_mode=pal ubi.mtd=2,2048 root=ubi0:rootfs rootfstype=ubifs ip=off lpj=1077248
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 48MB = 48MB total
[    0.000000] Memory: 44464KB available (3632K code, 444K data, 116K init, 0K highmem)
[    0.000000] SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] Console: colour dummy device 80x30
[... 205 lines cut ...]
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
[   16.040000] PCLK = 0 -> 0
[   16.040000] HTotal = 0 -> 1650
[   16.050000] VTotal = 0 -> 750
[   16.050000] ScanMode = I -> P
[   16.050000] PCLK = 0,(1650x750) PROGRESS CHANGE MODE
[   16.060000] VIC = 4
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547

```

MrBurnsAT, 2017-07-10:
same with 1.06.0000

```
DM36x initialization passed!
UBL Product Vesion : DJI-486M-UBL-1.0-rc0(2015-10-30)
Dji UBL Version: 1.51(Nov  2 2015 - 15:46:28)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
   DONE
Jumping to entry point at 0x81080000.

U-Boot Product Vesion : DJI-DEC-Uboot-1.0-rc0(2015-11-02)
U-Boot 2010.12-rc2-svn3214-Dji (Nov 02 2015 - 19:39:56)
Cores: ARM 486 MHz
DDR:   360 MHz
I2C:   ready
DRAM:  128 MiB
NAND:  128 MiB
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds

Loading from nand0, offset 0x4a0000
** Unknown image type
Wrong Image Format for bootm command
ERROR: can't get kernel image!

Loading from nand0, offset 0x900000
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
## Booting kernel from Legacy Image at 80700000 ...
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Kernel Product Vesion : DJI-GRC-Kernel-1.0-rc8(2014-11-21)
[    0.000000] Linux version 2.6.32.17-davinci1 (root@dji) (gcc version 4.3.3 (Sourcery G++ Lite 2009q1-203) ) #3 PREEMPT Fri Nov 21 12:02:46 HKT 2014
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: DaVinci DM36x EVM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] DaVinci dm36x_rev1.2 variant 0x8
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 12192
[    0.000000] Kernel command line: console=ttyS0,115200n8 rw dm365_imp.oper_mode=0 video=davincifb:vid0=0,10K:vid1=0,10K:osd0=1920x1080X16,8100K mem=48MB davinci_enc_mngr.ch0_output=COMPOSITE davinci_enc_mngr.ch0_mode=pal ubi.mtd=2,2048 root=ubi0:rootfs rootfstype=ubifs ip=off lpj=1077248
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 48MB = 48MB total
[    0.000000] Memory: 44464KB available (3632K code, 444K data, 116K init, 0K highmem)
[    0.000000] SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] Console: colour dummy device 80x30
[... 207 lines cut ...]
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
[   16.020000] PCLK = 0 -> 0
[   16.020000] HTotal = 0 -> 1650
[   16.030000] VTotal = 0 -> 750
[   16.030000] ScanMode = I -> P
[   16.030000] PCLK = 0,(1650x750) PROGRESS CHANGE MODE
[   16.040000] VIC = 4
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547

```

MrBurnsAT, 2017-07-10:
same with 1.03.00.21

```
DM36x initialization passed!
UBL Product Vesion : DJI-GRC-P3-SUPER-UBL-1.0-rc0(2015-05-12)
Dji UBL Version: 1.51(May 12 2015 - 11:29:40)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
   DONE
Jumping to entry point at 0x81080000.

U-Boot Product Vesion : DJI-GRC-Uboot-1.0-rc1(2015-01-27)
U-Boot 2010.12-rc2-svn-Dji (Jan 27 2015 - 18:45:11)
Cores: ARM 486 MHz
DDR:   360 MHz
I2C:   ready
DRAM:  128 MiB
NAND:  128 MiB
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds

Loading from nand0, offset 0x4a0000
** Unknown image type
Wrong Image Format for bootm command
ERROR: can't get kernel image!

Loading from nand0, offset 0x900000
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
## Booting kernel from Legacy Image at 80700000 ...
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Kernel Product Vesion : DJI-GRC-Kernel-1.0-rc8(2014-11-21)
[    0.000000] Linux version 2.6.32.17-davinci1 (root@dji) (gcc version 4.3.3 (Sourcery G++ Lite 2009q1-203) ) #3 PREEMPT Fri Nov 21 12:02:46 HKT 2014
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: DaVinci DM36x EVM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] DaVinci dm36x_rev1.2 variant 0x8
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 12192
[    0.000000] Kernel command line: console=ttyS0,115200n8 rw dm365_imp.oper_mode=0 video=davincifb:vid0=0,10K:vid1=0,10K:osd0=1920x1080X16,8100K mem=48MB davinci_enc_mngr.ch0_output=COMPOSITE davinci_enc_mngr.ch0_mode=pal ubi.mtd=2,2048 root=ubi0:rootfs rootfstype=ubifs ip=off lpj=1077248
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 48MB = 48MB total
[    0.000000] Memory: 44464KB available (3632K code, 444K data, 116K init, 0K highmem)
[    0.000000] SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] Console: colour dummy device 80x30
[... 211 lines cut ...]
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
[   15.980000] PCLK = 0 -> 0
[   15.980000] HTotal = 0 -> 1650
[   15.990000] VTotal = 0 -> 750
[   15.990000] ScanMode = I -> P
[   15.990000] PCLK = 0,(1650x750) PROGRESS CHANGE MODE
[   16.000000] VIC = 4
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547

```

MrBurnsAT, 2017-07-10:
same with 1.03.0020

MrBurnsAT, 2017-07-10:
LITTLE BIT Different but also not working 1.04.0030

```
UBL Product Vesion : DJI-GRC-P3-SUPER-UBL-1.0-rc0(2015-05-12)
Dji UBL Version: 1.51(May 12 2015 - 11:29:40)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
   DONE
Jumping to entry point at 0x81080000.

U-Boot Product Vesion : DJI-GRC-Uboot-1.0-rc1(2015-01-27)
U-Boot 2010.12-rc2-svn-Dji (Jan 27 2015 - 18:45:11)
Cores: ARM 486 MHz
DDR:   360 MHz
I2C:   ready
DRAM:  128 MiB
NAND:  128 MiB
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds

Loading from nand0, offset 0x4a0000
** Unknown image type
Wrong Image Format for bootm command
ERROR: can't get kernel image!

Loading from nand0, offset 0x900000
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
## Booting kernel from Legacy Image at 80700000 ...
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Kernel Product Vesion : DJI-GRC-Kernel-1.0-rc8(2014-11-21)
[    0.000000] Linux version 2.6.32.17-davinci1 (root@dji) (gcc version 4.3.3 (Sourcery G++ Lite 2009q1-203) ) #3 PREEMPT Fri Nov 21 12:02:46 HKT 2014
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: DaVinci DM36x EVM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] DaVinci dm36x_rev1.2 variant 0x8
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 12192
[    0.000000] Kernel command line: console=ttyS0,115200n8 rw dm365_imp.oper_mode=0 video=davincifb:vid0=0,10K:vid1=0,10K:osd0=1920x1080X16,8100K mem=48MB davinci_enc_mngr.ch0_output=COMPOSITE davinci_enc_mngr.ch0_mode=pal ubi.mtd=2,2048 root=ubi0:rootfs rootfstype=ubifs ip=off lpj=1077248
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 48MB = 48MB total
[    0.000000] Memory: 44464KB available (3632K code, 444K data, 116K init, 0K highmem)
[    0.000000] SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] Console: colour dummy device 80x30
[    0.000000] Calibrating delay loop (skipped) preset value.. 215.44 BogoMIPS (lpj=1077248)
[... 213 lines cut ...]
[   16.040000] HTotal = 0 -> 1650
[   16.050000] VTotal = 0 -> 750
[   16.050000] ScanMode = I -> P
[   16.050000] PCLK = 0,(1650x750) PROGRESS CHANGE MODE
[   16.060000] VIC = 4
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547

```

MrBurnsAT, 2017-07-10:
same with 1.02.0021

```
DM36x initialization passed!
UBL Product Vesion : DJI-GRC-P3-SUPER-UBL-1.0-rc0(2015-05-12)
Dji UBL Version: 1.51(May 12 2015 - 11:29:40)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
   DONE
Jumping to entry point at 0x81080000.

U-Boot Product Vesion : DJI-GRC-Uboot-1.0-rc1(2015-01-27)
U-Boot 2010.12-rc2-svn-Dji (Jan 27 2015 - 18:45:11)
Cores: ARM 486 MHz
DDR:   360 MHz
I2C:   ready
DRAM:  128 MiB
NAND:  128 MiB
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds

Loading from nand0, offset 0x4a0000
** Unknown image type
Wrong Image Format for bootm command
ERROR: can't get kernel image!

Loading from nand0, offset 0x900000
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
## Booting kernel from Legacy Image at 80700000 ...
   Image Name:   Linux-2.6.32.17-davinci1
   Created:      2014-11-21   4:02:48 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4024480 Bytes = 3.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Kernel Product Vesion : DJI-GRC-Kernel-1.0-rc8(2014-11-21)
[    0.000000] Linux version 2.6.32.17-davinci1 (root@dji) (gcc version 4.3.3 (Sourcery G++ Lite 2009q1-203) ) #3 PREEMPT Fri Nov 21 12:02:46 HKT 2014
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: DaVinci DM36x EVM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] DaVinci dm36x_rev1.2 variant 0x8
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 12192
[    0.000000] Kernel command line: console=ttyS0,115200n8 rw dm365_imp.oper_mode=0 video=davincifb:vid0=0,10K:vid1=0,10K:osd0=1920x1080X16,8100K mem=48MB davinci_enc_mngr.ch0_output=COMPOSITE davinci_enc_mngr.ch0_mode=pal ubi.mtd=2,2048 root=ubi0:rootfs rootfstype=ubifs ip=off lpj=1077248
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 48MB = 48MB total
[    0.000000] Memory: 44464KB available (3632K code, 444K data, 116K init, 0K highmem)
[    0.000000] SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] Console: colour dummy device 80x30
[... 208 lines cut ...]
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
[   16.040000] PCLK = 0 -> 0
[   16.040000] HTotal = 0 -> 1650
[   16.050000] VTotal = 0 -> 750
[   16.050000] ScanMode = I -> P
[   16.050000] PCLK = 0,(1650x750) PROGRESS CHANGE MODE
[   16.060000] VIC = 4
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
DummyRead68013 0.000000kb
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547
!!!!!!!!!!!!!!!!!!!!!!! open 68013 failed   VENDOR_ID = 0x0547

```

ruckusman, 2017-09-04:
This may sound obvious, however have you tried the 4 button reset followed by the GS_OFDM.bin on a usb key?
From there you may be able to update to another version on either a USB key if the App doesn't recognise the controller via USB.

I have recovered a failed HDMI board from an Inspire with precisely this technique, which would give no acknowledgement of connection to any tablets via USB

rufiooo, 2017-09-20:
Any updates in this? I've been trying to use USB key with GS_OFDM.bin but still no luck, update won't start.

ruckusman, 2017-09-21:
Have you done the 4 button reset prior to using the GS_OFDM.bin?

rufiooo, 2017-09-21:
Yup, (C1 + C2 + video rec + right jog dial) press and hold, then turn off and turn on controller.

ruckusman, 2017-09-21:
OK that should yield a white status LED - in the case of two Inspire remotes I had to recover recently the status LED went more cyan than white.

If you do that procedure, are you then able to connect to an Andriod device running the APP and use the downgrade function?

I will counsel against using an IOS device for remote firmware upgrades or downgrades because the two Inspire remotes were corrupted because of using an IOS device.

rufiooo, 2017-09-21:
I was trying everything - all buttons combinations, all BIN file versions. Controller is ok, because when I replaced HDMI module from master controller it was OK. So I updated firmware, connected to DJI GO etc. But as soon as I replaced this module back to original one, controller stoped to respond on USB (sticks and devices).

ruckusman, 2017-09-21:
Only thing I can offer is that the HDMI board that I recovered may have been a recoverable error with my process, though the symptoms of not registering on the USB port is a common symptom, it may have different root causes.

What colour is the status LED when you power the remote normally - without and buttons pressed?

rufiooo, 2017-09-21:
LED is red, when I power up craft than it becomes green and operates as it should (I can start motors etc). Just USB and HDMI ports are dead.

ruckusman, 2017-09-21:
OK then it sounds as though you need to re-initiate the process, try the four button reset followed by loading the GS_OFMDM.bin on a USB key.

From that point depending on which version is cached as to it will respond to firmware on a SUB key or needs to be connected to a tablet or phone. 

I don't recall if there was recognition on the device after resetting - using the downgrade on the device will inform you if the device recognises it and which versions are available to downgrade to, before then upgrading again

ruckusman, 2017-09-22:
I was just downgrading a remote and craft and have a simplified process for you

Put the GS_OFDM.bin on a USB key and with the remote turned off put it into the USB.

Power on the remote normally - then do the four button press, after up to 10 seconds the status LED will turn white and it should then proceed to load the GS_OFDM.bin from the USB key.

Let that process complete, it will pause several times during the process.

Once finished power off the remote, remove USB key, power back on and connect device.

Then using the downgrade option from the splash screen to reload firmware.

One tip, if the large ribbon cable isn't inserted correctly the C1 & C2 buttons will not work correctly.

Other than the above process failing to recover the HDMI board I cannot suggest any further actions

hope this helps

rufiooo, 2017-09-22:
No luck with this, USB is dead no matter what I do. But just another thought: what about microUSB port located next to HDMI? When I plug it into my desktop then it starts to install drivers - so this port works. And when I connect it to my android tablet via OTG USB adaptor then SimpleBGC (installed for alexmos gimbal config) app is trying to use this port.

Is there any option to use this microUSB to update firmware on I1 controller? I was trying to put in it OTG with pendrive with different bin files and start 4 button sequence but without any effect.

ruckusman, 2017-09-24:
The only reference I ever saw of the other USB was it being listed in a user manual for future expansion - that was many years ago and I've never seen or read of anything being manufactured for it since then.

There's a bunch of serial connection on the upper side of the HDMI Module, I can't offer any guidance on their use to you, however this thread delves into the intricacies of flashing firmware onto lightbridge modules with errors and may be a good starting place. It mentions bridging the BSEL and 3v3 pins on a lightbridge unit, both of which are also present on the HDMI board, which is a good thing.

https://phantompilots.com/threads/lightbridge-firmware-problem-and-solution-firmware-version-not-found-by-dji-assistant-tool.77073/

mefistotelis, 2017-10-30:
Back to what @MrBurnsAT tried to achieve - I think I get it now.

The firmware on the NAND flash consists of:
- u-boot bootloader partition
- ubifs partition
- kernel image
- failsafe kernel image

With the commands @MrBurnsAT used, he fixed u-boot and ubifs. The failsafe kernel image was functional, so he was also able to boot the board. But primary kernel image was damaged.
Since the failsafe kernel is based on lightbridge firmware, the device was recognized as lightbridge and acted as host over USB instead of as client.

We need a way to flash the primary kernel.

ruckusman, 2017-10-30:
This may or may not be of assistance - I had four Inspire remotes I was modifying to enable the Can bus port for use with the focus handwheel. 

Three went smoothly, the fourth remote had the same connection issues on the USB port before I began the modification, it would intermittently connect on the USB, but only one in 10 reboots at best.

I tried switching the module, which had corrupted firmware and was easily recovered on another remote with the 4 button reset.
On the troublesome remote the HDMI board(s) from other remotes worked intermittently.

The simplest way to determine if the USB port was active or not was to check for the ~5 volts from the main board.

It ended up being poor connection across the ribbon cable - it was difficult to determine which end.

The solution was add some thin sticky tape on the end to increase it's thickness marginally - I would counsel thinner tape rather than thicker as a first step so as not to stress the connector itself.

So the takeaway is to check for the 5 volts going to the USB to see if the issue is connection or firmware related

mefistotelis, 2017-10-30:
Ahh f#!&k. I found a way to write the kernel, but I used size as decimal integer - and it turned out the "nand write" command arbitrary decided it was a hex number. I cleared the part of flash (ubifs) which contained private key for encryption.

[61 more replies]