o-gs/dji-firmware-tools #41: Switching power zone to FCC via mobile app

mefistotelis, 2017-07-07:
Creating separate issue for FCC set by Dji Go to avoid mixing subjects.

 @MrBurnsAT commented on Jul 3
>
[image]
> Mefistotelis Maybe from go4 App if u Switch to FCC?

 @mefistotelis commented on Jul 4
>
>No idea what you mean. Can you use Dji Go 4 with Ph3?

 @MrBurnsAT commented on Jul 4
>
>Yes Go4 App Cam Work with p3.
>Little bit tricky but works. But i dont know exactly how i did it. Must try more
>
>GO4 App with P3
>[image]
>[image]

 @mingtaoxin commented on Jul 4
>
>how you can do it ? can you make some explanation ?

>@KyokushinPL commented on Jul 4
>
>i connected only to radio by this app:
>https://forums.hak5.org/index.php?/topic/41354-remove-ce-limits-from-dji->products/&do=findComment&comment=293797
>and i was able to set  it to US.
>
>but when i am trying to turn on the bird, then application is saying me i should use old dji go and i cant do anything.
>Maybe someone would be able to meter power after changing radio to US.
>
>I am also curious how long this change will affect - to next turn on, how it will be in litchi after turning off dji go 4.
>
>@MrBurnsAT  what version do you use?
>
>Update:
>OK folks, it can be done by the same method as in first link. I decompiled DJI go app for p3, there are same functions for country code. Its working same way, but i am too weak in java to put it together.

 @WindyYam commented on Jul 5
>
>ok, now i know where everything goes. I first post the DJI Go trick on Chinese dji forum on about 6.28 then i dont know who carry to here. It's done by tap flyc serial number view several times
>
>on disconnect the dji go app check TelephoneManager.getNetworkCountryIso() on android for mcc to set the RF power, a easy way is to set  gsm.operator.iso-country properties to "us" when mobile network is disabled
>
>but i dont know if it's buggy that mavic boost the rc power not the videofeed, when other drones boost both

 @MrBurnsAT commented on Jul 5
>Which Version of go App do u use?
>
>Go4 ?
>Need GoApp for Phantom 3

 @WindyYam commented on Jul 5
>
>The trick exists before GO 4.0.7 , dont know if its there for Go 3
>
>for Phantom user maybe the only way is to set gsm.operator.iso-country to us in root mode, and disable the phone network before start dji go, i have an app for this purpose but need Android root , dont know if you can download this
>
>https://pan.baidu.com/s/1jHPQdO6

@MrBurnsAT commented on Jul 5
>
>Thx could Download it.
>
>Ok. Than i must Root my Nvidia Shield

@MrBurnsAT commented on Jul 5
>
> Saw something like this in the dji Mobile sdk

mefistotelis, 2017-07-07:
Moving related posts.

@WindyYam commented on Jul 7
>hello guys, just in case you dont know, i create an dji config for switch to FCC for dji go and go4. 
>https://pan.baidu.com/s/1bP0CX4
>I dejava the dji go and found the reading sequence, looks like a back door DJI put in the code.
>
>also post on https://forums.hak5.org/index.php?/topic/41380-dji-configs-for-fcc-boost-and-32-channel/

@CirConFleX commented on Jul 7
> It's working with DJI Go 3 or only version 4 ?

@WindyYam commented on Jul 7
>i've found the config parser both in Go 3 and Go 4 Android, Think it might be able for both. 
There is also a SDR force boost version, but i'm not sure if it will do any damage to the sdr device, I got a better  signal for videofeed on my mavic pro, try at your own risk
download https://pan.baidu.com/s/1sljUQlv password: 9ajb

 @mefistotelis commented  on Jul 7
> @WindyYam, have you tested the RF power on any platform and confirmed that this config file really does anything?

 @WindyYam commented on Jul 7
> sorry i dont have a rf meter but im pretty sure the power is boosted on mavic and p3a

 @mefistotelis commented
> If you don't want to test it yourself, you should ask if someone is willing to do this. You may create a separate issue here to discuss your config and its impact. As long as the result is not confirmed, you should advertise it correctly, to avoid misleading the community.

 @pawelsky commented  on Jul 7
>Tested and I **DO NOT** confrim that (at least for GL300B, P3A, DJI Go 3.1.4/3.1.8). RF meters does not show any power increase on the GLB300B (tried both configs)

@CirConFleX commented  on Jul 7
> Thanks for testing !

 @WindyYam commented on Jul 7
> ok forget about my misleading post  i'll deal with it  later. just to tell you there is a dji config parser  located in package dji.pilot.public.c.a,  which deal with openallchannel,  updateurl, and sdr config. the sdr config is the last byte of this config , flag 0x01 is a sdrwrite assist which deals fcc force i think, flag 0x02 is the  sdr force boost flag, flag 0x04 and 0x08 currently unknown
>
> in my test with these flags i do have a signal improvement, i'll do a full test when i'm back home

pawelsky, 2017-07-07:
OK, so for now no solid evidence it has any effect on P3A and/or DJI Go 3.

Regarding Mavic and DJI Go 4 - what was the distance difference you observed with and without boost? What was the distance difference between CE and FCC setting? What was the distance difference with both?

WindyYam, 2017-07-07:
for what we observed, in stock settings we tested at each home and see about getting 1400m and  500m when the rc signal got empty, while there are still some video signal. 
switch to fcc, i tested not him , in both config way and countrycode way, i got enhanced rc signal to 2000m then video is empty.   there are other guys with mcc way got the same result
use the sdr boost config, i have 2800m with both get 3 signal marks, he has 1800 with 3 signal marks both.
and yeah we all live in shenzhen with bad signal interfernce. i have a reflect board on wavelength , he have not

pawelsky, 2017-07-07:
I'm assuming all tests with the same altitude?

WindyYam, 2017-07-07:
im pretty sure it's on the same altitude, the config and mcc can be set dynamicly, but seems cant set too much time or with too weak signal

pawelsky, 2017-07-07:
> but seems cant set too much time
Not sure if I understand what you mean, does it only work for a limited time after setting?

WindyYam, 2017-07-07:
no it means cant switch back and forth too much time

pawelsky, 2017-07-07:
In total, or within one session?

WindyYam, 2017-07-07:
in one session of cause

pawelsky, 2017-07-07:
OK, thanks for sharing.

DKoro1, 2017-07-14:
Hello guys. I dug for a couple of days and was able to implement Go4.1.3 for my p3p. I do not know if this will be useful. I have not flown yet. But I noticed one problem: if an sd card is inserted, the video signal is terminated abruptly, if the sd card is missing then everything works.

https://drive.google.com/file/d/0B7u2nMAlV1QxY05BeGdwSm1UNlk/view?usp=drivesdk

1) Turn on the remote control
2) connect Android, run Go4
3) enter the menu
4) include the drone

MrBurnsAT, 2017-07-14:
I Made it Work so.

1. Turn on the remote control
2. connect Android, run Go4
3. Turn in Drone (now message appears that p3 is connected and u must Switch to GoApp)
4. Turn Off Drone (now u see Phantom 4 Remote connected)
5. Klick Enter Device, go to Menue and Change CE to FCC Mode.
6. Turn in Drone again

Now u should have live View and so on and u should be able to Start the Drone

KyokushinPL, 2017-07-15:
Its not working. Each time i am turning on drone then i am getting a block screen with info to switch to dji go 3.

DKoro1, 2017-07-15:
On my apk?

KyokushinPL, 2017-07-15:
No, on modded 4.1.3 v22. Your is different?

DKoro1, 2017-07-15:
https://drive.google.com/file/d/0B7u2nMAlV1QxY05BeGdwSm1UNlk/view?usp=drivesdk
I mode this apk, and my p3p work with it.

KyokushinPL, 2017-07-15:
Its translated to russian?

DKoro1, 2017-07-15:
Yes.

KyokushinPL, 2017-07-15:
:|
Maybe you would be able to help the guys to mod that app here https://github.com/Bin4ry/deejayeye-modder/issues/10  ?

CirConFleX, 2017-07-16:
@KyokushinPL you think you can post the modded APK with all patchs please ?

WindyYam, 2017-07-20:
ok I've got my P3A back and done a little test on it with 3.1.5app and fw 1.10.90, it's regret to see the config does no extra effect on P3A performance, I further confirmed this by direct function calls to those sdr command, the callback returns fail. Also I've do direct call to countrycode and RC set power, but nothing interesting appears.
compare to mavic pro, the sdr calls return success, the countrycode return sucess and it does switched to fcc compared with stock mavic config

g1dtf, 2018-10-07:
From all of the above comments can I assume that the config files and app patches for FCC don't work on the lightbridge based P3A, P3pro and P4.