o-gs/dji-firmware-tools #43: Connecting Phantom 3 front USB to Linux

mefistotelis, 2017-07-11:
Anybody tried using the USB usually used for Assistant for anything else?

When I connect it to PuTTY on my Windows machine, I get a startup log. But when I connect it to Linux - I see no output.

Am I missing something, or is the serial device unsupported in Linux?

I trired both ```picocom``` and simple ```cat```:

<pre>#!/bin/bash
.
PORT=ttyACM0
.
n=0
while [ $n -lt 2147483647 ]; do
  if [ -e /dev/${PORT} ]; then
    break
  fi
  if [ $((n % 1000)) -eq 0 ]; then
    printf 'Waiting %d seconds for %s\n' $((n / 100)) ${PORT}
  fi
  sleep 0.01
  n=$((n+1))
done
.
printf 'Port %s found\n' ${PORT}
sleep 0.1
#sudo picocom /dev/${PORT} -b 9600 -d 8 -p n
#exit 0
.
sudo setserial /dev/${PORT} uart 16550A baud_base 9600 \
 spd_normal low_latency
sudo setserial -G /dev/${PORT}
.
printf 'Starting read from %s\n' ${PORT}
sudo cat /dev/${PORT} | od -x
</pre>

danieltroger, 2017-08-18:
With the phantom 3 standard and ubuntu 17.04 I also get no output.
```
root@mini:/dev/serial/by-id$ ls
usb-DJI_DJI_CONTROLLER_0001A0000000-if00
root@mini:/dev/serial/by-id# cat > usb-DJI_DJI_CONTROLLER_0001A0000000-if00 

^C
```

I suppose that you need to send some command in order to make it respond. Like configure some operating mode (Debug, Assistant, Simulator, whatever).

Have you looked at the drivers of the developer flight simulator for windows? (And the simulator itself)
Reverse-engineering them would probably be interesting.

mefistotelis, 2017-08-18:
>  With the phantom 3 standard and ubuntu 17.04 I also get no output.
<pre>cat > usb-DJI_DJI_CONTROLLER_0001A0000000-if00</pre>

It isn't possible to get input on screen with this command. It sends data from keyboard to the serial.

Anyway, you're right - I either didn't configured something, or the windows driver does makes some kind of initialization.

danieltroger, 2017-08-18:
Woops, sorry, I copied a wrong line from my terminal. I tried cat <device> and cat > <device> to see if it just needs *some* data to start replying.

I noticed that, after a fresh switching on,  192.168.1.2:5678 (in the p3's wifi, which should be the FC?) starts sending data if I connect to 192.168.1.1:2345 (RC) first.
Dunno why and how, but I imagine it needs some kind of initialization. I will try to launch the simulator on my windows machine when I have it again, and see if there's some output to the linux console after swapping the cable.

mefistotelis, 2021-06-15:
The USB connector is now explained here:
https://github.com/o-gs/dji-firmware-tools/wiki/P3X-ESC-center-board#external-interfaces

closed 2021-06-15 by mefistotelis