o-gs/dji-firmware-tools #45: Downgrade Problem after Craft was at Repair at DJI

MrBurnsAT, 2017-07-17:
Hello

Now im sure DJI does something with our Phantoms if they where sent in for Repair.

Every Craft comes at least with 1.10 back from Repair.

After that only One Version back (down to 1.9) is possible without problems.

If u go lower than 1.9 many different Problems appear.

Compass Problems are the main Problem.

After Downgrade to 1.8 all Compass values are at 0.

How can we Check whats the Problem. 
How to solve this Problem?

MrBurnsAT, 2017-07-17:
Im sure m1900 is the Problem!!!
[P3X_FW_LOG_AB (1).txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1153498/P3X_FW_LOG_AB.1.txt)

MrBurnsAT, 2017-07-17:
Here comes the Problem with 1.9 Not 1.8

MrBurnsAT, 2017-07-18:
Same Problem next Copter!
But this one wasnt at DJI

Only 1.10 was installed once.
(m1900) V255.255.255.254 

Will send LOG File as fast as i can.

ferraript, 2017-07-19:
if your target is FW 1.7.60, then I think the safest way is to cook its special version that doesn't contain package for m1900 and flash it with debug file

mefistotelis, 2017-07-19:
I don't think dji_fwcon.py is able to correctly build a firmware without one of the modules.

I recently used it to extract and re-pack all the DJI firmwares listed in CSV file (you can do this by changing EXEC_FLAG to 0x02), and the firmwares which had NULL modules failed the test because checksums were invalid. Looks like null modules have some kind of special influence on checksums.

But - you won't be sure until you try.

MrBurnsAT, 2017-07-19:
Would it be possible to take module 1900 from 1.10 and put it in 1.7 or so?

mefistotelis, 2017-07-19:
In case of some modules, having different versions could make them unable to communicate and therefore brick your drone; but for FPGA module like m1900 I don't think bricking is possible - the serial NAND memory is likely programmed without use of the FPGA chip itself.

So yes, you can merge modules from two versions, I can't see any problem with that.

MrBurnsAT, 2017-07-19:
So gone step by step up.

At 1.9 Problem goes away
[P3S_FW_LOG_AB(2).txt](https://github.com/mefistotelis/phantom-firmware-tools/files/1160097/P3S_FW_LOG_AB.2.txt)

ferraript, 2017-07-19:
>Would it be possible to take module 1900 from 1.10 and put it in 1.7 or so?

But I don't think this will help you because m1900 doesn't have that 255.255.255.254 version in FW 1.10

chazzzzz, 2017-07-30:
Why don't you try renaming M1900 target bin to "PFPGAFw3.bin" and put that on your card and load it that way to see if it clears up the version "255.255.255.254" ...It shouldn't matter what bin you get it from because what I'm seeing is they're the same version from at least fw ver 1.03 to 1.10...This is the way I downgraded my spare batteries...It acts a little differently while loading but it does work for THEM at least...Of course it's at your own risk but what it's doing this stuff...

'target':   'm1900',  Stores   firmware for FPGA air model 0  version=01.00.2144

MrBurnsAT, 2017-07-30:
Tried that already

No difference. It dont update the module. Because wrong Version. ;-)

KyokushinPL, 2017-08-10:
Strange
People are downgrading from 1.10 since long time.
https://phantompilots.com/threads/downgrading-firmware-on-p3p-from-1-10-9-to-1-7-6.105384/

Maybe you was on 1.11  and that caused a problem?

aka1ceman, 2018-01-20:
Can we revisit the m1900 issue now that we have keys tri look at newer fw? Can we tell what the problem stems from now? Is it possible to add a module by itself?