o-gs/dji-firmware-tools #48: P3P firmware upgrade failed - service.bin

acdjunior, 2017-08-21:
I have upgraded my phantom 3 pro and had issues with altitute limits. I decided to downgrade the firmware for the previous one. After this my P3P stopped to work and I tried to upgrade again without success. the log file shows the module 19 can not be upgraded, so I read in many differents foruns that I need the service.bin file to fix this problem. Can some one help me on this issue ?

acdjunior, 2017-08-23:
The log file shows: Done.

 [01785956]Firmware upgrading[4]...
 [01788975][19 00] Firmware upgrade start...
 [01819658][19 00] Firmware upgrade finish failed (step = 2, err = 0x08).
 [01849737]Done.

kiel1982, 2017-08-29:
did you find answer?

kiel1982, 2017-08-29:
I have same problem

mefistotelis, 2017-08-29:
Check these flashing methods:
https://github.com/mefistotelis/phantom-firmware-tools/wiki/Firmware-m1900

The OFDM board is quite cheap to buy used, so don't pursue it too hard.

ruckusman, 2017-09-04:
I've had a similar but slightly different issue with an Osmo with X5 attached - DJI changed the communications between the wifi module in the handle and the camera and I wanted to downgrade to previous versions to enable communication directly with the camera via telnet.

Anyway what I discovered is that using the FW_DEBUG files relevant to the firmware you are upgrading will allow unlimited downgrades in incremental steps. It also allows re-flashing with the same version.

So how to determine what the blank file should be called - I started with running fgrep looking for FW_DEBUG in the firmware. Sure enough it's there.

I got the clue here - so all kudos to mefistotelis
https://github.com/mefistotelis/phantom-firmware-tools/issues/4

In the case of the Osmo with X5 attached the blank file in the root directory needs to be called WM610_FC550_FW_DEBUG

You could try creating the relevant file and re-flashing the same downgrade firmware again, as it will re-load firmware, I believe in the case of the P3 Pro is P3X_FW_DEBUG.

Note that I have successfully downgraded and X5/Osmo combination from the most recent release firmware to the first release firmware in single increments, and am progressing forwards back up the releases. If there has been an error at any stage just reflashing with the same version has helped or else going down one version or up one version worked.

Hope this helps