o-gs/dji-firmware-tools #49: Locating the bitrate table

ruckusman, 2017-08-23:
Has anyone been able to locate the bitrate table for the camera bitrates?

I've posted here seeking some assistance outlining the process

Basically I'm trying to locate it within the OSMO_FC550_FW_V01.03.00.40_m0100_part_sys.a9s which has been extracted with the tools from this site

https://dashcamtalk.com/forum/threads/changing-the-default-bitrate-of-ambarella-based-cameras-using-aft.10220/page-2

The tools from that site don't work with the A9s obviously, but it's a possibility that the bitrate table view tool can be utilised if a way can be found to determine the start offset of the table within the file.

I'm basing this effort upon specs from here for the Xiaomi Yi 4k camera specs posted here 
http://elproducente.com/4k-xiaomi-yi-action-camera-2-review/

and the project located here which has enabled higher bitrates using autoexec.ash scripts reminiscent of the original gopro hacks
https://github.com/irungentoo/Xiaomi_Yi_4k_Camera

I have tested high speed MicroSD cards at being capable of over 20MB/s, so with good quality high speed cards we should be able to emulate the Xiaomi Yi 4K camera data speeds at least up to 100Mb/s

mefistotelis, 2017-08-23:
So what pattern can be used to find the table? Or what the Ambarella firmware does with it (which Ambarella SDK API call uses it)?

ruckusman, 2017-08-24:
This thread regarding various A7 based dashcams as well as the successful modification to the DJI P2 Vision Plus camera is the source of my inspiration. 

https://dashcamtalk.com/forum/threads/app-aft-1-3-2-ambarella-firmware-editor-development-paused.6029/page-21

Yes I am searching to find a pattern to identify the table within the file - hopefully those experienced dashcam modders can assist.

I don't know which API call utilises the table, I am fairly certain that it sets the video recording compression type and data rate level passed to the Ambarella SOC.

Mid 2016 there was an Inspire firmware update which increased the maximum video data rate(s) and likely changed it to CBR as I've check various recent files which display a data rate of 59.96 Mbps CBR.

I already have a potential mod for an Inspire X3 camera to use a high speed full size SD card which will allow higher data rates.

mefistotelis, 2017-08-24:
Then here's a place to start - this comes from ```P3X_FW_V01.01.0008_m0100_part_sys.idb```.

<pre>int VideoTuner_VidEncDefaultSetting()
{
  if ( word_EA97178 != 34 )
  {
    sub_E8D6354(1, "VideoTuner_VidEncDefaultSetting", 1657, "Configure video tuner video encode in wrong state(%d)");
    return -1;
  }
    sub_E8D6354(0,"VideoTuner_VidEncDefaultSetting",1661," Configure video encode start...");
    VideoEncode_FrameRateDivisor = vid_settings[2].field_14;
    VideoEncode_DataFmt = 0;
    VideoEncode_StreamID_ViewZoneID = 0;
    VideoEncode_StreamID_StreamID = 0;
    VideoEncode_Window_Width = vid_settings[2].Width;
    VideoEncode_Window_Height = vid_settings[2].Height;
    VideoEncode_IsIntervalCap = 0;
    VideoEncode_RotateConfig = 0;
    VideoEncode_FrameRate_Interlace = vid_settings[2].field_8;
    VideoEncode_FrameRate_NumUnitsInTick = vid_settings[2].field_10;
    VideoEncode_FrameRate_TimeScale = vid_settings[2].field_C;
    h264_pBitsBufAddr = dword_12298D4C;
    h264_BitsBufSize = 0x4000000;
    h264_M = vid_settings[2].field_1C;
    h264_N = vid_settings[2].field_20;
    h264_IsCabac = 1;
    h264_QualityLevel = 131;
    h264_BitRate = vid_settings[2].Bitrate;
    h264_BitRateSetting = vid_settings[2].field_24;
    byte_EAB167D = 51;
    if ( SLOBYTE(vid_settings[2].field_24) == 3 )
    {
      h264_VbrPercent = 50;
      h264_VbrMinRatio = 90;
      h264_VbrMaxRatio = 110;
    }
    h264_IdrInterval = vid_settings[2].field_18;
    h264_EnableReadRawData = 1;
    h264_EnableSaveYuvData = 0;
    h264_EnableSaveVidData = 1;
    pf_strcpy(h264_RawSeqFilePath, "C:\\VideoTuner\\VidEncMem.raw");
    pf_strcpy(h264_YuvSeqFilePath, "C:\\VideoTuner\\VidEncMem.yuv");
    pf_strcpy(h264_VidBitStreamFilePath, "C:\\VideoTuner\\VidEncMem.h264");
    sub_E8D6354(0, "VideoTuner_VidEncDefaultSetting", 1702, " Configure video encode done...");
    return 0;
}</pre>

danieltroger, 2017-08-24:
Wow! That sounds very promising! Maybe one could change
```
    VideoEncode_Window_Width = vid_settings[2].Width;
    VideoEncode_Window_Height = vid_settings[2].Height;
```
to get the phantom 3 standard record 4k?

danieltroger, 2017-08-24:
Where is `vid_settings` defined?

mefistotelis, 2017-08-24:
I defined it today, so it's not in the symbols commited to repository yet.
<details>
<pre>; struct VideoEncSettings vid_settings[41]
vid_settings	VideoEncSettings <0, 3840, 2160, 30000000, 0, 30000, 1001, 1, 4, 1, 8,\
				  0x101, 0, 0, 0, 0, 0,	0, 0, 0, 0, 0, 0>
		VideoEncSettings <0, 3840, 2160, 30000000, 0, 30000, 1001, 1, 4, 1, 8,\
				  0x101, 0, 0, 0, 0, 0,	0, 0, 0, 0, 0, 0>
		VideoEncSettings <0, 3840, 2160, 30000000, 0, 30000, 1001, 1, 4, 1, 8,\
				  0x101, 0, 0, 0, 0, 0,	0, 0, 0, 0, 0, 0>
		VideoEncSettings <0, 3840, 2160, 30000000, 0, 20000, 1001, 1, 4, 1, 8,\
				  0x101, 0, 0, 0, 0, 0,	0, 0, 0, 0, 0, 0>
		VideoEncSettings <0, 4096, 2160, 30000000, 0, 25000, 1000, 1, 4, 1, 8,\
				  0x101, 0, 0, 2000000,	0, 25000, 1000,	1, 4, 1, 8, \
				  0x101>
		VideoEncSettings <0, 4096, 2160, 30000000, 0, 24000, 1001, 1, 4, 1, 8,\
				  0x101, 0, 0, 2000000,	0, 24000, 1001,	1, 4, 1, 8, \
				  0x101>
		VideoEncSettings <2, 3840, 2160, 60000000, 0, 30000, 1001, 1, 4, 1, 8,\
				  0x101, 432, 240, 2000000, 0, 30000, 1001, 1, 4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 3840, 2160, 30000000, 0, 25000, 1000, 1, 4, 1, 8,\
				  0x101, 432, 240, 2000000, 0, 25000, 1000, 1, 4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 3840, 2160, 60000000, 0, 24000, 1001, 1, 4, 1, 8,\
				  0x101, 432, 240, 2000000, 0, 24000, 1001, 1, 4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 2560, 1440, 16000000, 0, 60000, 1001, 1, 4, 1, 8,\
				  0x101, 432, 240, 2000000, 0, 30000, 1001, 2, 4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 3840, 1920, 30000000, 0, 30000, 1001, 1, 4, 1, 8,\
				  0x101, 480, 240, 2000000, 0, 30000, 1001, 1, 4, 1, \
				  8, 0x101>
		VideoEncSettings <0, 2704, 1520, 60000000, 0, 30000, 1001, 1, 4, 1, 8,\
				  0x1000101, 432, 240, 2000000,	0, 30000, 1001,	2, 4, \
				  1, 8,	0x101>
		VideoEncSettings <0, 2704, 1520, 60000000, 0, 25, 1, 1,	4, 1, 8, \
				  0x1000101, 432, 240, 2000000,	0, 30000, 1001,	2, 4, \
				  1, 8,	0x101>
		VideoEncSettings <0, 2704, 1520, 60000000, 0, 24000, 1001, 1, 4, 1, 8,\
				  0x101, 432, 240, 2000000, 0, 30000, 1001, 2, 4, 1, \
				  8, 0x101>
		VideoEncSettings <0, 1920, 1080, 16000000, 0, 120000, 1001, 1, 4, 1, \
				  8, 0x101, 0, 0, 2000000, 0, 30000, 1001, 4, 4, 1, 8,\
				  0x101>
		VideoEncSettings <2, 1920, 1080, 26000000, 0, 60000, 1001, 1, 4, 1, 8,\
				  0x101, 1280, 720, 2000000, 0,	30000, 1001, 2,	4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 1920, 1080, 16000000, 0, 50000, 1000, 1, 4, 1, 8,\
				  0x101, 432, 240, 2000000, 0, 30000, 1001, 2, 4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 1920, 1080, 16000000, 0, 48000, 1001, 1, 4, 1, 8,\
				  0x101, 432, 240, 2000000, 0, 30000, 1001, 2, 4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 1920, 1080, 16000000, 0, 30000, 1001, 1, 4, 1, 8,\
				  0x101, 1280, 720, 2000000, 0,	30000, 1001, 1,	4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 1920, 1080, 16000000, 0, 25000, 1000, 1, 4, 1, 8,\
				  0x101, 848, 480, 2000000, 0, 25000, 1000, 1, 4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 1920, 1080, 16000000, 0, 24000, 1001, 1, 4, 1, 8,\
				  0x101, 848, 480, 2000000, 0, 24000, 1001, 1, 4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 1920, 960,	16000000, 0, 60000, 1001, 1, 4,	1, 8, \
				  0x101, 432, 240, 2000000, 0, 30000, 1001, 2, 4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 1920, 960,	16000000, 0, 30000, 1001, 1, 4,	1, 8, \
				  0x101, 848, 480, 2000000, 0, 30000, 1001, 1, 4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 1280, 720,	16000000, 0, 240000, 1001, 1, 4, 1, 8,\
				  0x101, 432, 240, 2000000, 0, 30000, 1001, 8, 4, 1, \
[... 29 lines cut ...]
				  0x101, 1920, 1080, 16000000, 0, 30000, 1001, 1, 4, \
				  1, 8,	0x101>
		VideoEncSettings <2, 1280, 720,	16000000, 0, 60000, 1001, 1, 4,	1, 8, \
				  0x101, 1280, 720, 16000000, 0, 60000,	1001, 1, 4, 1,\
				  8, 0x101>
		VideoEncSettings <0, 3840, 2160, 8000000, 0, 2000, 1001, 1, 4, 1, 8, \
				  0x101, 0, 0, 2000000,	0, 2000, 1001, 1, 4, 1,	8, \
				  0x101>
		VideoEncSettings <2, 1920, 1080, 16000000, 0, 60000, 1001, 1, 4, 1, 8,\
				  0x101, 432, 240, 2000000, 0, 60000, 1001, 1, 4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 3840, 2160, 30000000, 0, 24000, 1001, 1, 4, 1, 8,\
				  0x101, 432, 240, 2000000, 0, 24000, 1001, 1, 4, 1, \
				  8, 0x101>
		VideoEncSettings <0, 4000, 3000, 8000000, 0, 2000, 1001, 1, 4, 1, 8, \
				  0x101, 0, 0, 2000000,	0, 2000, 1001, 1, 4, 1,	8, \
				  0x101>
		VideoEncSettings <0, 2560, 1440, 16000000, 0, 30000, 1001, 1, 4, 1, 8,\
				  0x101, 360, 240, 2000000, 0, 30000, 1001, 1, 4, 1, \
				  8, 0x101>
		VideoEncSettings <2, 1920, 1080, 20000000, 0, 30000, 1001, 1, 4, 1, 8,\
				  0x101, 1920, 1080, 16000000, 0, 30000, 1001, 1, 4, \
				  1, 8,	0x101>
</pre>
</details>

mefistotelis, 2017-08-24:
Btw, creating an empty file "0x40003000.dji" on the sd-card root will trigger creation of "CameraLogCur.log" somewhere in hidden "MISC" folder.

No idea what the log will contain, it is described only as "log record config file enable".

ruckusman, 2017-08-24:
Wonderful work Mefistotelis - it seems the video encode bitrates are perhaps set differently to the A7 chip unless the values you're seeing are loaded from the settings table.

At any rate I will find which firmware increased the Inspire camera data rates and do a comparison between it and the one previous so that we cam make sense of the values you have discovered.

Earlier today I was trying load the sys partition converted to elf binary into a debugger but got thwarted by permissions errors with the debugger.

BTW the dashcams modders have methods of pushing firmware updates onto their bricked A7 based cams using a windows driver called AmbaUSB. This may potentially be one method to update only the camera firmware or recover from badly modded camera firmware.

https://www.goprawn.com/forum/main-forum/894-tool-for-unbricking-a12-a9-a7-a5-based-cams-ambausb-for-elf-bin-firmware-update-from-pc

lsusb on my linux machine doesn't recognise anything on the USB port so the above drivers may work to push firmware.

Also of note is the autoexec.ash methods used by both the Xiaomi Yi 4K camera and Gopros, which may potentially be a method to assign non-permanent changes for testing.

This is one method to potentially read out the current settings and possibly to push new settings
https://github.com/irungentoo/Xiaomi_Yi_4k_Camera/tree/master/dump_camera_settings

I'm going to test the telnet daemon on the X3 camera attached to an osmo see if it is already started, or if the technique here will enable it
https://github.com/irungentoo/Xiaomi_Yi_4k_Camera/tree/master/telnet

mefistotelis, 2017-08-24:
>> dashcams modders have methods of pushing firmware updates onto their bricked A7 based cams using a windows driver called AmbaUSB

Thank you, that's useful info.

> lsusb on my linux machine doesn't recognise anything on the USB port

I don't think anyone confirmed that the micro-USB on gimbal is connected to Ambarella. It would seem likely, but on P3 Adv Ambarella is connected directly to OFDM, and the connection there is USB-based. So there are also reasons to be unsure.

ruckusman, 2017-08-25:
Normally with all of the craft powered the USB on the cameras works as USB mass storage when an SD card is in place, so if the firmware comes from the SD card then there is an access route which may perhaps need to be enabled/configured.
I suspect that upon detecting the USB cable type, normal or OTG with the resistor between lines 4 & 5 the USB functionality may differ. The dashcam forums make mention of using OTG cables for flashing firmware onto bricked cams for recovery

ruckusman, 2017-08-25:
Three ports are open with an Osmo Wifi connection
ftp port 21 - not much to be seen as it's the busybox linux server and doesn't have access to any of the Ambarella file system that I am able to discover -
telnet port 23 - responds to connection then immediately slams the door shut
TCP port 5678 - which has various listed uses, nothing definitive, I assume this is for the control transactions between the Phone/App and the osmo unit itself - this may yield something useful

danieltroger, 2017-08-25:
> Btw, creating an empty file "0x40003000.dji" on the sd-card root will trigger creation of "CameraLogCur.log" somewhere in hidden "MISC" folder.

I tried it out, very interesting log. Check it out here: https://leetfil.es/paste/a9c92f37
Another file, `elog_0`, was also created in the SD card root, no idea what it contains. Looks like nothing: https://leetfil.es/file/8812234f
I would like to get into the AmbaShell of the RTOS, but I don't know how. Seems like it would be possible with some kind of inter process communication, maybe?

The gopro hacks worked by entering commands into that shell, and something in some link of @ruckusman which I looked into worked by patching /dev/mtdblock1. If I run `strings` on it, AmbaShell is definitely contained there, and other interesting stuff too. (The same strings as you can find in the logfile are all there, so it must be the RTOS?)

Now how far I have figured out that the camera system works on the p3standard:
all those findings where mainly made by exploring the three linux systems over telnet:
192.168.1.1, 192.168.1.2 & 192.168.1.3
And by reading stuff on the internet. Maybe it is useful for you:
So, as you know, the DJI go app connects to mainly three ports in normal operation, it also connects over ftp to send firmware update files and to playback videos:

192.168.1.1:2345 - which is communication with the RC's computer that is _not_ the linux box. In the linux system the process `apsrv` opens port 2345 and maps it to `/dev/ttyACM0`. If you pass some arguments (can't remember) to apsrv, when calling it manually, you can get a hexdump of the communication to stdout or a given filename.

192.168.1.2:5678 - that is the gimbal top board's linux computer. An equivalent of `apsrv`, `clisrv`, runs there, and does exactly the same - but with the serial link of the flight controller, this is how DJI GO get's the telemetry data.

Then there is 192.168.1.3:9000 and 9001 (UDP), which is the `cam_server`-process on the Ambarella *linux*-side, which you can telnet into. It sends a h264 stream with UDT, you can recieve it with https://github.com/hardening/Cave/blob/master/phantom3-video.cpp and pipe it to some player (like ffplay) and play it back on you computer.
I think this is possible with osmo also, I read some russian blog on it.

apsrv, clisrv and cam_server also seem to contain some firmware-upgrade related code.
Then there is some usb-related code in cam_server too, since I aren't so skilled in deassembling/compiling/whatever I haven't really figured anything out yet.
I also want to know more about the ambarella's usb-interfaces, lsusb lists two devices

```
/tmp/SD0 # lsusb
Bus 001 Device 002: ID 0525:a4a2
Bus 001 Device 001: ID 1d6b:0002
```
one is for the network:
```
/tmp/SD0 # ifconfig
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

usb0      Link encap:Ethernet  HWaddr 3E:13:79:04:80:52  
          inet addr:192.168.1.3  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1038 errors:0 dropped:31 overruns:0 frame:0
          TX packets:739 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:70327 (68.6 KiB)  TX bytes:318248 (310.7 KiB)
```
And no clue about the other one.
When connecting something to the usb-port located on the gimbal top board, the gimbal starts to beep, the motors get switched off, and you can access the files on the sd-card.
Funny thing is that it is impossible to telnet into 192.168.1.3 if the usb-cable is connected, but the computer isn't halted - I ran a script that writes the time into a file every second and there where no interruptions. Maybe the network-usb get's reused for mounting the sd-card.

However, my main goal at the moment, is, to somehow communicate with the ambarella RTOS.
I did some testing about how things work by switching off the linux systems with `halt`, here are my findings:
the ambarella RTOS *must* be controlled by the gimbal top board (maybe clisrv?), because if I kill cam_server or switch the ambarella linux off, I still can start and stop recording, change settings, etc in the go app - there's just no video feed.
And, by the way, CameraLogCur.log gets flooded with

```
[0m
[0m[01482114][CA9_0]  [0;31m***[DJI_IPC] DjiIpcSend[L1517]: Send msg failed(833)!, Rval:-1

[0m
[0m[01482381][CA9_0]  [0;31m***[DJI_IPC] IPC_Rpmsg_Send[L647]: Send msg failed(897)!, Rval:-1
```
If I switch off 192.168.1.2, the camera is still recording but I'm obviously unable to communicate with it any longer because 192.168.1.2 does the wifi-stuff.
And if I _restart_ it, the gimbal starts beeping and shuts off, sometimes goes on again for a short moment, and I then have to restart the drone to get it working again or connect a USB cable, wait some time, and disconnect it again.

So to get into the Ambarella RTOS there are x possibilities:
1. Figure out how this IPC thing works and enter over the communication channel that cam_server uses to receive the h264 stream.
2. Figure out how clisrv IP-Communicates with the RTOS and get into Amba_Shell
3. Figure out how to connect over USB with a computer
[... cut ...] with the xiami hack? But how do we figure out the right commands?

Or, we just patch the bitrate and resolution in the firmware - when we know how to understand the vid_settings table.
This we could do over either /dev/mtd(block)X live from the unix thing or over the normal firmware-upgrade method.
Problem is we could brick the Ambarella system and have a hard time reflashing it.

So much to a script kiddies attempt to break the limits of a sophisticated system. Hope this is to some use for you.

ruckusman, 2017-08-25:
Very Interesting work - if you're seeing the cam_server you're looking at another kernel and file system to the one I've been exploring - I have some clues for you. Sorry for the long message - attachments failed to work

I managed to get telnet access by looking into the messages located in /var/log - spoofing the wifi MAC address got me straight in - see the messages

The OSMO only has the one ip address running, so it looks as though you may have accessed other kernels - see dmesg partition details - below > below

MESSAGES ->
Jan  1 00:00:02 (none) syslog.info syslogd started: BusyBox v1.22.1
Jan  1 00:00:02 (none) kern.info kernel: [    2.600000] Loading modules backported from Linux version master-2014-05-22-0-gf2032ea
Jan  1 00:00:02 (none) kern.info kernel: [    2.610000] Backport generated by backports.git backports-20140320-37-g5c33da0
Jan  1 00:00:02 (none) kern.info kernel: [    2.870000] cfg80211: Calling CRDA to update world regulatory domain
Jan  1 00:00:03 (none) kern.info kernel: [    2.870000] cfg80211: World regulatory domain updated:
Jan  1 00:00:03 (none) kern.info kernel: [    2.880000] cfg80211:  DFS Master region: unset
Jan  1 00:00:03 (none) kern.info kernel: [    2.880000] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
Jan  1 00:00:03 (none) kern.info kernel: [    2.890000] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
Jan  1 00:00:03 (none) kern.info kernel: [    2.900000] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
Jan  1 00:00:03 (none) kern.info kernel: [    2.910000] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
Jan  1 00:00:03 (none) kern.info kernel: [    2.920000] cfg80211:   (5170000 KHz - 5250000 KHz @ 160000 KHz), (N/A, 2000 mBm), (N/A)
Jan  1 00:00:03 (none) kern.info kernel: [    2.920000] cfg80211:   (5250000 KHz - 5330000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
Jan  1 00:00:03 (none) kern.info kernel: [    2.930000] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
Jan  1 00:00:03 (none) kern.info kernel: [    2.940000] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
Jan  1 00:00:03 (none) kern.info kernel: [    2.950000] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
Jan  1 00:00:03 (none) kern.debug kernel: [    3.800000] ath: EEPROM regdomain: 0x0
Jan  1 00:00:03 (none) kern.debug kernel: [    3.800000] ath: EEPROM indicates default country code should be used
Jan  1 00:00:03 (none) kern.debug kernel: [    3.800000] ath: doing EEPROM country->regdmn map search
Jan  1 00:00:03 (none) kern.debug kernel: [    3.800000] ath: country maps to regdmn code: 0x3a
Jan  1 00:00:03 (none) kern.debug kernel: [    3.800000] ath: Country alpha2 being used: US
Jan  1 00:00:03 (none) kern.debug kernel: [    3.800000] ath: Regpair used: 0x3a
Jan  1 00:00:03 (none) kern.debug kernel: [    3.820000] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
Jan  1 00:00:03 (none) kern.info kernel: [    3.840000] cfg80211: Calling CRDA for country: US
Jan  1 00:00:03 (none) kern.info kernel: [    3.850000] cfg80211: Regulatory domain changed to country: US
Jan  1 00:00:04 (none) kern.info kernel: [    3.850000] cfg80211:  DFS Master region: FCC
Jan  1 00:00:04 (none) kern.info kernel: [    3.860000] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
Jan  1 00:00:04 (none) kern.info kernel: [    3.870000] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 3000 mBm), (N/A)
Jan  1 00:00:04 (none) kern.info kernel: [    3.870000] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz), (N/A, 1700 mBm), (N/A)
Jan  1 00:00:04 (none) kern.info kernel: [    3.880000] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz), (N/A, 2300 mBm), (0 s)
Jan  1 00:00:04 (none) kern.info kernel: [    3.890000] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 3000 mBm), (N/A)
Jan  1 00:00:04 (none) kern.info kernel: [    3.900000] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
Jan  1 00:00:04 (none) kern.info kernel: [    3.930000] ieee80211 phy0: Atheros AR9340 Rev:0 mem=0xb8100000, irq=47
Dec 30 07:19:33 (none) kern.info kernel: [    7.140000] device wlan0 entered promiscuous mode
Dec 30 07:19:35 (none) kern.info kernel: [    8.510000] br-lan: port 1(wlan0) entered forwarding state
Dec 30 07:19:35 (none) kern.info kernel: [    8.510000] br-lan: port 1(wlan0) entered forwarding state
Dec 30 07:19:35 (none) kern.info kernel: [    8.540000] device usb0 entered promiscuous mode
Dec 30 07:19:35 (none) daemon.info udhcpd[541]: udhcpd (v1.22.1) started
Dec 30 07:19:35 (none) daemon.err udhcpd[541]: can't open '': No such file or directory
Dec 30 07:19:35 (none) daemon.notice telnetd[556]: limit mac 38:bc:1a:e0:6e:0d
Dec 30 07:19:48 (none) daemon.info hostapd: wlan0: STA 38:bc:1a:e0:6e:0d IEEE 802.11: authenticated
Dec 30 07:19:48 (none) daemon.info hostapd: wlan0: STA 38:bc:1a:e0:6e:0d IEEE 802.11: associated (aid 1)
Dec 30 07:19:48 (none) daemon.info hostapd: wlan0: STA 38:bc:1a:e0:6e:0d WPA: pairwise key handshake completed (RSN)
Dec 30 07:19:50 (none) kern.info kernel: [   23.520000] br-lan: port 1(wlan0) entered forwarding state
Dec 30 07:19:50 (none) daemon.info udhcpd[541]: Sending OFFER of 192.168.1.20
Dec 30 07:19:50 (none) daemon.info udhcpd[541]: Sending ACK to 192.168.1.20
[... cut ...]
echo  >> /tmp/SD0/info.txt
ls -l /usr/sbin/ >> /tmp/SD0/info.txt
echo  >> /tmp/SD0/info.txt
ls -l /lib/ >> /tmp/SD0/info.txt
echo  >> /tmp/SD0/info.txt
ls -l /usr/lib/ >> /tmp/SD0/info.txt
echo  >> /tmp/SD0/info.txt
ls -l /var/ >> /tmp/SD0/info.txt
echo  >> /tmp/SD0/info.txt
ls -l /run/ >> /tmp/SD0/info.txt
echo  >> /tmp/SD0/info.txt
ls -l /tmp/ >> /tmp/SD0/info.txt
echo  >> /tmp/SD0/info.txt
echo END_OF_FILE >> /tmp/SD0/info.txt
busybox unix2dos -d /tmp/SD0/info.txt
echo ambainfo.sh script done ...

ruckusman, 2017-08-26:
OK so I also had fuzzy eyes and a foggy brain when I wrote the message, many things became clear to me this morning as I was paddelboarding.

Thank you Mefistotelis for editing the message, it was a mess once I'd finished posting it, I tried twice to clean it up without success.

Great test on the write speeds on the SD card BTW - I have a mate with a high spec card capable of 22MB/s from a PC USB port, which I will test once I have access to the SD card.

Firstly, running nmap only discovers the 192.168.1.2 IP address for the OSMO, previously I could FTP only, before discovering the MAC address restriction within the /var/log/message file.

Dec 30 07:19:35 (none) daemon.notice telnetd[556]: limit mac 38:bc:1a:e0:6e:0d

Spoofing that MAC address allowed me immediate telnet access to the kernel available on 192.168.1.2. 
For the record, that is basically a router in operation, complete with OpenWrt and Red Hat copyright notices within various string searches. The only running process of note is hostapd for routing network traffic.

I should have thought to just use FTP to copy the 10 partitions located within /dev/mtb, I will do that later today.

I will also attempt to mount them within the router kernel and see what they reveal

I haven't yet tried running nmap again with the spoofed MAC address to see if the router reveals the other IP addresses & kernels/file systems etc. 
I believe you are much further into the whole system than I am thus far.

Looking at the partitions it expands I'd say the router on 192.168.1.2 expands them as a master process, and then the other kernels spawn from there. Perhaps they are running on different cores as the Ambarella is after all a multi core SoC, two A9's, one A11, and an image and video DPS's.

The file system which you are seeing looks very much like it could be the RTOS kernel as you have direct access to the SD card (c:) and the AmbaFS (a:) - I may be incorrect on this.

Anyway

Remember the dumps you have created are the file systems located within the memory ranges, it may be possible to mount them and read them, otherwise 

Use the dumps of the files systems, try following the instructions from the dashcams threads to locate the bitrate table if it is within the binary lumps.

autoexec.ash scripts are passed to RTOS to be executed after booting - this was the method pioneered by the Gopro hacks. 
From there we ma be able to construct an autoexec.ash script to modify the specified memory ranges as the Xiaomi Yi 4k camera script does.
It basically tells the RTOS to alter the specified memory ranges with different values.

I will have to revise my readings, however there are ways in which changes can be made semi permanent so that the scripts don't have to be present on the root directory of the SD card.

That is a good low risk method to test new settings

I would like to be able to record in Mjpeg as an option as there is less I frame pulsing within the GOP which ruins many shots, although an 80Mb/s second bitrate is a step in the right direction.
Bring on 120Mb/s with ultra high speed cards if the USB bus is capable.

I am also aiming towards removing some of the so called safety features implemented by DJI like the prop obstruction code which has meant I have had to rebuild two smashed Inspires recently due to false positives and the craft falling from the sky. Next project...

It should also be possible to add different resolutions by modifying current less desirable ones, 2.7K is desirable to many, though that would best be achieved in conjunction with App mods to list the changes to the user Good thing is that now that is possible on Andriod.

To not, from when I was repairing Phantoms, there was one physical difference between the P3 Advanced and P3 Professional, which was a chip not installed in the gimbal on the advanced. 
From memory this was the DM368 used to handle the low latency video subsampling for transmission allowing the Ambarella to encode 4K.
The advanced had a lower bitrate video downlink as a result also.

For example I recall an Inspire firmware update which changed the data path through the gimbal onto different pins of the video downlink, if the craft and gimbal did not match there was no video stream.

I do not recall if the DM368 was put into the loop or excluded from it, I will test later with some tape over the contacts to see where that stream originates from in the current firmware.

I cannot speak as to the internal structure of the P3 Standard, however I'd question the assumption about there being more than two SoC's, but don't take my word as gospel if you have disassembled the camera/gimbal unit and found more than on SoC.

At this point if you have a windows machine try the AmbaUSB drivers from links posted above and connect to the gimbal USB to see what functionality they enable, as this is one potential method from recovering from firmware modification issues.

mefistotelis, 2017-08-26:
> I would like to get into the AmbaShell of the RTOS, but I don't know how.

In Ph3 Pro/Adv, it can be accessed by serial interface U0 (service pads on the board).
https://github.com/mefistotelis/phantom-firmware-tools/wiki/Firmware-m0100#debug-serial-interface-u0

ruckusman, 2017-08-26:
Seems I may have had the hierarchy wrong for the RTOS and Linux  see the notes in the script description on this page _ RTOS runs bare metal with Linux virtualised - I suppose there's no guarantee that is how DJI have it configured.

https://github.com/EvoStream/evostream_addons/blob/master/bash_scripts/Ambarella.A9-EMS-StartStreaming.sh

One element confuses me slightly, I am seeing a linux kernel operating a router essentially and you are seeing a linux kernel operating also.
If however you are able to execute that shell script and it returns positive results for some of the commands only normally available from autoexec.ash scripts then I would tender that you are working on the bare metal operating system already.
You have all of the camera control commands, which are absent from my file system

danieltroger, 2017-08-29:
@ruckusman:

> Great test on the write speeds on the SD card BTW - I have a mate with a high spec card capable of 22MB/s from a PC USB port, which I will test once I have access to the SD card.

I also tried an UHS2 card which is cabable of 50MB/s write and 150MB/s read and also only got around 10MB/s write from the amba linux. But maybe the RTOS has got a faster sd card access.

> For the record, that is basically a router in operation, complete with OpenWrt and Red Hat copyright notices within various string searches.

Yes. On the p3sta this router is connected to the wifi antenna and is responsible for the video downlink.

> The only running process of note is hostapd for routing network traffic.

Sure? No clisrv or apsrv running?

> I should have thought to just use FTP to copy the 10 partitions located within /dev/mtb, I will do that later today.

Pretty sure you can't read them over ftp because they are special files. I transferred them with netcat and telnet.

> I believe you are much further into the whole system than I am thus far.

Seems so, I believe you haven't discovered the ambarella system yet.

> The file system which you are seeing looks very much like it could be the RTOS kernel as you have direct access to the SD card (c:) and the AmbaFS (a:) - I may be incorrect on this.

It is indeed on the ambarella chip, but it for sure is not the RTOS, as I don't have access to the AmbaShell but rather the linux shell.

> Remember the dumps you have created are the file systems located within the memory ranges, it may be possible to mount them and read them, otherwise

That's a good idea, will try it.

> Use the dumps of the files systems, try following the instructions from the dashcams threads to locate the bitrate table if it is within the binary lumps.

I'll read through them soon, thanks.

> autoexec.ash scripts are passed to RTOS to be executed after booting - this was the method pioneered by the Gopro hacks. (...)

Yes, but as I said, the memory dump from the ambarella chip which includes the RTOS doesn't contain the string "autoexec.ash", so I don't think it will work.

> From when I was repairing Phantoms, there was one physical difference between the P3 Advanced and P3 Professional, which was a chip not installed in the gimbal on the advanced.
From memory this was the DM368 used to handle the low latency video subsampling for transmission allowing the Ambarella to encode 4K.
The advanced had a lower bitrate video downlink as a result also.

Yeah, same thing with p3 sta and p3 4k, the standard does the encoding on the ambarella SoC, I'd be curios if one could kill the livestream and record 4k instead…

> I cannot speak as to the internal structure of the P3 Standard, however I'd question the assumption about there being more than two SoC's, but don't take my word as gospel if you have disassembled the camera/gimbal unit and found more than on SoC.

I am >100% sure about that "assumption", and, as @mefistotelis said
> We have photos of all Ph3 Std boards:
https://github.com/mefistotelis/phantom-firmware-tools/wiki/DJI-Hardware

You can look on the wiki.
This is the camera encoder board with the ambarella chip, I call it 192.168.1.3:
https://github.com/mefistotelis/phantom-firmware-tools/wiki/WM321-Camera-encoder-board

And this is the gimbal top board (192.168.1.2): https://github.com/mefistotelis/phantom-firmware-tools/wiki/WM321-Gimbal-top-board

> One element confuses me slightly, I am seeing a linux kernel operating a router essentially and you are seeing a linux kernel operating also.
If however you are able to execute that shell script and it returns positive results for some of the commands only normally available from autoexec.ash scripts then I would tender that you are working on the bare metal operating system already.
You have all of the camera control commands, which are absent from my file system

Exactly. It seems like you (so far) only have gotten access to the OpenWRT router, which, in my case, is the gimbal top board. I'm in the linux system of the ambarella chip and want to get into the RTOS, because the recording settings and all the nice 'n neat stuff is in there.

Did you get further meanwhile?

-----------------------

> In Ph3 Pro/Adv, it can be accessed by serial interface U0 (service pads on the board).
https://github.com/mefistotelis/phantom-firmware-tools/wiki/Firmware-m0100#debug-serial-interface-u0

So, I ordered an USB to FTTI adapter and tried my luck.
I tried some of the pads on the gimbal top board which either were the openwrt router (U2), something that returned garbage, no matter what baud rate (U1) and some pads returning nothing (U1_RX_T and U1_TX_T).

Then I figured that on the old camera encoder board the service pads are labeled, and there there is U0.

Horray.

So I tried to solder wires to those pads which are in an area of like 0.5cm^2 and put hot glue onto them to keep them there.
But it was impossible to get the drone going. The gimbal always failed it's self-test because of the connected wires obstructing the camera, went crazy, and rebooted the aircraft. And some mosquitoes made fun of me.
So completely desperate I tried to get the hot glue I had used to make the cables stay during the gimbal dance away, so that I could try again with lighter cables, and accidentally ripped off the U0 RX service pad. There's a little piece of copper left but I'm out of luck because I don't have the right equipment to resolder it.
So now I have to think of another way to get in :/

danieltroger, 2017-08-29:
But @ruckusman are you sure there isn't an 192.168.1.1 or 192.168.1.3 on the Osmo?
See here: http://www.g0l.ru/blog/n4101

ruckusman, 2017-08-30:
Unfortunately I am certain that the IP addresses to the Ambarella chip are now obscured. I suspect that in response to the hacking of the OSMO DJI altered the configuration of the router to counter the hack.
There is only the one IP address on this camera, there is another virtual interface which is assigned IPv6 - accessing that interface and the OS behind it is challenging my very limited knowledge of networking, a challenge to say the least.

For compatibility with the Inspire which this camera usually runs on the firmware was recently updated to the latest.
So I would recommend caution in updating any firmware as it looks that DJI has closed the access, much like what is happening with the Mavic and other next generation offerings.

I do have broken X3 cameras which need repair that are on much older firmware which should still be open, that doesn't solve the access issue with this X5 unfortunately, though I will use them to test increasing the bitrate.

If you are able to access a file system with anything DJI there as per the OSMO hack then you have access to everything you need already.

ruckusman, 2017-08-30:
OH I forgot to mention that I have used dd to transfer a few of the MTD partitions - then ftp, only a few of them would transfer because of insufficient space - however they are only the file system of the OpenWRT which I already have full access to.

ruckusman, 2017-08-31:
This thread contains an interesting technique to enable serial access via the USB port - not sure it may be useful for you, however if there is an assistant program you can capture the HEX string to enable the USB connection and then pass that via Putty or another console program begin chatting...

https://phantompilots.com/threads/lightbridge-firmware-problem-and-solution-firmware-version-not-found-by-dji-assistant-tool.77073/

ruckusman, 2017-08-31:
I forgot to mention there is a very complete thread here also on console access to the P3's components

https://phantompilots.com/threads/i-bricked-my-p3s-with-ftp-range-mod-lets-fix-it-together.81432/

mefistotelis, 2018-11-06:
The bitrate table of all Ph3s can now be exported and edited with amba_sys_hardcoder.py.

closed 2018-11-06 by mefistotelis