o-gs/dji-firmware-tools #54: gimbal wheel not working

pljakobs, 2017-10-12:
I know this isn't entirely the right forum or form, but it seems if anyone has an idea, it might be you folks:

I have two Phantom 3 Standards and therefore obviously also two remotes. 
With one of them, I cannot control the gimbal with the gimbal wheel. When I switch the remotes between copters, the problem stays with the remote.
I can control the gimbal with the blue ring in the app.
Here's the strangest part: when I go into controller calibration, I can see that the app can see the gimbal wheel position just fine.
I assume that the signal path (or data path) for the gimbal wheel is split at some point, with one side feeding into the WiFi and being displayed in the app and the other side going to the RC Link, controlling the actual gimbal. I have traced the signal from the poti to the controller (I assume the QFP is a uC) pin on the acquisiton board and that is, as expected, fine.

I assume that the A/D conversion is not the problem since the analog signal does not split before that uC
I would think that after this point, everything is handled in the digital domain, but how would then just one single signal be cleanly cut out of the picture?

Oh, With my other controller / copter, I had inadvertently set the Gimbal Tilt Speed to 0 in gimbal advanced settings which led to the same effect: gimbal would not move by way of the gimbal wheel but the blue ring still worked. On the faulty system, this is set to the default 50, though.

Does anyone have any pointers where to go from here?

kind regards

pj

mefistotelis, 2017-10-12:
Looks like software problem.

It is possible that besides the "gimbal speed" which you can see, there's another parameter of similar function. To make sure it's not the case, I'd try factory reset the RC configuration (I don't know how to do that in Ph3 std; in pro/adv RC, you do that by shutting down RC with proper key combination).

It is also possible that the RC firmware is the issue - try up/downgrading the RC (again, I don't know how to do that in Ph3 std; in pro/adv, you press and hold a few seconds the Dji Go icon in top right, but I imagine in std this may be done by SD card reader).

pljakobs, 2017-10-12:
@mefistotelis ok, there's a hidden feature of the app I hadn't seen yet.
Tried 1.9.20 - got stuck at 100% (why do devs even display "100%" in a progress indicator?)
now trying the same again.

Thanks for the feedback, will post result

danieltroger, 2017-10-12:
I think you can also upgrade (don't know about downgrading) the RC with dji assistant 2 over the RC USB port - surprisingly.

pljakobs, 2017-10-12:
All right, I have now successfully re-applied 1.9.20 but to no avail. The gimbal wheel still doesn't work.
From an architecture perspective: am I correct that there is no analog signal beyond the point where the gimbal wheel signal enters that first pqfp on the acquistion board?

ruckusman, 2017-10-15:
This may or may not be helpful, however I have found that parameter settings which were available with certain App/Firmware versions sometimes linger even when they have been changed/moved/eliminated in different versions.

For instance I have an Inspire remote which is on the latest firmware, that has the ability to use the C1 button to switch the gimbal control wheel between pitch and yaw control. 
This was a setting on a very old firmware/App combination which has long since been removed.
That setting still shows in the App as available, but only on that remote. 
So on another remote of the same firmware version, using the same App - the above is not an available option.

It is possible that the App version may be playing a role in not changing the value of the gimbal wheel speed in RC settings, despite seeming to work.

Have you tried modifying that setting without the craft powered on?

I also recently stumbled upon an App version which didn't display the gimbal wheel speed setting at all when the craft was connected.

DJI continuously and frequently make errors with firmware and Apps so you may have discovered a combination of previous to current firmware upgrades which gives your error.

Switching between iOS and Android can yield different results also.

Is there an assistant like the P2 remote assistant, I cannot recall the precise program name, with which to calibrate the remote. 
Reason I mention it, is because P2's with misbehaving remotes which calibrated just fine in the P2 assistant, but still misbehave, would be corrected by using the remote assistant, for no good reason at all that I could discern

mefistotelis, 2017-10-15:
> am I correct that there is no analog signal beyond the point where the gimbal wheel signal enters that first pqfp on the acquistion board?

Looking at the board:
https://github.com/mefistotelis/phantom-firmware-tools/wiki/GL358-Main-board
I can see LMV61 op amp above STM32 072C micro-controller. If the signal enters op amp, it is still analog. It gets digital inside STM32.

ruckusman, 2017-10-15:
from my reading of it, he can even see the travel of the gimbal control wheel in the App

quote "Here's the strangest part: when I go into controller calibration, I can see that the app can see the gimbal wheel position just fine."

which was why I proposed something bizarre happening in firmware/App/settings.

pj can of course confirm or correct my reading

pljakobs, 2017-10-15:
ruckusman that is correct, in the controller calibration, the gimbal wheel movement shows as a blew bar, travelling to the +/- 100% marks.
in a way, it would make more sense for this to be a software issue, since I can't see how hardware would be to blame, but if so, it would have to be in the controller, not the drone itself.

pj

ruckusman, 2017-10-15:
Hi pj - 
Try the Phantom 2 Vision Plus RC Assistant from this page

http://www.dji.com/phantom-2-vision-plus/info#downloads

with any luck it will recognise the controller via the micro USB port and allow you to do the calibration.
The P3 Standard remotes and WIFI video transmission are directly derived from the Vision +.
I suspect the internals are the same board, the one difference being the relocation of the Wifi module to inside the remote.

Obviously avoid any firmware update messages if you encounter any as I could not guarantee that a firmware change via that software wouldn't prevent it from connecting to a tablet/phone.

Otherwise a post here in this thread may be useful as these guys have delved very deeply into the P3S remote - some very knowledgeable users amongst them

https://phantompilots.com/threads/i-bricked-my-p3s-with-ftp-range-mod-lets-fix-it-together.81432/

pljakobs, 2017-10-16:
Ruckusman, thanks, that's good advice, I'll see if I get around it on the weekend. I'm travelling all week.

pljakobs, 2017-10-28:
finally got to try the RC Assistant.

- installed RC assistant from the link above
- installed USB drivers (Windows 10 x64)
- R/C shows up as com port
- RC Assistant does not recognize it

not sure if there's anything missing locally or if the P3s R/C is just not supported..

notsolowki, 2017-10-28:
Idk if this will help you or not peter but i have managed to create the
same problem along time ago when i changed the parameters in flyc_params.
Did you change a bunch of settings in flyc_params? Im not even sure if you
have that option with p2

pljakobs, 2017-10-28:
@notsolowki I got this controller "as is" used. If the previous owner has changed anything or not, I am not aware. In fact, I am not even sure I can ftp into the controller, I tried today, but didn't get a valid directory listing (Firmware 1.9.20)