o-gs/dji-firmware-tools #59: Problem with extracting bin file

eagleman111, 2017-10-29:
Hi,
I am trying to extract bin file by running "python amba_fwpak.py -vv -x -m fw.bin" seems it fails and this is the result:
fw.bin: Opening for extraction
fw.bin: Header:
{   'crc32': '52503030',
    'model_name': b'xV4\x12'
                  b'\x01',
    'ver_info': '54.69-4349'}
fw.bin: Detection finished with entry larger than file; expecting 1 entries
fw.bin: Warning: Detection finished with unusual condition, verify files
fw.bin: After detection, expecting 1 entries
fw.bin: Entries:
[{   'crc32': '5053000C',
    'dt_len': 79}]
fw.bin: Post Header:
{   'part_size': '502b098b '
                 '0000098b '
                 '00000000 '
                 '00000000 '
                 '0001003a '
                 '010a0005 '
                 '000002b2 '
                 '0000a2e8 '
                 '0000a2e8 '
                 '42ec6231 '
                 '6f4cc0e1 '
                 'cd35c144 '
                 '76d1db20 '
                 '42ec6231 '
                 '6f4cc0e1'}
fw.bin: Warning: Invalid magic value in partition 0 header; will try to extract anyway.
fw.bin: Entry 0
{   'build_date': '1-00-29',
    'crc32': 'CD35C144',
    'dt_len': 16777226,
    'flag1': '0000A350',
    'flag2': 'CF3CF4DE',
    'magic': '0000A350',
    'mem_addr': '0000A59A',
    'padding': 'a1abed14 '
               'ab584309 '
               '76d47c45 '
               'cf3cf4de '
               'a1abed14 '
               'ab584309 '
               '76d47c45 '
               '00011019 '
               '03020ffe '
               '000148ea '
               '00140300 '
               '00140300 '
               '1363f4b6 '
               'c7e68699 '
               '158aa9ac '
               'ae3c7f83 '
               'aa01b1f7 '
               'f77bed82 '
               '02dd09fa '
               'c58b8de1 '
               '0001000b '
               '01010323 '
               '00154bea '
               '00008f5d '
               '00008f5d '
               'c89251f8 '
               'da4eedb2 '
               'efdc210b '
               'a1b53c3f '
               'c89251f8 '
               'da4eedb2 '
               'efdc210b '
               'a1b53c3f '
               '0001002b '
               '00070f00 '
               '0015db47 '
               '000056a0 '
               '000056a0 '
               '1179a648 '
               '99119989 '
               '2e383858 '
               '671d5267 '
               '1179a648 '
               '99119989 '
               '2e383858 '
               '671d5267 '
               '0001004b '
               '00070f00 '
               '001631e7 '
               '000056a0 '
               '000056a0 '
               '1179a648 '
               '99119989 '
               '2e383858 '
               '671d5267 '
               '1179a648',
    'version': '30417.56096'}
fw.bin: Extracting entry  0, pos      108, len 16777226 bytes
fw.bin: Warning: partition 0 truncated, 3210405 out of 16777226 bytes
fw.bin: Warning: Entry 0 data checksum mismatch; got 37CD7A27, expected CD35C144.
fw.bin: Warning: Entry 0 cummulative checksum mismatch; got 3C94CF30, expected 5053000C.
fw.bin: Warning: Entry 0 date makes no sense.
fw.bin: Warning: partition 0 header uses values from padded area in an unknown manner.
fw.bin: Warning: Total cummulative checksum mismatch; got C36B30CF, expected 52503030.

Any help?

mefistotelis, 2017-10-29:
Your ```fw.bin``` is clearly not the file you can extract with this tool.

eagleman111, 2017-10-29:
it is the "MATRICE600PRO_FW_V01.00.00.80.bin" file from "supported_firmwares.csv" from your repository

mefistotelis, 2017-10-29:
Why do you want to extract it with Ambarella extractor?

eagleman111, 2017-10-29:
I want to extract it to be able to edit parameters, and discover it

mefistotelis, 2017-10-29:
ok, then you should probably use a correct extractor. The firmware package you're trying was not made by Ambarella.

eagleman111, 2017-10-29:
mmmmm, so your tools are not compatible with m600pro?

mefistotelis, 2017-10-29:
All the firmwares within ```supported_firmwares.csv``` are supported.

eagleman111, 2017-10-30:
the "MATRICE600PRO_FW_V01.00.00.80.bin" file from "supported_firmwares.csv" however it is not extracting. Any help?

mefistotelis, 2017-10-30:
I'm getting mixed messages from you. The struggles you're having suggest you are not up to the task, but if you're working on Matrice platform, you should have at least a basic technical knowledge.

I will assume you just had a bad day or temporary mind freeze. Here's your answer:

Ambarella chips are used for cameras, like Action Cams and also the cameras mounted in DJI drones. What you have there is a firmware package for the whole DJI drone, not for Ambarella camera. The firmware package was made by DJI. You have to use DJI extractor on that firmware. If there is a camera module firmware within the DJI firmware package, then you could extract it as next step - but you mentioned you want to edit flyc_params, which has nothing to do with camera.

After you extract the modules, you can look at this project wiki for details of each module.

eagleman111, 2017-10-30:
1-I got you, so all dji drones which dont have camera installed with it, no need to use amba_&.py, right?
2- I extracted the bin file and now have the modules, ande trying to extract the parameter file by this command "dji_flyc_param_ed.py -vv -x -m fw_m0306.bin"
It show me that "Flight controller parameters array signature not detected in input file"
This is the log file:
fw_m0306.bin: Opening for extraction
Traceback (most recent call last):
  File "dji_flyc_param_ed.py", line 630, in <module>
    main(sys.argv[1:])
  File "dji_flyc_param_ed.py", line 611, in main
    flyc_extract(po,fwmdlfile)
  File "dji_flyc_param_ed.py", line 475, in flyc_extract
    raise ValueError("Flight controller parameters array signature not detected in input file.")
ValueError: Flight controller parameters array signature not detected in input file."

mefistotelis, 2017-10-30:
Do you have the module unencrypted?

eagleman111, 2017-10-31:
I dont think so, all what i did is extract the bin file, then run the "dji_flyc_param_ed.py -vv -x -m" on "fw_m0306.bin" file, Do I have to decryot it first? and how?

mefistotelis, 2017-10-31:
We don't have a key. You need decrypted module.  Hope you are reading details about this module on the wiki.

eagleman111, 2017-11-01:
I read alot wiki helped me alot, but not in decrypting it :( Any help?

mefistotelis, 2017-11-01:
Have you seen the word "unencrypted" in some places?

closed 2017-12-16 by mefistotelis