o-gs/dji-firmware-tools #60: Phantom 3 Pro USB board (GL300A version)

familienhansen, 2017-11-23:
OK. So I got this non working USB board and would like to flash it, but Im not quite sure how. I know how to hook it up to the TTL adaptor though.

I previously flash a bad wifi module on my P2 Vision+ with the TTL adaptor, Tera term, and a guide. 
Im expecting this to be almost the same, but Im missing the guide that points me to the firmware files, and the commands to flash it.

Is it possible to get some pointers in here?

I also have a bad gimbal top board from a Phantom 3 pro. Has anyone succeeded flashing them?

Thanks in advance!

mefistotelis, 2017-11-23:
There are instructions for flashing m1300 in the "Wiki" section of this project. The procedure for m0800 is identical; we don't have images of partitions which are not included in FW update though.

familienhansen, 2017-11-23:
Thank you! 
Yes, that's about as far I got. Its just that I'm totally blank regarding how to get the images. I'm on a win7 box, and have absolutely no knowledge of Linus or Linux commands. Is that required, or is there another way?

mefistotelis, 2017-11-23:
You can use Python scripts on Windows too.
The instruction does contain links to alternatives. Just read it fully.

familienhansen, 2017-11-23:
Thanks. Will reread it and see if it makes more sense this time. I hope its OK if I return though :-)

Thumbs up for all your work with this. Quite impressive!

closed 2017-12-16 by mefistotelis