o-gs/dji-firmware-tools #61: P3A FIRMWARE BRICKED - NEED HELP

CezarBR, 2017-11-25:
Hello friends, I will try to summarize the situation of my Phantom 3 Advanced for you and I apologize if any words sound strange, I am using the translator.

to a few days I had a bad behavior in the application DJI GO forcing me to restart the application, after that I realized that the application asked for the update of the RC control firmware. and also noticed that the AC aircraft also exhibited different behavior, it did not connect to the RC remote control. I tried several procedures before taking the decision to force the downgrade of the aircraft, and I did it many times, I made several attempts of downgrades and unsuccessfully all upgrade ... Now my aircraft when it is turned on blinks twice the gimbal led and totally extinguishes , it does not read any firmware package and the D - DD D - DD beep as if the aircraft was connected to a computer by the USB cable. PLEASE HELP ME, THE DRONE IS MY WAY TO MAKE MONEY AND I'M 10 DAYS STOPPED ....

THANK YOU ALL

mefistotelis, 2017-11-25:
Try not to shout in the title.

We need DAT log from the aircraft, and last update log from the camera.

CezarBR, 2017-11-25:
thousand apologies for the title screaming.
the current status of the aircraft is as follows:
the aircraft is not reading any firmware, when connecting the aircraft it gimbal led two times and completely erases, regardless of whether there is any * .bin file in the SDCARD.
I tried several times to force upggrades and downgrades and yesterday after I forge the downgrade to version v1.1.9 downloaded from this link: http://download.dji-innovations.com/downloads/phantom_3/en/Phantom_3_Advanced_Firmware_v1.1.9_en.zip
after that the gimbal led has cleared and it does not read any other firmware package nor does it produce any log.
--------
One of the last logs that I could save was more than 3 days ago and from that last log I tried several times the downgrades until I was as I mentioned above.
--------- log

https://drive.google.com/file/d/1aWE1zsFkTUJnCjRbYklfQ6b_0laZvQ9L/view?usp=drivesdk      <-----------link to last saved log file

I apologize if I am speaking strangely or repetitively, but I am using a translator and the words do not always go the way I want and I am trying to be very clear not to confuse voices. I would like to know if there is any way to reinstall the firmware from an application for computers with win10, and I would also like to know what the USB port that is located on the drone / front of the drone is. I'm super lay in the subject I only know to follow tutorials. so again, I apologize

CezarBR, 2017-11-26:
Hi, I have news ... I have made a great progress. created courage and dismounted my aircraft, so I started with the gimbal and after that I dismounted that part where the sonar and the positioning camera are and I discovered that this part was too tight, I loosened the screws of it and my aircraft connected to the RC. PERFECT .... but when checking which firmware was installed and I saw 1.1.9+ .. the memory card is not reading. does not start any memory card upgrade. the gimbal led remains off. only blinks when the drone is turned on but soon goes out.

CezarBR, 2017-11-26:
maybe some way to update using the windows command prompt through the USB cable. almost working ...

mefistotelis, 2017-11-26:
The sd-card is controlled by Ambarella chip. Since we don't have any documentation of that chip, we don't know how to re-program it.

Your best bet is to find a new [Gimbal top board](https://github.com/mefistotelis/phantom-firmware-tools/wiki/WM320-Gimbal-top-board) and replace it.

If you think you have enough skill to figure out how to program it - that requires soldering wires to the service pads with serial interface and entering bootloader mode.

MrBurnsAT, 2017-11-26:
I think his problem is that he downgradet below 1.2 without MVOM0Fw.bin

Now i think ofmd and Gimbal cant communicate and thats the problem 

I would try to install DM368Fw.bin (or maybe PDM368Fw3.bin) from 1.2

After that it maybe can communicate again

CezarBR, 2017-11-26:
I really do not have any skills ... I do not even have a fingernail of the skill and knowledge that you have. I contacted you because on those days I was trying to "fix" my drone this forum was the most complete I found all over the internet ... many people came to tell me that my problems were hardware and I was able to prove that no ... but one of the mistakes I made was to force downgrade to version 1.1.9 where it ended up disconnecting the top of the gimbal. I really apologize for not understanding much about and I'm sure that without your help I will not be able to do much

mefistotelis, 2017-11-26:
> I would try to install DM368Fw.bin (or maybe PDM368Fw3.bin) from 1.2

There is no DM368 module in Ph3 Adv. It would be hard to update non-existing hardware.

> I really do not have any skills ...

Then buy a new gimbal, or the gimbal top board if you are able to disconnect the old one and connect a replacement.

> I really apologize for not understanding much about and I'm sure that without your help I will not be able to do much

Note that users here are not interested in teaching someone with no knowledge. If you wish to learn electronics, there are courses for that. If you're looking for people more interested in sharing the basics on Phantom drones, go to phantompilots.com.

What you are trying to do is possible, but noone did it before. There is no instruction.

CezarBR, 2017-11-26:
Ok thx

CezarBR, 2017-11-26:
Danke für Ihre Freundlichkeit. Bleib bei Gott

closed 2017-11-26 by CezarBR