o-gs/dji-firmware-tools #62: Flash m0800 kernel to P3P gimbal topboard

familienhansen, 2017-12-07:
Just a quick question.. 

Can you put the kernel on the SD-card and flash it from there, or do you have to solder on a USB connector?

Thanks in advance!

mefistotelis, 2017-12-07:
If both linux'es (Ambarella and daVinci) are working, you can use sd-card slot for Ambarella:
https://github.com/mefistotelis/phantom-firmware-tools/wiki/Firmware-m0800#by-firmware-module-in-camera

If not, you have to either solder the slot or use transfer over serial (yModem). For kernel (<4MB) - it will take like 20 minutes, but is doable. For UbiFS (~100MB) - transfer over serial becomes almost impossible.

familienhansen, 2017-12-07:
Awesome. Thanks! (again)

familienhansen, 2017-12-07:
Yikes. I guess I need to transfer the file with yModem...

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds
Dji-Pro #
Unknown command 'ry 'help'
Dji-Pro # mw.b 0x80008000 0xFF 0x460000
Dji-Pro # usb start
(Re)start USB...
USB:   scanning bus for devices... No USB Device found
       scanning bus for storage devices... 0 Storage Device(s) found
Dji-Pro #

familienhansen, 2017-12-07:
Done! Only took like 10 mins with yModem, so not a big problem :-)

I just noticed this:

Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Is it something to worry about, or is it normal?

mefistotelis, 2017-12-07:
> *** Warning - bad CRC, using default environment

This is normal, you should not save env.

familienhansen, 2017-12-07:
Cool. Guess I just need to test it now. Thank you!

closed 2017-12-16 by mefistotelis