o-gs/dji-firmware-tools #63: Data / flight data

aka1ceman, 2017-12-23:
With all the new worry about cybersecurity, is there a way we can look further into the Phantom to see if and where flight data records are recorded and what if any gets sent back to DJI?

mefistotelis, 2017-12-23:
> is there a way we can look further into the Phantom to see if and where flight data records are recorded

For flight records on SD-card - we can fully read them with Wireshark dissectors.

For other info received by the RC and mobile device - we can capture internal communication between drone components, and again look at it with Wireshark. The internal packets are often called DUPC or DUML.

> what if any gets sent back to DJI?

Some people are capturing all packets sent by the Dji Go app, others are decompiling the app to look into specifics there.

We know that Dji Go sends a lot of data to several chinese companies and services.
We also know that for several months now  the new firmwares are broadcasting drone ID info when flying.

Dji recently said both of these will have "opt-out" option. I don't know whether this option was added, as I'm using old software.

aka1ceman, 2017-12-23:
If I am to sell my out of warranty unit, and want privacy, how hard would it be to wipe the data?

closed 2021-06-15 by mefistotelis