o-gs/dji-firmware-tools #64: Need u-boot, ubl and kernel images for P3C and P3S

okstuv, 2017-12-29:
I've successfully extracted and decrypted the m0800 module from P3X firmware bins, but don't see an equivalent module in P3S or P3S firmware bins.

Can I simply use the u-boot, ulmage and ubl_297m_1 from the /dji/data/ folder extracted from m0800 in P3X firmware to flash my failed P3S and P3C Davinci SoC?

BTW, It looks like the NAND location and Davinci SoC's boot up and kernel hasn't changed much at all since LB1 and P2V+ hardware...

mefistotelis, 2017-12-29:
You can only update hardware which exists. Where exactly did you found Davinci chip in P3C or P3S?

Make sure to read all the related wikis, ie:
https://github.com/mefistotelis/phantom-firmware-tools/wiki/WM320-Gimbal-top-board#variants

okstuv, 2017-12-29:
Thx for the reply. 

After going over the hardware again in the wiki (WM321 for P3C) and looking over the boards on hand, looks like you're right. That secondary video processing (and therefore the Davinci chip) hardware isn't on the P3C. My assumption was based on how P2 Vision Plus has a Davinci chip, so I figured all P3's did too...

Re: P3X, it doesn't look like a header needs to be skipped/removed for uImage when re-flashing ther kernel. Please confirm.

Any tips on debugging the P3C's ARM kernel bootup? I haven't been able to dig up much other than P3Pro resources like yours and try to generalize them to P3Std. In fact haven't nailed down what pads to connect to for serial connection...

mefistotelis, 2017-12-29:
>  P3X, it doesn't look like a header needs to be skipped/removed for uImage when re-flashing ther kernel. 

That it true. Only u-boot needs header removal.

> Any tips on debugging the P3C's ARM kernel bootup?

You mean Ambarella? There are several ARM chips within the drone.
So you probably want to tap to this board:
https://github.com/mefistotelis/phantom-firmware-tools/wiki/WM321-Camera-encoder-board

To power the board outside of the drone, you probably only need 5V.
If you board doesn't have names of service pads, you can check the names on photo on the wiki.

When you have any findings, make sure to share them - there is limited info on P3C because noone shared it. I will update the wiki to any findings.

okstuv, 2017-12-29:
Ok, thanks. Will work on the p3c Ambarella and share what I find.

https://github.com/mefistotelis/phantom-firmware-tools/wiki/WM321-Camera-encoder-board

okstuv, 2017-12-30:
a bit off topic for this issue, but tangentially relevant: where should i look for extracting binaries off the NAND associated with the DaVinci processor?

I applied some of the P3X Davinci wiki article to fixing the Video Transmission module of my P2VP and was able to reflash the kernel, only to get stuck at bootup due to a bad encrypted partition:

> If the encrypted partition gets damaged, Linux will stop booting - kernel module responsible for encryption will freeze. In such case, the system freezes soon after the message:
> 
>    encrypt device:at88 found

I don't have the binary on hand for the Phantom 2 vision plus DM365, but I'm game to extracting it off a good module if you point me to the right direction on how to undertake that endeavor.

Below is the before and after where I had a bad kernel, then fixed that only to find the encrypted partition needs flashing too:

Before and successful serial kernel upload:
`Dji-Pro # DM36x initialization passed!
UBL Product Vesion : DJI-GSP-UBL-1.0-rc10(2014-08-15)
Dji UBL Version: 1.51(Aug 15 2014 - 17:05:12)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
Uboot Checksum:0x7E25B44B
Actua Checksum:0xA4F61296
Checksum error ...
Reset ...DM36x initialization passed!
UBL Product Vesion : DJI-GSP-UBL-1.0-rc10(2014-08-15)
Dji UBL Version: 1.51(Aug 15 2014 - 17:05:12)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
Uboot Checksum:0x7E25B44B
Actua Checksum:0x7E25B44B

U-Boot Product Vesion : DJI-GSP-Uboot-1.0-rc4(2014-07-23)
U-Boot 2010.12-rc2-svn-Dji (Jul 23 2014 - 11:14:40)
Cores: ARM 432 MHz
DDR:   297 MHz
I2C:   ready
DRAM:  128 MiB
NAND:  128 MiB
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
*** Warning - bad CRC, using default environment

Net:   Ethernet PHY: GENERIC @ 0xff
DaVinci-EMAC
Press ESC to abort autoboot in 1 seconds

Loading from nand0, offset 0x4a0000
** Unknown image type
Wrong Image Format for bootm command
ERROR: can't get kernel image!

Loading from nand0, offset 0x900000
** Unknown image type
Wrong Image Format for bootm command
ERROR: can't get kernel image!
Dji-Pro #
Dji-Pro # loady 0x80008000 115200
## Ready for binary (ymodem) download to 0x80008000 at 115200 bps...
CCxyzModem - CRC mode, 1(SOH)/3949(STX)/0(CAN) packets, 4 retries
## Total Size      = 0x003db260 = 4043360 Bytes
Dji-Pro #
Dji-Pro # nand erase 0x04a0000 0x460000

NAND erase: device 0 offset 0x4a0000, size 0x460000
Erasing at 0x8e0000 -- 100% complete.
OK
Dji-Pro # nand write 0x80008000 0x04a0000 0x460000

[... 134 lines cut ...]
[    1.530000] ths7303 1-002c: chip found @ 0x58 (DaVinci I2C adapter)
[    1.540000] ths7303 1-002c: ths7303 write failed
[    1.550000] ths7303: probe of 1-002c failed with error -121
[    1.550000] tvp5150_init
[    1.560000] vpfe_init
[    1.560000] vpfe-capture: vpss clock vpss_master enabled
[    1.560000] vpfe-capture vpfe-capture: v4l2 device registered
[    1.570000] vpfe-capture vpfe-capture: video device registered
[    1.580000] tvp5150_probe
[    1.580000] tvp5150 1-005d: tvp5150 1-005d decoder driver registered !!
[    1.590000] vpfe-capture vpfe-capture: v4l2 sub device tvp5150 registered
[    1.600000] vpfe-capture vpfe-capture: v4l2 sub device cat6023 register fails
[    1.610000] vpfe_register_ccdc_device: DM365 ISIF
[    1.610000] DM365 ISIF is registered with vpfe.
[    1.620000] Trying to register davinci display video device.
[    1.620000] layer=c17f6c00,layer->video_dev=c17f6d64
[    1.630000] Trying to register davinci display video device.
[    1.630000] layer=c17f7000,layer->video_dev=c17f7164
[    1.640000] davinci_init:DaVinci V4L2 Display Driver V1.0 loaded
[    1.650000] watchdog watchdog: heartbeat 1 sec
[    1.650000] TCP cubic registered
[    1.660000] NET: Registered protocol family 17
[    1.660000] lib80211: common routines for IEEE802.11 drivers
[    1.670000] ksocket init
**[    2.310000] encrypt device:at88 found**`

mefistotelis, 2017-12-30:
> where should i look for extracting binaries off the NAND associated with the DaVinci processor?

If you can connect USB to your DaVinci - mount a SD card reader and simply dd the whole NAND to sd-card, ie:
```
    dd if=/dev/mtd3 of=/tmp/sdcard/mtd_whole bs=1048576
```

If no USB - transfer it over serial. Use ModemY. Instead of one huge transfer which will take hours, divide the file to parts and transfer part at a time. It is best not to restart until all parts are extracted, otherwise UbiFS will need repairing.

```
cd /tmp

echo '#!/bin/sh
for N in $(seq 0 7)
do
    dd if=/dev/mtd3 of=/tmp/mtd_part$N bs=1048576 count=16 skip=$(($N * 16))
    lsz --ymodem -b -8 /tmp/mtd_part$N
    rm /tmp/mtd_part$N
done' > do.sh
chmod a+x do.sh

./do.sh
```

After you have all the parts, you can merge them together and extract proper partitions, using ```dd``` with proper ```skip=```. Flash map for Ph3 gimbal is here, your is likely identical:
https://github.com/mefistotelis/phantom-firmware-tools/wiki/Firmware-m0800#structure

okstuv, 2017-12-30:
Will try to find USB information for this board, else the ymodem approach. 

Thanks for the pointers and code, much appreciated!

okstuv, 2017-12-31:
I'm stumped or too dumb for this...

I hit escape during the Davince bootup to get to u-boot prompt and I'm limited to u-boot's very limited command set (listed below). As you can see no support for DD or CD, etc... 

```
Dji-Pro # help
?       - alias for 'help'
askenv  - get environment variables from stdin
base    - print or set address offset
boot    - boot default, i.e., run 'bootcmd'
bootd   - boot default, i.e., run 'bootcmd'
bootm   - boot application image from memory
bootp   - boot image via network using BOOTP/TFTP protocol
chpart  - change active partition
cmp     - memory compare
coninfo - print console devices and information
cp      - memory copy
crc32   - checksum calculation
dhcp    - boot image via network using DHCP/TFTP protocol
echo    - echo args to console
editenv - edit environment variable
env     - environment handling commands
exit    - exit script
false   - do nothing, unsuccessfully
fatinfo - print information about filesystem
fatload - load binary file from a dos filesystem
fatls   - list files in a directory (default /)
go      - start application at address 'addr'
help    - print command description/usage
i2c     - I2C sub-system
iminfo  - print header information for application image
imxtract- extract a part of a multi-image
itest   - return true/false on integer compare
loadb   - load binary file over serial line (kermit mode)
loads   - load S-Record file over serial line
loady   - load binary file over serial line (ymodem mode)
loop    - infinite loop on address range
md      - memory display
mdc     - memory display cyclic
mm      - memory modify (auto-incrementing address)
mtdparts- define flash/nand partitions
mtest   - simple RAM read/write test
mw      - memory write (fill)
mwc     - memory write cyclic
nand    - NAND sub-system
nboot   - boot from NAND device
nfs     - boot image via network using NFS protocol
nm      - memory modify (constant address)
ping    - send ICMP ECHO_REQUEST to network host
printenv- print environment variables
reset   - Perform RESET of the CPU
run     - run commands in an environment variable
saveenv - save environment variables to persistent storage
saves   - save S-Record file over serial line
setenv  - set environment variables
showvar - print local hushshell variables
sleep   - delay execution for some time
source  - run script from memory
test    - minimal test like /bin/sh
tftpboot- boot image via network using TFTP protocol
true    - do nothing, successfully
ubi     - ubi commands
usb     - USB sub-system
usbboot - boot from USB device
version - print monitor version

```

mefistotelis, 2018-01-05:
u-boot has limited support of dumping things. the ```dd```  command and script is to be used from within Linux. use ```stop```  to get clear command prompt, without all the transcoding services.

okstuv, 2018-01-05:
Great. Thanks again. I'll give this a try.

ttkirst, 2018-06-12:
Can someone please assist me, I need a full NAND dump of the p2v+ WiFi module as almost everything is corrupt and I’ve tried flashing various other images without luck. Root fs is also corrupt... Any help would be greatly appreciated!

ttkirst, 2018-06-12:
Just for further information... it is a WM301 DM368 unit with the 432Mhz and 297Mhz processor and RAM...

mefistotelis, 2021-06-14:
DaVinci flashing is now explained here:
https://github.com/o-gs/dji-firmware-tools/wiki/Flashing-firmware-on-DaVinci-media-processors

closed 2021-06-14 by mefistotelis

guewald, 2024-05-09:
I'm getting stuck on ( encrypt device:at88 found ) and i'm unable to stop kernel loadingto write comands.... power the boar pressing esc but it loads and after this i'm unable to write any command... using extra putty.

mefistotelis, 2024-05-09:
RX wire good, TX wire bad?

guewald, 2024-05-09:
all wires good... check and resolder many times...
its funny that on the initialization log the ESC optio to abort boot does not apear

mefistotelis, 2024-05-09:
Then your only option is to flash an u-boot which is configured to allow the key control.

guewald, 2024-05-09:
i will try another versions, tks

guewald, 2024-05-10:
No success .... i have 3 versions... two of them does not allow stop boot and one of them dont boot 

`OK
OK

Starting kernel ...

[    0.000000] Kernel Product Vesion : DJI-GSP-Kernel-1.1-rc1(2014-09-15)
[    0.000000] Linux version 2.6.32.17-davinci1 (root@dji) (gcc version 4.3.3 (Sourcery G++ Lite 2009q1-203) ) #3 PREEMPT Mon Sep 15 13:24:10 HKT 2014
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: DaVinci DM36x EVM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] DaVinci dm36x_rev1.2 variant 0x8
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 12192
[    0.000000] Kernel command line: console=ttyS0,115200n8 rw dm365_imp.oper_mode=0 video=davincifb:vid0=0,10K:vid1=0,10K:osd0=1920x1080X16,8100K mem=48MB davinci_enc_mngr.ch0_output=COMPOSITE davinci_enc_mngr.ch0_mode=pal ubi.mtd=2,2048 root=ubi0:rootfs rootfstype=ubifs ip=off lpj=1077248
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 48MB = 48MB total
[    0.000000] Memory: 44448KB available (3648K code, 447K data, 116K init, 0K highmem)
[    0.000000] SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:245
[    0.000000] Console: colour dummy device 80x30
[    0.000000] Calibrating delay loop (skipped) preset value.. 215.44 BogoMIPS (lpj=1077248)
[    0.000000] Mount-cache hash table entries: 512
[    0.000000] CPU: Testing write buffer coherency: ok
[    0.000000] DaVinci: 8 gpio irqs
[    0.000000] NET: Registered protocol family 16
[    0.030000] EVM: tvp5146 SD video input
[    0.110000] bio: create slab <bio-0> at 0
[    0.110000] DM365 IPIPE initialized in Continuous mode
[    0.120000] SCSI subsystem initialized
[    0.120000] usbcore: registered new interface driver usbfs
[    0.120000] usbcore: registered new interface driver hub
[    0.120000] usbcore: registered new device driver usb
[    0.130000] vpss vpss: dm365_vpss vpss probed
[    0.130000] vpss vpss: dm365_vpss vpss probe success
[    0.130000] dm365_afew_hw_init
[    0.140000] ch0 default output "COMPOSITE", mode "PAL"
[    0.140000] VPBE Encoder Initialized
[    0.140000] cfg80211: Using static regulatory domain info
[    0.140000] cfg80211: Regulatory domain: US
[    0.140000]  (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp)
[    0.140000]  (2402000 KHz - 2472000 KHz @ 40000 KHz), (600 mBi, 2700 mBm)
[    0.140000]  (5170000 KHz - 5190000 KHz @ 40000 KHz), (600 mBi, 2300 mBm)
[    0.140000]  (5190000 KHz - 5210000 KHz @ 40000 KHz), (600 mBi, 2300 mBm)
[    0.140000]  (5210000 KHz - 5230000 KHz @ 40000 KHz), (600 mBi, 2300 mBm)
[    0.140000]  (5230000 KHz - 5330000 KHz @ 40000 KHz), (600 mBi, 2300 mBm)
[    0.140000]  (5735000 KHz - 5835000 KHz @ 40000 KHz), (600 mBi, 3000 mBm)
[    0.140000] cfg80211: Calling CRDA for country: US
[    0.140000] LogicPD encoder initialized
[    0.140000] Switching to clocksource timer0_1
[    0.150000] musb_hdrc: version 6.0, pio, host, debug=0
[    0.170000] musb_hdrc: USB Host mode controller at fec64000 using PIO, IRQ 12
[    0.170000] musb_hdrc musb_hdrc: MUSB HDRC host driver
[    0.170000] musb_hdrc musb_hdrc: new USB bus registered, assigned bus number 1
[    0.170000] usb usb1: configuration #1 chosen from 1 choice
[    0.170000] hub 1-0:1.0: USB hub found
[    0.170000] hub 1-0:1.0: 1 port detected
[    0.170000] NET: Registered protocol family 2
[    0.170000] IP route cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.170000] TCP established hash table entries: 2048 (order: 2, 16384 bytes)
[    0.170000] TCP bind hash table entries: 2048 (order: 1, 8192 bytes)
[    0.170000] TCP: Hash tables configured (established 2048 bind 2048)
[    0.170000] TCP reno registered
[    0.170000] NET: Registered protocol family 1
[    0.170000] RPC: Registered udp transport module.
[    0.170000] RPC: Registered tcp transport module.
[... 76 lines cut ...]
[    6.660000] ths7303 1-002c: ths7303 write failed
[    6.660000] ths7303: probe of 1-002c failed with error -110
[    6.670000] tvp5150_init
[    6.670000] vpfe_init
[    6.680000] vpfe-capture: vpss clock vpss_master enabled
[    6.680000] vpfe-capture vpfe-capture: v4l2 device registered
[    6.690000] vpfe-capture vpfe-capture: video device registered
[    6.700000] tvp5150_probe
[    6.700000] tvp5150 1-005d: tvp5150 1-005d decoder driver registered !!
[    6.700000] vpfe-capture vpfe-capture: v4l2 sub device tvp5150 registered
[    6.710000] vpfe-capture vpfe-capture: v4l2 sub device cat6023 register fails
[    6.720000] vpfe_register_ccdc_device: DM365 ISIF
[    6.730000] DM365 ISIF is registered with vpfe.
[    6.730000] Trying to register davinci display video device.
[    6.740000] layer=c17f6c00,layer->video_dev=c17f6d64
[    6.750000] Trying to register davinci display video device.
[    6.750000] layer=c17f7000,layer->video_dev=c17f7164
[    6.760000] davinci_init:DaVinci V4L2 Display Driver V1.0 loaded
[    6.760000] watchdog watchdog: heartbeat 1 sec
[    6.770000] TCP cubic registered
[    6.770000] NET: Registered protocol family 17
[    6.780000] lib80211: common routines for IEEE802.11 drivers
[    6.780000] ksocket init
[    7.710000] encrypt device:at88 found
`