o-gs/dji-firmware-tools #65: Flashing Adv firmware over Pro - is it possible ?

petr222, 2018-01-05:
Hi, we have several Phantom 3 Advanced drones which were upgraded
to Pro by changing gimbals or camera control boards and flashing Pro firmware.
After a several hours flashing of firmware, all Pro camera control boards NAND died. 
Therefore we bought some Advanced control boards where is no NAND
memory to fail. But there is our problem:

It was possible to change from Adv to Pro by SD card with P3X firmware, which was 
flashed. But Phantom 3 with Professional firmware ignore P3S (Advanced) BIN file and it ignore
all known debug files.

Is there a way how to flash Advanced BIN file to board which was flashed with Pro BIN file?
If it works from Adv to Pro, it should work from Pro to Ad.

In DJI Go there is no error, only message that camera firmware update is required.

mefistotelis, 2018-01-05:
Have you tried copying the file, providing both "P3X*.bin" and "P3S*.bin" on the sd-card?

If this won't be enough, open the "P3S*.bin" file with hex editor and change the string at byte 28 from "P3X" to "P3S".

petr222, 2018-01-05:
First one - which I am trying to repair now - is genuine Professional with gimbal where control board is replaced by Advanced version. It look like camera is not correctly recognized - no picture, no camera control and it look like drone does not see SD card either. 

Is there a picture available which resistors must be desoldered and where they must be soldered ?

Is it possible that without moving these resistors camera does not work ? All three Advanced cameras
have same problem, therefore I think it is not a problem of bad Advanced control board.

Second drone - which we will try tomorrow - is genuine Advanced with Pro firmware flashed in (that's when two Professional control boards died). I will try same Advanced cameras on it tomorrow.

mefistotelis, 2018-01-05:
These boards can be switched and it causes no issues, as long as FW on the new top board is the same as on the rest of the drone.

Which resistors - you'd have to compare photos of the boards. The resistors are near USB multiplexer chips. I made partial schematics of it here:
https://github.com/mefistotelis/dji-hardware-schematics

petr222, 2018-01-06:
I tried Advanced control board on second drone (Advanced converted to Professional) and there is same problem - no picture, no camera control, no sd card recognised.

As camera upgrade is required, I will try to lower drone firmware (using damaged Pro control board) to try if it help.

petr222, 2018-01-06:
Look like Adv control board nr 1. have a problem, as nr 2, started to flash Advanced firmware to Advanced body (with Pro firmware installed). 

Is there way to diagnose gimbal control boards ? Drone say camera firmware is required, but camera and sd card is not recognized, therefore flashing firmware is not possible. Green LED is flashing on gimbal, red start to flash after a while. No error in DJI Go, only message that camera firmware update is required.

Firmware version in gimbal control board is unknown, drone firmware is 1.10.

mefistotelis, 2018-01-06:
I'm pretty sure the camera will update its own firmware, even if not connected to the rest of the drone.

To diagnose, solder to serial interface of the Ambarella chip.

petr222, 2018-01-06:
Is serial interface similar to this guide ?
https://phantompilots.com/threads/fixing-flash-in-dm365-chip-within-ph3-pro-gimbal.127310/

mefistotelis, 2018-01-06:
> Is serial interface similar to this guide ?

Serial interface similar to a guide? no.

Please read my wiki on m0100, especially the "interfaces" section.

petr222, 2018-01-06:
Now I propably get it - U0_T and U0_R (3.3V UART of Ambarella chip; allows access to Ambarella system console)?

About these resistors near USB and Ambarella and  DaVinci - does it mean that Pro control board can be turned to Advanced by moving these resistors ? As faulty NAND is used to boot DaVinci which is not used in Advanced ersion.

mefistotelis, 2018-01-06:
> About these resistors near USB and Ambarella and DaVinci - does it mean that Pro control board can be turned to Advanced by moving these resistors ?

I believe yes. I never tried it though. I've seen two resistors, but there may be more.

petr222, 2018-01-07:
Is in necessary to assemble whole gimbal to flash firmware?
Or it look like Pro control board does not like Advanced firmware.

 [00000821][01 00] Firmware upgrade finished failed (step = 3, err = 0xffffffff).

mefistotelis, 2018-01-07:
> Is in necessary to assemble whole gimbal to flash firmware?

I don't know. Very likely yes, because without the rest of the drone Ambarella is continuously restarting.

petr222, 2018-01-13:
With serial port output it look like it does endless circle
without visible error - boot, flash, reboot, boot, flash, reboot, boot, flash, reboot.....

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 396000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 30857142
SD1    freq: 24000000
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x4 0x789aedcf
System WARM boot.
Detecting done ......

[00000000][CA9_0] [DJI] Start WDT of 20 seconds for boot up......

[00000000][CA9_0] [Version 1.0.11824], Oct 16 2015, 14:06:46

[00000000][CA9_0] [DJI] SD Updater Version [v01.23.4920] Build Mar  4 2016 16:57:05

[00000328][CA9_0] [DJI] Stop WDT.

[00000328][CA9_0] [DJI] Firmware updating......

[00000329][CA9_0] Card In Slot.
[00000630][CA9_0] SD clock source, frequency = 216000000
[00000630][CA9_0] SDCalcClk for SD card initiation:: Actual Clk = 140625, pCLK = 128
[00000648][CA9_0] start to check media
[00000654][CA9_0] AmbaSD_InitCard - SD_CARD OK
[00000659][CA9_0] Packet Detected / c:\P3S_FW_V01.11.0020.bin /.

[00000761][CA9_0] [DJI] Try Ddr Param 0.

[00000762][CA9_0] Back up partion data ......
[00036810][CA9_0] Image "System Software" is found!
[00036810][CA9_0] Image "DSP uCode" is found!
[00036810][CA9_0] Image "System ROM Data" is found!
[00036810][CA9_0] Image "Linux Kernel" is found!
[00036810][CA9_0] Image "Linux Root FS" is found!
[00036810][CA9_0] Start firmware CRC check...

[00036810][CA9_0] Checking System Software
[00040190][CA9_0] Verifying image CRC ... done
[00040190][CA9_0]       length:         9154368
[00040190][CA9_0]       crc32:          0x845822c8
[00040190][CA9_0]       ver_num:        1.0
[00040190][CA9_0]       ver_date:       2016/9/23
[00040190][CA9_0]       img_len:        9154112
[00040190][CA9_0]       mem_addr:       0x0e600000

[00040190][CA9_0] Checking DSP uCode
[00041624][CA9_0] Verifying image CRC ... done
[00041624][CA9_0]       length:         3889408
[00041624][CA9_0]       crc32:          0x4f07273c
[00041624][CA9_0]       ver_num:        1.0
[00041624][CA9_0]       ver_date:       2016/9/23
[00041624][CA9_0]       img_len:        3889152
[00041624][CA9_0]       mem_addr:       0x0ff00000

[00041624][CA9_0] Checking System ROM Data
[... 205 lines cut ...]
[00048548][CA9_0]       img_len:        5038608
[00048548][CA9_0]       mem_addr:       0x00508000

[00048548][CA9_0] Checking Linux Root FS
[00058505][CA9_0] Verifying image CRC ... done
[00058505][CA9_0]       length:         27001088
[00058505][CA9_0]       crc32:          0x65014d0d
[00058505][CA9_0]       ver_num:        1.0
[00058505][CA9_0]       ver_date:       2016/9/23
[00058505][CA9_0]       img_len:        27000832
[00058506][CA9_0]       mem_addr:       0x00001000

[00058983][CA9_0] Program "System Software" to NAND flash ...
[00062210][CA9_0] Done
[00062414][CA9_0] Program "DSP uCode" to NAND flash ...
[00063764][CA9_0] Done
[00064369][CA9_0] Program "System ROM Data" to NAND flash ...
[00068394][CA9_0] Done
[00068659][CA9_0] Program "Linux Kernel" to NAND flash ...
[00070419][CA9_0] Done
[00071846][CA9_0] Program "Linux Root FS" to NAND flash ...
[00081188][CA9_0] Done
[00081521][CA9_0] [DJI] Done!

[00081521][CA9_0] [DJI] Reboot......

petr222, 2018-01-14:
Tried second Adv board - there is nothing on serial output from Ambarella,

Measured all Voltage test pads - correct Voltage is here.

mefistotelis, 2018-01-14:
> With serial port output it look like it does endless circle without visible error - boot, flash, reboot,..

Do you have SD-card with firmware update inserted, or is the endless update circle going on its own, without sd-card?

It looks like the update was successful, so all we need to do is to end the circle.

petr222, 2018-01-14:
Yes, there is a bin on the SD card.

Without firmware, it does this:

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 396000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 30857142
SD1    freq: 24000000
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x4 0x789aedcf
System WARM boot.
Detecting done ......

[00000000][CA9_0] [DJI] Start WDT of 20 seconds for boot up......

[00000000][CA9_0] [Version 1.0.11824], Oct 16 2015, 14:06:46

[00000000][CA9_0] [DJI] SD Updater Version [v01.23.4920] Build Mar  4 2016 16:57:05

[00000328][CA9_0] [DJI] Stop WDT.

[00000328][CA9_0] [DJI] Firmware updating......

[00000328][CA9_0] Card In Slot.
[00000629][CA9_0] SD clock source, frequency = 216000000
[00000629][CA9_0] SDCalcClk for SD card initiation:: Actual Clk = 140625, pCLK = 128
[00000647][CA9_0] start to check media
[00000653][CA9_0] AmbaSD_InitCard - SD_CARD OK
[00000658][CA9_0] Can't detect firmware.
[00000699][CA9_0] Firmware  program fail--- ARM DATA ABORT EXCEPTION ---
Oops: CPU Exception!
pc : [<0001134c>]    lr : [<00000000>]
sp : 0006f328  ip : 00000000  fp : 00000000
r10: 0006b400  r9 : 00000000  r8 : 00000000
r7 : 00000000  r6 : 00000000  r5 : 00000000
r4 : 0006f394  r3 : 00000000  r2 : 00000000
r1 : fff90c5b  r0 : ffffffa4
Flags: NzCv
IRQs on  FIQs on  M
---------------------------------------------

petr222, 2018-01-14:
Look like there is a problem with the firmware in control board,
as it try to flash every version on the SD card, not knowing
which version is installed:

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 396000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 30857142
SD1    freq: 24000000
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x4 0x789aedcf
System WARM boot.
Detecting done ......

[00000000][CA9_0] [DJI] Start WDT of 20 seconds for boot up......

[00000000][CA9_0] [Version 1.0.11824], Oct 16 2015, 14:06:46

[00000000][CA9_0] [DJI] SD Updater Version [v01.23.4920] Build Mar  4 2016 16:57:05

[00000328][CA9_0] [DJI] Stop WDT.

[00000328][CA9_0] [DJI] Firmware updating......

[00000328][CA9_0] Card In Slot.
[00000629][CA9_0] SD clock source, frequency = 216000000
[00000629][CA9_0] SDCalcClk for SD card initiation:: Actual Clk = 140625, pCLK = 128
[00000647][CA9_0] start to check media
[00000653][CA9_0] AmbaSD_InitCard - SD_CARD OK
[00000658][CA9_0] Packet Detected / c:\P3S_FW_V01.02.0007.bin /.

[00000760][CA9_0] [DJI] Try Ddr Param 0.

[00000760][CA9_0] Back up partion data ......
[00038021][CA9_0] Image "System Software" is found!
[00038021][CA9_0] Image "DSP uCode" is found!
[00038021][CA9_0] Image "System ROM Data" is found!
[00038021][CA9_0] Image "Linux Kernel" is found!
[00038021][CA9_0] Image "Linux Root FS" is found!
[00038021][CA9_0] Start firmware CRC check...

[00038021][CA9_0] Checking System Software
[00041353][CA9_0] Verifying image CRC ... done
[00041353][CA9_0]       length:         9035584
[00041353][CA9_0]       crc32:          0x43e88314
[00041353][CA9_0]       ver_num:        1.0
[00041353][CA9_0]       ver_date:       2015/6/13
[00041353][CA9_0]       img_len:        9035328
[00041353][CA9_0]       mem_addr:       0x0e600000

[00041353][CA9_0] Checking DSP uCode
[00042787][CA9_0] Verifying image CRC ... done
[00042787][CA9_0]       length:         3889408
[00042787][CA9_0]       crc32:          0xd6c85dda
[00042787][CA9_0]       ver_num:        1.0
[00042787][CA9_0]       ver_date:       2015/6/13
[00042787][CA9_0]       img_len:        3889152
[00042787][CA9_0]       mem_addr:       0x0ff00000

[00042787][CA9_0] Checking System ROM Data
[00047072][CA9_0] Verifying image CRC ... done
[00047072][CA9_0]       length:         11624704
[00047072][CA9_0]       crc32:          0xc6dfedab
[00047072][CA9_0]       ver_num:        1.0
[00047072][CA9_0]       ver_date:       2015/6/13
[00047072][CA9_0]       img_len:        11624448
[00047073][CA9_0]       mem_addr:       0x03c00000

[00047073][CA9_0] Checking Linux Kernel
[00049026][CA9_0] Verifying image CRC ... done
[00049026][CA9_0]       length:         5301856
[00049026][CA9_0]       crc32:          0xec552c27
[00049026][CA9_0]       ver_num:        1.0
[00049026][CA9_0]       ver_date:       2015/6/13
[00049026][CA9_0]       img_len:        5301600
[00049026][CA9_0]       mem_addr:       0x00508000

[00049026][CA9_0] Checking Linux Root FS
[00059025][CA9_0] Verifying image CRC ... done
[00059025][CA9_0]       length:         27132160
[00059025][CA9_0]       crc32:          0xd546d3e8
[00059025][CA9_0]       ver_num:        1.0
[00059025][CA9_0]       ver_date:       2015/6/13
[00059025][CA9_0]       img_len:        27131904
[00059025][CA9_0]       mem_addr:       0x00001000

[00059498][CA9_0] Program "System Software" to NAND flash ...
[00062694][CA9_0] Done
[00062900][CA9_0] Program "DSP uCode" to NAND flash ...
[00064256][CA9_0] Done
[00064860][CA9_0] Program "System ROM Data" to NAND flash ...
[00068865][CA9_0] Done
[00069143][CA9_0] Program "Linux Kernel" to NAND flash ...
[00071008][CA9_0] Done
[00072426][CA9_0] Program "Linux Root FS" to NAND flash ...
[00081866][CA9_0] Done
[00082199][CA9_0] [DJI] Done!

[00082199][CA9_0] [DJI] Reboot......

---------------------------------------------

mefistotelis, 2018-01-14:
The messages `Can't detect firmware.` and `Firmware %s program fail` come from module m0101 - they don't exist in m0100.

I don't know for sure what the relation is between these modules - I assumed that m0100 is the proper Ambarella firmware, and m0101 is just a loader.

But here the board rewrites all partitions, and after a reboot it still only has the m0101 accessible... maybe the issue is that it writes m0101 over and over, not m0100?

Try extracting modules from the firmware, and putting m0100 alone on sd-card, renaming it to `PFC300XFw3.bin`.

petr222, 2018-01-14:
PFC300XFw3.bin - cannot detect firmware
FC300XFw.bin - cannot detect firmware

mefistotelis, 2018-01-14:
Oh right, I forgot it's Advanced - `PFC300SFw3.bin`. You can try it, thoug I don't think there will be much difference:

After looking at the m0101 code, I think the loader is damaged and cannot end the flashing circle. After `Firmware %s program fail` the next message should be `[DJI] Reboot......` - but the firmware crashes before it is able to display it.

So, the loaders function is to do the following actions, in order:
1. Finds FW update on sd-card
2. updates the application FW (m0100)
3. marks next boot to be to application
4. reboot

Looks like something bad happens at point 3 - the loader is not able to switch back to application, and crashes.

So the firmware is probably updated successfully, and would work if only the loader would let it run. But the loader isn't able to do this.

How to fix that - I have no idea.

petr222, 2018-01-14:
Actually, there is a difference.

PFC300SFw3.bin was flashed only once, after reboot it cannot detect firmware on card,
but ended in reboot loop like this - without flashing m100 firmware again:

[00000659][CA9_0] Can't detect firmware. is repeated after reboot.

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 396000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 30857142
SD1    freq: 24000000
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x4 0x789aedcf
System WARM boot.
Detecting done ......

[00000000][CA9_0] [DJI] Start WDT of 20 seconds for boot up......

[00000000][CA9_0] [Version 1.0.11824], Oct 16 2015, 14:06:46

[00000000][CA9_0] [DJI] SD Updater Version [v01.23.4920] Build Mar  4 2016 16:57:05

[00000328][CA9_0] [DJI] Stop WDT.

[00000328][CA9_0] [DJI] Firmware updating......

[00000328][CA9_0] Card In Slot.
[00000629][CA9_0] SD clock source, frequency = 216000000
[00000629][CA9_0] SDCalcClk for SD card initiation:: Actual Clk = 140625, pCLK = 128
[00000647][CA9_0] start to check media
[00000653][CA9_0] AmbaSD_InitCard - SD_CARD OK
[00000658][CA9_0] [DJI] Try Ddr Param 0.

[00000658][CA9_0] Back up partion data ......
[00036387][CA9_0] Image "System Software" is found!
[00036387][CA9_0] Image "DSP uCode" is found!
[00036387][CA9_0] Image "System ROM Data" is found!
[00036387][CA9_0] Image "Linux Kernel" is found!
[00036387][CA9_0] Image "Linux Root FS" is found!
[00036387][CA9_0] Start firmware CRC check...

[00036387][CA9_0] Checking System Software
[00039762][CA9_0] Verifying image CRC ... done
[00039762][CA9_0]       length:         9154368
[00039762][CA9_0]       crc32:          0x845822c8
[00039762][CA9_0]       ver_num:        1.0
[00039762][CA9_0]       ver_date:       2016/9/23
[00039762][CA9_0]       img_len:        9154112
[00039762][CA9_0]       mem_addr:       0x0e600000

[00039762][CA9_0] Checking DSP uCode
[00041197][CA9_0] Verifying image CRC ... done
[00041197][CA9_0]       length:         3889408
[00041197][CA9_0]       crc32:          0x4f07273c
[00041197][CA9_0]       ver_num:        1.0
[00041197][CA9_0]       ver_date:       2016/9/23
[00041197][CA9_0]       img_len:        3889152
[00041197][CA9_0]       mem_addr:       0x0ff00000
[... 87 lines cut ...]
r1 : fff90c5b  r0 : ffffffa4
Flags: NzCv
IRQs on  FIQs on  M
---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 396000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 43200000
SD1    freq: 24000000
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x0 0x789aedcb
System WARM boot.
Continue to booting up ......

---------------------------------------------

mefistotelis, 2018-01-14:
Now the issue it - I don't know how the messages look on a fully working board. Without anything to compare to, I can't be sure when it goes wrong.

Either the m0101 module should switch boot to m0100, or should execute m0100 by itself.

petr222, 2018-02-15:
I now know why one of Advanced gimbal boards reset itself:

[00073665][CA9_0] Program "System Software" to NAND flash ...
[00076911][CA9_0] Done
[00077119][CA9_0] Program "DSP uCode" to NAND flash ...
[00078483][CA9_0] Done
[00079100][CA9_0] Program "System ROM Data" to NAND flash ...
[00083164][CA9_0] Done
[00083433][CA9_0] Program "Linux Kernel" to NAND flash ...
[00085213][CA9_0] Done
[00086650][CA9_0] Program "Linux Root FS" to NAND flash ...
[00092676][CA9_0] Initial bad block. <block 642>
[00096084][CA9_0] Done
[00096418][CA9_0] [DJI] Done!

Is there a way to fix it, or is in necessary to replace whole board ?

mefistotelis, 2018-02-15:
This doesn't seem to match subject of this issue.

Anyway, if it persists after several tries, looks like the NAND is damaged in a wrong place (most NAND errors can be recovered, as long as they don't touch initial blocks). We don't know much about Ambarella, so it would be hard to recover.

On the other hand, I don't know whether Ambarella even uses the Linux part - in GoPros, this part is mostly used for GUI on LCD screen. Video processing and update code are located within System Software, and that uses RTOS which does not need RootFS.

So - it it won't work, replace the board.

Nekenieh, 2018-04-16:
Hi, I succesfully dowgraded and reanimated a P3P with module 1500/800 problem through the cheap USB-TTL converter method thanks to you guys and Mr. Sunsetcatcher.
The only thing i still have is a vage brown spot in my pictures so i am wondering if i can interchange the camera board(not the gimbal topboard) with an P3 advanced one? Are they compatable? Thanks

mefistotelis, 2018-04-16:
> can interchange the camera board(not the gimbal topboard) with an P3 advanced one

They're identical. The difference between pro and adv is only in the top board.

Nekenieh, 2018-04-17:
That’s very good to know! Anyway i took the camera sensor off and showed a dirty spot. Cleaned it and problem solved. Thanks!!

valbewe, 2019-11-11:
Hi. @petr222 could you fix your gimbal board? I have a phantom 3 advanced with exactly the same issue (No camera, gimbal is moving at initalisation then it switches off and is hanging loose. In DJI Go there is no error, only message that camera firmware update is required, ...). 
I've attached an uart to usb converter to the serial interface (U0_T and U0_R pins) and the log looks the same like yours: boot, flash, reboot, boot, flash, reboot, boot, flash, reboot.....

Is this mabye the behavior when the firmware versions of the different modules are not compatible to each other?

szupi, 2020-01-07:
Hi, @mefistotelis, At first, this repo is excellent, plenty of useful material -
[image]
 great job I have to say.
I would like to convert my p3 pro gimbal board to p3 advanced, since I have one with damaged goldpins for NAND of DM365 chip.  I can read that it is a matter of moving two 0ohm resistors. I tried to compare my p3 proboard with advanced ones pictures found in your hardware specification as well as ones found on the Internet, however no significant success.
I located USB multiplexers, as aforementioned resistors are supposed to be next to, but I also found on phantom [pilots thread](https://phantompilots.com/threads/p3-advanced-with-a-4k-camera.41243/post-691249) with differences between these two boards, and it seems to be diffrent what I expected.
Is it possible you can give me a hand here and mark what to be moved and where? I attached a good quality picture of my board focused on the are I supposed to find the resistors.
Thanks,
Łukasz

mefistotelis, 2020-01-07:
I once considered changing one of my boards, and then described some of it to someone from PhantomPilots; but I never really did that.

First, about differences in the thread: those show elements which exist on Pro and are not populated on Adv. You don't have to care too much about these. If the conversion succeeds, you may remove one chip - buck converter - to make sure those elements won't consume any power; but it isn't even necessary.

For the elements you need to change - I think the two resistors are enough, but I never confirmed that. Which resistors - I can't remember exactly, but I remember the resistors needed to be rotated 90 degrees. So the footprint was like:

&#9632;   &#9632;
&#9632;

And resistor was places across two contacts, different two on Pro and Adv.

I marked that on the partial schematic here:
https://github.com/o-gs/dji-hardware-schematics/tree/master/phantom_3_pro_gimbal_top_board
Though only on schematic - not on footprint; you'll have to find the proper connection yourself.

Also, to make the "transformation", you will really need two changes: resistors, and software. You will have to burn Ph3 Adv module on Ambarella chip. You will have to do a slight modification to headers of the firmware, otherwise Ph3 Pro will refuse to burn it.

szupi, 2020-01-10:
@mefistotelis I did reistors modification, would you give me some clues regarding fw burning, shall I do it using sd card? I suppose I have to extract individual modules and put it back together to one bin file afterwards, right?
Best regards,
Ł.

mefistotelis, 2020-01-10:
You only need Ambarella module. You can compare modules from the same FW version from Pro and Adv to get more info.

You will have to change header of the Ambarella module. Extract it, change header in *.a9h file, then re-pack. But don't re-pack to complete package, burn only Ambarella. Info on burning individual modules is on the wiki.

szupi, 2020-01-11:
I was struggling with corrupted python installation whole day, but eventually managed it.
so, basically I extracted latest fw for p3s and p3x, then extracted m0100 module, compared, modified p3s to pretend p3x, then merged all p3s ambarella files into one m0100 module using amba_fwpak.py and what do I do now? I could not find anything about burning individual modules in wiki :(

mefistotelis, 2020-01-11:
Good progress.
The wiki:
https://github.com/o-gs/dji-firmware-tools/wiki/Flashing-firmware-via-SD-card#by-firmware-module-in-camera

szupi, 2020-01-11:
Damn... I was there (reading this part) several times yesterday... 2 AM is definetly not advantageous for logic thinking ;) I even found this article where author clearly writes about name change https://dji.retroroms.info/howto/modp3params

I am getting back to work, thanks :)

szupi, 2020-01-11:
Ok, seems to be succesful. Thank you a lot. I have several more boards, where amba does not seem to be responding. Do you have any experience with bringing them alive? I suppose it is not so easy, right?
Cheers,
Ł.

mefistotelis, 2020-01-11:
Great to hear that; congrats!

For Ambarella - I don't have experience with fixing it - I never got one which was damaged. Also, the public information about that chip is limited - no datasheet nor programming guide. You can find documentation for older Ambarella chips, I'm sure the flashing methods didn't change much.

You definitely should start by tapping to serial interface and checking whether it displays anything.

In case you'll figure out anything - please share, so that other people may get further with this.

szupi, 2020-01-18:
Ok, so I dug my boards and I found ones that I spoke of. Most of them after powering up, are uart silent (tx and rx on another side of board where ambarella sits). However there is one that is reacting, led is blinking, ambarella has access to sd card, however the live feed has horizontal lines. It is P3S. I tried to upgrade the firmware using dji container file and it failed on m0100 module, then I used extracted module, given it proper name and tried SD upgrade, it failed as well:

```
[00012475]========== remo-con disconnect. boot(15) ============
 [00012522]Firmware [C:\PFC300SFw3.bin] detected, card sn [0xb06f61aa].
 [00012573][01 00] Firmware upgrade start...
 [00012408]========== remo-con disconnect. boot(1) ============
 [00012461]Firmware [C:\PFC300SFw3.bin] detected, card sn [0xb06f61aa].
 [00012511][01 00] Firmware upgrade start...
 [00012423]========== remo-con disconnect. boot(1) ============
 [00012474]Firmware [C:\PFC300SFw3.bin] detected, card sn [0xb06f61aa].
 [00012524][01 00] Firmware upgrade start...
 [00012413]========== 2020.01.17 22:28:05. boot(1) ============
 [00012472]Firmware [C:\PFC300SFw3.bin] detected, card sn [0xb06f61aa].
 [00012524][01 00] Firmware upgrade start...
 [00012397]========== 2020.01.17 22:29:24. boot(1) ============
 [00012454]Firmware [C:\PFC300SFw3.bin] detected, card sn [0xb06f61aa].
 [00012511][01 00] Firmware upgrade start...
```
and so on.

Interestingly, I cannot see anything upgrade releted being printed on uart interface during upgrade.
UART, board alone without SD card:
 
```
---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 132000000
DDR    freq: 396000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 24000000
SD1    freq: 24000000
---------------------------------------------

Start WDT timer OK.
Magic: 0x18c8d012 0x5888c81c 0xa40d814
System COLD boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 396000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 24000000
SD1    freq: 24000000
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x1 0x789aedca
System WARM boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

[... 817 lines cut ...]
System WARM boot.
Rolling Back done ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 396000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 30857142
SD1    freq: 24000000
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x15 0x789aedde
System WARM boot.
Rolling Back done ......
```
I've just found this site http://dc.p-mc.eu/tools/a7tool with ambarela A7 programming tools, I try to use it.

TaruTaruKing, 2020-09-03:
How exactly do I obtain "FC300XFw.bin" to flash the m0100 only? Thanks

szupi, 2020-09-03:
You need to use python fw dissector available in published tools here. It is well described about how to use.Br, 
Temat: Re: [o-gs/dji-firmware-tools] Flashing Adv firmware over Pro - is it possible ? (#65)Data: 2020-09-03 12:41Nadawca: "TaruTaruKing" Adresat: "o-gs/dji-firmware-tools" ; DW: "szupi" ; "Comment" ; 
How exactly do I obtain "FC300XFw.bin" to flash the m0100 only? Thanks

TaruTaruKing, 2020-09-03:
So it isn't as simple as dropping the file on the SD card and putting it in the gimbal board?

TaruTaruKing, 2020-09-03:
> 
> So it isn't as simple as dropping the file on the SD card and putting it in the gimbal board?

Why can't I just download the file somewhere if so? Haha

szupi, 2020-09-03:
Flashing is exactly so simple, however you asked about obtaining this module thus I answered how to derive it from a few file :)Temat: Re: [o-gs/dji-firmware-tools] Flashing Adv firmware over Pro - is it possible ? (#65)Data: 2020-09-03 12:57Nadawca: "TaruTaruKing" Adresat: "o-gs/dji-firmware-tools" ; DW: "szupi" ; "Comment" ; 

You need to use python fw dissector available in published tools here. It is well described about how to use.Br, Temat: Re: [o-gs/dji-firmware-tools] Flashing Adv firmware over Pro - is it possible ? (#65)Data: 2020-09-03 12:41Nadawca: "TaruTaruKing" Adresat: "o-gs/dji-firmware-tools" ; DW: "szupi" ; "Comment" ; How exactly do I obtain "FC300XFw.bin" to flash the m0100 only? Thanks

So it isn't as simple as dropping the file on the SD card and putting it in the gimbal board?

TaruTaruKing, 2020-09-03:
Lmfao, very true. I haven't slept, so I'm just trying to find the file and give it a shot before I call it quits lol

szupi, 2020-09-03:
I think I might have this file somewhere, however I cannot assure you that I will be able to pick the correct one.Temat: Re: [o-gs/dji-firmware-tools] Flashing Adv firmware over Pro - is it possible ? (#65)Data: 2020-09-03 12:59Nadawca: "TaruTaruKing" Adresat: "o-gs/dji-firmware-tools" ; DW: "szupi" ; "Comment" ; 

You need to use python fw dissector available in published tools here. It is well described about how to use.Br, Temat: Re: [o-gs/dji-firmware-tools] Flashing Adv firmware over Pro - is it possible ? (#65)Data: 2020-09-03 12:41Nadawca: "TaruTaruKing" Adresat: "o-gs/dji-firmware-tools" ; DW: "szupi" ; "Comment" ; How exactly do I obtain "FC300XFw.bin" to flash the m0100 only? Thanks

So it isn't as simple as dropping the file on the SD card and putting it in the gimbal board?

Why can't I just download the file somewhere if so? Haha

TaruTaruKing, 2020-09-03:
Nothing like Russian roulette at seven in the morning lol I have a Phantom 3 Pro if that matters lol

TaruTaruKing, 2020-09-03:
I read somewhere that the guy flashed firmwares backwards and then forwards and it fixed his gimbal board with undetected devices 8 and 15, but it's taking forever...

szupi, 2020-09-03:
Just state what you are up to, describe what you've done so far and where you are, I will try to help.Temat: Re: [o-gs/dji-firmware-tools] Flashing Adv firmware over Pro - is it possible ? (#65)Data: 2020-09-03 13:06Nadawca: "TaruTaruKing" Adresat: "o-gs/dji-firmware-tools" ; DW: "szupi" ; "Comment" ; 

I think I might have this file somewhere, however I cannot assure you that I will be able to pick the correct one.Temat: Re: [o-gs/dji-firmware-tools] Flashing Adv firmware over Pro - is it possible ? (#65)Data: 2020-09-03 12:59Nadawca: "TaruTaruKing" Adresat: "o-gs/dji-firmware-tools" ; DW: "szupi" ; "Comment" ; You need to use python fw dissector available in published tools here. It is well described about how to use.Br, Temat: Re: [o-gs/dji-firmware-tools] Flashing Adv firmware over Pro - is it possible ? (#65)Data: 2020-09-03 12:41Nadawca: "TaruTaruKing" Adresat: "o-gs/dji-firmware-tools" ; DW: "szupi" ; "Comment" ; How exactly do I obtain "FC300XFw.bin" to flash the m0100 only? Thanks

Nothing like Russian roulette at seven in the morning lol I have a Phantom 3 Pro if that matters lol

TaruTaruKing, 2020-09-03:
[01026609]Version checking[3]...
 [01026741][03 05][05] v34.2.0.9 -> v34.2.0.9 need upgrade.
 [01026860][03 06][05] v2.4.14.5 -> v2.4.14.5 need upgrade.
 [01026938][04 00][05] v1.44.0.0 -> v1.44.0.0 need upgrade.
 [01027078][11 00][05] v1.7.15.1 -> v1.7.15.1 need upgrade.
 [01027244][12 00][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
 [01027345][12 01][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
 [01027472][12 02][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
 [01027608][12 03][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
 **[01040184][15 00][00] device not detected.**
 [01040307][17 00][05] v1.1.1.7 -> v1.1.1.7 need upgrade.
 [01040466][17 01][05] v1.0.2.7 -> v1.0.2.7 need upgrade.
 [01040564][19 00][00] v255.255.255.254 -> v1.0.8.96
 [01040633][01 00][05] v1.30.5036 -> v1.30.5036 need upgrade
 [01040708][01 01][05] v1.30.5036 -> v1.30.5036 need upgrade
 **[01053283][08 00][00] device not detected.**
 [01053378][09 00][05] v3.0.0.10 -> v3.0.0.10 need upgrade.
 [01053524]Packet upgrade failed at version checking.

Trying to fix these lol

szupi, 2020-09-03:
Have you resoldered jumpers/resistors on the board?Temat: Re: [o-gs/dji-firmware-tools] Flashing Adv firmware over Pro - is it possible ? (#65)Data: 2020-09-03 13:08Nadawca: "TaruTaruKing" Adresat: "o-gs/dji-firmware-tools" ; DW: "szupi" ; "Comment" ; 
I read somewhere that the guy flashed firmwares backwards and then forwards and it fixed his gimbal board with undetected devices 8 and 15, but it's taking forever...

TaruTaruKing, 2020-09-03:
Nah, this was a common problem after firmware updates from what I've read. I believe it is firmware related, so I am just working backwards from 1.11.0200 to 1.03.0200 and then I will go forwards. Just thought I could try those files if that didn't work as a last ditch effort before I toss the board. I could bake it or use a hot air blower on it I guess, but I do think it's firmware related. I found this post that made me attempt the stepped firmware downgrade/upgrade, but it's pretty tedious lol

"Hi Guys,

I know this is an old thread, but thought it worthwhile sharing with you how I fixed this without sending the aircraft back to DJI, or replacing any hardware.
I posted this last night on another forum, but I'm not sure about the rules of link sharing from other forums(?), so I'll just add the entire post here:

Right, I've had this issue myself, and have just managed to fix it without RMA or hardware replacement.
It's pretty much taken me all weekend though, researching this and other forums.

So I tried different firmware versions on the aircraft, the two firmware versions available on the controller, nothing.

Then I found this link showing how to help 'force' upgrade or downgrade firmware:
forcing fw downgrade with P3X_FW_DEBUG · Issue #4 · mefistotelis/phantom-firmware-tools · GitHub
It's a very simple thing to enable, just create a file called P3X_FW_DEBUG in the root of the SD card you are using to do the firmware upgrade.
I did that by creating a text file, naming it as above, and removing the .txt extension.

Reading in the above link, mefistotelis states that he believes that long jumps in firmware upgrades are not ideal.

So, and this was the painful bit, I decided to try and flash the firmware in order by version steps - it took bloody ages:
I found this link on one of these forums where you can download all the versions of firmware for the P3P:

DJI Phantom Accessories & Tips - Phantom Help
I downloaded them all, and started with the current version, going one by one down to a much older version (not the oldest version):
v1.10.9
v1.09.6
v1.08.8
v1.07.6
v1.06.4
v1.05.3
v1.04.1
v1.03.2
v1.02.6

All of them failed with the long dirty tone sounding on the aircraft at the end of the upgrade process, but I persevered.
Noting that each time, the aircraft would mostly boot as normal, but still showing 'No Image Transmission' when connected.
Also, a few times, the aircraft would not properly start at all, and instead would start beeping loudly 'like an alarm' clock and no lights showing on the aircraft as some posters have commented.
BUT, the upgrades would still run through, so I kept going, and after applying the next version down, the loud beeping would stop and the aircraft lights would come back on.
Eventually, I got to the earliest version (v1.02.6) with none of the upgrades reporting successful, but then I started incrementing firmware versions from oldest to newest as follows:

from v1.02.6
v1.03.2
v1.04.1
v1.05.3
v1.06.4
v1.07.6
v1.08.8
and on this version, the firmware update was successful - for the first time!

Not quite believing it had worked, I decided to power on the craft and controller (which was running v1.5 firmware).

Lo and behold, the **** picture is back and working again!
I can't believe it! I am going to cancel the RMA that I raised today, while I was halfway through this process and lost hope - convinced it was a hardware issue!

I hope the above process works for anyone else out there who has tried many different ways to fix their issue.

It just goes to show, it's not 'definitely a hardware issue' if you get both the:
[00028828][15 00][00] device not detected.
[00041697][08 00][00] device not detected.
at the same time.

Best of luck!"

szupi, 2020-09-03:
So this is wrong neighbourhood. You need to hook your board to the uart. Provided your dm365 and man chip are not permanently damaged. You know that In this I downgraded my p3x to p3s, right?Temat: Re: [o-gs/dji-firmware-tools] Flashing Adv firmware over Pro - is it possible ? (#65)Data: 2020-09-03 13:10Nadawca: "TaruTaruKing" Adresat: "o-gs/dji-firmware-tools" ; DW: "szupi" ; "Comment" ; 
[01026609]Version checking[3]...
[01026741][03 05][05] v34.2.0.9 -&gt; v34.2.0.9 need upgrade.
[01026860][03 06][05] v2.4.14.5 -&gt; v2.4.14.5 need upgrade.
[01026938][04 00][05] v1.44.0.0 -&gt; v1.44.0.0 need upgrade.
[01027078][11 00][05] v1.7.15.1 -&gt; v1.7.15.1 need upgrade.
[01027244][12 00][05] v1.10.0.0 -&gt; v1.10.0.0 need upgrade.
[01027345][12 01][05] v1.10.0.0 -&gt; v1.10.0.0 need upgrade.
[01027472][12 02][05] v1.10.0.0 -&gt; v1.10.0.0 need upgrade.
[01027608][12 03][05] v1.10.0.0 -&gt; v1.10.0.0 need upgrade.[01040184][15 00][00] device not detected.
[01040307][17 00][05] v1.1.1.7 -&gt; v1.1.1.7 need upgrade.
[01040466][17 01][05] v1.0.2.7 -&gt; v1.0.2.7 need upgrade.
[01040564][19 00][00] v255.255.255.254 -&gt; v1.0.8.96
[01040633][01 00][05] v1.30.5036 -&gt; v1.30.5036 need upgrade
[01040708][01 01][05] v1.30.5036 -&gt; v1.30.5036 need upgrade[01053283][08 00][00] device not detected.
[01053378][09 00][05] v3.0.0.10 -&gt; v3.0.0.10 need upgrade.
[01053524]Packet upgrade failed at version checking.
Trying to fix these lol

TaruTaruKing, 2020-09-03:
You did that for fun I guess? What do you mean hook my board to the uart? I took intro to C in college, but that's about the extent of my programming knowledge lol And that was 7 years ago

TaruTaruKing, 2020-09-03:
But yeah, I don't even know how I ended up in this thread lol

szupi, 2020-09-03:
> You did that for fun I guess?

No, solder pads for NAND chip were damaged, so it was theonly feasible option to have anything.

> But yeah, I don't even know how I ended up in this thread lol

https://phantompilots.com/threads/fixing-flash-in-dm365-chip-within-ph3-pro-gimbal.127310/
Been there, read all? Good luck!

ivankika1982, 2020-11-07:
I get this without an sd card
`Code:
---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 132000000
DDR    freq: 396000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 24000000
SD1    freq: 24000000
---------------------------------------------

Start WDT timer OK.
Magic: 0x4a40d81e 0x484a581e 0x588ad81e
System COLD boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 396000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 24000000
SD1    freq: 24000000
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x1 0x789aedca
System WARM boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 396000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 24000000
SD1    freq: 24000000
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x2 0x789aedc9
System WARM boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
[... 365 lines cut ...]

[00000329][CA9_0] [DJI] Firmware updating......

[00000329][CA9_0] Card In Slot.
[00000630][CA9_0] SD clock source, frequency = 216000000
[00000630][CA9_0] SDCalcClk for SD card initiation:: Actual Clk = 140625, pCLK = 128
[00000653][CA9_0] start to check media
[00000657][CA9_0] AmbaSD_InitCard - SD_CARD OK
[00000673][CA9_0] Can't detect firmware.
[00000742][CA9_0] Firmware  program fail.--- ARM DATA ABORT EXCEPTION ---
Oops: CPU Exception!
pc : [<0001134c>]    lr : [<00000000>]
sp : 0006f328  ip : 00000000  fp : 00000000
r10: 0006b400  r9 : 00000000  r8 : 00000000
r7 : 00000000  r6 : 00000000  r5 : 00000000
r4 : 0006f394  r3 : 00000000  r2 : 00000000
r1 : fff90c5b  r0 : ffffffa4
Flags: NzCv
IRQs on  FIQs on  M
---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 396000000

rufiooo, 2021-01-02:
> 
> 
> @mefistotelis I did reistors modification, would you give me some clues regarding fw burning, shall I do it using sd card? I suppose I have to extract individual modules and put it back together to one bin file afterwards, right?
> Best regards,
> Ł.

Hello Łukasz,
could you share a photo with your resistors modification (I'm not that electronic skilled to find it by the diagram...)? I've got P3P board with teared out NAND chip and leads on board as well, so replacing it is not an option.
Thanks in advance - Rufiooo