o-gs/dji-firmware-tools #66: dji_flyc_param_ed.py do not work with A3 FC  bin

sincoder, 2018-01-22:
$ python3 ../dji_flyc_param_ed.py -vv -m A3_FW_V02.00.00.01_m0306.bin -x
A3_FW_V02.00.00.01_m0306.bin: Opening for extraction
Traceback (most recent call last):
  File "../dji_flyc_param_ed.py", line 716, in <module>
    main(sys.argv[1:])
  File "../dji_flyc_param_ed.py", line 697, in main
    flyc_extract(po,fwmdlfile)
  File "../dji_flyc_param_ed.py", line 558, in flyc_extract
    raise ValueError("Flight controller parameters array signature not detected in input file.")
ValueError: Flight controller parameters array signature not detected in input file.

sincoder, 2018-01-22:
the bin file was been decrypted

mefistotelis, 2018-01-22:
I can confirm it doesn't work.

Is A3 based on the same uC as P3X? If not, it likely needs different base address.

sincoder, 2018-01-22:
Thanks, it works

> $ python3 ../dji_flyc_param_ed.py -vv -x -m N3_FW_V01.01.01.00_m0306.bin -b 0x420000
> N3_FW_V01.01.01.00_m0306.bin: Opening for extraction
> N3_FW_V01.01.01.00_m0306.bin: Matching parameters array at 0x0015c834: 1102 entries, format 2017
> N3_FW_V01.01.01.00_m0306.bin: Extracting parameters array at 0x0015c834: 1102 entries

closed 2018-01-22 by sincoder