o-gs/dji-firmware-tools #69: Unlock 100 MBit/s for P3P 4K possible?

marck1991, 2018-02-05:
Hi all,

the sensor of the P3P should be technically able to encode 4K video with 100MBit/s (H.264 Level 5.1).
DJI however limited it to 60MBit/s wich results in some compression artifacts in some scenes. 
Do you see any possibility to unlock this software limitation?
The datasheet should be: http://www.ambarella.com/uploads/docs/A9-product-brief.pdf

mefistotelis, 2018-02-05:
That would need enabling and testing.
Possible, if only someone is willing to spend time on analysing m0100 and finding where compression profiles are.

What you linked to is just a promotional leaflet, we do not have a real datasheet.

marck1991, 2018-02-05:
Thanks for you answer @mefistotelis .
I‘m willing to spend time on this, if I am able to to this.
A short google on the SDK hasn’t given any useful results. 
Do you have an address where to get it?

mefistotelis, 2018-02-05:
I don't have it, but I do have similar one. Contact me via e-mail or slack.