o-gs/dji-firmware-tools #70: p3a dji_fwcon problem

kydcs, 2018-02-22:
Hi Steve:
    My python edition is 2.7,beacuse my system is win7.Because python 3.63 can`t run at win7.
About" import configparser" problem,i already solve the problem,but dji_fwcon file have any more command wrong!  Thankyou!
Traceback (most recent call last):
  File "dji_fwcon.py 

", line 672, in <module>
    main(sys.argv[1:])
  File "dji_fwcon.py 

", line 669, in main
    raise NotImplementedError('Unsupported command.')
NotImplementedError: Unsupported command.

mefistotelis, 2018-02-22:
> Hi Steve:

Who is Steve? You?

> My python edition is 2.7,beacuse my system is win7.Because python 3.63 can`t run at win7.

Not true in the slightest.

> About" import configparser" problem,i already solve the problem,but dji_fwcon file have any more command wrong!

You can backport dji_fwcon to the old Python if you want, sure. We're not really interested in having/supporting the backported version, though.

> raise NotImplementedError('Unsupported command.')

Are you using correct parameters? Use "--help" for help.

closed 2018-02-22 by kydcs

tomm10000, 2018-07-31:
CMD console 
C:\PS3P>python -m pip install pycryptodome

C:\PS3P>python.exe dji_fwcon.py -v -x -p P3X_FW_V01.11.0020.bin
C:\PS3P>python.exe dji_flyc_param_ed.py -vv -x -m P3X_FW_V01.11.0020_m0306.bin
C:\PS3P>python.exe dji_flyc_param_ed.py -vv -u -m P3X_FW_V01.11.0020_m0306.bin