o-gs/dji-firmware-tools #73: Flight Parameter Extracted From Mavic FW

chazzzzz, 2018-03-27:
Hey Mefistotelis,

I can't seem to get the flight parameters extracted using “ dji_flyc_param_ed.py -vv -x -b 0x420000 -m wm220_0306_v03.02.30.13_20170405.pro.fw.sig” Maybe you can tell me what I’m doing wrong please?

[image]

chazzzzz, 2018-04-20:
Anybody Please? I'm trying to change one default parameter this way because when I change it in assistant it reverts back to the default when the Mavic is power cycled...This script works nicely for the phantom 3 but I cant get it this work for the Mavic. Thank You

mefistotelis, 2018-04-20:
Just checked on current script:
```sh-4.4$ ./dji_flyc_param_ed.py -vv -x -b 0x420000 -m wm220_0306_v03.02.35.05_20170525.pro.bin
wm220_0306_v03.02.35.05_20170525.pro.bin: Opening for extraction
wm220_0306_v03.02.35.05_20170525.pro.bin: Matching parameters array at 0x00152b50: 1153 entries, f
ormat 2017
wm220_0306_v03.02.35.05_20170525.pro.bin: Extracting parameters array at 0x00152b50: 1153 entries
sh-4.4$```
You must have encrypted file.

chazzzzz, 2018-04-20:
Thanks for the reply...I downloaded from a few different sources with the same result. Can you send me a link to an unencypted file?

mefistotelis, 2018-04-21:
Try this:
https://github.com/fvantienen/dji_rev

mefistotelis, 2018-11-06:
The firmware can now be decrypted with dji_mvfc_fwpak.py.

closed 2018-11-06 by mefistotelis