o-gs/dji-firmware-tools #75: DJI Phantom 3 advanced no image transmission and no record

LukeM2602, 2018-05-14:
I have got a Phantom 3 adv that won’t record or transmit a signal to the phone I have tried different solutions like to downgrade or try another sd card but non of it won’t work. There is no light on gimble board but the gimble functions correctly any solutions like the soldering way to flash the chip if possiable

mefistotelis, 2018-05-14:
You need to figure out a way to flash Ambarella A9 firmware. You may use A7 datasheet as a reference, as A9 one is hard to get. If it errors out while flashing, you'll have to replace the NAND chip.
Please write about your progress, this is interesting subject.

mefistotelis, 2018-05-14:
I don't have any board damaged this way.

LukeM2602, 2018-05-15:
It might work if you put a custom script on the memory of the gimble it’s self via the service points

mefistotelis, 2023-01-09:
Not much info here, closing.

closed 2023-01-09 by mefistotelis