o-gs/dji-firmware-tools #76: nofly array update issue

sincoder, 2018-05-15:
I am try remove some items from the json file , or remove all items , not work 
$ python2 ../../../phantom-firmware-tools/dji_flyc_nofly_ed.py  -u -vv -m P3SE_FW_V01.05.0040_m0306.bin
P3SE_FW_V01.05.0040_m0306.bin: Opening for update
P3SE_FW_V01.05.0040_m0306.bin: Matching zones array at 0x000bc894: 1628 entries
P3SE_FW_V01.05.0040_m0306.bin: Matching zones array at 0x000cab06: 1463 entries
P3SE_FW_V01.05.0040_m0306.bin: Warning: multiple (2) matches found for fly zones array with alignment 0x02
P3SE_FW_V01.05.0040_m0306.bin: Matching coords array at 0x0008ce64: 62 entries
P3SE_FW_V01.05.0040_m0306.bin: Matching coords array at 0x0009c604: 4183 entries
P3SE_FW_V01.05.0040_m0306.bin: Warning: multiple (2) matches found for fly coords array with alignment 0x02
P3SE_FW_V01.05.0040_m0306.bin: Loading JSON file...
P3SE_FW_V01.05.0040_m0306.bin: Updating no fly zones array at 0x000bc894: 1628 entries
Traceback (most recent call last):
  File "../../../phantom-firmware-tools/dji_flyc_nofly_ed.py", line 508, in <module>
    main(sys.argv[1:])
  File "../../../phantom-firmware-tools/dji_flyc_nofly_ed.py", line 499, in main
    flyc_nofly_update(po,fwmdlfile)
  File "../../../phantom-firmware-tools/dji_flyc_nofly_ed.py", line 382, in flyc_nofly_update
    eprint("{}: Warning: no fly zone not found in fw: area_id={:s}".format(po.mdlfile,pvparprop['area_id']))
ValueError: Unknown format code 's' for object of type 'int'

sincoder, 2018-05-15:
It's best to have an option to clear all the items  ^_^

mefistotelis, 2018-05-15:
Please note that the update function is unfinished. It will not work unless you write it.

Also, even if/when the update will be implemented, removing an item from the list won't cause to remove it from firmware - what I planned here is to allow updating the zones (ie. changing radius, type etc.), not removing it.

Still, what you reported is a legitimate bug. I will fix it, so that it will get back to showing "not implemented" error.

sincoder, 2018-05-15:
Thank you, I also noticed that it have not implemented after i read the code

closed 2018-05-15 by sincoder