o-gs/dji-firmware-tools #77: Ph3 Pro/Adv - Recording communication between OFDM and ESC Center Board

mefistotelis, 2018-06-01:
This is not really an issue, but rather a tutorial.

There is a 6-pin ribbon cable which connects OFDM Board to ESC Center Board:
- On OFDM side, the connector is marked as `CENTER` - see [OFDM board photos](https://github.com/o-gs/dji-firmware-tools/wiki/P3X-OFDM-Receiver-board#board-view).
- On ESC Center side, it is merged with VPS cable into one connector marked `VIEW/OFDM`, though the words are a bit separated from it - see [ESC center board photos](https://github.com/o-gs/dji-firmware-tools/wiki/P3X-ESC-center-board#board-view).

Here's the pinout on OFDM side:
[image]

And pinout from ESC Center side:
[image]

To record communication between Flight Controller and OFDM, you need to connect two USB-to-TTL converters to the `RX`, `TX` and `GND` pins of OFDM side, or to the `RX1`, `TX1` and `GND` pins on ESC Center side. You need two USB-to-TTL converters, as you need two "receive" lines - you will dump communication in both directions.

You can pinch wires, or put needles into connectors, or prepare a new ribbon cable which will extend the existing one, and add a second route for the 3 pins that need connecting to USB-to-TTL device.

Here's an example of the extension ribbon cable for the OFDM side:
[image]

And here's an example of the complete setup:
[image]

After all the connections are done, you can use `comm_serial2pcap.py` to dump the communication to a PCAP file on PC.
Example for PC running Linux (recommended):
```
sudo ./comm_serial2pcap.py -b 256000 -u 3 -e -vvv -w p3x_ofdm_to_center_run_001.pcap /dev/ttyUSB0 /dev/ttyUSB1
```

The PCAP files can be opened with Wireshark, after the [proper dissectors are installed](https://github.com/o-gs/dji-firmware-tools/tree/master/comm_dissector).

mefistotelis, 2018-06-02:
I ordered mine from China, ie:
https://www.aliexpress.com/wholesale?catId=0&initiative_id=AS_20180602085713&SearchText=6+pin+1.25mm+cable
https://www.aliexpress.com/wholesale?minPrice=&maxPrice=&isBigSale=n&isFreeShip=y&isNew=n&isFavorite=n&isMobileExclusive=n&isLocalReturn=n&shipFromCountry=&shipCompanies=&SearchText=usb+to+ttl&CatId=0&g=y&SortType=price_asc&initiative_id=SB_20180602085930&needQuery=n&groupsort=1

dnapapas, 2018-06-02:
Ok, thanks for the info. In the meantime I will start putting together a linux box. Any suggestions that would be most useful for what we are doing? I only have Win10 machines, and android devices.

mefistotelis, 2018-06-02:
You mean distribution? Any will do. I mostly use Ubuntu.

dnapapas, 2018-06-03:
Ok, got it. I was just looking at the boards you are talking about above. They look nothing like the P4, they look like P3.

mefistotelis, 2018-06-03:
Oh, I see. I assumed you're using Ph3. My knowledge on Ph4 is very limited.

dnapapas, 2018-06-04:
@mefistotelis  [https://www.rcgeeks.co.uk/blog/phantom-4-basic-teardown-whats-inside(url)
This will give you a good idea of the Ph4 internal layout.

mefistotelis, 2018-06-04:
> if there are any other items I may need for future testing that you can think of, please let me know.

Soldering equipment, wires... not much more I could think of.

> Do you know if there are any internal Ph4 board photos around?

I will do a little search, maybe it is time to add some of its boards to the wiki. It will take me some time though.

dnapapas, 2018-06-04:
Next time I take this thing apart I will take some close up hi res photos of the boards. I have plenty of soldering equipment and I can solder very well so once I get those parts I should be set then.
I am going to start my own site for saving and sharing all the info I find on the P4. Problem is, I dont have servers setup for dns. Also, I am only capable of doing very basic web page design. Would you be interested in helping out with that? Website name I have registered is, dronietech.com

mefistotelis, 2018-06-04:
> I am going to start my own site for saving and sharing all the info I find on the P4.

Wouldn't it be better to use any of existing methods?

Ie. you could create your own Github project and make a wiki similar to what this project has. You only need to use Git and Markdown language, plus we can easily exchange our pages easily.

You could also expand the RetroRoms subsite which many folks form Slack group are working on:
https://dji.retroroms.info/

For me - I like the Github method. Is is easy for anyone to clone, plus I don't have to deal with layout or server support. Github will probably function much longer than I'll be willing to support my own servers.

dnapapas, 2018-06-04:
Just something basic where we could keep everything organized in one place. Maybe some sort of message board, photo storage area, and anything else you would think to be useful. Different areas for different aircrafts and projects etc. It has been a very long time since I made any kind of website and I was just experimenting with using free webpage builders, html editing, free perl scripts, but I really have no knowledge of how to do it properly. If I did attempt to do it, it would be a learning process, but I can probably get through it. I can build linux servers with apache webserver and the such and make it all work but, I don't have the knowledge or know how to make it secure.

dnapapas, 2018-06-04:
> @mefistotelis EDIT: back to the needed equipment - in case we will want to dump packets in-flight, some kind of Pi clone with two serial interfaces would be useful. Not sure if any of those has two UARTS though, my Pi Zero only comes with one :/.
This stuff is all new to me. I have never done it before but Im sure I can and I want to learn how to do it. So, I will purchase anything I will need to do this if you don't mind just explaining some things

mefistotelis, 2018-06-04:
> This stuff is all new to me.

Raspberry Pi Zero is just a small computer, on which you can install Linux and use the Python script (comm_serial2pcap.py) to dump packets. It is small enough to be put into a flying drone. With additional buck converter, it could even be powered from the drone battery.

> I will purchase anything I will need to do this if you don't mind just explaining some things

I'm not sure if we will need packet dumps during flight; we don't need it for figuring out FCC, as the mode switch happens before takeoff.

dnapapas, 2018-06-04:
> @mefistotelis Wouldn't it be better to use any of existing methods?

>Ie. you could create your own Github project and make a wiki similar to what this project has. You only need to use Git and Markdown language, plus we can exchange our pages easily.

>You could also expand the RetroRoms subsite which many folks form Slack group are working on:
https://dji.retroroms.info/

>For me - I like the Github method. Is is easy for anyone to clone, plus I don't have to deal with layout or server support. Github will probably function much longer than I'll be willing to support my own servers.

Thats fine, I can do that. Not sure about how to use github, I just have to get my head wrapped around it for a minute., Im just interested in having my own site for learning purposes really, and just be able to share my flight videos and photos and some other interests. And just have a tech area where I could share what Im currently working on.

dnapapas, 2018-06-04:
> @mefistotelis I'm not sure if we will need packet dumps during flight; we don't need it for figuring out FCC, as the mode switch happens before takeoff.

Well I will get something so we have it just incase we would need it for something. I will try to find something with 2 serial interfaces and 2 UARTS, if there is such a device.

danieltroger, 2018-06-04:
I've got an odroid xu4 as webserver (with dyndns) if you want to have some space on it I'd happily give you access. It's currently running on a 32GB µSD-card but I could add a hard drive if you need more storage.
Aand it's currently set up as a LAMP server on ubuntu 18.04 with PHP7.
Bandwith is (realistically) 90mbit/s down and 35mbit/s up.

dnapapas, 2018-06-04:
Ok, sorry for the dumb question but could we use 2 Pi devices for a 2 UART solution? I assume they need to be on the same device but I figure I would ask anyways.

dnapapas, 2018-06-04:
Is this something that can work? USB hub and USB to RS232 converters w/ftdi chipset.
Something else I found,
[image]

dnapapas, 2018-06-04:
@danieltroger I appreciate your offer and I will probably take you up on it if I cant get things going here on my own. My problem is, I am not good at all in web page design.

dnapapas, 2018-06-19:
@mefistotelis  Ok, my parts came in alot sooner than expected. I ordered 10 each of the wires and plugs and 4 of the usb to ttl adapters.
[image]
[image]

mefistotelis, 2018-06-19:
Ok, great. Do you know where to connect these to your Ph4?

dnapapas, 2018-06-20:
@mefistotelis  No sir, thats where I will need your assistance. If you can guide me, I will get it done. Im doing this for you guys so just tell me what you want me to do. The only thing I need is a way to adjust power output. But lets dig into this thing and see what else can be found.

mefistotelis, 2018-06-20:
I will look at available Ph4 board photos when I'm able. Since it is Lightbridge like in Ph3, I expect the same hardware abnd the same connections, just fit on different boards.

mefistotelis, 2018-07-07:
It looks like I won't be able to determine the specific place to record packets in Ph4.

I analysed its internals; published the photos I used here:
https://github.com/o-gs/dji-firmware-tools/wiki/DJI-Hardware#phantom-4

So the Lightbridge is placed on 3-in-1 board, and it is connected to Flight Controller via Flat Film Wire. This is the connection we want to monitor; but since it's now Flat Film and not ribbon, it is hard to monitor directly. There are probably test pads which we could solder wires to, but the photos I have are not enough to figure out which pads are the interface we want.

pawelsky, 2018-07-08:
How about connecting to the flight controller board, where the serials meet? If I had to guess I would say that on [this picture](https://raw.githubusercontent.com/wiki/o-gs/dji-firmware-tools/WM330-photos/WM330-Flight-Controller-boardv10a-btm.png) the test points for most serials are labelled (with T#/R#).